raspberrypi/linux #753: fresh install update to latest firmware getting amixer error

amazon83, 2015-01-02:
$amixer get master
amixer: Unable to find simple control 'master',0

bug?

pelwell, 2015-01-02:
That's not a lot to go on.

1) Which branch are you on?
2) What ALSA devices should you have in your system?
3) What do you expect 'master' to be?
4) What do you get from `aplay -l` now (and with the old configuration, if possible)?

amazon83, 2015-01-02:
1. tested in Raspberry pi B+ (RS components) / A+ (Element 14)
2. All fresh install debian run all apt-update, apt-upgrade, apt dist-upgrade, rpi-update
3. all amixer command control volume not working.
4. 
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

uname -a
Linux pi 3.12.35+ #730 PREEMPT Fri Dec 19 18:31:24 GMT 2014 armv6l GNU/Linux

pelwell, 2015-01-02:
Your system is only showing the default analogue audio devices, and these don't have a control called 'master'. So, going back to question 2, do you have any other audio devices in your system that used to work but aren't working now?

amazon83, 2015-01-02:
Below show my another pi A+ using HDMI, also getting same problem. i try amixer ssget sset command also give me return amixer: Unable to find simple control. all my audio working fine. just the command (amixer) not working.

tested in Raspberry pi  A+ (Element 14)
All fresh install debian run all apt-update, apt-upgrade, apt dist-upgrade, rpi-update
all amixer command control volume not working.

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

uname -a
Linux pia 3.12.35+ #730 PREEMPT Fri Dec 19 18:31:24 GMT 2014 armv6l GNU/Linux

amazon83, 2015-01-02:
I can just use alsamixer to control my volume, but in programming we have to use amixer to control the volume + - , hope the team can get this fix. 

3.12.35+ #730 PREEMPT Fri Dec 19 18:31:24 GMT 2014 armv6l GNU/Linux

pelwell, 2015-01-02:
But 'master' is not a control exported by the bcm2835 driver, so I don't know how you expect this to work. `amixer controls` lists the controls supported, and `amixer contents` gives more detail.

Did/do you perhaps have some custom settings in /etc/asound.conf ?

amazon83, 2015-01-02:
No, all fresh install

pelwell, 2015-01-02:
Do you have any evidence that `amixer get master` has ever worked for you?

amazon83, 2015-02-08:
please close this, it fixed.

closed 2015-02-09 by popcornmix