raspberrypi/linux #756: add bcm2835-pwm to bcm2708.c

btanghe, 2015-01-05:
In linux kernel release 3.19-rc1, a bcm2835-pwm driver is added.
To load the driver properly with board files (bcm2708.c), this patch has to be added to the file

diff --git a/arch/arm/mach-bcm2708/bcm2708.c b/arch/arm/mach-bcm2708/bcm2708.c
index 6349cbe..2f05901 100644
--- a/arch/arm/mach-bcm2708/bcm2708.c
+++ b/arch/arm/mach-bcm2708/bcm2708.c
@@ -226,6 +226,10 @@ static struct clk sdhost_clk = {
 #endif
 };
 
+static struct clk pwm_clk = {
+        .rate = 9200000,
+};
+
 static struct clk_lookup lookups[] = {
 	{			/* UART0 */
 	 .dev_id = "dev:f1",
@@ -243,6 +247,10 @@ static struct clk_lookup lookups[] = {
 	 }, {	/* BSC1 */
 		 .dev_id = "bcm2708_i2c.1",
 		 .clk = &sdhost_clk,
+	 },
+	{
+	 .dev_id = "bcm2835-pwm.1",
+	 .clk = &pwm_clk,
 	 }
 };
 
@@ -632,6 +640,21 @@ static struct platform_device bcm2708_bsc0_device = {
 	.resource = bcm2708_bsc0_resources,
 };
 
+static struct resource bcm2835_pwm_resources[] = {
+	{
+	 .start = PWM_BASE,
+	 .end = PWM_BASE + SZ_4K - 1,
+	 .flags = IORESOURCE_MEM,
+	 }
+};
+
+static struct platform_device pwm_bcm2835_device = {
+	.name = "bcm2835-pwm",
+	.id = 1,
+	.resource = bcm2835_pwm_resources,
+	.num_resources = ARRAY_SIZE(bcm2835_pwm_resources),
+};
+
 
 static struct resource bcm2708_bsc1_resources[] = {
 	{
@@ -957,6 +980,8 @@ void __init bcm2708_init(void)
 	spi_register_board_info(bcm2708_spi_devices,
 			ARRAY_SIZE(bcm2708_spi_devices));
 #endif
+
+	bcm_register_device(&pwm_bcm2835_device);
 }
 
 static void timer_set_mode(enum clock_event_mode mode,
diff --git a/arch/arm/mach-bcm2708/include/mach/platform.h b/arch/arm/mach-bcm2708/include/mach/platform.h
index 2e7e1bb..d3975cf 100644
--- a/arch/arm/mach-bcm2708/include/mach/platform.h
+++ b/arch/arm/mach-bcm2708/include/mach/platform.h
@@ -76,6 +76,7 @@
 #define BSC1_BASE		 (BCM2708_PERI_BASE + 0x804000) /* BSC1 I2C/TWI */
 #define USB_BASE                 (BCM2708_PERI_BASE + 0x980000) /* DTC_OTG USB controller */
 #define MCORE_BASE               (BCM2708_PERI_BASE + 0x0000)   /* Fake frame buffer device (actually the multicore sync block*/
+#define PWM_BASE		 (BCM2708_PERI_BASE + 0x20C000)
 
 #define ARMCTRL_BASE             (ARM_BASE + 0x000)
 #define ARMCTRL_IC_BASE          (ARM_BASE + 0x200)           /* ARM interrupt controller */

pelwell, 2015-01-05:
3.18.y is currently available in the FIRMWARE=next branch. In it we have enabled device tree support by default, although it can be disabled using a config.txt entry. The main motivation for this switch is to move away from the need to add platform devices to bcm2708.c; instead, we just need to patch the relevant .dts files and recompile them. As a result we would be reluctant to add this patch unless there was a compelling reason to do so.

Documentation of our device tree support is available [here](https://github.com/raspberrypi/documentation/blob/master/configuration/device-tree.md), and there is an active forum thread discussing it [here](http://www.raspberrypi.org/forums/viewtopic.php?f=29&t=93015).

Since you have developed this for bcm2835, I presume you already have a suitable device tree description of the pwm device. It would help us in evaluating the driver if you could:
1) provide any relevant device tree source, and
2) recommend some simple test code that we can use.

btanghe, 2015-01-06:
The device tree description is send to the linux kernel mailing list https://lkml.org/lkml/2015/1/5/410
If you want to enable gpio pin 18 as pwm, you can apply the patch below (hw tested).
You can add the second pwm channel on pin 19 (only available on b-plus) on the same way.
Because their is no suitable clk driver, I've added a fixed-clock to the dts file.
The pwm clock isn't enabled by default. I'm using a userspace prog to enable the pwm clock.
I'm using the bcm2835_pwm_set_clock(BCM2835_PWM_CLOCK_DIVIDER_2) function of the http://www.airspayce.com/mikem/bcm2835/ library.
The next thing on my todo list is to look at the clk driver.

diff --git a/arch/arm/boot/dts/bcm2835-rpi-b-plus.dts b/arch/arm/boot/dts/bcm2835-rpi-b-plus.dts
index e479515..4a085bc 100644
--- a/arch/arm/boot/dts/bcm2835-rpi-b-plus.dts
+++ b/arch/arm/boot/dts/bcm2835-rpi-b-plus.dts
@@ -20,11 +20,16 @@
 };
 
 &gpio {
-       pinctrl-0 = <&gpioout &alt0 &i2s_alt0 &alt3>;
+       pinctrl-0 = <&gpioout &alt0 &i2s_alt0 &alt3 &pwm1_alt5>;
 
        /* I2S interface */
        i2s_alt0: i2s_alt0 {
                brcm,pins = <18 19 20 21>;
                brcm,function = <4>; /* alt0 */
        };
+
+       pwm1_alt5: pwm1_alt5 {
+               brcm,pins = <19>;
+               brcm,function = <2>;  /* alt5 */
+       };
 };

diff --git a/arch/arm/boot/dts/bcm2835-rpi-b.dts b/arch/arm/boot/dts/bcm2835-rpi-b.dts
index bafa46f..6ae6349 100644
--- a/arch/arm/boot/dts/bcm2835-rpi-b.dts
+++ b/arch/arm/boot/dts/bcm2835-rpi-b.dts
@@ -13,11 +13,16 @@
 };
 
 &gpio {
-       pinctrl-0 = <&gpioout &alt0 &i2s_alt2 &alt3>;
+       pinctrl-0 = <&gpioout &alt0 &i2s_alt2 &alt3 &pwm_alt5>;
 
        /* I2S interface */
        i2s_alt2: i2s_alt2 {
                brcm,pins = <28 29 30 31>;
                brcm,function = <6>; /* alt2 */
        };
+       
+       pwm_alt5: pwm_alt5 {
+               brcm,pins = <18>;
+               brcm,function = <2>;  /* alt5 */
+       };
 };

btanghe, 2015-01-06:
My previous comment is screwed, my apologies
```
diff --git a/arch/arm/boot/dts/bcm2835-rpi-b-plus.dts b/arch/arm/boot/dts/bcm2835-rpi-b-plus.dts
index e479515..4a085bc 100644
--- a/arch/arm/boot/dts/bcm2835-rpi-b-plus.dts
+++ b/arch/arm/boot/dts/bcm2835-rpi-b-plus.dts
@@ -20,11 +20,16 @@
 };
 
 &gpio {
-       pinctrl-0 = <&gpioout &alt0 &i2s_alt0 &alt3>;
+       pinctrl-0 = <&gpioout &alt0 &i2s_alt0 &alt3 &pwm1_alt5>;
 
        /* I2S interface */
        i2s_alt0: i2s_alt0 {
                brcm,pins = <18 19 20 21>;
                brcm,function = <4>; /* alt0 */
        };
+
+       pwm1_alt5: pwm1_alt5 {
+               brcm,pins = <19>;
+               brcm,function = <2>;  /* alt5 */
+       };
 };

diff --git a/arch/arm/boot/dts/bcm2835-rpi-b.dts b/arch/arm/boot/dts/bcm2835-rpi-b.dts
index bafa46f..6ae6349 100644
--- a/arch/arm/boot/dts/bcm2835-rpi-b.dts
+++ b/arch/arm/boot/dts/bcm2835-rpi-b.dts
@@ -13,11 +13,16 @@
 };
 
 &gpio {
-       pinctrl-0 = <&gpioout &alt0 &i2s_alt2 &alt3>;
+       pinctrl-0 = <&gpioout &alt0 &i2s_alt2 &alt3 &pwm_alt5>;
 
        /* I2S interface */
        i2s_alt2: i2s_alt2 {
                brcm,pins = <28 29 30 31>;
                brcm,function = <6>; /* alt2 */
        };
+       
+       pwm_alt5: pwm_alt5 {
+               brcm,pins = <18>;
+               brcm,function = <2>;  /* alt5 */
+       };
 };
```

notro, 2015-01-06:
> The device tree description is send to the linux kernel mailing list https://lkml.org/lkml/2015/1/5/410

You have set ```status = "ok"```. Probably not what you want? ```status = "disabled"``` is used for all the other devices since it's the board DTS file that decides which devices to enable. And doesn't onboard audio break when using pwm?
You are also changing the clock naming convention in that file: ```pwm@3``` vs. ```clock@3```

btanghe, 2015-01-07:
I've corrected the comments in v2 (https://lkml.org/lkml/2015/1/7/219) Thanks.
Indeed, because of the shared (18 and 19 are also used for pcm) pins, you have to choose between audio and pwm.

gohai, 2015-08-16:
I am terribly interested to run the bcm2835-pwm module with downstream kernels. Anyone got this working already? (I just looked at the Pi's 4.1 kernel, where the module Kconfig still depends solely on ARCH_BCM2835.)

pelwell, 2015-08-17:
I've created a PR with the basic DTB support and a pair of overlays. It would be great if you could give them a try.

pelwell, 2015-08-17:
As the commit message and overlay README say, these overlays (like the driver) assume the clock has been configured some other way, either by some external program or by playing a sound over the analogue audio interface (which will leave the clock configured). Note that if you choose the latter method, this will leave the PWM functions enabled on the audio output, which a) may not work and b) may sound really bad. It would be safer to use raspi-gpio (or similar) to select a different function for pins 40 and 45.

btanghe, 2015-08-18:
The commit looks fine for me, you only have to worry about the clock-frequency parameter in the dts file. This frequency depends on the configuration of the pwm clock.
This configuration (as mentioned above) has to be done by a external program and is not enabled by default.

pelwell, 2015-08-18:
Thanks for the feedback - the commit has been merged.

Does that mean we can close this issue now?

btanghe, 2015-08-18:
No problem. 
You can definitely close the issue.

closed 2015-08-18 by btanghe

gohai, 2015-10-10:
@btanghe Now that Eric Anholt's clock rework for bcm2835 is about to get included into mainline, is there anything that can be done to your pwm driver to drop the requirement of using a user-space program to enable the (audio) clock? I'd *love* to be able to at least use the single PWM channel cleanly for some applications!

Link to to the (already acked) commit
https://github.com/anholt/linux/commit/0d56b97a140ba1d505564fad4c92e0f6a8d79718