raspberrypi/linux #757: 3.18 DT mode: no ACT led activity

notro, 2015-01-05:
In DT mode the led platform device is not registered, so no led blinking (non-DT mode works fine).
I tried adding the leds node from arch/arm/boot/dts/bcm2835-rpi-b.dts:
```
	leds {
		compatible = "gpio-leds";

		act {
			label = "ACT";
			gpios = <&gpio 16 1>;
			default-state = "keep";
			linux,default-trigger = "heartbeat";
		};
	};

```
but there is no blinking, even though it's registred:
```
$ cat /sys/class/leds/ACT/trigger
none mmc0 timer oneshot [heartbeat] backlight gpio cpu0 default-on
```

Has anyone looked into this?

Another thing, can we have LEDS_GPIO builtin? It's always loaded anyway and that would give the possibility of heartbeat blinking before rootfs is mounted.

popcornmix, 2015-01-06:
LEDS_GPIO=y I have done.
@pelwell may be able to help with leds.

notro, 2015-01-06:
@popcornmix Thanks.

PeterPablo, 2015-01-06:
Just for future reference, https://github.com/raspberrypi/linux/commit/a5dda238581142cee553307bf89c2d3ab8ee7dc7 is the commit of @popcornmix.

pelwell, 2015-01-06:
I got as far as reproducing the problem yesterday, and I will continue to work on it today.