raspberrypi/linux #758: 3.18.1 compile error using gcc 4.8

MrEngman, 2015-01-06:
Tried using gcc 4.8 to configure linux 3.18.1 to compile a wifi module and it fails with an error running command **make modules_prepare**

Made sd card with 3.18.1, git cloned linux and firmware ready to enable compiling modules.

Ran sudo apt-get update/upgrade and installed gcc-4.8 and g++-4.8. Configured to use gcc-4.8 etc using 

    sudo rm /usr/bin/gcc
    sudo ln -s /usr/bin/gcc-4.8 /usr/bin/gcc
    sudo rm /usr/bin/g++
    sudo ln -s /usr/bin/g++-4.8 /usr/bin/g++
    sudo rm /usr/bin/cpp
    sudo ln -s /usr/bin/cpp-4.8 /usr/bin/cpp
    sudo rm /usr/bin/gcov
    sudo ln -s /usr/bin/gcov-4.8 /usr/bin/gcov

Then tried 

    cd /hom/pi/src/linux
    make mrproper && zcat /proc/config.gz > .config && make modules_prepare

**make modules_prepare** failed with the following error.

     ....
      CC      kernel/bounds.s
      GEN     include/generated/bounds.h
      CC      arch/arm/kernel/asm-offsets.s
    arch/arm/kernel/asm-offsets.c:53:2: error: #error Your compiler is too buggy; it is known to miscompile kernels
     #error Your compiler is too buggy; it is known to miscompile kernels
      ^
    arch/arm/kernel/asm-offsets.c:54:2: error: #error and result in filesystem corruption and oopses.
     #error and result in filesystem corruption and oopses.
      ^
    Kbuild:81: recipe for target 'arch/arm/kernel/asm-offsets.s' failed
    make[1]: *** [arch/arm/kernel/asm-offsets.s] Error 1
    Makefile:980: recipe for target 'prepare0' failed
    make: *** [prepare0] Error 2

Anyone explain why this is happening? gcc reports version **gcc version 4.8.2 (Raspbian 4.8.2-21~rpi3rpi1)**. Kernel appears to be compiled with **gcc version 4.8.3** I then installed gcc-4.7 etc and it compiles OK.

popcornmix, 2015-01-06:
Kernel is cross-compiled with:
```
arm-linux-gnueabihf-gcc (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) 4.8.3 20140303 (prerelease)
```
Looks like >= 4.8.0 and < 4.8.3 is considered buggy by kernel.

This is really a raspbian issue. Did you install gcc-4.8 from raspian/jessie?

It should be 4.8.3-13 which is okay (https://packages.debian.org/jessie/gcc-4.8)

asb, 2015-01-06:
It's not an upstream Raspbian issue - it's an issue with the version we
(Raspberry Pi) are distributing. gcc-4.8 came in from the Collabora work on
Webkit and the new browser. Cross compiling the kernel really is the way we
recommend if possible.

popcornmix, 2015-01-06:
Can we bump gcc-4.8 to the jessie version (4.8.3-13)

MrEngman, 2015-01-06:
I installed gcc 4.8 using apt-get install which I assume is using standard sources.list included in 3.18.1 which is

    deb http://mirrordirector.raspbian.org/raspbian/ wheezy main contrib non-free rpi

Attempting to install 4.8.3-13 by manually downloading packages is a real pain as it requires a whole load of dependencies downloading as well.

Cross compiling is not possible for me but I am not interested in fully compiling the kernel but just setting it up for compiling particular wifi modules as per my original post.

I've managed to break the original 3.18.1 image, or at least apt-get by adding the following to the /etc/apt/sources.list

    deb http://ftp.de.debian.org/debian jessie main

so I'm in the process of regenerating my SD card and starting again.

MrEngman, 2015-01-07:
I'm finding it impossible to install gcc 4.8.3-13.

Installing the jessie 4.8.3-13 package requires a load of dependent packages and trying to install them is bringing up a variety of other issues such as, additional dependencies, installed packages it say cannot be removed or replaced. Then it suggested running something like apt-get autoremove and that started removing hundreds of packages, > 700, and that has killed my pi again. 

Any suggestions for installing gcc 4.8.3-13 and related packages on my Pi?

Off to re-install the 3.18.1+ image on my SD card again.

popcornmix, 2015-01-07:
It seemed to work okay for me following this:
http://somewideopenspace.wordpress.com/2014/02/28/gcc-4-8-on-raspberry-pi-wheezy/
However it easy to go wrong when mixing wheezy/jessie repos.

MrEngman, 2015-01-07:
And it worked for me as well. Thanks for the pointer. Much simpler than trying to install manually and it works as well!!

NicoHood, 2016-01-08:
I am trying to recompile the 4.4 kernel with my elementary os (based on ubuntu 14.04) with gcc 4.8.4 but I get the same error.

However on my raspberry jessie installation the compilation seems to work fluently (so far, its so slow).

I've read that you need 4.8.3 or higher, but isnt 4.8.4 good enough? It might be related to the 4.4 kernel. Raspbian has gcc 4.9.2 installed (wow!)

popcornmix, 2016-01-09:
@MrEngman I assume this is no longer an issue with jessie and can be close?

MrEngman, 2016-01-09:
I have no idea as I now cross-compile the drivers I compile on a PC as I found other issues as well, such as some drivers causing the kernel to crash when compiled on a Pi even when compiled with no warnings.

I cross-compile my drivers, including for 4.4.0-rc7/8, using the compilers from the raspberry pi tools repo and no long see any of the problems I did compiling on a Pi.

Fell free to close if you want to.

closed 2016-01-09 by popcornmix