raspberrypi/linux #762: Password visible when entered at command line prompt

danielgrimshaw, 2015-01-19:
Whenever the keycode ctrl, shift, 'u' is entered at a prompt for a password on the command line, then the characters a,d,c,b,f, and u are printed to the console when typed. They are printed underlined. The e key will also be printed when pressed, but stops all further characters from being printed. Pressing the enter key appears to clear the displayed password and type a newline character into the string, but not end the input command. This may be a problem with the linux kernel itself. I am running a fully updated version of rasbian on a revision 2 model B pi.

popcornmix, 2015-01-19:
I can't immediately reproduce this. Are you saying after boot and typing in username, that you then type ctrl-shift-u and then a and the a is visible?
Are you using a local (keyboard) console, or ssh/uart?
Do you get the same behaviour on another linux system (e.g. ubuntu on PC?)

Ferroin, 2015-01-19:
I can confirm that this IS NOT the case on a amd64 Gentoo Box running a 3.18 series kernel.  My first thought on this is some issue with either the keymaps or whatever systemd uses in place of getty.

danielgrimshaw, 2015-01-19:
This only seems to occur in the terminal on the x-server. I am using a local console.

Ferroin, 2015-01-19:
Then this sounds like a bug in whatever terminal emulator you're using, or possibly (less likely) in pty drivers.

popcornmix, 2015-01-20:
If it doesn't happen from initial (before X) console then I'm pretty sure it's not a kernel issue, but a bug in LXTerminal.

nadrimajstor, 2015-01-20:
IMHO, it is by design intended behavior. Ctrl+Shift+u followed by a appropriate hex code is a way to input unicode characters in GUIs.

closed 2015-11-14 by danielgrimshaw