raspberrypi/linux #763: Compile error on 3.12 with official toolchain compiler

judilsteve, 2015-01-20:
Please forgive me if this is a user error as this is my first time compiling a kernel, but I'm attempting to cross-compile the 3.12 branch using the compiler in the toolchain found at raspberrypi/tools using my laptop which is running LMDE. Here are the steps that I have taken:

git clone https://github.com/raspberrypi/linux.git --depth 1
cd linux
cp arch/arm/configs/bcmrpi_defconfig .config
make menuconfig (unticked x64 and saved)
make-kpkg clean
export CONCURRENCY_LEVEL=3
export ARCH=arm
export CROSS_COMPILE=/home/james/kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf
sudo fakeroot make-kpkg --initrd kernel_image kernel_headers

The compile fails at drivers/char/broadcom/vc_sm/vmcs_sm.o with output as in the following pastebin:
pastebin.com/p31ku4nz

pelwell, 2015-01-20:
It looks as though the wrong compiler is being used. Have you tried using just make instead of make-kpkg (using the environment variables you have already set)?:
```
make bcmrpi_defconfig zImage
```

pelwell, 2015-01-20:
I note that make-kpkg has the command-line options --arch and --cross-compile, and that the documentation for --arch says:
> The same effect  can  be  achieved  by setting the environment variable KPKG_ARCH.
which makes me thing that your ARCH setting may be ignored. Assuming that plain "make" works, you might want to try different combinations of environment variables and command-line options.

judilsteve, 2015-01-20:
Thanks for your quick and helpful reply. Turns out my CROSS_COMPILE
variable was incorrect. Changing it to
'/home/james/kernel/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/arm-linux-gnueabihf-
resulted in a successful compilation via plain make. Now I just have to go
read the manpage for make-kpkg to figure out which parameters I need to
emulate the CROSS_COMPILE and ARCH variables. Thanks very much once again
for all your help, you can go ahead and close this ticket.

pelwell, 2015-01-20:
You're welcome.

closed 2015-01-20 by pelwell