raspberrypi/linux #764: rtl8188eu wifi driver requires seperate firmware file.

MrEngman, 2015-01-21:
Just updated my pi from 3.12.36+ to 3.18.3+ with rpi-update. Rebooted and the kernel appeared to hang. Looks like a problem with the rtl8188eu wifi driver.

    [    6.834444] r8188eu: module is from the staging directory, the quality is unknown, you have been warned.
    [    7.522870] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_A_CUT_1T1R_RomVer(0)
    [    7.844997] random: nonblocking pool is initialized
    [    8.002964] usbcore: registered new interface driver r8188eu
    [   12.342025] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
    [   12.881522] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
    [   22.879338] r8188eu 1-1.2:1.0: Direct firmware load for rtlwifi/rtl8188eufw.bin failed with error -2
    [   22.879388] r8188eu 1-1.2:1.0: Firmware rtlwifi/rtl8188eufw.bin not available

The rtl8188eu wifi driver in the drivers/staging/rtl8188eu is compiled in and the module is included in the 3.18.3+ image. However, it does not work as it requires a separate firmware file rtl8188eufw.bin in directory /lib/firmware/rtlwifi and the firmware binary is not included in the image.

A good example why the firmware should not have been stripped out of the source code and moved to a separate file.

Please add in the firmware file rtl8188eufw.bin

popcornmix, 2015-01-21:
Can you point me at where rtl8188eufw.bin should come from?

MrEngman, 2015-01-22:
You can find it here. https://github.com/lwfinger/rtl8188eu

popcornmix, 2015-01-22:
@asb what's your view? Should kernel driver firmware be distributed with rpi-update?

notro, 2015-01-22:
> A good example why the firmware should not have been stripped out of the source code and moved to a separate file.

It seems that firmware isn't allowed in the kernel source anymore:
https://github.com/lwfinger/rtl8188eu/issues/50#issuecomment-39025676
http://git.kernel.org/cgit/linux/kernel/git/firmware/linux-firmware.git/tree/rtlwifi

PeterPablo, 2015-01-22:
After some digging I found this [commit](https://git.kernel.org/cgit/linux/kernel/git/firmware/linux-firmware.git/commit/?id=ffc47f18780aeda6c15108d1765e36a543ee4cc4) to `linux-firmware` at `kernel.org`. That should be it?! (I am wet behind the ears concerning the kernel, so please bear with me.)

zaazbb, 2015-02-06:
does anybody build a rtl8188eu.ko for rapi-3.18.5, can you share it to me? 
thanks.

PeterPablo, 2015-02-06:
When I upgraded to 3.18.x two days ago the firmware was included and everything worked without manual interaction. What kind of problems do you face?

dinau, 2015-02-07:
Hi @zaazbb,

Try this,

    sudo wget https://github.com/lwfinger/rtl8188eu/raw/c83976d1dfb4793893158461430261562b3a5bf0/rtl8188eufw.bin  -O /lib/firmware/rtlwifi/rtl8188eufw.bin
then do reboot.
As @MrEngman says above,
Kernel 3.18.5+ #748 dose not have 'rtl8188eufw.bin' file too.

organom, 2015-02-12:
Thanks zaazbb, it worked perfectly ;)

matthuisman, 2015-03-12:
Also experienced this problem and dinau's fix worked.
Will this be fixed in upcoming versions?

mrkoz, 2015-03-12:
Spent an hour looking for this very solution, thanks @dinau you rock.

Diaoul, 2015-03-22:
Any updates on this? Having to manually download the driver is a real pain.

XECDesign, 2015-03-22:
Really, this is a problem that will resolve itself when we move to Jessie. In the meantime, it would be nice if the relevant package was backported, but I don't think it will happen in time.

jvinterberg, 2015-03-24:
@Dinau thank you very much for that oneliner! saved me from going out to buy yet another 3 dongles.

appy10, 2015-04-10:
but when we transfer the bin file into lib/firmware/rtlwifi,the access is denied,so how to transfer it...

organom, 2015-04-10:
@appy10 you have to use the sudo has described. If you don't have sudo installed/configured, you can login as root or change user to root as: 'su root' and then run the command 'wget https://github.com/lwfinger/rtl8188eu/raw/c83976d1dfb4793893158461430261562b3a5bf0/rtl8188eufw.bin  -O /lib/firmware/rtlwifi/rtl8188eufw.bin' (without the sudo)

appy10, 2015-04-10:
thnks...the command worked with sudo..

Nagadhiraj, 2015-04-10:
after transfering  rtl8188eufw.bin  to /lib/firmware/rtlwifi ....but still not working. When i check by using dmesg  : it says  : 
 r8188 1-1.4:1.0: firmware, attempted to load /lib/firmware/rtlwifi/rtl8188eufw.bin, but failed with error -22
 r8188 1-1.4:1.0: Direct firmware load for /rtlwifi/rtl8188eufw.bin failed with error -22

blachanc, 2015-04-25:
@ dinau 
your solution Did the job for me too.
Thanks  a lot !!
Dinau's solution:
sudo wget https://github.com/lwfinger/rtl8188eu/raw/c83976d1dfb4793893158461430261562b3a5bf0/rtl8188eufw.bin  -O /lib/firmware/rtlwifi/rtl8188eufw.bin

Ben

kumar98, 2015-05-04:
hai Dinau...Thasnks a lot....u r superb man.. It gave me a relax now...

mator, 2015-07-30:
Can you please add this firmware file to firmware package? Thanks!

XECDesign, 2015-07-30:
Done, please test.

closed 2015-08-01 by XECDesign

neko1984, 2015-08-29:
@Dinau Many thanks. It works like a charm on 4.1.6+ kernel :)

mator, 2015-08-30:
# pkgfile -s rtl8188eufw.bin
core/linux-firmware
aur/dkms-8188eu

thanks

tradback, 2015-09-30:
@dinau Many many thanks!  I've been looking for this solution for about 2 weeks now.  It also saved me another wifi usb purchase.  I had been sure it would work as the wifi-usb I had was working with Openelec and Raspbian (after update/upgrade).  But it wouldn't work with volumio.  But I tried this, and damn it works!  Thanks again!

mrtipale, 2016-07-03:
Worked for me. Many thanks to @dinau ! I had almost given hopes on that wifi dongle. :)

amateusz, 2016-10-24:
@dinau's solution worked for me.
FYI: armbian upgrade (4.7.3 -> 4.8.4) broke it I suppose. (I'm on orange pi a20)
strange enough "rtlwifi" dir wasn't even present..