raspberrypi/linux #766: SIGILL on using GDB, in ld-linux-armhf.so.3

sneakernets, 2015-01-23:
Disclaimer: I'm using the latest firmware...

I am working on [zdoom](https://github.com/rheit/zdoom), a Doom source port. However, when I attach zdoom to GDB on the Raspberry Pi, I get this:

```
Program received signal SIGILL, Illegal instruction.
0xb6fe8acc in ?? () from /lib/ld-linux-armhf.so.3
```

`bt` produces nothing of worth.

This is interesting, in that the program "runs" fine otherwise, but not with `gdb`.

Compiler: gcc 4.6.3-14+rpi1
Linux: 3.18.3+

popcornmix, 2015-01-23:
I believe this is normal (I've seen this when debugging xbmc for a couple of years).
I suspect the code is deliberately executing the illegal instruction with a signal handler enabled (e.g. to determine whether optional instructions are supported).
See:
http://stackoverflow.com/questions/15071625/gdb-ddd-program-received-signal-sigill

sneakernets, 2015-01-23:
I actually checked for this too, but even when I use  `handle SIGILL pass nostop noprint`, the program dies in `gdb` with the same error. I suppose I should have noted that in the first post. 

This has been hair-pullingly frustrating to figure out on my own for the past two weeks, so I threw in the towel today and submitted it as an issue. :grin:

nbrachet, 2015-01-30:
I have experienced this issue too after a recent upgrade -- sadly I don't remember which one.

I traced this SIGILL down to a static initialization in my code. Still frustrating issue.

popcornmix, 2015-01-31:
@nbrachet A firmware/kernel upgrade, or other packages?

nbrachet, 2015-01-31:
@popcornmix I'm afraid both.

nbrachet, 2015-01-31:
If that helps the problem occurs whether I build with g++-4.6 or g++-4.8.

popcornmix, 2015-01-31:
If it was a firmware/kernel update that caused it, then downgrading should fix it. You can revert to an older version:
http://raspberrypi.stackexchange.com/questions/15917/restore-kernel-version/15919#15919

sneakernets, 2015-02-02:
If it helps, this started to be an issue around mid-december 2014.

nbrachet, 2015-02-06:
I've tracked down the firmware version in which this bug appeared: [cad071a](https://github.com/Hexxeh/rpi-firmware/commit/cad071ac50ddf8a9f3c7a5d808cfa0daf5effb45), and it present up to the latest [93cb9a4](https://github.com/Hexxeh/rpi-firmware/commit/93cb9a45741681987b207c68b63db7ab72b2a4cf)

sneakernets, 2015-02-06:
Yeah, it looks like 3.18 firmware update still has some rough edges. Thanks for letting me know!

popcornmix, 2015-02-07:
Unfortunately the kernel change was the 3.12 to 3.18 (so literally tens of thousands of commits...)
Can you determine the instruction generating the exception (e.g. with "disas")

popcornmix, 2015-02-07:
One thing I've notice is 3.18 kernel reports:
Features	: half thumb fastmult vfp edsp java tls 

The older kernel reports:
Features	: swp half thumb fastmult vfp edsp java tls 

The (deprecated) swp feature was removed here: 
https://github.com/raspberrypi/linux/commit/58171bf2af6b547a560b304f6ab2b9edf1c31d5a

So if the instruction is a swp instruction this may be relevant.

nbrachet, 2015-02-09:
Here's the disas:

```
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y

Starting program: test

Program received signal SIGILL, Illegal instruction.
0xb6fe8acc in ?? () from /lib/ld-linux-armhf.so.3
(gdb) disas 0xb6fe8acc,0xb6fe8aff
Dump of assembler code from 0xb6fe8acc to 0xb6fe8aff:
=> 0xb6fe8acc:	ldr	r3, [pc, #164]	; 0xb6fe8b78
   0xb6fe8ad0:	cmp	r1, #0
   0xb6fe8ad4:	push	{r4}		; (str r4, [sp, #-4]!)
   0xb6fe8ad8:	add	r3, pc, r3
   0xb6fe8adc:	mov	r2, r0
   0xb6fe8ae0:	beq	0xb6fe8b5c
   0xb6fe8ae4:	mov	r0, #76	; 0x4c
   0xb6fe8ae8:	ldr	r12, [pc, #140]	; 0xb6fe8b7c
   0xb6fe8aec:	mul	r0, r0, r1
   0xb6fe8af0:	add	r12, pc, r12
   0xb6fe8af4:	add	r0, r0, #56	; 0x38
   0xb6fe8af8:	add	r0, r0, r12
   0xb6fe8afc:	adds	r12, r2, #0
End of assembler dump.
```

popcornmix, 2015-02-09:
Can you post the binary that does this? (e.g. to dropbox or google drive)

nbrachet, 2015-02-09:
https://dl.dropboxusercontent.com/u/1020551/rtpraspivid

Offending code is:
```
    int traceroute(...)
    {
#ifndef __ARM_EABI__
        static const uint16_t pid = getpid();
#else
        // gdb (raspberrypi) trips with SIGILL!?!
        // see https://github.com/raspberrypi/linux/issues/766
               const uint16_t pid = getpid();
#endif
...
```
When pid is defined as a static (as it is in the executable I posted to dropbox) the executable SIGILL in gdb.

Note that the same problem happens if I replace the call to getpid() with a call to:
```
static uint16_t foo()
{
    return 1234;
}
```
But it does not happen if I replace the call to getpid() with a constant:
```
    static const uint16_t pid = 1234;
```

It does feel like a stack corruption... Could gdb be corrupting the stack somehow?

popcornmix, 2015-02-09:
I suspect that you do have a memory corruption issue in your code.
You've basically got undefined behaviour, and gdb just happens to cause the memory layout to fail in a different way.

valgrind may help identify memory corruption issues in your code.

I assume your code is based on RaspiVid? Can you identify the exact change you did that provoked the crash? Was it linking with a new library or different build flags? Was it adding a new line of code?

nbrachet, 2015-02-09:
That was my thinking too... until @sneakernets reported the exact same issue. Down to the same address where the crash occurs.

I did try valgrind but it is useless on the pi: http://stackoverflow.com/questions/20066215/valgrind-unrecognizes-memcmp-instruction-in-raspberry-pi.

The only change I can pinpoint was upgrading the firmware.

popcornmix, 2015-02-09:
The memcmp issue with valgrind is easily fixed. See comment on stackoverflow:

>And ok, it's easier than that. /usr/lib/arm-linux-gnueabihf/libcofi_rpi.so is itself preloaded. So just edit /etc/ld.so.preload on the raspberry, and comment out the line, which should give you the memcmp from glibc instead

nbrachet, 2015-02-09:
:flushed: Will try...

nbrachet, 2015-02-09:
valgrind output is clean:

```
==8094== Memcheck, a memory error detector
==8094== Copyright (C) 2002-2011, and GNU GPL'd, by Julian Seward et al.
==8094== Using Valgrind-3.7.0 and LibVEX; rerun with -h for copyright info
==8094== Command: ../../bin/rtpraspivid
==8094== 
==8094== Warning: noted but unhandled ioctl 0xc410 with no size/direction hints
==8094==    This could cause spurious value errors to appear.
==8094==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==8094== Warning: noted but unhandled ioctl 0xc400 with no size/direction hints
==8094==    This could cause spurious value errors to appear.
==8094==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==8094== Warning: noted but unhandled ioctl 0xc40d with no size/direction hints
==8094==    This could cause spurious value errors to appear.
==8094==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.

rtpraspivid Camera App v1.3.12-nick
...
```

jessedog3, 2015-02-10:
I had the same problem (down to the address of the illegal instruction), but running the program then attaching gdb worked.

Running [this](http://byuu.org/emulation/higan/) with the diff [here](http://board.byuu.org/phpbb3/viewtopic.php?f=4&t=297)

There is no stack corruption before loading a rom, at which point things break (hence gdb), but gdb dies before even creating windows.

jvcleave, 2015-02-21:
I am getting the same behavior with an App compiled on the RPI2

What I find weird is that I can run the same binary compiled on an RPI - compiling it on the RPI2 (g++ 4.8) and 
```
Program received signal SIGILL, Illegal instruction.
0x76fe8acc in ?? () from /lib/ld-linux-armhf.so.3
```

RizzoRat, 2015-03-02:
Issue confirmed after dist-upgrade to 3.18.7
SIGILL in ld-linux-armhf.so.3  + 0xf30c
Didn't trace much yet, but a simple "hello world" does the job though

RizzoRat, 2015-03-02:
did a damn lot of kernel reverts/updates. ( though it's pretty clear I just state it is not directly related to ld-2.13.so (ld-linux-armhf.so.3) itself, which apparently did not change)

3.12.36 still works, issue definetely got introduced with 3.18 (as of 20th Jan) - unfortunately (lots of diffs. sidenote: I  recompiled test project after each rbpi-update to make sure)

nbrachet, 2015-03-04:
@popcornmix anything I can do to help track this problem down? It is getting very frustrating...
I updated the code and it brought another example of this problem with a "simple":
```
            static int64_t last_second = -1;
```

hogliux, 2015-03-17:
I'm seeing the same problem. Attaching after the process has started doesn't help as I get the same illegal instruction after my program starts any thread.

khuey, 2015-03-26:
This is caused by gdb attempting to use an OABI syscall instruction to invoke the __ARM_NR_breakpoint syscall.  The Raspberry Pi kernel is built without CONFIG_OABI_COMPAT, so the kernel only looks at r7 for the syscall number and throws a SIGILL because the syscall # is invalid.  gdb has some logic to determine when to use a newer style breakpoint (a special illegal instruction).  I didn't look at why that is failing.  But if you rebuild the kernel with CONFIG_OABI_COMPAT=y gdb will work.

popcornmix, 2015-03-26:
Thanks @khuey that sounds plausible. I'll look into this tomorrow.

khuey, 2015-03-26:
I didn't check if this is what is tripping gdb up, but despite the name, readelf says that ld-2.13.so is not built with hard float enabled.

popcornmix, 2015-03-27:
@khuey do we also want CONFIG_FPE_NWFPE/CONFIG_FPE_FASTFPE?

nbrachet, 2015-03-27:
@khuey Thank you for the tip! Why would this trip gdb only in certain cases though? I can use gdb. It only crashes for some executables.

khuey, 2015-03-27:
Unfortunately I don't know.  It's not needed to run gdb, certainly.

khuey, 2015-03-27:
@nbrachet Good question!  If you strace a gdb session do you see a PTRACE_POKEDATA/PTRACE_POKETEXT call with 0xef9f0001 as an argument?  What is the output of running readelf -hV on your ld-linux-armhf.so.3?

nbrachet, 2015-03-27:
No PTRACE_POKEDATA/PTRACE_POKETEXT with -xef9f0001:
```
nick@raspberrypib:~ $ strace -o x gdb userland.git/build/bin/raspivid
GNU gdb (GDB) 7.4.1-debian
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /home/nick/userland.git/build/bin/raspivid...done.
(gdb) run
...
nick@raspberrypib:~ $ grep POKE x | grep 0xef9f0001
nick@raspberrypib:~ $ 
```

Many PTRACE_POKEDATA but none with 0xef9f0001:
```
nick@raspberrypib:~ $ grep POKE x
ptrace(PTRACE_POKEDATA, 3949, 0xb6fe8ac8, 0xe7f001f0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6fe8ac8, 0xe12fff1e) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6fdbc80, 0xe7f001f0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6fdbc80, 0xe597313c) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6fe8ac8, 0xe7f001f0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6ecc19c, 0x180) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6ecc1a0, 0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eca184, 0x100) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eca184, 0x100) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eca184, 0x100) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eca184, 0x100) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6fe8ac8, 0xe12fff1e) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6fdc65c, 0xe7f001f0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eca184, 0x100) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6fdc65c, 0xeb00190f) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eb36b4, 0xe7f001f0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eb36b8, 0xe7f001f0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6fe8ac8, 0xe7f001f0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6ff863c, 0x10101) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eca184, 0x100) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6d2c67c, 0x1000100) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eca184, 0x100) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6d2c6a4, 0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6d2c6a8, 0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6d2c6ac, 0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6d2c6b0, 0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6ecc1a4, 0) = 0
ptrace(PTRACE_POKEDATA, 3952, 0xb6eb36b4, 0xe12fff1e) = 0
ptrace(PTRACE_POKEDATA, 3952, 0xb6eb36b8, 0xe12fff1e) = 0
ptrace(PTRACE_POKEDATA, 3952, 0xb6fe8ac8, 0xe12fff1e) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eb56cc, 0xe7f001f0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eb56cc, 0xe2840f85) = 0
ptrace(PTRACE_POKEDATA, 3952, 0xb6eb36b4, 0xe7f001f0) = 0
ptrace(PTRACE_POKEDATA, 3952, 0xb6eb36b8, 0xe7f001f0) = 0
ptrace(PTRACE_POKEDATA, 3952, 0xb6fe8ac8, 0xe7f001f0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb652c6a4, 0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb652c6a8, 0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb652c6ac, 0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb652c6b0, 0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6ecc1a4, 0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb652c67c, 0x1000100) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eca184, 0x100) = 0
ptrace(PTRACE_POKEDATA, 3953, 0xb6eb36b4, 0xe12fff1e) = 0
ptrace(PTRACE_POKEDATA, 3953, 0xb6eb36b8, 0xe12fff1e) = 0
ptrace(PTRACE_POKEDATA, 3953, 0xb6fe8ac8, 0xe12fff1e) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eb56cc, 0xe7f001f0) = 0
ptrace(PTRACE_POKEDATA, 3949, 0xb6eb56cc, 0xe2840f85) = 0
ptrace(PTRACE_POKEDATA, 3953, 0xb6eb36b4, 0xe7f001f0) = 0
[... 49 lines cut ...]
  Flags:                             0x5000002, has entry point, Version5 EABI
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         7
  Size of section headers:           40 (bytes)
  Number of section headers:         23
  Section header string table index: 22

Version symbols section '.gnu.version' contains 28 entries:
 Addr: 0000000000000618  Offset: 0x000618  Link: 4 (.dynsym)
  000:   0 (*local*)       0 (*local*)       0 (*local*)       3 (GLIBC_PRIVATE)
  004:   3 (GLIBC_PRIVATE)   3 (GLIBC_PRIVATE)   2 (GLIBC_2.4)     2 (GLIBC_2.4)  
  008:   2 (GLIBC_2.4)     3 (GLIBC_PRIVATE)   2 (GLIBC_2.4)     2 (GLIBC_2.4)  
  00c:   2 (GLIBC_2.4)     3 (GLIBC_PRIVATE)   2 (GLIBC_2.4)     3 (GLIBC_PRIVATE)
  010:   2 (GLIBC_2.4)     3 (GLIBC_PRIVATE)   3 (GLIBC_PRIVATE)   3 (GLIBC_PRIVATE)
  014:   2 (GLIBC_2.4)     3 (GLIBC_PRIVATE)   2 (GLIBC_2.4)     3 (GLIBC_PRIVATE)
  018:   3 (GLIBC_PRIVATE)   2 (GLIBC_2.4)     3 (GLIBC_PRIVATE)   3 (GLIBC_PRIVATE)

Version definition section '.gnu.version_d' contains 3 entries:
  Addr: 0x0000000000000650  Offset: 0x000650  Link: 5 (.dynstr)
  000000: Rev: 1  Flags: BASE   Index: 1  Cnt: 1  Name: ld-linux-armhf.so.3
  0x001c: Rev: 1  Flags: none  Index: 2  Cnt: 1  Name: GLIBC_2.4
  0x0038: Rev: 1  Flags: none  Index: 3  Cnt: 2  Name: GLIBC_PRIVATE
  0x0054: Parent 1: GLIBC_2.4
```

khuey, 2015-03-27:
@nbrachet so after playing around for a while I noticed that if OS/ABI (in the readelf -h output) says "UNIX - System V" gdb works, but if it says "UNIX - GNU" gdb gets the SIGILL.  Does that match what you see?  I'll try to trace through gdb later today.

khuey, 2015-03-27:
Yeah, that's what it is.  gdb has logic to pick the EABI out of the binary, that runs only if the OS/ABI is ELFOSABI_NONE.

https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;a=blob;f=gdb/arm-tdep.c;h=630a2077d9cb4bfdc101061f36de3b51afd7ca13;hb=HEAD#l9956

khuey, 2015-03-27:
I submitted a patch to gdb.

https://sourceware.org/ml/gdb-patches/2015-03/msg00925.html

sneakernets, 2015-03-30:
@khuey : thanks for tracking this down!

popcornmix, 2015-04-02:
Latest rpi-update firmware has CONFIG_OABI_COMPAT. Please test.

nbrachet, 2015-04-03:
Seems to work for me in:
```
$ uname -a
Linux raspberrypi 3.18.10+ #774 PREEMPT Wed Mar 25 13:58:34 GMT 2015 armv6l GNU/Linux
$ cat /boot/.firmware_revision 
40185a95ac04ffcec406d9e1ef934406d7221939
```

Thanks!

popcornmix, 2015-04-03:
@sneakernets are you happy with this now?

sneakernets, 2015-04-06:
Looks like it's working for me. Thanks!

closed 2015-04-06 by popcornmix

popcornmix, 2016-08-31:
@khuey do you know if your patch has made it into released gdb?
Is it in raspbian jessie? Is CONFIG_OABI_COMPAT currently required?

khuey, 2016-08-31:
It did make in into gdb mainline https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;a=commitdiff;h=d403db2720ef6ac091dd1c0101ffc60242199528 and the 7.10 release.

I don't know if it's in raspbian jessie.  Debian jessie is still on 7.7.1.

If raspbian is not shipping 7.10 or later, and has not manually taken my patch, then taking raspberrypi/firmware#652 will regress this.

popcornmix, 2016-08-31:
@khuey Thanks for checking. Hopefully it will make it to stable at some point.

XECDesign, 2016-08-31:
@popcornmix debian is not likely to bump gdb version in stable. Any reason not to patch 7.7.1 in our releases?

popcornmix, 2016-09-01:
@XECDesign yes that is an option.

@Ferroin mentioned requesting this patch be applied to debian as it seems like a bug fix:
See: https://github.com/raspberrypi/firmware/issues/652

If that happens we should get the patched version by default, but if not we can add a custom patch.

Ferroin, 2016-09-01:
Short term, I'd suggest just doing a custom patch.  I have not yet had a chance to file a bug against Debian (things kind of blew up at work shortly after I posted that comment), and it may take a while to convince them to include it, if they decide to include it at all.

plugwash, 2016-09-01:
Shiftplusone just pointed me at this. So far I have failed to reproduce it. Specificafically I have failed to get the compiler to spit out a "UNIX - GNU" binary.

XECDesign, 2016-09-02:
@plugwash Just gave this a go myself. Compiled a kernel with CONFIG_OABI_COMPAT=n and a version of GDB with the patch applied. Using zdoom linked earlier as an example, without installing that version of GDB:
```
pi@raspberrypi:~/zdoom/build $ gdb zdoom
GNU gdb (Raspbian 7.7.1+dfsg-5) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from zdoom...done.
(gdb) run
Starting program: /home/pi/zdoom/build/zdoom 

Program received signal SIGILL, Illegal instruction.
_dl_debug_initialize (ldbase=1685382482, ns=195904) at dl-debug.c:47
47	dl-debug.c: No such file or directory.
(gdb) 
```
Installing the patched version of GDB and trying again...
```
pi@raspberrypi:~/zdoom/build $ sudo dpkg -i ../../dev/gdb_7.7.1+dfsg-5+rpi1_armhf.deb 
(Reading database ... 70074 files and directories currently installed.)
Preparing to unpack .../gdb_7.7.1+dfsg-5+rpi1_armhf.deb ...
Unpacking gdb (7.7.1+dfsg-5+rpi1) over (7.7.1+dfsg-5) ...
Setting up gdb (7.7.1+dfsg-5+rpi1) ...
Processing triggers for man-db (2.7.0.2-5) ...

pi@raspberrypi:~/zdoom/build $ gdb zdoom
GNU gdb (Raspbian 7.7.1+dfsg-5+rpi1) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from zdoom...done.
(gdb) run
Starting program: /home/pi/zdoom/build/zdoom 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
ZDoom 2.9pre-1230-gbf5f10a - 2016-09-02 10:48:38 +0200 - SDL version
Compiled on Sep  2 2016

M_LoadDefaults: Load system defaults.
Cannot find a game IWAD (doom.wad, doom2.wad, heretic.wad, etc.).
Did you install ZDoom properly? You can do either of the following:

1. Place one or more of these wads in ~/.config/zdoom/.
2. Edit your ~/.config/zdoom/zdoom.ini and add the directories of your
iwads to the list beneath [IWADSearch.Directories]
[Inferior 1 (process 13140) exited with code 0377]
(gdb)
```

plugwash, 2016-09-03:
did you compile that zdoom binary yourself? if so with what compiler on what version of raspbian?

XECDesign, 2016-09-03:
Yes, straight forward compile using the toolchain from raspbian jessie.
```
sudo apt-get install libsdl2-dev
git clone https://github.com/rheit/zdoom.git
cd zdoom/
mkdir build
cd build/
cmake ../
make -j4
```
```
pi@raspberrypi:~/zdoom/build $ apt-cache policy gcc
gcc:
  Installed: 4:4.9.2-2
  Candidate: 4:4.9.2-2
  Version table:
 *** 4:4.9.2-2 0
        500 http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status
pi@raspberrypi:~/zdoom/build $ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 4.9.2-10' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.9.2 (Raspbian 4.9.2-10) 
```

frixhub, 2016-11-05:
Good day,

I'm developing on the Allwinner A20 & A33 CPUs, which contains ARM cores.
I've first noticed the segfault during remote debugging, using gdbserver,
from my host PC to the target. Debugging the the application directly on
the target did not cause this problem. Performing the remote debugging from
a 32-bit host to the 32-bit target made no difference. That seemed to 
eliminate the possibility of a 64/32 bit compatibility issue.

My setup:
Host PC:
  Kubuntu 16.04 64-bit
  Toolchain: gcc-linaro-6.1.1-2016.08
           : gdb version (Linaro_GDB-2016.08) 7.11.1.20160801-git
            Obtained from: https://releases.linaro.org/components/toolchain/binaries          
Target:
  Ubuntu 16.04 ARMHF
  : GDB version: (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1

Illustrating the problem
------------------------
Program: main.c
  #include <stdio.h>
  int main (void)
  {
    printf("Hallo world!\n");
    return 0;
  }
Compile on host PC:
  $ arm-linux-gnueabihf-gcc -g main.c -o main_32
File signature:
  $ file main_32
    main_32: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked,
          interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, 
          BuildID[sha1]=cc1a4a1ff17da63f9f8362c84dffe54ee7fff4ff, not stripped

Remote target:
The executable (main_32) needs to be copied to the remote target
$ gdbserver :5555 main_32  
  Process main_32 created; pid = 904
  Listening on port 5555
  Remote debugging from host 192.168.7.2

Host PC:
$ arm-linux-gnueabihf-gdb ./main_32
(gdb) target remote 192.168.7.1:5555 
  Remote debugging using 192.168.7.1:5555
  Reading /lib/ld-linux-armhf.so.3 from remote target...
  warning: File transfers from remote targets can be slow. Use "set sysroot" to access files locally 
                 instead.
  Reading /lib/ld-linux-armhf.so.3 from remote target...
  Reading symbols from target:/lib/ld-linux-armhf.so.3...Reading /lib/ld-2.23.so from remote target...
  Reading /lib/.debug/ld-2.23.so from remote target...
  (no debugging symbols found)...done.
  0xb6fd7a40 in ?? () from target:/lib/ld-linux-armhf.so.3

(gdb) continue
  Continuing.
  Program received signal SIGSEGV, Segmentation fault.
  0xb6fd9822 in ?? () from target:/lib/ld-linux-armhf.so.3

To try and figure out where the segfault occurred the debug symbols need to be loaded.
The gdb command 'set debug-file-directory' can be used to accomplish this. It seems that this command 
is only relative to the host PC filesystem and not according to the remote target file system.
With 'set sysroot' the debugger can also be instructed to load all data locally instead of transferring it
runtime over the network.

Make sure the debug symbols are on the remote file system:
# apt-get install libc6-dbg
Copy the full contents of the SD card residing on the Allwinner A20 rootfs to a local directory on
the host PC: /opt/gnueabihf_rootfs

Illustrating the solution
-------------------------
Remote target:
$ gdbserver :5555 main_32 
  Process main_32 created; pid = 912
  Listening on port 5555
  Remote debugging from host 192.168.7.2
  
  Hallo world!
  Child exited with status 0

Host PC:
$ arm-linux-gnueabihf-gdb ./main_32
  (gdb) set debug-file-directory /opt/gnueabihf_rootfs/usr/lib/debug
  (gdb) set sysroot /opt/gnueabihf_rootfs
  (gdb) target remote 192.168.7.1:5555

  Remote debugging using 192.168.7.1:5555
  Reading symbols from /opt/gnueabihf_rootfs/lib/ld-linux-armhf.so.3...done.
  Reading symbols from /opt/gnueabihf_rootfs/usr/lib/debug/lib/arm-linux-gnueabihf/ld-2.23.so...done.
  0xb6fd7a40 in _start ()
   
  (gdb) continue
  Continuing.
  Cannot parse expression `.L966 4@r4'.
  warning: Probes-based dynamic linker interface failed.
  Reverting to original interface.

Even though there is a warning regarding "Probes-based dynamic linker interface failed", the program
keeps on running correctly. The thread describing the GDB patch about the linker interface can be
read at: https://sourceware.org/ml/gdb-patches/2015-08/msg00629.html

It appears the work around was introduced around August 2015, but if you do not specify the path to
the debug symbols, the debugger still segfaults.

The 'set debug-file-directory' and 'set sysroot' can be added to the .gdbinit file to be applied
automatically when you start a remote debug session.

Note: Activating the kernel option: CONFIG_OABI_COMPAT  did not fix the problem on the Allwinner A20 or A33.

Regards,
Frikkie Thirion

masoudr, 2018-09-01:
Hi,
I'm still facing this problem can anybody confirm what is the cause of the problem? kernel itself? I'm trying to cross-compile my app with `linaro toolchain` and using its debugger `arm-linux-gnueabihf-gdb` to debug remotely.