raspberrypi/linux #770: pps gpio stopped working on latest update

jacs90, 2015-01-27:
Something has happened in the bcm2708 code somewhere between the following two versions

Was: Linux raspi-2 3.12.32+ #721 PREEMPT Fri Nov 7 16:50:31 GMT 2014 armv6l
Now: Linux raspi-2 3.18.3+ #742 PREEMPT Mon Jan 26 18:27:18 GMT 2015 armv6l

that is stopping pps_gpi working with bcm2708.pps_gpio_pin=18 set in /boot/cmdline.txt

I should be getting "bcm2708: GPIO 18 setup as pps-gpio device "  in the kernel dmesg but its no there anymore. I have check that pps_gpio is getting loaded

Thanks

Chris

pelwell, 2015-01-27:
If you haven't disabled device tree ("device_tree=" in your config.txt), and I hope you haven't, then you need to add "dtoverlay=pps-gpio" to your config.txt. You can then delete the cmdline option. See /boot/overlays/README for an explanation.

jacs90, 2015-01-27:
Many thanks for that.
 
/boot/overlays/README  is not currently download via  rpi-update.

pps seem to be working now but how its working out which pin  PPS  is on I have no idea.
Shouldn't there be something like

device_tree_param=pps_gpio_pin=18 in config.txt ?

Currently dmsg is giving
pps pps0: new PPS source pps.-1

which is the default  pin

Chris

pelwell, 2015-01-27:
The README (also available [here](http://www.raspberrypi.org/forums/viewtopic.php?p=675661#p675661)) explains that you can specify an explicit pin using:
```
dtoverlay=pps-gpio,gpiopin=18
```
but that 18 is the default.

I don't think that -1 is a pin value, it is a part of the device name, and using a parameterised overlay doesn't get you the "GPIO.18" name.

jacs90, 2015-01-27:
Thanks for your reply's all makes sense now.

closed 2015-01-27 by jacs90