raspberrypi/linux #771: 3.14 compilation fail on drivers/mmc/host/sdhci-bcm2708.c

peerchemist, 2015-01-28:
I have just encountered this error while compiling.

I guess it was introduced with this commit: https://github.com/raspberrypi/linux/commit/df259363827ca5dacf4802c5aaea43ed3f167fc9 as it did not happen before.

> drivers/mmc/host/sdhci-bcm2708.c: In function 'sdhci_bcm2708_probe':
drivers/mmc/host/sdhci-bcm2708.c:1187:36: error: assignment of member 'missing_status' in read-only object
   
> sdhci_bcm2708_ops.missing_status = sdhci_bcm2708_missing_status;
                                                          ^                                                    

> drivers/mmc/host/sdhci-bcm2708.c:1191:41: error: assignment of member 'spurious_crc_acmd51' in read-only object
   
> sdhci_bcm2708_ops.spurious_crc_acmd51 = sdhci_bcm2708_quirk_spurious_crc_acmd51;
                                                                      ^

 ^ points to = 
I have tried to keep formatting but failed.

peerchemist, 2015-01-30:
Can anyone replicate this?

popcornmix, 2015-01-31:
Nope.
```
git checkout rpi-3.14.y
ARCH=arm make bcmrpi_defconfig
ARCH=arm make
```
completed fine for me.

peerchemist, 2015-02-01:
Tested again, clean this time.
It works, so false alarm. I still have to find out why it did not work, when I do I will post it.

closed 2015-02-01 by peerchemist