raspberrypi/linux #772: use of spidev.h returns EINVAL

oklitz, 2015-01-28:
Hello !
I'm using arch linux: 3.18.3-3-ARCH #1 PREEMPT with newest wiringpi-git 69.0a9fdeb-1 packet from the AUR.
spi is enabled in config.txt
module spi_bcm2708 is loaded and devices are in /dev/spidev0.0 and /dev/spidev0.1

fd = wiringPiSPISetup (0, 500000);
if (fd == -1)
    errExit("wiringPiSPISetup");

--function call returns without errors

uint8_t buff = 0b11110000;
ret = wiringPiSPIDataRW (0, &buff, 1);
if (ret == -1)
    errExit("wiringPiSPIDataRW");
buff = 0b00001111;
ret = wiringPiSPIDataRW (0, &buff, 1);
if (ret == -1)
    errExit("wiringPiSPIDataRW");

the second function call returns -1 with [EINVAL Invalid argument]
the first byte is shifted out correctly (i have a logic analyser on the spi bus)

echo 123 > /dev/spidev0.0 works fine

This code worked with older kernels, but after the jump to 3.18 it doesn't.
But i can't say it exactly because i have used the libwiringPi the last time in last year.

I have tried to fill the struct spi_ioc_transfer from /usr/include/linux/spi/spidev.h
with valid data like the libwiringpi does but same result.

Do i make a mistake ? Does something changed in the kernel ?
Is somebody able to reproduce the error ?

regards Marco Oklitz

closed 2015-01-29 by oklitz

j-marjanovic, 2015-03-08:
Hi, were you able to resolve the problem? I am getting the same result (-EINVAL on ioctl) with the folowing code:

```C
void read_usedw(void){
	struct spi_ioc_transfer	xfer[1];
	char tx_buf[20];
	char rx_buf[20];
	int fd;
	int status;

	memset(rx_buf, 0, sizeof(rx_buf));

	xfer[0].tx_buf = tx_buf;
	xfer[0].rx_buf = rx_buf;
	xfer[0].len    = 1 + 8;

	fd = open("/dev/spidev0.0", O_RDWR);
	if (fd < 0) {
	        perror("open() failed");
                return;
	}
		
	status = ioctl(fd, SPI_IOC_MESSAGE(1), xfer);
	
	if (status < 1) {
		perror("ioctl(SPI_IOC_MESSAGE) failed");
		goto close_file;
	}

	printf("Resp: ");
	for(int i=0; i<9; i++){
		printf("%02x ", (unsigned char)rx_buf[i]);
	}

	printf("\n\n");

close_file:
	close(fd);
}
```

pelwell, 2015-03-08:
You need to update your Wiring Pi library, then it will work again (a kernel change makes it necessary to zero a datastructure before the ioctl).

j-marjanovic, 2015-03-08:
Yeah, after zeroing the spi_ioc_transfer structure it works. Thanks for the tip!