raspberrypi/linux #776: System freeze when sudo modprobe i2c_bcm2708

jacqu, 2015-01-31:
Bug appeared since last update (January 30th 2015). Not seen before.

I verified that the bug is not in:
pi@raspberrypi ~ $ uname -r
3.12.28+

Reproducible on various RPIs:

pi@raspberrypi ~ $ sudo rmmod i2c_bcm2708
pi@raspberrypi ~ $ sudo modprobe i2c_bcm2708
--> System freeze.

I2c has been configured at 400kHz:
pi@raspberrypi ~ $ cat /etc/modprobe.d/i2c.conf
options i2c_bcm2708 baudrate=400000

pi@raspberrypi ~ $ uname -a
Linux raspberrypi 3.12.35+ #730 PREEMPT Fri Dec 19 18:31:24 GMT 2014 armv6l GNU/Linux

pi@raspberrypi ~ $ dpkg -s libc6 | grep ^Version
Version: 2.13-38+rpi2+deb7u7

popcornmix, 2015-01-31:
I'm not seeing this:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 3.18.5+ #744 PREEMPT Fri Jan 30 18:19:07 GMT 2015 armv6l GNU/Linux
pi@raspberrypi:~ $ sudo rmmod i2c_bcm2708
Error: Module i2c_bcm2708 is not currently loaded
pi@raspberrypi:~ $ sudo modprobe i2c_bcm2708
pi@raspberrypi:~ $ lsmod
Module                  Size  Used by
i2c_bcm2708             6004  0 
joydev                  9766  0 
evdev                  11000  3 
uio_pdrv_genirq         3666  0 
uio                     9897  1 uio_pdrv_genirq
```
Any non-standard settings in cmdline.txt or config.txt?

pelwell, 2015-01-31:
You don't show a full load/unload/load cycle - have you tested that?

popcornmix, 2015-01-31:
Works for me
```
pi@raspberrypi:~ $ sudo modprobe i2c_bcm2708
pi@raspberrypi:~ $ sudo rmmod i2c_bcm2708
pi@raspberrypi:~ $ sudo modprobe i2c_bcm2708
pi@raspberrypi:~ $ sudo rmmod i2c_bcm2708
pi@raspberrypi:~ $ lsmod
Module                  Size  Used by
joydev                  9766  0 
evdev                  11000  3 
uio_pdrv_genirq         3666  0 
uio                     9897  1 uio_pdrv_genirq
pi@raspberrypi:~ $ sudo modprobe i2c_bcm2708
pi@raspberrypi:~ $ lsmod
Module                  Size  Used by
i2c_bcm2708             6004  0 
joydev                  9766  0 
evdev                  11000  3 
uio_pdrv_genirq         3666  0 
uio                     9897  1 uio_pdrv_genirq
```

jacqu, 2015-01-31:
Note that I can reproduce this bug on 3 different RPIs. I have a backup SD card with an older version of the kernel, 3.12.28+ instead of 3.12.35+. And with the same RPI, the SD with the older kernel doesn't have this bug.

On my system, the i2c module are loaded at boot time. For that, you have to issue these commands:
echo i2c-bcm2708 >> /etc/modules
echo i2c-dev >> /etc/modules

Also remove the blacklisting of the i2c-bcm2708 module:
pi@raspberrypi /etc/modprobe.d $ cat raspi-blacklist.conf 
# blacklist i2s, spi and i2c by default (many users don't need them)

blacklist spi-bcm2708
#blacklist i2c-bcm2708
blacklist snd_soc_bcm2708_i2s

Furthermore I run i2c at full speed (400kHz). To do so:
echo options i2c_bcm2708 baudrate=400000 > /etc/modprobe.d/i2c.conf

If I have some time, I will try a diff between the 3.12.28+ and 3.12.35+ source code of the i2c-bcm2708 driver.

jacqu, 2015-01-31:
FYI, this is the last modification of the drivers/i2c/busses/i2c-bcm2708.c file (tweaking the spinlocks may trigger system freeze):
@@ -265,10 +265,11 @@ static int bcm2708_i2c_master_xfer(struct i2c_adapter *adap,
bi->nmsgs = num;
bi->error = false;
- spin_unlock_irqrestore(&bi->lock, flags);
-
bcm2708_bsc_setup(bi);
+ /* unlockig _after_ the setup to avoid races with the interrupt routine */
+ spin_unlock_irqrestore(&bi->lock, flags);
+
ret = wait_for_completion_timeout(&bi->done,
msecs_to_jiffies(I2C_TIMEOUT_MS));
if (ret == 0) {

jacqu, 2015-02-02:
Good news: with today's kernel update, the problem disappeared .

root@raspberrypi:/etc/init.d# uname -a
Linux raspberrypi 3.18.5+ #744 PREEMPT Fri Jan 30 18:19:07 GMT 2015 armv6l GNU/Linux

popcornmix, 2015-02-02:
Cool. There was another kernel update today - can you check if that is still okay.

mw46d, 2015-02-05:
Somewhat related:-( It looks like I lost my i2c bus completely with the upgrade to 3.18.5+:-(
root@issabove:/boot# uname -a
Linux issabove 3.18.5+ #744 PREEMPT Fri Jan 30 18:19:07 GMT 2015 armv6l GNU/Linux
root@issabove:/boot# i2cdetect -l
root@issabove:/boot#
that worked before the upgrade:-(

The i2c modules are loaded:
root@issabove:/boot# lsmod | fgrep i2c
i2c_bcm2708             6004  0 
i2c_dev                 6709  0 

But the spi- module does not want to load:-(
root@issabove:/boot# modprobe -v spi-bcm2708
insmod /lib/modules/3.18.5+/kernel/drivers/spi/spi-bcm2708.ko 
ERROR: could not insert 'spi_bcm2708': No such device

My last working kernel was:
Linux version 3.12.29+  #714 PREEMPT Wed Oct 1 23:11:38 BST 2014

mw46d, 2015-02-05:
same problem still with
pi@issabove ~ $ uname -a
Linux issabove 3.18.5+ #748 PREEMPT Wed Feb 4 21:24:41 GMT 2015 armv6l GNU/Linux

pelwell, 2015-02-05:
You need to add:
```
dtparam=i2c_arm=on,spi=on
```
to /boot/config.txt, and reboot. If you've done an apt-get upgrade then your new raspi-config can do that for you.

See: http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=97314&p=676583

jacqu, 2015-02-05:
I did what pelwell suggests. And below is the result of rmmoding then modprobing of the i2c driver.

Furthermore, I specified in /etc/modprobe.d/i2c.conf that I want i2c to run at 400kHz:
```
echo options i2c_bcm2708 baudrate=400000 > /etc/modprobe.d/i2c.conf
```
But if you read the end of dmesg ouptut, you see that this option is not taken into account:
```
BSC1 Controller at 0x20804000 (irq 79) (baudrate 100000)
```
This used to work before.
```
[  250.316413] ------------[ cut here ]------------
[  250.316483] WARNING: CPU: 0 PID: 3178 at drivers/clk/clk.c:850 __clk_disable+0x68/0x74()
[  250.316498] Modules linked in: ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 nf_nat nf_conntrack_ipv4 nf_defrag_ipv4 xt_state nf_conntrack iptable_filter ip_tables x_tables rfcomm bnep bluetooth rfkill i2c_dev snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd ch341 8192cu i2c_bcm2708(-) spi_bcm2708 usbserial uio_pdrv_genirq uio
[  250.316642] CPU: 0 PID: 3178 Comm: rmmod Not tainted 3.18.5+ #744
[  250.316727] [<c00151fc>] (unwind_backtrace) from [<c0012710>] (show_stack+0x20/0x24)
[  250.316770] [<c0012710>] (show_stack) from [<c05553e0>] (dump_stack+0x20/0x28)
[  250.316804] [<c05553e0>] (dump_stack) from [<c0022e7c>] (warn_slowpath_common+0x7c/0x9c)
[  250.316828] [<c0022e7c>] (warn_slowpath_common) from [<c0022f58>] (warn_slowpath_null+0x2c/0x34)
[  250.316855] [<c0022f58>] (warn_slowpath_null) from [<c045dc28>] (__clk_disable+0x68/0x74)
[  250.316881] [<c045dc28>] (__clk_disable) from [<c045dd54>] (clk_disable+0x34/0x40)
[  250.316922] [<c045dd54>] (clk_disable) from [<bf01e064>] (bcm2708_i2c_remove+0x44/0x5c [i2c_bcm2708])
[  250.316977] [<bf01e064>] (bcm2708_i2c_remove [i2c_bcm2708]) from [<c03671a4>] (platform_drv_remove+0x28/0x40)
[  250.317009] [<c03671a4>] (platform_drv_remove) from [<c0364e5c>] (__device_release_driver+0x68/0xbc)
[  250.317032] [<c0364e5c>] (__device_release_driver) from [<c0365690>] (driver_detach+0x120/0x124)
[  250.317054] [<c0365690>] (driver_detach) from [<c0364b78>] (bus_remove_driver+0x5c/0xb0)
[  250.317075] [<c0364b78>] (bus_remove_driver) from [<c0365f58>] (driver_unregister+0x38/0x58)
[  250.317097] [<c0365f58>] (driver_unregister) from [<c036728c>] (platform_driver_unregister+0x1c/0x20)
[  250.317132] [<c036728c>] (platform_driver_unregister) from [<bf01ec8c>] (bcm2708_i2c_exit+0x14/0x1c [i2c_bcm2708])
[  250.317181] [<bf01ec8c>] (bcm2708_i2c_exit [i2c_bcm2708]) from [<c00877ac>] (SyS_delete_module+0x140/0x1c0)
[  250.317212] [<c00877ac>] (SyS_delete_module) from [<c000e8c0>] (ret_fast_syscall+0x0/0x48)
[  250.317225] ---[ end trace 2664190f8ad9f09b ]---
[  257.054305] bcm2708_i2c_init_pinmode(1,2)
[  257.054341] bcm2708_i2c_init_pinmode(1,3)
[  257.059459] bcm2708_i2c 20804000.i2c: BSC1 Controller at 0x20804000 (irq 79) (baudrate 100000)
```

pelwell, 2015-02-05:
Yesterday's update brought some new parameters. What you want is:
```
dtparam=i2c_arm=on,i2c_arm_baudrate=400000
```
It's all in /boot/overlays/README.

jacqu, 2015-02-05:
I rebooted with the parameters you suggested. But with no luck.
```
pi@raspberrypi ~ $ dmesg|grep baudrate
[    7.163477] bcm2708_i2c 20804000.i2c: BSC1 Controller at 0x20804000 (irq 79) (baudrate 100000)
pi@raspberrypi ~ $ cat /boot/config.txt
# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
arm_freq=900

# for more options see http://elinux.org/RPi_config.txt
start_x=1
gpu_mem=128
core_freq=250
sdram_freq=450
over_voltage=2
dtparam=spi=on
dtparam=i2c_arm=on,i2c_arm_baudrate=400000
```

jacqu, 2015-02-05:
BTW, the kernel warning is deterministic on rmmoding the i2c driver.

pelwell, 2015-02-05:
For those options to work, you need the latest firmware. If you haven't updated in the last 24 hours, do so. If you have, try this:
```
strings /boot/bcm2708-rpi-b-plus.dtb | grep baudrate
strings /boot/start.elf | grep baudrate
```
If the board type is different, or you are using a different loader (perhaps start_x) then feel free to change the filenames above, but all ought to be kept in sync.

You're right about the warning - I'll check that out.

jacqu, 2015-02-05:
If have upgraded 1 hour ago (apt-get update;apt-get upgrade).
I still think I'm lagging behind the latest version because:
```
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 3.18.5+ #744 PREEMPT Fri Jan 30 18:19:07 GMT 2015 armv6l GNU/Linux

```
And just above, mw64d has:
```
pi@issabove ~ $ uname -a
Linux issabove 3.18.5+ #748 PREEMPT Wed Feb 4 21:24:41 GMT 2015 armv6l GNU/Linux
```
The results of your commands, plus the same one with start_x.elf  is blank.

pelwell, 2015-02-05:
Ah, yes, you need to rpi-update to get the latest release. The Raspbian (apt-get upgrade) releases change less frequently, but I would anticipate a point release fairly soon.

jacqu, 2015-02-05:
The 400kHz param is now working. Thanks.
```
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 3.18.5+ #748 PREEMPT Wed Feb 4 21:24:41 GMT 2015 armv6l GNU/Linux
pi@raspberrypi ~ $ dmesg|grep baudrate
[    6.437895] bcm2708_i2c 20804000.i2c: BSC1 Controller at 0x20804000 (irq 79) (baudrate 400000)
pi@raspberrypi ~ $ sudo rmmod i2c_bcm2708
pi@raspberrypi ~ $ dmesg
[...][  185.840180] ------------[ cut here ]------------
[  185.840251] WARNING: CPU: 0 PID: 3170 at drivers/clk/clk.c:850 __clk_disable+0x68/0x74()
[  185.840266] Modules linked in: ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 nf_nat nf_conntrack_ipv4 nf_defrag_ipv4 xt_state nf_conntrack iptable_filter ip_tables x_tables bnep rfcomm bluetooth rfkill i2c_dev snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd ch341 8192cu i2c_bcm2708(-) spi_bcm2708 usbserial uio_pdrv_genirq uio
[  185.840412] CPU: 0 PID: 3170 Comm: rmmod Not tainted 3.18.5+ #748
[  185.840479] [<c00151fc>] (unwind_backtrace) from [<c0012710>] (show_stack+0x20/0x24)
[  185.840519] [<c0012710>] (show_stack) from [<c05556d8>] (dump_stack+0x20/0x28)
[  185.840559] [<c05556d8>] (dump_stack) from [<c0022e7c>] (warn_slowpath_common+0x7c/0x9c)
[  185.840590] [<c0022e7c>] (warn_slowpath_common) from [<c0022f58>] (warn_slowpath_null+0x2c/0x34)
[  185.840623] [<c0022f58>] (warn_slowpath_null) from [<c045df20>] (__clk_disable+0x68/0x74)
[  185.840652] [<c045df20>] (__clk_disable) from [<c045e04c>] (clk_disable+0x34/0x40)
[  185.840699] [<c045e04c>] (clk_disable) from [<bf01e064>] (bcm2708_i2c_remove+0x44/0x5c [i2c_bcm2708])
[  185.840763] [<bf01e064>] (bcm2708_i2c_remove [i2c_bcm2708]) from [<c03671f0>] (platform_drv_remove+0x28/0x40)
[  185.840797] [<c03671f0>] (platform_drv_remove) from [<c0364ea8>] (__device_release_driver+0x68/0xbc)
[  185.840824] [<c0364ea8>] (__device_release_driver) from [<c03656dc>] (driver_detach+0x120/0x124)
[  185.840853] [<c03656dc>] (driver_detach) from [<c0364bc4>] (bus_remove_driver+0x5c/0xb0)
[  185.840880] [<c0364bc4>] (bus_remove_driver) from [<c0365fa4>] (driver_unregister+0x38/0x58)
[  185.840908] [<c0365fa4>] (driver_unregister) from [<c03672d8>] (platform_driver_unregister+0x1c/0x20)
[  185.840947] [<c03672d8>] (platform_driver_unregister) from [<bf01ec8c>] (bcm2708_i2c_exit+0x14/0x1c [i2c_bcm2708])
[  185.841004] [<bf01ec8c>] (bcm2708_i2c_exit [i2c_bcm2708]) from [<c00877ac>] (SyS_delete_module+0x140/0x1c0)
[  185.841040] [<c00877ac>] (SyS_delete_module) from [<c000e8c0>] (ret_fast_syscall+0x0/0x48)
[  185.841057] ---[ end trace 20ef05c1697a217c ]---
```

pelwell, 2015-02-05:
bec4f0aff550047407c86eab7f2699ed99c5176d should have fixed it.

mw46d, 2015-02-05:
OK, my PiGlow is working again as well;-)

I tried to search/google for my problem but that forum post did not show up:-( And since I started with the normal `apt-get update && apt-get upgrade', I did not expect the boot parameters to change:-( 

Thanks,
mw46d

pelwell, 2015-02-05:
We realised some time ago that we should enable Device Tree by default. If our DT usage was going to be in the true spirit of DT, that meant abandoning the "always on" semantics of the old board support code; this necessarily meant that some configuration changes would be needed after the update. After some debate we decidedit was better to require some user intervention after this update rather than attempt to change things automatically behind the scenes, because it is both safer and educational.

pelwell, 2015-03-09:
Can we close this now?

mw46d, 2015-03-09:
yes, thanks.

jacqu, 2015-03-10:
It didn't show up in kernels compiled after the 5th February. Many thanks for your reactivity !

pelwell, 2015-03-10:
Thanks for the clear reports.

closed 2015-03-10 by pelwell