raspberrypi/linux #777: external USB drive: blk_update_request: I/O error and the USB drive gets another /dev/sd node

f18m, 2015-01-31:
Hi,
I found a quite-reproducible problem (happens quite often!) with the following setup:

- HW: Raspberry model B+, powered by its own 5V, 2A power supply
- USB HUB 3.0 (Trascend TS-HUB3K), powered by its own 12V, 1.5A power supply (15W)
- 2 external hard disks (the problem happens also with just 1 hard disk!)

- root@rasptorrent:~# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 7 (wheezy)"
NAME="Raspbian GNU/Linux"
VERSION_ID="7"
VERSION="7 (wheezy)"
ID=raspbian
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

- root@rasptorrent:~# uname -a
Linux rasptorrent 3.18.5+ #744 PREEMPT Fri Jan 30 18:19:07 GMT 2015 armv6l GNU/Linux

(did "rpi-update" on Jan 31 2015)

HOW TO REPRODUCE: just producing traffic on the USB external drives at some point produces:

[ 1116.317108] sd 1:0:0:0: [sdb]  
[ 1116.317199] Result: hostbyte=0x07 driverbyte=0x00
[ 1116.317220] sd 1:0:0:0: [sdb] CDB: 
[ 1116.317233] cdb[0]=0x28: 28 00 13 7b 13 08 00 00 f0 00
[ 1116.317297] blk_update_request: I/O error, dev sdb, sector 326832904

(several times, for different sectors)

and the USB device (e.g., /dev/sdb1) is destroyed and a new dev node is created (e.g., /dev/sdc1).

The dwc_otg driver seems latest one:

root@rasptorrent:~# dmesg | grep dwc_otg
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708.boardrev=0x10 bcm2708.serial=0xee394e21 smsc95xx.macaddr=B8:27:EB:39:4E:21 bcm2708_fb.fbswap=1 bcm2708.disk_led_gpio=47 bcm2708.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
[    1.583112] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
[    2.027416] WARN::dwc_otg_hcd_init:1047: FIQ DMA bounce buffers: virt = 0xdbc14000 dma = 0x5bc14000 len=9024
[    2.054741] dwc_otg: Microframe scheduler enabled
[    2.087234] dwc_otg bcm2708_usb: DWC OTG Controller
[    2.097477] dwc_otg bcm2708_usb: new USB bus registered, assigned bus number 1
[    2.110137] dwc_otg bcm2708_usb: irq 32, io mem 0x00000000
[    2.173907] usb usb1: Manufacturer: Linux 3.18.5+ dwc_otg_hcd
[    2.215058] dwc_otg: FIQ enabled
[    2.215085] dwc_otg: NAK holdoff enabled
[    2.215098] dwc_otg: FIQ split-transaction FSM enabled
[    2.696903] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    3.257234] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    3.617035] usb 1-1.2: new low-speed USB device number 4 using dwc_otg
[    4.117189] usb 1-1.3: new low-speed USB device number 5 using dwc_otg
[    4.647235] usb 1-1.5: new high-speed USB device number 6 using dwc_otg
[    5.287832] usb 1-1.5.2: new high-speed USB device number 7 using dwc_otg
[    5.757505] usb 1-1.5.4: new high-speed USB device number 8 using dwc_otg
[ 1116.837726] usb 1-1.5.4: new high-speed USB device number 9 using dwc_otg

and it seems that "FIQ enabled", "NAK holdoff enabled" " FIQ split-transaction FSM enabled" features are there.

I'm going to attach the dmsg and lsusb outputs. Please let me know if I can try some fix.

f18m, 2015-01-31:
DMESG OUTPUT:

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.18.5+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #744 PREEMPT Fri Jan 30 18:19:07 GMT 2015
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine model: Raspberry Pi Model B+
[    0.000000] cma: Reserved 8 MiB at 0x1b800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 114688
[    0.000000] free_area_init_node: node 0, pgdat c084d794, node_mem_map db474000
[    0.000000]   Normal zone: 896 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 113792
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708.boardrev=0x10 bcm2708.serial=0xee394e21 smsc95xx.macaddr=B8:27:EB:39:4E:21 bcm2708_fb.fbswap=1 bcm2708.disk_led_gpio=47 bcm2708.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 437208K/458752K available (5926K kernel code, 358K rwdata, 1876K rodata, 340K init, 734K bss, 21544K reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[    0.000000]     vmalloc : 0xdc800000 - 0xff000000   ( 552 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc07a6a60   (7803 kB)
[    0.000000]       .init : 0xc07a7000 - 0xc07fc000   ( 340 kB)
[    0.000000]       .data : 0xc07fc000 - 0xc085582c   ( 359 kB)
[    0.000000]        .bss : 0xc085582c - 0xc090d0c8   ( 735 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:522
[    0.000025] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483648000ns
[    0.000076] Switching to timer-based delay loop, resolution 1000ns
[    0.000360] Console: colour dummy device 80x30
[    0.001434] console [tty1] enabled
[    0.001481] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.001556] pid_max: default: 32768 minimum: 301
[    0.001934] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.003011] Initializing cgroup subsys memory
[    0.003101] Initializing cgroup subsys devices
[    0.003159] Initializing cgroup subsys freezer
[    0.003213] Initializing cgroup subsys net_cls
[    0.003263] Initializing cgroup subsys blkio
[    0.003387] CPU: Testing write buffer coherency: ok
[    0.003502] ftrace: allocating 19476 entries in 58 pages
[    0.111287] Setting up static identity map for 0x55c8f8 - 0x55c954
[    0.114119] devtmpfs: initialized
[    0.131804] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.134975] pinctrl core: initialized pinctrl subsystem
[    0.137561] NET: Registered protocol family 16
[    0.143094] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[    0.171058] cpuidle: using governor ladder
[    0.201120] cpuidle: using governor menu
[    0.201635] bcm2708.uart_clock = 3000000
[    0.204744] No ATAGs?
[    0.204808] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.204872] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.204936] mailbox: Broadcom VideoCore Mailbox driver
[    0.205100] bcm2708_vcio: mailbox at f200b880
[    0.205244] bcm_power: Broadcom power driver
[    0.205291] bcm_power_open() -> 0
[    0.205321] bcm_power_request(0, 8)
[    0.706070] bcm_mailbox_read -> 00000080, 0
[... 218 lines cut ...]
[ 1116.938756] usb 1-1.5.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1116.938774] usb 1-1.5.4: Product: NS1066
[ 1116.938791] usb 1-1.5.4: Manufacturer: Norelsys
[ 1116.938808] usb 1-1.5.4: SerialNumber: 0123456789ABCDE
[ 1116.947231] usb-storage 1-1.5.4:1.0: USB Mass Storage device detected
[ 1116.957029] scsi host2: usb-storage 1-1.5.4:1.0
[ 1118.488809] scsi 2:0:0:0: Direct-Access     NORELSYS 106X             1C   PQ: 0 ANSI: 6
[ 1118.493957] sd 2:0:0:0: [sdc] 976773168 512-byte logical blocks: (500 GB/465 GiB)
[ 1118.494854] sd 2:0:0:0: [sdc] Write Protect is off
[ 1118.494893] sd 2:0:0:0: [sdc] Mode Sense: 43 00 00 00
[ 1118.495731] sd 2:0:0:0: [sdc] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[... cut ...]
[ 1176.624954] Buffer I/O error on dev sdb1, logical block 15543735, async page read
[ 1179.680094] Buffer I/O error on dev sdb1, logical block 87877518, async page read
[ 1179.777933] Buffer I/O error on dev sdb1, logical block 87877518, async page read
[ 1181.891037] Buffer I/O error on dev sdb1, logical block 87877518, async page read
[ 1181.911066] Buffer I/O error on dev sdb1, logical block 87877518, async page read
[ 1181.931217] Buffer I/O error on dev sdb1, logical block 87877518, async page read

f18m, 2015-01-31:
LSUSB OUTPUT:

Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. 
Bus 001 Device 004: ID 046d:c018 Logitech, Inc. Optical Wheel Mouse
Bus 001 Device 005: ID 045e:009d Microsoft Corp. Wireless Optical Desktop 3.0
Bus 001 Device 006: ID 8564:4000  
Bus 001 Device 007: ID 2537:1068  
Bus 001 Device 009: ID 2537:1066

P33M, 2015-02-01:
Is there a firmware revision where this behaviour first started? I.e. can you identify which revision of firmware is the first bad one?

f18m, 2015-02-01:
Unfortunately no. I have my raspberry only since a few days. Before buying the powered USB hub, it gave the same problem (back then it was using a raspbmc distribution with kernel 3.12 but the errors in dmesg were similar/identical). Moreover, after /dev/sd node recreation I could also hear the typical noise of the external hard drive having problems starting up... for this reason I thought the problems were due to insufficient power and thus I bought the USB hub. 

Now the problem in dmesg is similar/identical but it doesn't really seem a power issue: the hard disk stay on even when attached just to the USB hub and additionally, after the /dev/sd node recreation the hard disk emits no noise...

An additional thing: both the hard disk work just fine under Windows (and also under Linux on my workstation) even under heavy load... I think it must be something with dwc_otg driver...

f18m, 2015-02-01:
Another note: I also tried to power everything from the USB hub (15W should be enough for the 2 external HDD and the raspberry) using a micro USB cable that goes from the hub to the raspberry micro-USB connection. The same problem happens.

Moreover, it also happens apparently randomly even when writing small files on the disk periodically (I have  a script that writes 512B every 30 seconds)... for example just now I see:

(....all init messages.....)
[   38.667175] fuse init (API version 7.23)
[19013.716559] sd 0:0:0:0: [sda]
[19013.716599] Result: hostbyte=0x07 driverbyte=0x00
[19013.716618] sd 0:0:0:0: [sda] CDB:
[19013.716631] cdb[0]=0x28: 28 00 04 27 27 7c 00 00 f0 00
[19013.716688] blk_update_request: I/O error, dev sda, sector 69674876
[19013.806554] sd 0:0:0:0: [sda]
[19013.806596] Result: hostbyte=0x07 driverbyte=0x00
[19013.806615] sd 0:0:0:0: [sda] CDB:
[19013.806628] cdb[0]=0x28: 28 00 04 27 28 6c 00 00 10 00
[19013.806682] blk_update_request: I/O error, dev sda, sector 69675116
[19013.865329] usb 1-1.5.2: USB disconnect, device number 7
[19013.866655] sd 0:0:0:0: [sda]
[19013.866691] Result: hostbyte=0x01 driverbyte=0x00
[19013.866712] sd 0:0:0:0: [sda] CDB:
[19013.866724] cdb[0]=0x28: 28 00 04 27 28 7c 00 00 f0 00
[19013.866779] blk_update_request: I/O error, dev sda, sector 69675132
[19013.866944] sd 0:0:0:0: [sda]
[19013.866970] Result: hostbyte=0x01 driverbyte=0x00
[19013.866988] sd 0:0:0:0: [sda] CDB:
[19013.866998] cdb[0]=0x28: 28 00 04 27 29 6c 00 00 10 00
[19013.867047] blk_update_request: I/O error, dev sda, sector 69675372

that is, in idle conditions, after from second 38 to second 19013 no problem, then suddenly the USB failure.

f18m, 2015-02-05:
Tried with another hard disk (again, over USB). I left it idle (mounted on the filesystem but without read/writes from userspace) for ~3days: no problem.
Then I started writing/reading:

[90665.519028] sd 0:0:0:0: [sda]
[90665.519069] Result: hostbyte=0x07 driverbyte=0x00
[90665.519088] sd 0:0:0:0: [sda] CDB:
[90665.519101] cdb[0]=0x28: 28 00 27 25 b9 38 00 00 f0 00
[90665.519167] blk_update_request: I/O error, dev sda, sector 656783672
[90665.618980] sd 0:0:0:0: [sda]
[90665.619022] Result: hostbyte=0x07 driverbyte=0x00
[90665.619041] sd 0:0:0:0: [sda] CDB:
[90665.619053] cdb[0]=0x28: 28 00 27 25 ba 28 00 00 10 00
[90665.619110] blk_update_request: I/O error, dev sda, sector 656783912
[90665.628528] usb 1-1.5.3: USB disconnect, device number 7
[90665.639075] sd 0:0:0:0: [sda]
[90665.639117] Result: hostbyte=0x01 driverbyte=0x00
[90665.639136] sd 0:0:0:0: [sda] CDB:
[90665.639151] cdb[0]=0x28: 28 00 27 25 ba 38 00 00 f0 00
[90665.639209] blk_update_request: I/O error, dev sda, sector 656783928
[90665.639486] EXT4-fs warning (device sda5): ext4_end_bio:317: I/O error -5 writing to inode 2616323 (offset 173670400 size 32768 starting block 119737000)
[90665.639525] Buffer I/O error on device sda5, logical block 5320096
[90665.639549] Buffer I/O error on device sda5, logical block 5320097
[90665.639569] Buffer I/O error on device sda5, logical block 5320098
[90665.639587] Buffer I/O error on device sda5, logical block 5320099
[90665.639605] Buffer I/O error on device sda5, logical block 5320100
[90665.639621] Buffer I/O error on device sda5, logical block 5320101
[90665.639638] Buffer I/O error on device sda5, logical block 5320102
[90665.639657] Buffer I/O error on device sda5, logical block 5320103
[90665.639711] EXT4-fs warning (device sda5): ext4_end_bio:317: I/O error -5 writing to inode 2616323 (offset 175374336 size 32768 starting block 119737416)
[90665.639736] Buffer I/O error on device sda5, logical block 5320512
[90665.639757] Buffer I/O error on device sda5, logical block 5320513
[90665.639823] EXT4-fs warning (device sda5): ext4_end_bio:317: I/O error -5 writing to inode 2616323 (offset 196313088 size 32768 starting block 119742528)
[90665.639899] EXT4-fs warning (device sda5): ext4_end_bio:317: I/O error -5 writing to inode 2616323 (offset 223379456 size 32768 starting block 119749136)

By the way I'm a C/C++ developer... if you could point me to the right point of the code writing those 

"sd 0:0:0:0: [sda]\nResult: hostbyte=0x07 driverbyte=0x00"

I could help you with understanding the failure reason perhaps (althought I don't know much about USB driver routines!)...

f18m, 2015-02-26:
Actually I found that this bug is possibly related to the following long-standing issue in Linux kernel:

 https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=625922

MeeGeeVee, 2015-03-01:
I have a similar issue 

Running crashplan on a raspberry pi

Raspberry PI B with own powersupply - booting from SDCard 
Seagate Expansion 2TB USB hard drive 1 - Powered by RPI
NTFS partition was reformatted and did a full scandisk in windows (without errors : chkdsk /F /R /X without error) 

USB drive disconnecting as device 4 and reconnecting as 5 
First on Linux RPI 3.12.28+ 
Now on Linux version 3.18.8+

Details: 
First running on : 
Linux RPI 3.12.28+ #709 PREEMPT Mon Sep 8 15:28:00 BST 2014 armv6l
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.12.28+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #709 PREEMPT Mon Sep 8 15:28:00 BST 2014
 ...
    2.395822] Freeing unused kernel memory: 140K (c05bd000 - c05e0000)
[    2.428097] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    2.436572] Indeed it is in host mode hprt0 = 00001101
[    2.648426] usb 1-1: New USB device found, idVendor=0424, idProduct=9512
[    2.657030] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.666672] hub 1-1:1.0: USB hub found
[    2.672437] hub 1-1:1.0: 3 ports detected
[    2.958257] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    3.078587] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00
[    3.087064] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.114979] smsc95xx v1.0.4
[    3.183817] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-bcm2708_usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:4e:3e:35
[    3.308275] usb 1-1.3: new high-speed USB device number 4 using dwc_otg
[    3.421303] usb 1-1.3: New USB device found, idVendor=0bc2, idProduct=2312
[    3.437809] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.455266] usb 1-1.3: Product: Expansion
[    3.468109] usb 1-1.3: Manufacturer: Seagate 
[    3.474056] usb 1-1.3: SerialNumber: NA49AKKC
[    3.494274] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[    3.516889] scsi0 : usb-storage 1-1.3:1.0
[    3.973285] udevd[156]: starting version 175
[    4.751726] scsi 0:0:0:0: Direct-Access     Seagate  Expansion        0636 PQ: 0 ANSI: 6
[    4.789472] sd 0:0:0:0: [sda] 3907029167 512-byte logical blocks: (2.00 TB/1.81 TiB)
[    4.818478] sd 0:0:0:0: [sda] Write Protect is off
[    4.824932] sd 0:0:0:0: [sda] Mode Sense: 2b 00 10 08
[    4.838110] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, supports DPO and FUA
[    4.891840]  sda: sda1 sda2 sda3 sda4
[    4.915313] sd 0:0:0:0: [sda] Attached SCSI disk
[    5.078484] bcm2708_spi bcm2708_spi.0: master is unqueued, this is deprecated
[    5.258377] bcm2708_spi bcm2708_spi.0: SPI Controller at 0x20204000 (irq 80)
[    6.139475] bcm2708-i2s bcm2708-i2s.0: Failed to create debugfs directory
...
[31672.472087] usb 1-1.3: USB disconnect, device number 4
[31672.479681] sd 0:0:0:0: [sda] Unhandled error code
[31672.479722] sd 0:0:0:0: [sda]  
[31672.479736] Result: hostbyte=0x01 driverbyte=0x00
[31672.479749] sd 0:0:0:0: [sda] CDB: 
[31672.479760] cdb[0]=0x2a: 2a 00 24 0b cb 98 00 00 f0 00
[31672.479805] end_request: I/O error, dev sda, sector 604752792
[31672.479827] Buffer I/O error on device sda1, logical block 75593843
[31672.479840] lost page write due to I/O error on sda1
[31672.479871] Buffer I/O error on device sda1, logical block 75593844
[31672.479884] lost page write due to I/O error on sda1
[31672.479903] Buffer I/O error on device sda1, logical block 75593845
[31672.479915] lost page write due to I/O error on sda1
[31672.479933] Buffer I/O error on device sda1, logical block 75593846
[31672.479944] lost page write due to I/O error on sda1
[31672.479964] Buffer I/O error on device sda1, logical block 75593847
[31672.479976] lost page write due to I/O error on sda1
[31672.479994] Buffer I/O error on device sda1, logical block 75593848
[... 184 lines cut ...]
[22342.765735] sd 1:0:0:0: [sdb] Write Protect is off
[22342.765772] sd 1:0:0:0: [sdb] Mode Sense: 2b 00 10 08
[22342.766749] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, supports DPO and FUA
[22342.787645]  sdb: sdb1 sdb2 sdb3 sdb4
[22342.793223] sd 1:0:0:0: [sdb] Attached SCSI disk
[22348.173746] buffer_io_error: 60 callbacks suppressed
[22348.173789] Buffer I/O error on dev sda1, logical block 786449, async page read
[22348.174803] Buffer I/O error on dev sda1, logical block 786449, async page read
[22348.178741] Buffer I/O error on dev sda1, logical block 786449, async page read
[22348.180791] Buffer I/O error on dev sda1, logical block 786449, async page read
[22350.133189] Buffer I/O error on dev sda1, logical block 786449, async page read
[22350.134229] Buffer I/O error on dev sda1, logical block 786449, async page read
[22350.138038] Buffer I/O error on dev sda1, logical block 786449, async page read
[22350.140092] Buffer I/O error on dev sda1, logical block 786449, async page read
[22352.270635] Buffer I/O error on dev sda1, logical block 61321656, async page read
[22352.272957] Buffer I/O error on dev sda1, logical block 61321656, async page read
[22836.221680] buffer_io_error: 38 callbacks suppressed
[22836.221723] Buffer I/O error on dev sda1, logical block 242143488, async page read
[22836.254434] Buffer I/O error on dev sda1, logical block 61321656, async page read
[22836.255735] Buffer I/O error on dev sda1, logical block 61321656, async page read
[22836.263045] Buffer I/O error on dev sda1, logical block 786449, async page read
>`

I'm not a linux guru, but trying my best to understand things , and solve where possible.
All your assistance is more then welcome

Razican, 2015-04-16:
I confirm that this happens with Raspbery Pi 2 B running:
```Linux 3.18.11-v7+ #777 SMP PREEMPT Sat Apr 11 17:30:37 BST 2015 armv7l GNU/Linux```

I have a RAID attached via USB 3.0 (limited to 2.0 by RPi hardware) And it seems to happen under heavy traffic load. Is a 2x2TB in RAID 0 (total of 4TB drive). It fails when checking torrent downloads.

MeeGeeVee, 2015-04-17:
In my case it was 100% power related.

I was able to test the drive and generate different load conditions with the command 'badblocks' and tuning of the parameters -d -c- b ( fe: sudo badblocks  -e 100 -v /dev/sda1 -s -c 32 -d 50). As a result the drive failed on different blocks or not at all. This was an indication that the individual blocks were ok. 

Changing the powersupply of the RPI (2A) did not help neither.

After buying a USB3 Y-cable to power the drive from another USB port, all  got solved and my drive is still running fine for weeks now. http://www.ebay.com/sch/i.html?_from=R40&_nkw=usb3+y-cable

Hope this helps your case too.

Razican, 2015-04-17:
In my case, the USB RAID is self-powered, so the power should not be any problem.

MeeGeeVee, 2015-04-17:
Worth to try some badblocks tests to see if you can reproduce a failure? "sudo badblocks -e 100 -v /dev/sda1 -s" will test at maximum speed. If failure should occur each time in different blocks it would put the hard drive out of the scope.

Razican, 2015-04-17:
After checking it with badblocks, after about 6 hours of checking it stopped after this error message:

```
[21989.512626] usb 1-1.5: reset high-speed USB device number 4 using dwc_otg
[21989.603287] usb 1-1.5: USB disconnect, device number 4
[21989.620170] scsi 0:0:0:0: rejecting I/O to offline device
[21989.625727] scsi 0:0:0:0: [sda] killing request
[21989.630473] scsi 0:0:0:0: [sda]  
[21989.633890] Result: hostbyte=0x01 driverbyte=0x00
[21989.638692] scsi 0:0:0:0: [sda] CDB: 
[21989.642420] cdb[0]=0x88: 88 00 00 00 00 00 50 d4 47 80 00 00 00 80 00 00
[21989.649557] blk_update_request: I/O error, dev sda, sector 1356089216
[21989.656584] Buffer I/O error on dev sda1, logical block 169478128, async page read
[21989.664568] Buffer I/O error on dev sda1, logical block 169478128, async page read
[21989.672388] Buffer I/O error on dev sda1, logical block 169478128, async page read
[21989.680280] Buffer I/O error on dev sda1, logical block 169478128, async page read
[21989.688367] Buffer I/O error on dev sda1, logical block 169478129, async page read
[21989.696346] Buffer I/O error on dev sda1, logical block 169478129, async page read
[21989.704281] Buffer I/O error on dev sda1, logical block 169478129, async page read
[21989.717096] Buffer I/O error on dev sda1, logical block 169478129, async page read
[21989.730112] Buffer I/O error on dev sda1, logical block 169478130, async page read
[21989.742960] Buffer I/O error on dev sda1, logical block 169478130, async page read
[21991.573600] usb 1-1.5: new high-speed USB device number 5 using dwc_otg
[21991.695072] usb 1-1.5: New USB device found, idVendor=152d, idProduct=0539
[21991.703942] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=5
[21991.713205] usb 1-1.5: Product: USB to ATA/ATAPI Bridge
[21991.720824] usb 1-1.5: Manufacturer: JMicron
[21991.727026] usb 1-1.5: SerialNumber: 239C7E548FFF
[21991.738885] usb-storage 1-1.5:1.0: USB Mass Storage device detected
[21991.747727] scsi host1: usb-storage 1-1.5:1.0
[21992.754883] scsi 1:0:0:0: Direct-Access     Sharkoon  H/W RAID0            PQ: 0 ANSI: 5
[21992.767773] sd 1:0:0:0: Attached scsi generic sg0 type 0
[21992.778207] sd 1:0:0:0: [sdb] Very big device. Trying to use READ CAPACITY(16).
[21992.788120] sd 1:0:0:0: [sdb] 7813857280 512-byte logical blocks: (4.00 TB/3.63 TiB)
[21992.800149] sd 1:0:0:0: [sdb] Write Protect is off
[21992.806690] sd 1:0:0:0: [sdb] Mode Sense: 28 00 00 00
[21992.807146] sd 1:0:0:0: [sdb] No Caching mode page found
[21992.814189] sd 1:0:0:0: [sdb] Assuming drive cache: write through
[21992.823579] sd 1:0:0:0: [sdb] Very big device. Trying to use READ CAPACITY(16).
[21993.203112]  sdb: sdb1
[21993.209125] sd 1:0:0:0: [sdb] Very big device. Trying to use READ CAPACITY(16).
[21993.221187] sd 1:0:0:0: [sdb] Attached SCSI disk
[28490.293973] buffer_io_error: 90 callbacks suppressed
[28490.301253] Buffer I/O error on dev sda1, logical block 36, async page read
```

The last two were me trying to do a ```ls``` in the mounted /dev/sda1, that now is called /dev/sdb1, so it won't work.

f18m, 2015-04-18:
By the way I still haven't solved this problem. Just FYI: I see the same problem on both Raspberry PI model B+ and on my BeagleBone.

The hard disks triggering this problem are miscellaneous:
 - WDC WD5000LPVT (500 GB) enclosed in USB3 enclosure (reports itself as Norelsys NS1068)
 - Toshiba MQ01ABD050 (500 GB) enclosed in USB3 enclosure (reports itself as Norelsys NS106X)
 - Toshiba MQ01UBD100 (1 TB) with its own Toshiba USB3 enclosure

It may help if you write down the manufacturers and model names of the hard disks that trigger this problem (the "Result: hostbyte=* driverbyte=*" error message). Thanks!

f18m, 2015-04-18:
Another comment: on my desktop computer running Ubuntu Trusty 14.04 no problem are reported when transferring large files between the disks reported above.

analyser-, 2015-04-21:
I'm not sure if it is related, however I'll share anyway, since it can help someone.

When I was assembly my rpi file server, with two external hard drives plugged into an usb hub, both in those cheap enclosures, I was not be able to use them till I've used a power source of 5.3 V (to the usb hub), instead 5.0 V. So, I strongly recommend trying a higher voltage power source.

Blindfreddy, 2015-04-21:
I have the same error using a WD 2TB USB3 disk, refer https://www.google.de/url?url=https://github.com/raspberrypi/linux/issues/885&rct=j&q=&esrc=s&sa=U&ei=qz82VZIhwaewAYDNgKAN&ved=0CBoQFjAB&sig2=OFhTaepwrroA5QRZuaWnUQ&usg=AFQjCNEzwcQ-hGcApolnu3RnIxP6Et4JdA

HDD: DetailsHDD D 
  iManufacturer           1 Western Digital
  iProduct                2 My Passport 0748

I actually have two Pi's, each with same model USB HDD.  1st on is a Model B, 2nd one a Model 2B.  Works fine on the older Model B, but the problem occurs on the newer Model 2 B.

Kernel and Firmware on older Model B
uname -a
Linux rpi1 3.18.7+ #755 PREEMPT Thu Feb 12 17:14:31 GMT 2015 armv6l GNU/Linux
sudo /opt/vc/bin/vcgencmd version
Feb 14 2015 22:17:16 
Copyright (c) 2012 Broadcom
version 7789db485409720b0e523a3d6b86b12ed56fd152 (clean) (release)

Kernel and Firmware on newer Model 2 B
uname -a
Linux rpi4 3.18.11-v7+ #780 SMP PREEMPT Mon Apr 20 14:59:10 BST 2015 armv7l GNU/Linux

sudo /opt/vc/bin/vcgencmd version
Apr 20 2015 15:02:24 
Copyright (c) 2012 Broadcom
version 423cbd6f9f4f6097643cb90ddfd2c2e4c01ef3fd (clean) (release)

MeeGeeVee, 2015-04-23:
> BlindFreddy : 
As stated before I'm not a specialist and my incident was power related. Reading your case I can only assist you to suggest some validations if your case isn't power related too. Interestiging in that you have 2 RPI and 2 drives.
- Are you able to change the drives from one RPI to each other. 
- Are you able to switch the power supply from one RPI to another
- I believe Western Digital Passport drive gets the power from the USB, are you able to use a y-cable to power the failing drive externally (http://www.ebay.com/sch/i.html?_from=R40&_nkw=usb3+y-cable). 
- Does 'sudo badblocks -e 100 -v /dev/sda1 -s' runs fine or fails each time at the same point ?

The result of the test should perhaps show it the issue remains related to HD, RPI/OS version or power supply .

Razican, 2015-04-24:
I don't think it's a power related error, since it worked perfectly in Raspberry Pi B, but it's not working in Raspberry Pi 2 B. In any case, I might be able to try it again in Raspberry Pi B this weekend.

Blindfreddy, 2015-04-24:
Yes, I can try it out using Pi B and Pi 2B and the two drives.  I cannot try external power y-cable - don't have one and also, since it has the drive has been working flawelessly on Pi B using internal power for well over a year, I think it is reasonable to expect it to work on Pi 2B.

popcornmix, 2015-04-24:
Pi2 can use more power that Pi1, so I don't think you can infer a power supply that is good for Pi1 is necessarily good for Pi2.

Razican, 2015-04-24:
But in my case, the HDD RAID is self powered.

pelwell, 2015-04-24:
FYI, Adafruit have released a tutorial on adding an external HDD to a Pi, and they say:
> The Raspberry Pi doesn't have enough power to run most external hard drives or SSDs, so unless you have a drive with its own power supply, you'll probably need a powered USB hub.

f18m, 2015-04-24:
In my case, I'm using a powered USB hub (certified for USB 3.0, so that the
hub comes with a 15W power adapter)... all these issues make me feel that
it's not really a power supply issue.

Perhaps to be sure the only way would be attaching an oscope to the USB
lines and see directly if the signal strength over USB D+ and D- lines is
good enough or not.

Honestly, I suspect some driver issue, either on Linux driver side or in
the hard disk / USB adaptor firmware.
Probably Windows is capable of handling such situation (indeed most vendors
only test their hard drives with Windows only), but Linux no.

Just my 2cents,
Francesco M.

igniz-, 2015-04-27:
Hi all, I'm having this very same problem but not on a raspberry, just a regular intel computer. My HDD is attached to a SATA-to-USB adaptor and it shows this exact same behaviour. It actually started after an update to the system, including a Kernel update, so the problem must be there, since no hardware was changed, and I even tested two different linux distros (opensuse 13.2 & ubuntu 15.04) both does exaclty the same. Today I will try a regression to an older linux distro an see if the problem "solves" just to be sure of this.
Hope this narrows the source of the problem a litte.

Regards!

Razican, 2015-04-29:
I can confirm (almost 100% sure) that it is not power related. I have bought a 5V 2A PSU for the Raspberry Pi 2 and a powered USB hub to connect the storage there. As I said before the storage is self powered, so there is no doubt that all has enough power. But it continues to fail. This time after more than 2 hours of use, but it disconects the device and adds it as a new drive (even if UUID is the same).

In my change to the USB hub I also ran into #824. In any case I could do a workaround for that (changing the root delay, as said in the issue comments). The thing is that as we have seen, under heavy load, the device simply disapears from /dev.

Here is, again the dmesg:
```
[ 7991.378571] sd 0:0:0:0: [sda]  
[ 7991.381813] Result: hostbyte=0x00 driverbyte=0x08
[ 7991.386578] sd 0:0:0:0: [sda]  
[ 7991.389753] Sense Key : 0x3 [current] 
[ 7991.393605] sd 0:0:0:0: [sda]  
[ 7991.396781] ASC=0x3 ASCQ=0x0
[ 7991.399702] sd 0:0:0:0: [sda] CDB: 
[ 7991.403251] cdb[0]=0x88: 88 00 00 00 00 00 f5 d5 a1 f0 00 00 00 f0 00 00
[ 7991.410232] blk_update_request: I/O error, dev sda, sector 4124418544
[ 8022.335762] usb 1-1.5.4: reset high-speed USB device number 5 using dwc_otg
[ 8022.417011] usb 1-1.5.4: USB disconnect, device number 5
[ 8022.435619] sd 0:0:0:0: [sda]  
[ 8022.438833] Result: hostbyte=0x01 driverbyte=0x00
[ 8022.443595] sd 0:0:0:0: [sda] CDB: 
[ 8022.447166] cdb[0]=0x88: 88 00 00 00 00 00 f5 d5 a2 e0 00 00 00 10 00 00
[ 8022.454133] blk_update_request: I/O error, dev sda, sector 4124418784
[ 8022.465193] sd 0:0:0:0: [sda]  
[ 8022.468442] Result: hostbyte=0x01 driverbyte=0x00
[ 8022.473207] sd 0:0:0:0: [sda] CDB: 
[ 8022.476761] cdb[0]=0x88: 88 00 00 00 00 00 f5 d5 a2 f0 00 00 00 f0 00 00
[ 8022.483721] blk_update_request: I/O error, dev sda, sector 4124418800
[ 8024.296853] usb 1-1.5.4: new high-speed USB device number 6 using dwc_otg
[ 8024.330635] Buffer I/O error on dev sda1, logical block 539903643, async page read
[ 8024.341511] Buffer I/O error on dev sda1, logical block 539903643, async page read
[ 8024.352686] Buffer I/O error on dev sda1, logical block 539903643, async page read
[ 8024.363942] Buffer I/O error on dev sda1, logical block 539903643, async page read
[ 8024.375431] Buffer I/O error on dev sda1, logical block 539903643, async page read
[ 8024.387017] Buffer I/O error on dev sda1, logical block 539903643, async page read
[ 8024.398615] Buffer I/O error on dev sda1, logical block 539903643, async page read
[ 8024.410269] Buffer I/O error on dev sda1, logical block 539903643, async page read
[ 8024.418369] usb 1-1.5.4: New USB device found, idVendor=152d, idProduct=0539
[ 8024.418379] usb 1-1.5.4: New USB device strings: Mfr=1, Product=2, SerialNumber=5
[ 8024.418386] usb 1-1.5.4: Product: USB to ATA/ATAPI Bridge
[ 8024.418392] usb 1-1.5.4: Manufacturer: JMicron
[ 8024.418398] usb 1-1.5.4: SerialNumber: 239C7E548FFF
[ 8024.419830] usb-storage 1-1.5.4:1.0: USB Mass Storage device detected
[ 8024.420271] scsi host1: usb-storage 1-1.5.4:1.0
[ 8024.478674] Buffer I/O error on dev sda1, logical block 539903643, async page read
[ 8024.490996] Buffer I/O error on dev sda1, logical block 539903643, async page read
[ 8025.417797] scsi 1:0:0:0: Direct-Access     Sharkoon  H/W RAID0            PQ: 0 ANSI: 5
[ 8025.431029] sd 1:0:0:0: [sdb] Very big device. Trying to use READ CAPACITY(16).
[ 8025.442048] sd 1:0:0:0: [sdb] 7813857280 512-byte logical blocks: (4.00 TB/3.63 TiB)
[ 8025.454605] sd 1:0:0:0: [sdb] Write Protect is off
[ 8025.461347] sd 1:0:0:0: [sdb] Mode Sense: 28 00 00 00
[ 8025.462178] sd 1:0:0:0: [sdb] No Caching mode page found
[ 8025.469414] sd 1:0:0:0: [sdb] Assuming drive cache: write through
[ 8025.479051] sd 1:0:0:0: [sdb] Very big device. Trying to use READ CAPACITY(16).
[ 8025.494139] sd 1:0:0:0: Attached scsi generic sg0 type 0
[ 8025.968816]  sdb: sdb1
[ 8025.978257] sd 1:0:0:0: [sdb] Very big device. Trying to use READ CAPACITY(16).
[ 8025.993777] sd 1:0:0:0: [sdb] Attached SCSI disk
```

Blindfreddy, 2015-05-04:
Some news:  I can no longer reproduce the error.  I have not updated the config on the Model 2B, yet the problem is no longer occurring.  I did fix some errors in the filesystem, so maybe that is the root cause of the buffer I/O errors.  Here is what I did:

1. Attached USB HDD to 'old' Model B Pi (powered by Pi), then ran fsck without flags which reported clean fs (ext4).

2. Still connected to old Pi, ran fsck -v -f  to generate I/O load - this was known to cause the buffer I/O errors on the new Model 2B Pi; as expected, no buffer I/O errors on the 'old' Model B, but interestingly, there were several problems with the fs despite it having previously reported to be clean....;  fixed the errors, reran fsck -v -f, no errors. Refer details below.

3. Discconnected USB HDD and connected to 'new' Model 2B Pi and again ran fsck -v -f: no errors, no more buffer I/O errors.

4. Ran a very large rsync job (1TB data) across LAN from the 2nd, identical USB HDD connected to a third Pi (Model B), no errors. See attached dmesg output.

So I cannot reproduce the error anymore. Only change is to have a properly clean FS.  Perhaps someone else can run fsck with -f and then try to reproduce the error.

fsck output from 3. above:

fsck from util-linux 2.20.1
e2fsck 1.42.5 (29-Jul-2012)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Entry '#34705188' in /lost+found (12) has deleted/unused inode 34705188.  Clear<y>? yes
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
Inode bitmap differences:  -34705188
Fix<y>? yes
Free inodes count wrong for group #4236 (7628, counted=7629).
Fix<y>? yes
Free inodes count wrong (98690663, counted=98690664).
Fix<y>? yes

/dev/sda1: ***** FILE SYSTEM WAS MODIFIED *****

     7403928 inodes used (6.98%, out of 106094592)
      188820 non-contiguous files (2.6%)
        9418 non-contiguous directories (0.1%)
             # of inodes with ind/dind/tind blocks: 2999026/2997395/2997945
             Extent depth histogram: 1454181/9001/7
   371609074 blocks used (87.57%, out of 424369023)
           0 bad blocks
         773 large files

     1527971 regular files
      214931 directories
     1452884 character device files
     1334802 block device files
     1533953 fifos
        2409 links
        9582 symbolic links (8937 fast symbolic links)
     1329796 sockets
------------
     7406328 files

Razican, 2015-05-08:
I tried to switch to my old Pi B and I got the same error. I get the same error with the powered USB hub.

cageyz, 2015-05-21:
same problem with my Pi 2 B and WD disk(powered)

cageyz, 2015-06-11:
I have solved this problem. The ns1066 chipset (used by orico 6518US3) has some compatibility Issues so that when I changed the disk dock to orico 6518S3(which uses ASM1153E chipset), there is no disconnection anymore. Leave ns1066 far away from Raspberry PI.

kursion, 2015-06-24:
For me it was a power issue. The RPi was power by my labtop. I couldn't mount my External HDD. Finally I used my Samsung charger and it worked instantly.

popcornmix, 2015-06-24:
@f18m any change with latest 4.0 kernel? (from rpi-update)

Ealdwulf, 2015-10-12:
I can reproduce this issue (Buffer I/O error on dev sda, logical block XXXX, async page read) by using dd to flash an sd card with one of these: http://www.amazon.co.uk/Integral-MicroSD-USB-Card-Reader/dp/B001JPPQS6
That's on kernel 3.18-ck1-v7. If I get a chance to check on v4, I'll update.
It doesn't happen every time, but when it does the symptom is that dd slows to a crawl (20kb/s)

mujss, 2015-11-05:
I had exactly the same problem. The setup was that the raspberry pi would connect to 2 external HDDs and the drives were shared over the network. Most often when data to the HDDs were read or written to, i would get these errors and the HDD would be no longer mounted. The fix then would be to reboot the pi.
I had done lots of googling for a number of days and changed USB cables, changed powered USB hub, changed HDDs but no luck. 
What eventually worked for me was when i created a cron job on the pi to read the HDD every minute. And now 26 days later, i havent seen that happen again. 

*/1 * * * * ls /media/USBHDD1/

dcramps, 2016-01-11:
Seeing a very similar issue on one of my drives.

>[  741.577436] usb 1-1.5: USB disconnect, device number 6
>[  741.588838] sd 1:0:0:0: [sdb] UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=0x00
>[  741.588876] sd 1:0:0:0: [sdb] CDB: opcode=0x28 28 00 39 5d 6a c0 00 00 f0 00
>[  741.588895] blk_update_request: I/O error, dev sdb, sector 962423488
>[  741.589075] sd 1:0:0:0: [sdb] UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=0x00
>[  741.589094] sd 1:0:0:0: [sdb] CDB: opcode=0x28 28 00 39 5d 6b b0 00 00 10 00
>[  741.589105] blk_update_request: I/O error, dev sdb, sector 962423728
>[  741.996670] usb 1-1.5: new high-speed USB device number 8 using dwc_otg
>[  742.024710] Buffer I/O error on dev sdb1, logical block 44, async page read
>[  742.026325] Buffer I/O error on dev sdb1, logical block 44, async page read
>[  742.027198] Buffer I/O error on dev sdb1, logical block 44, async page read
>[  742.028831] Buffer I/O error on dev sdb1, logical block 44, async page read
>[  742.029558] Buffer I/O error on dev sdb1, logical block 44, async page read
>[  742.031026] Buffer I/O error on dev sdb1, logical block 44, async page read
>[  742.031721] Buffer I/O error on dev sdb1, logical block 44, async page read
>[  742.097779] usb 1-1.5: New USB device found, idVendor=2537, idProduct=1066
>[  742.097802] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
>[  742.097813] usb 1-1.5: Product: NS1066
>[  742.097824] usb 1-1.5: Manufacturer: Norelsys
>[  742.097835] usb 1-1.5: SerialNumber: 0123456789ABCDE
>[  742.098643] usb-storage 1-1.5:1.0: USB Mass Storage device detected
>[  742.099025] scsi host3: usb-storage 1-1.5:1.0
>[  743.631380] scsi 3:0:0:0: Direct-Access     NORELSYS 106X             AB51 PQ: 0 ANSI: 6
>[  743.636549] sd 3:0:0:0: [sdd] 3907029168 512-byte logical blocks: (2.00 TB/1.81 TiB)
>[  743.636914] sd 3:0:0:0: Attached scsi generic sg1 type 0
>[  743.638483] sd 3:0:0:0: [sdd] Write Protect is off
>[  743.638508] sd 3:0:0:0: [sdd] Mode Sense: 43 00 00 00
>[  743.646199] sd 3:0:0:0: [sdd] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
>[  743.665058]  sdd: sdd1 sdd2
>[  743.667688] sd 3:0:0:0: [sdd] Attached SCSI disk

This is happening when there is a lot of activity on the drive. Drive is in a USB enclosure, self powered. Pi is powered by a 5V/2A adapter.

PeterPablo, 2016-01-11:
You might want to try adding `max_usb_current=1` to your `/boot/config.txt` ([context](http://raspberrypi.stackexchange.com/questions/27708/is-setting-max-usb-current-1-to-give-more-power-to-usb-devices-a-bad-idea)).

M-Tomohiro, 2016-01-20:
I think cause is heat.
Do you set heatsink on CPU and LAN9514?
LAN9514 is USB and Ethernet controller.

JamesH65, 2016-01-20:
Neither of those devices requires a heatsink.

jackpgao, 2016-05-30:
same question. model 2 B with USB 3.0 HUB. The problem is the USB3.0. 2.0 works well.

tehfink, 2016-06-16:
Also seeing this issue with a rpi model 2B & USB 3 self-powered hub, and 4GB USB seagate drive.

humbledude, 2016-08-10:
I uses 2 external disks and only one disk has the same problem.
Problem is occur on IPTime external disk case + HDD.
It has external power adapter, so I connected the disk directly to rpi.
I recently bought rpi3 and this issue was not reproduced on my previous rpi1 B+
I guess this could be related to heat, its hot summer here and it seems to be reproduced after some time after reboot.

[123285.840778] usb 1-1.5: reset high-speed USB device number 5 using dwc_otg
[123286.223323] sd 0:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[123286.223363] sd 0:0:0:0: [sdb] tag#0 CDB: opcode=0x28 28 00 4d 00 09 08 00 00 08 00
[123286.223390] blk_update_request: I/O error, dev sdb, sector 1291847944

dminca, 2017-04-02:
Having the same issue on a virused USB Flash Kingston DT101 G2 (16GiB). I think the virus affected it's boot sectors and now the Kernel is unable to read the size of the device in order to mount it properly.

`dmesg -T` output:
```sh
# connect the usb
[Sun Apr  2 14:51:35 2017] usb 1-1.2: new high-speed USB device number 12 using ehci-pci
[Sun Apr  2 14:51:35 2017] usb-storage 1-1.2:1.0: USB Mass Storage device detected
[Sun Apr  2 14:51:35 2017] scsi host6: usb-storage 1-1.2:1.0
[Sun Apr  2 14:51:37 2017] scsi 6:0:0:0: Direct-Access     GENERIC  USB Mass Storage 1.00 PQ: 0 ANSI: 4 CCS
[Sun Apr  2 14:51:37 2017] sd 6:0:0:0: Attached scsi generic sg2 type 0
[Sun Apr  2 14:51:37 2017] sd 6:0:0:0: [sdb] Read Capacity(10) failed: Result: hostbyte=0x00 driverbyte=0x08
[Sun Apr  2 14:51:37 2017] sd 6:0:0:0: [sdb] Sense Key : 0x6 [current]
[Sun Apr  2 14:51:37 2017] sd 6:0:0:0: [sdb] ASC=0x28 ASCQ=0x0
[Sun Apr  2 14:51:37 2017] sd 6:0:0:0: [sdb] 0 512-byte logical blocks: (0 B/0 B)
[Sun Apr  2 14:51:37 2017] sd 6:0:0:0: [sdb] 0-byte physical blocks
[Sun Apr  2 14:51:37 2017] sd 6:0:0:0: [sdb] Write Protect is off
[Sun Apr  2 14:51:37 2017] sd 6:0:0:0: [sdb] Mode Sense: 45 00 00 00
[Sun Apr  2 14:51:37 2017] sd 6:0:0:0: [sdb] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[Sun Apr  2 14:51:37 2017] sd 6:0:0:0: [sdb] Read Capacity(10) failed: Result: hostbyte=0x00 driverbyte=0x08
[Sun Apr  2 14:51:37 2017] sd 6:0:0:0: [sdb] Sense Key : 0x6 [current]
[Sun Apr  2 14:51:37 2017] sd 6:0:0:0: [sdb] ASC=0x28 ASCQ=0x0
[Sun Apr  2 14:51:37 2017] sd 6:0:0:0: [sdb] Attached SCSI removable disk
[Sun Apr  2 14:51:44 2017] usb 1-1.2: USB disconnect, device number 12

# disconnect the usb
[Sun Apr  2 14:51:44 2017] usb 1-1.2: USB disconnect, device number 12
```

`lsusb` output (dimmed):
```sh
Bus 001 Device 012: ID 0951:1665 Kingston Technology Digital DataTraveler SE9 64GB
```

Since it didn't read the size, he didn't even attach the filesystem `/dev/sdX`, output of `lsblk`:
```sh
NAME              MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda                 8:0    0 232.9G  0 disk
└─sda1              8:1    0 232.9G  0 part
  ├─root--vg-boot 254:0    0   3.1G  0 lvm  /boot
  ├─root--vg-root 254:1    0    50G  0 lvm  /
  ├─root--vg-swap 254:2    0   500M  0 lvm  [SWAP]
  └─root--vg-home 254:3    0   150G  0 lvm  /home
sr0                11:0    1  1024M  0 rom
```

**CONCLUSION** boot sectors affected, USB Flash is bricked. :wastebasket:

adde88, 2017-04-23:
Hi from the OpenWRT community! ✋ 
I was curious to if a solution was found to this problem?
I'm having the **exact** same problem on my device on kernel versions: _3.18.20_ to _3.18.36_ !
It seems to be related to the SD-card. Replacing my SD-card with a USB-thumb drive (mounted exactly similar), "fixes" the problem. But i miss my SD-card. 😭 

Same errors messages as above shows, and the sd-card disconnects from the device, and then re-appears a second later with a new identifier each time (/dev/sda -> /dev/sdb - /dev/sdc)
Running badblocks triggers it every time. And very often when reading/writing to the device it happens, but never on standby.

I know this thread is about Raspberry Pi, not OpenWRT. But i had to ask if you guys found a solution, because i'm pretty desperate to get this fixed. 😃

P33M, 2017-05-04:
For everyone experiencing random device disconnects, what happens if you put a USB hub between the Pi and the device in question?

ben-ba, 2017-05-28:
Same problem here with
OSMC (up to date)
RaspberryPi 3
USB 2.0 Hub (Hama 1 to 4 - [https://de.hama.com/00078472/hama-usb-20-hub-14-with-power-supply-black-cardboard-box?srcLang=0&targetLang=1](url) )

`osmc@osmc:~$ sudo lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 3: Dev 30, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 3: Dev 33, If 0, Class=Mass Storage, Driver=usb-storage, 480M
            |__ Port 4: Dev 34, If 0, Class=Mass Storage, Driver=usb-storage, 480M`

two usb drives on hub, both WD MyPassport, 4TB, 
one (sda1) NTFS, the other one (sbb1) is ext4 formatted.
`/dev/sda1 /media/6080669B80667804 fuseblk rw,nosuid,nodev,relatime,user_id=0,group_id=0,default_permissions,allow_other,blksize=4096 0 0
/dev/sdb1 /media/MyPassport2 ext4 rw,noatime,data=ordered 0 0`

Getting trouble when i rsync all data (nearly 4TB) from the "NTFS" drive to the "ext4" drive, dmsg output:

`[  +0,001428] sd 17:0:0:0: [sdb] Very big device. Trying to use READ CAPACITY(16).
[  +0,001299] sd 17:0:0:0: [sdb] Attached SCSI disk
[  +1,116244] buffer_io_error: 3498 callbacks suppressed
[  +0,000016] Buffer I/O error on dev sdc1, logical block 38374017, async page read
[  +0,000207] Buffer I/O error on dev sdc1, logical block 38374049, async page read
[  +0,000109] Buffer I/O error on dev sdc1, logical block 38374017, async page read
[  +0,003382] Buffer I/O error on dev sdc1, logical block 38374081, async page read
[  +0,000358] Buffer I/O error on dev sdc1, logical block 38374113, async page read
[  +0,000115] Buffer I/O error on dev sdc1, logical block 38374081, async page read
[  +0,005591] Buffer I/O error on dev sdc1, logical block 38374145, async page read
[  +0,000228] Buffer I/O error on dev sdc1, logical block 38374177, async page read
[  +0,000840] Buffer I/O error on dev sdc1, logical block 38374145, async page read
[  +0,002943] Buffer I/O error on dev sdc1, logical block 38374209, async page read
[  +0,160145] scsi 16:0:0:1: Attached scsi generic sg3 type 13
[  +0,047971]  sda: sda1
[  +0,001202] sd 16:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[  +0,001244] sd 16:0:0:0: [sda] Attached SCSI disk
[  +4,783473] buffer_io_error: 3893 callbacks suppressed
[  +0,000017] Buffer I/O error on dev sdc1, logical block 38520130, async page read
[  +0,006422] Buffer I/O error on dev sdc1, logical block 38520194, async page read
[  +0,000415] Buffer I/O error on dev sdc1, logical block 38520226, async page read
[  +0,000144] Buffer I/O error on dev sdc1, logical block 38520194, async page read
[  +0,004098] Buffer I/O error on dev sdc1, logical block 38520258, async page read
[  +0,000234] Buffer I/O error on dev sdc1, logical block 38520290, async page read
[  +0,000098] Buffer I/O error on dev sdc1, logical block 38520258, async page read
[  +0,003047] Buffer I/O error on dev sdc1, logical block 38520322, async page read
[  +0,000409] Buffer I/O error on dev sdc1, logical block 38520354, async page read
[  +0,000153] Buffer I/O error on dev sdc1, logical block 38520322, async page read`

u see now sdc1 because the target drive, with ext4 get auto remounted as read only.

a manually remount did the trick for a few hours copying data...

all my drive in detail:
`osmc@osmc:~$ parted -l
Model: WD My Passport 259D (scsi)
Disk /dev/sda: 4001GB
Sector size (logical/physical): 512B/4096B
Partition Table: gpt
Disk Flags: 

Number  Start   End     Size    File system  Name         Flags
 1      1049kB  4001GB  4001GB  ntfs         My Passport  msftdata

Model: WD My Passport 25E2 (scsi)
Disk /dev/sdb: 4001GB
Sector size (logical/physical): 512B/4096B
Partition Table: gpt
Disk Flags: 

Number  Start   End     Size    File system  Name         Flags
 1      1049kB  4001GB  4001GB  ext4         MyPassport2

Model: SD SL64G (sd/mmc)
Disk /dev/mmcblk0: 63,9GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start   End     Size    Type     File system  Flags
 1      1049kB  256MB   255MB   primary  fat32        lba
 2      257MB   63,9GB  63,6GB  primary  ext4

osmc@osmc:~$ 
`

JamesH65, 2017-09-13:
@p33m Is this covered by other USB issues? Can it be closed?

M-Tomohiro, 2017-09-13:
hi, I explain about my case.

１ｓｔ:"environment"
raspberry pi 2 model B.
SSD connect to USB port.
OS ubuntu for raspberry pi.

Test case is copy 50GB(1000 files) to SSD from PC.
error :  Buffer I/O error on dev sdxx.

2nd:"environment"
raspberry pi 2 model B.(not 1st device, it mean the i have two raspberry pi.  )
SSD connect to USB port.(device is same as １ｓｔ)
OS ubuntu for raspberry pi.(device is same as １ｓｔ)

Test case is copy 50GB(1000 files) to SSD from PC.
It worked very well.

Different point is raspberry pi only. other all environment same as 1st.
So, In my understanding it is a problem of raspberry pi HW.

JamesH65, 2017-12-04:
@p33m @M-Tomohiro Any further progress on this one?

M-Tomohiro, 2017-12-06:
@JamesH65 Yes, there is no progress.

Dreaky255, 2018-02-21:
Same issue for me, and solved.
It was a power issue : My raspberry and hard drive (2Tb) were both powered through an usb hub-powered (2.5A). 
I had a DC power switch (jack) to the hub to easily power on and off my raspberry. The switch was actually too resistent to let pass enough power to the whole thing. It was working with a 1TB hard drive, but not with my brand new 2TB.
After deleting the switch and directly powering the hub with the DC adaptor, problem has disappeared.

Hope it helps some of you.

Sry for my poor english.

JamesH65, 2018-06-27:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

adde88, 2018-07-04:
I would like this issue to remain open, as i've yet to find a fix for it anywhere. And that would be really awesome to get. :)
But if it needs to be closed, i also respect that.

JamesH65, 2018-07-04:
@adde88 I presume that you are using an up to date Raspbian, and that everything is sufficiently well powered?

adde88, 2018-07-04:
No, i'm not. That's why i'm ok with you closing it.
The reason i wanted this open is because i'm having the exact same issue on another emedded device, running OpenWRT. (Me and alot of other people)
My device is running the exact same version of the kernel.
Based on that my gut is telling me that there's "something" with this kernel version.
I also haven't found another place where this issue has been discussed.
So therefore i was hoping that someone found what's the cause of this. If it's power-related, or a kernel-bug, anything would be interesting to find out. As it could lead to the bug being fixed on our devices as well.

But again, i respect your decision to close this, considering my issue is not directly related to the Raspberry Pi.

JamesH65, 2018-07-05:
OK, since this is no longer an RPi issue (as far as I can tell), I'll close it.

closed 2018-07-05 by JamesH65

pelwell, 2018-07-05:
Note that closed issues can still be searched and commented on.

risa2000, 2018-08-31:
FWIW I have just "discovered" the same problem. I am running latest Raspbian Stretch `Linux version 4.14.62-v7+`, on `Raspberry Pi 3 Model B Rev 1.2`. When connecting my brand new USB<>SATA adapter cable called "Qicent" (https://www.amazon.de/gp/product/B01MAWO4V2) which is identified by kernel log as:

```
[    4.541428] usb 1-1.4: new high-speed USB device number 8 using dwc_otg
[    4.672042] usb 1-1.4: New USB device found, idVendor=2537, idProduct=1068
[    4.672055] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    4.672063] usb 1-1.4: Product: NS106X
[    4.672072] usb 1-1.4: Manufacturer: Norelsys
[    4.672080] usb 1-1.4: SerialNumber: 0123456789ABCDE
[    4.672813] usb-storage 1-1.4:1.0: USB Mass Storage device detected
[    4.675619] scsi host1: usb-storage 1-1.4:1.0
...
[    9.258261] scsi 1:0:0:0: Direct-Access     Generic  HDD              0    PQ: 0 ANSI: 0
[    9.259226] sd 1:0:0:0: Attached scsi generic sg1 type 0
[    9.261147] sd 1:0:0:0: [sdb] Attached SCSI removable disk
```

It seems the `Manufacturer` and `Product` is similar to the already reported bridge chips above. The `sudo lsusb -v -d 2537:1068` shows:

```
Bus 001 Device 008: ID 2537:1068
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x2537
  idProduct          0x1068
  bcdDevice            1.00
  iManufacturer           1 Norelsys
  iProduct                2 NS106X
  iSerial                 3 0123456789ABCDE
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
[... 472 lines cut ...]
[  149.206806] ---[ end trace 406cdc4a71053c24 ]---
[  183.088969] ------------[ cut here ]------------
[  183.088998] WARNING: CPU: 0 PID: 499 at fs/fs-writeback.c:2162 __mark_inode_dirty+0x2f4/0x408
[  183.089004] bdi-block not registered
[  183.089009] Modules linked in: aes_arm_bs crypto_simd cryptd algif_skcipher af_alg bnep hci_uart btbcm serdev bluetooth ecdh_generic sg cdc_acm brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd fixed uio_pdrv_genirq uio ip_tables x_tables ipv6
[  183.089140] CPU: 0 PID: 499 Comm: rs:main Q:Reg Tainted: G        WC      4.14.62-v7+ #1134
[  183.089145] Hardware name: BCM2835
[  183.089175] [<8010ffd8>] (unwind_backtrace) from [<8010c240>] (show_stack+0x20/0x24)
[  183.089191] [<8010c240>] (show_stack) from [<80787284>] (dump_stack+0xd4/0x118)
[  183.089208] [<80787284>] (dump_stack) from [<8011da6c>] (__warn+0xf8/0x110)
[  183.089221] [<8011da6c>] (__warn) from [<8011dacc>] (warn_slowpath_fmt+0x48/0x50)
[  183.089236] [<8011dacc>] (warn_slowpath_fmt) from [<802bb818>] (__mark_inode_dirty+0x2f4/0x408)
[  183.089256] [<802bb818>] (__mark_inode_dirty) from [<80429628>] (f2fs_set_page_dirty_nobuffers+0x114/0x144)
[  183.089275] [<80429628>] (f2fs_set_page_dirty_nobuffers) from [<8042978c>] (f2fs_set_data_page_dirty+0x134/0x148)
[  183.089294] [<8042978c>] (f2fs_set_data_page_dirty) from [<8022c284>] (set_page_dirty+0x84/0xcc)
[  183.089312] [<8022c284>] (set_page_dirty) from [<804222e0>] (f2fs_write_end+0x11c/0x2a4)
[  183.089328] [<804222e0>] (f2fs_write_end) from [<8021e830>] (generic_perform_write+0x118/0x1b0)
[  183.089343] [<8021e830>] (generic_perform_write) from [<802210c8>] (__generic_file_write_iter+0x19c/0x1f8)
[  183.089361] [<802210c8>] (__generic_file_write_iter) from [<8040621c>] (f2fs_file_write_iter+0xe4/0x218)
[  183.089380] [<8040621c>] (f2fs_file_write_iter) from [<8028b05c>] (__vfs_write+0xe0/0x138)
[  183.089396] [<8028b05c>] (__vfs_write) from [<8028b27c>] (vfs_write+0xb4/0x1bc)
[  183.089412] [<8028b27c>] (vfs_write) from [<8028b4cc>] (SyS_write+0x54/0xb0)
[  183.089429] [<8028b4cc>] (SyS_write) from [<80108060>] (ret_fast_syscall+0x0/0x28)
[  183.089437] ---[ end trace 406cdc4a71053c25 ]---
```

oskar-skog, 2018-09-23:
Could it be the USB-SATA adapter that's buggy?  I've had severe issues with a Revoltec enclosure, several dead sectors when using it.

risa2000, 2018-09-24:
> Could it be the USB-SATA adapter that's buggy?
It could, but when I tried it in normal PC (Windows 7), there was no problem (but I probably did not have other devices on the same USB hub either).
The fact that it completely resets the USB interface points to the hardware bug, rather than software, but whether it is the bridge chip, or the USB hub in RPi, I do not know, nor if it is really HW bug, or some linux kernel mechanism.

adde88, 2018-12-13:
I've done ALOT of debugging on my OpenWRT device. (Pineapple NANO)
I managed to cross-compile the latest kernel (4.14.81) in the hope that this would squash the bug.
But the same error keeps dropping. Though i'm noticing a pattern.

In ALL of the error messages it failed while using the SCSI OPCODE: 0x28 (READ)
So, i'm currently trying to locate exactly which driver is failing, as the kernel have loaded both OHCI and EHCI.
I also tried to uninstall the "scsci-generic" module.
It's not using any MMC driver to manage the SD-card.

Here's some info about the device from lsusb:
```
root@Pineapple:~# lsusb
Bus 001 Device 008: ID 05e3:0745 Genesys Logic, Inc. Logilink CR0012
Bus 001 Device 003: ID 0cf3:9271 Qualcomm Atheros Communications AR9271 802.11n
Bus 001 Device 002: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@Pineapple:~# lsusb -v -d 05e3:0745

Bus 001 Device 008: ID 05e3:0745 Genesys Logic, Inc. Logilink CR0012
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0745 Logilink CR0012
  bcdDevice            9.33
  iManufacturer           0
  iProduct                1 USB Storage
  iSerial                 2 000000000933
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength           12
  bNumDeviceCaps          1
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      Link Power Management (LPM) Supported
Device Status:     0x0000
  (Bus Powered)
```

I'll provide more feedback if  i find anything of interest.

[2 more replies]