raspberrypi/linux #782: adjtimex calls failing on 3.18.5

rmenessec, 2015-02-01:
The new code in 3.18.4 for the bcm2708 pps_gpio_pin parameter was extremely welcome, but now I'm stuck on that kernel.

Immediately after upgrading to 3.18.5 yesterday, I started getting these failures from chronyd:
```
Jan 31 13:03:50 localhost chronyd[3496]: chronyd version 1.31 starting
Jan 31 13:03:50 localhost chronyd[3496]: Frequency 38.867 +/- 0.011 ppm read from /var/lib/chrony/chrony.drift
Jan 31 13:03:50 localhost chronyd[3496]: Fatal error : adjtimex failed for set_frequency, freq_ppm=3.8867e+01 required_freq=-3.8867e+01 required_tick=10000
Jan 31 13:10:19 localhost chronyd[3564]: chronyd version 1.31 starting
Jan 31 13:10:19 localhost chronyd[3564]: linux_hz directive is no longer supported
Jan 31 13:10:19 localhost chronyd[3564]: Frequency 38.867 +/- 0.011 ppm read from /var/lib/chrony/chrony.drift
Jan 31 13:10:19 localhost chronyd[3564]: Fatal error : adjtimex failed for set_frequency, freq_ppm=3.8867e+01 required_freq=-3.8867e+01 required_tick=10000
```
I found a few references to chronyd failing on the adjtimex call when presented with odd kernel HZ values or the NOHZ option. I tested booting with 'nohz=off' and I tested specifying the kernel HZ value in chrony.conf ('linux_hz', seen above), but got nowhere. Help?

chron0, 2015-02-07:
Hi, I've stumbled over a very similar issue with 3.10.67 on a Odroid C1 though. I don't know it it's related but since I've found my problem here too, maybe my solution will help there too. I need to tweak kernel/time/ntp.c to get chrony to work again, see https://apollo.open-resource.org/mission:log:2015:02:06:patching-kernel-ntp-to-fix-odroid-armv7a-chrony-adjtimex-bug

kukabu, 2015-02-08:
It's upstream bug. https://bugzilla.kernel.org/show_bug.cgi?id=92481

pelwell, 2015-02-08:
Thanks for that.

chron0, 2015-02-08:
Patched, compiled, tested: Works (for Odroid C1 at least). Thanks for the update kukabu.

vk4tux, 2015-02-08:
chrono pls contact vk4tux at bigpond dot com re: C1?

chron0, 2015-02-08:
Should have mail now :)

popcornmix, 2015-02-08:
Please run rpi-update and test.
That includes the patch referenced by @kukabu.

clivem, 2015-02-08:
Is this patch going to be committed to rpi-3.18.y now it has "passed" testing? ;)

pelwell, 2015-02-08:
It's already in there: 772fa1092af0c1109890fe93a151b36c012afca1

popcornmix, 2015-02-12:
@rmenessec Can you test and close if okay?

rmenessec, 2015-02-13:
Testing with today's 3.18.y (.7). chrony starts and runs again, but my PPS signal is unstable. I'm unclear whether this might be a problem with the patch, changes from .6 to .7, or none of the above. Any thoughts?

Ruffio, 2016-08-10:
@rmenessec has this issue been resolved? If yes, please close this issue. Thanks.

rmenessec, 2016-08-10:
@Ruffio I'm not aware of any action that's been taken to resolve this, as you can see from the thread.

Ruffio, 2016-08-10:
@rmenessec I see multiple references on 25 Feb 2015 plus a request to test it. Have you tried with the latest firmware? Current kernel version 4.4 and you refer to 3.18. Could you test with latest version?

rmenessec, 2016-08-20:
I don't even have the system in question any more. The voltage converter failed a year ago. What happened to asking me to test back in the beginning of 2015?

Ruffio, 2016-08-21:
@rmenessec, how should I know? I'm an independent C# developer that just happens to think that RPi is great product, and I want to contribute to make it better. I can't fix anything on RPi, but I can help with administrative tasks, such as takng a look on all old issues and ask people if they are still relevant. IMHO if the real issue has been resolved or doesn't show up anymore/no more relevant, then the issue should be closed as there is no real issue. Alot of the old and untouched issues (for a year) has been closed now after I asked people to retest. I think that I have posted/asked people to verify their issue 350 times now, which you would have noticed if you have followed/watched Fully and not just when you are referenced. So why haven't you been asked before to test? I guess that nobody before me has taken this task on themself to ping people/old issues.

closed 2016-08-23 by popcornmix