raspberrypi/linux #786: 3.18.y booting issue

ha5di, 2015-02-02:
In our distribution, piCore we are booting an initrd.gz file instead of having root file system on SD card partition. WIth kernel/firmware before 3.18.y w/o DT it works with the thes settings:

config.txt:

#-----------------------------------------------------------------------
# Specify iniramfs
ramfsfile=6.1beta1.gz
ramfsaddr=0x00a00000
#-----------------------------------------------------------------------

cmdline.txt:

dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/ram0 elevator=deadline rootwait nortc loglevel=3 noembed

Unfortunately it doesn't work with 3.18.y I get invalid root message. Can you advice how to boot it? 

Thanks... Bela

pelwell, 2015-02-02:
First of all, as a temporary measure you can disable DT by adding:
```
device_tree=
```
to your config.txt.

To help me investigate this, could you upload your initrd.gz somewhere where I can download it for testing purposes?

pelwell, 2015-02-02:
Are you running the latest firmware? Before disabling DT could you take a look in /proc/device-tree/chosen - there should be a "linux,initrd-start" and "linux,initrd-end". If so, run "od -Ax -tx1" on them.

Finally (for now), can you capture the output of dmesg with and without DT.

ha5di, 2015-02-02:
Hi Phil,

thanks for the fast response. Will retest it with the current firmware and kernel and be back (I tested last a week ago 3.18.3+).

Béla

ghollingworth, 2015-02-04:
Try using 

ramfsaddr=-1

amtssp, 2015-02-05:
Thanks Ghollingworth 

That was a good suggestion it seems to boot now, but I then another problem occurred - probably an error on my side.

pelwell, 2015-02-06:
Are you happy to close this?

ha5di, 2015-02-06:
Thanks for all. ramfsaddr=-1 solved the booting issue. Our startup procedure now hangs on at udev daemon start, but it is another issue.

I'm closing this issue.

ha5di, 2015-03-01:
It was working fine with 3.8.6 and dated kernel. Now with 3.18.8 and the current firmware it doesn't work anymore. With root=/dev/ram0 in cmdline.txt and ramfsfile=... in config.txt kernel do not find root file system. Adding ramfsaddr=-1 kernel do not boot, no messages only the rainbow coloured screen.

How to boot initrd now?

closed 2015-03-01 by ha5di