raspberrypi/linux #787: Pi NoIR Camera is hanging. Only power off solves the issues.

brguy, 2015-02-02:
########################
Background Information:
########################
I built a smart door solution with a RPi B+ (Raspbian) which makes use of three sensors (PIR, Gyroscope [I2C] and an Ultrasound). Depending on the sensor status managed by a Python script,  I start recording a video by using the raspivid command and/or play sounds on Bluetooth speakers. I also have a USB WIFI dongle connected to the RPi.

This is how I call it in my .sh file:
raspivid -w 800 -h 600 -t 15000 -o file.h264 -n -rot 270

########################
Problem description:
########################
The camera works fine for a while (I cannot precise for how long [2 hours maybe] nor what triggers the issue), but then it stops working. The funny thing is that I get no error message. 

########################
What did I do to identify the root cause:
########################
- I tried to manually kill the child process (Raspivid) but it simply doesn't work. If I kill the parent process (.sh file) then my child process gets assigned to the PID 1 (init.d).

- I tried recording a new video in a new terminal window but the new process hangs too. 

- Rebooting the Pi doesn't work neither. It says it is rebooting but it does not (not even with the -f option).

Have you ever faced this issue? How can I get it fixed?
Thanks!

JamesH65, 2015-02-02:
Can you try it without the rot command? Then without -n if the rot change
makes no difference.

Also, check you are not running out of space on the SD card.

brguy, 2015-02-02:
Hi James,

I tried removing the parameters you mentioned and no difference.
SD card space is alright. Still 4GB left and each video is no bigger than 1,2 MB.
Thanks!

brguy, 2015-02-04:
Hi everyone,
Any ideas?
Thanks!

lurch, 2015-02-23:
If you comment out the call to raspivid, does the rest of your application then work as expected for more than 2 hours? IMHO the first step in trying to debug a complicated problem like this, is to try and narrow down in exactly which combinations (you mentioned you're also using a whole bunch of sensors) things do and don't work as expected.

With so many things connected to your Pi, I assume you're using a decent power supply?

dflogeras, 2016-06-23:
Was this problem ever resolved?  I am seeing (what I think) is the same thing.

Ruffio, 2016-08-10:
@brguy has this issue been resolved? If yes, please close this issue. Thanks.

dflogeras, 2016-08-11:
My setup is still locking up (same issue as OP), as outlined in this thread https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=152967

Has something changed recently (I guess since my last post in Jun) that might have fixed this?

dflogeras, 2016-08-20:
I had some time this week, and retested on latest (4.4.17-g157e000) sources, and the problem is still present.

@6by9 any chance to look at the output of vgdbg I put in the aforementioned thread post?

6by9, 2016-08-22:
@dflogeras You're not reporting exactly the same issue as here. You're using Motion via V4L2. This is raspivid.
I'll comment on your forum post regarding the vcdbg output.

6by9, 2016-08-22:
Regarding the original issue, somehow the camera subsystem is locking up during shutdown, and as it only supports a single client, it won't start up again without rebooting the GPU.
It's not something I have seen, but it is likely to be a failure at some critical moment - there were a lot of cleanups of the code to address these on production platforms, so I wonder if some have been missed from the Pi branch. Finding them could be tricky though.

JamesH65, 2017-05-17:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

Alternatively, use the forums for further questions.

closed 2017-05-17 by JamesH65