raspberrypi/linux #788: 3.18.5: HiFiBerry Amp not working at all

hifiberry, 2015-02-02:
The device_tree_overlay for the HiFiBerry Amp did not really work in 3.18.4. However using device_tree= it was possible to load the modules. 
In the new 3.18.5 kernel, this isn't working at all.

The drivers are loaded, but won't be initialised. Any idea where to start debugging this?

pelwell, 2015-02-02:
Is this the only one of your boards that isn't working? If so, that would tend to point the finger at the codec.

There are a few things you can try:

1) set "dtdebug=1", reboot, and check the output of "sudo vcdbg log msg" for DT errors.

2) After the reboot, capture the output of dmesg, lsmod, and the contents of /etc/modules /etc/modprobe.d/raspi-blacklist.conf and put them somewhere downloadable.

3) Take a look in /proc/device-tree for the nodes you are expecting and see if they are there. You could also capture the content into a .tar file and put it somewhere downloadable.

4) "sudo modprobe i2c-dev", then "i2cdetect -y 1" to see if the codec is showing up.

hifiberry, 2015-02-02:
Thanks, I will look into this. I'm not sure if it is the only bard with this issue.

pelwell, 2015-02-02:
If I get some time I can test some of the others (I don't have an "Amp").

hifiberry, 2015-02-02:
Hi Phil, this is something we have to change ;-) Send a mail with your address to sales at hifiberry.com, I will make sure that you get an Amp+.

pelwell, 2015-02-02:
Done - thank you!

hifiberry, 2015-02-02:
I don't have a /proc/device-tree. Is there anything I need to enable this?

BTW: I think i've found the problem, just don't know how exactly to fix it. 
The 
MODULE_DEVICE_TABLE
was missing for the Amp. I've added it, but the module still won't be loaded automatically (modprobe works). Any hint?

hifiberry, 2015-02-02:
@pelwell : What compilation option do you use for the standard kernel? If I make the kernel using make bcmrpi_defconfig, I don't have a /proc/device-tree, but it is there when using the kernel that comes with rpi-update.

notro, 2015-02-02:
You need to tag the kernel. This is from my build script log:
```
/home/pi/tools/mkimage/mkknlimg /home/pi/rpi-firmware/fbtft/workdir/linux/arch/arm/boot/zImage /home/pi/rpi-firmware/fbtft/workdir/out/kernel.img
```
https://github.com/raspberrypi/tools/blob/master/mkimage/mkknlimg

pelwell, 2015-02-02:
Thanks, Notro. Yes, that's right.

The way the kernel boot process works,  you have to choose between DT and ATAGs. Passing a DT blob to a kernel that doesn't expect it is an easy way to stop your Pi booting.

To solve the problem of backwards compatibility, we needed a way of knowing if a kernel would understand the DTB, but that is hard when the kernel is compressed. By requiring a special trailer on the kernel to enable DT support, we moved the problem to the build process, which can spare a second or two to decompress and analyse the kernel image.

Had you got that far, the vcdbg output would have said that the kernel lacked the required trailer.

hifiberry, 2015-02-02:
Ok, I see: "001182.934: No kernel trailer (run mkknlimg to fix) - assuming not DT-capable". I will have a closer look into this tomorrow.

hifiberry, 2015-02-03:
Looks like there was still some code missing in the Amp driver to fully work with device_tree_overlays.
I've created a patch for this:
https://github.com/raspberrypi/linux/pull/790

pelwell, 2015-02-06:
Are you happy to close this?

hifiberry, 2015-02-06:
Yes, latest kernel works. Thank you.

closed 2015-02-06 by hifiberry