raspberrypi/linux #793: rpi 3.18.y-rebase and Kernel compile with touchscreen

MrChristo59, 2015-02-04:
Hello all,
I have recompiled the new kernel for rasp 2 with the eGalax driver enables in menuconfig.
But rasbian won't load, I get the 4 color wallpaper...
Is their something different to do for the new pi ?

popcornmix, 2015-02-04:
Did you use the right config? Start with:
ARCH=arm make bcm2709_defconfig

MrChristo59, 2015-02-04:
Yes I did, everything went well.

popcornmix, 2015-02-04:
Did you replace kernel7.img (rather than kernel.img?)

MrChristo59, 2015-02-04:
Oh dear, i'm such a mess.
I did replace kernel7.img but didn't understand why there was a "7" so i renamed it...
I think it will work better now.

Thanks for your help and your good work

MrChristo59, 2015-02-04:
Actually, that didn't fix the problem... 
Does the "./imagetool-uncompressed.py" need to be update to create the kernel7.img ?

popcornmix, 2015-02-04:
No, just renaming Image or zImage to kernel7.img should be enough.
If you need device tree then you need to run mkknlimg:
https://github.com/raspberrypi/tools/blob/master/mkimage/mkknlimg

but I'd still expect kernel to boot without DT without doing that.

Can you try just building an unchanged kernel with the bcm2709_defconfig and no other changes?

MrChristo59, 2015-02-04:
Ok I try that.

MrChristo59, 2015-02-04:
Compiling...
Here is the code I use to compile kernel, in case you see something
strange. It worked for 3.18.y for the rasp B+.

"
make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabi- bcm2709_defconfig
make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabi- menuconfig // to
enable eGalax driver
make ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabi- -k -j6

make modules_install ARCH=arm CROSS_COMPILE=/usr/bin/arm-linux-gnueabi-
INSTALL_MOD_PATH=../modules/

./imagetool-uncompressed.py ../../linux/arch/arm/boot/Image

Then copy kernel.img, modules and firwmare path.
"

Compiled on a Mint distrib

popcornmix, 2015-02-04:
Don't use ./imagetool-uncompressed.py
Just copy Image (or zImage for a smaller kernel) to kernel7.img.

MrChristo59, 2015-02-04:
I worked. Sorry you told me that before but I didn't understand.
It runs... But still a problem, when I start x, control are disabled... No
touchscreen and not even usb control (mouse/keyboard).
I guess that's another issue

MrChristo59, 2015-02-04:
maybe you can help me with that error : libkmod: "ERROR
../libkmod/libkmod.c:554 kmod_search_moddep: could not open moddep file
'/lib/modules/3.18.5-v7/modules.dep.bin'"

2015-02-04 18:53 GMT+01:00 Christopher Coat <christo.coat@gmail.com>:

popcornmix, 2015-02-04:
Looks like you are not setting the modules path. E.g.
make ARCH=arm INSTALL_MOD_PATH=../modules modules_install

MrChristo59, 2015-02-04:
Oh dear, it was simply a copy failure.... Today is not a good day for me...
Anyway it works very well and I want to *thank you a lot.*

For the people who come here and want to make it works.
Simply use my kernel and modules for 3.18.5-rebase :
https://1fichier.com/?7uwem02si9

closed 2015-02-21 by popcornmix