raspberrypi/linux #797: making rpi-3.18.y met some errors "unrecognized command line option"

zaazbb, 2015-02-05:
os: ubuntu 14.04
rpi: rpi-3.18.y- a6cf3c99bc89e2c010c2f78fbf9e3ed478ccfd46
cross compiler: gcc-4.7-linaro-rpi-gnueabihf (from internet)

1, download "2015-01-31-raspbian.zip" from raspberry pi website, write it to sd card, copy /proc/config.gz to pc, unzip it, rename to .config, and copy it to rpi-3.18.y/
2, add new touchscreen driver code, and enable it in "make menuconfig"
3, make  it, 
```text
make ARM=arm CROSS_COMPILE=/home/jf/work/raspberry/gcc-4.7-linaro-rpi-gnueabihf/bin/arm-linux-gnueabihf-
```

when building, met some errors as below:
```txt
scripts/kconfig/conf --silentoldconfig Kconfig
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/basic/bin2c
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf --silentoldconfig Kconfig
  SYSTBL  arch/x86/syscalls/../include/generated/asm/syscalls_32.h
  SYSHDR  arch/x86/syscalls/../include/generated/asm/unistd_32_ia32.h
  SYSHDR  arch/x86/syscalls/../include/generated/asm/unistd_64_x32.h
  SYSTBL  arch/x86/syscalls/../include/generated/asm/syscalls_64.h
  SYSHDR  arch/x86/syscalls/../include/generated/uapi/asm/unistd_32.h
  SYSHDR  arch/x86/syscalls/../include/generated/uapi/asm/unistd_64.h
  SYSHDR  arch/x86/syscalls/../include/generated/uapi/asm/unistd_x32.h
  HOSTCC  arch/x86/tools/relocs_32.o
  HOSTCC  arch/x86/tools/relocs_64.o
  HOSTCC  arch/x86/tools/relocs_common.o
  HOSTLD  arch/x86/tools/relocs
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  WRAP    arch/x86/include/generated/asm/clkdev.h
  WRAP    arch/x86/include/generated/asm/cputime.h
  WRAP    arch/x86/include/generated/asm/dma-contiguous.h
  WRAP    arch/x86/include/generated/asm/early_ioremap.h
  WRAP    arch/x86/include/generated/asm/mcs_spinlock.h
  WRAP    arch/x86/include/generated/asm/scatterlist.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  CC      kernel/bounds.s
arm-linux-gnueabihf-gcc: error: unrecognized command line option ‘-m64’
arm-linux-gnueabihf-gcc: error: unrecognized command line option ‘-mno-red-zone’
arm-linux-gnueabihf-gcc: error: unrecognized command line option ‘-mcmodel=kernel’
arm-linux-gnueabihf-gcc: error: unrecognized command line option ‘-mno-sse’
arm-linux-gnueabihf-gcc: error: unrecognized command line option ‘-mno-mmx’
arm-linux-gnueabihf-gcc: error: unrecognized command line option ‘-mno-sse2’
arm-linux-gnueabihf-gcc: error: unrecognized command line option ‘-mno-3dnow’
make[1]: *** [kernel/bounds.s] Error 1
make: *** [prepare0] Error 2
```

oniongarlic, 2015-02-05:
You have a typo, it should be "ARCH=arm" not "ARM=arm"

zaazbb, 2015-02-05:
yes, I modify it , rebuild,  met a new error:
```test
fs/btrfs/extent_io.c: In function ‘btrfs_free_io_failure_record’:
fs/btrfs/extent_io.c:2166:13: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
make[2]: *** No rule to make target `fs/btrfs/volumes.o', needed by `fs/btrfs/btrfs.o'.  Stop.
make[1]: *** [fs/btrfs] Error 2
make: *** [fs] Error 2
```

pelwell, 2015-02-05:
It's probably worth a:
```
make ARCH=arm clean
```
to get the world back into a sane state.

zaazbb, 2015-02-05:
make clean also met a error.
```text
jf@ubuntu:~/work/raspberry/linux-rpi-3.18.y$ make ARCH=arm clean
  CLEAN   .
  CLEAN   arch/arm/kernel
./scripts/Makefile.clean:17: fs/fscache/Makefile: File name too long
make[2]: stat: fs/fscache/Makefile: File name too long
make[2]: *** No rule to make target `fs/fscache/Makefile'.  Stop.
make[1]: *** [fs/fscache] Error 2
make: *** [_clean_fs] Error 2
```

pelwell, 2015-02-05:
What kind of filing system are you compiling in?

zaazbb, 2015-02-05:
I do not know, I am under ubuntu 32bit,

zaazbb, 2015-02-05:
ubuntu 14.04 32bit, ext4.
Ext4 (version 1.0) — Mounted at Filesystem Root.

pelwell, 2015-02-05:
The fact that your kernel is complaining about clock skew is worrying. You often see that using NFS when the clock differs between the server and the client.

I think this isn't a problem in the source, and that you have managed to get your installation into a peculiar state. I suggest you commit your changes, then use git format-patch to generate a patch. Then delete the entire tree, git clone it afresh, git am to apply your patch, then try the build steps again.

zaazbb, 2015-02-05:
I'll try it tomorrow. 
thank you very much.

zaazbb, 2015-02-05:
I delete the linux dir, and reunzip the linux-05851b0ab113773aeefbe98bee94298fa3cf646c.zip,
create default config
```text
jf@ubuntu:~/work/raspberry/linux-05851b0ab113773aeefbe98bee94298fa3cf646c$ make bcmrpi_defconfig ARCH=arm CROSS_COMPILE=/home/jf/work/raspberry/gcc-4.7-linaro-rpi-gnueabihf/bin/arm-linux-gnueabihf-
```
then, make.
```text
jf@ubuntu:~/work/raspberry/linux-05851b0ab113773aeefbe98bee94298fa3cf646c$ make bcmrpi_defconfig ARCH=arm CROSS_COMPILE=/home/jf/work/raspberry/gcc-4.7-linaro-rpi-gnueabihf/bin/arm-linux-gnueabihf-
```
first,  stoped at a broken linke file
```text
In file included from fs/btrfs/super.c:60:0:
fs/btrfs/backref.h:23:19: fatal error: ulist.h: No such file or directory
compilation terminated.
make[2]: *** [fs/btrfs/super.o] Error 1
make[1]: *** [fs/btrfs] Error 2
make: *** [fs] Error 2
```
i rename it as ulist(copy).h, and download a new ulist.h file.
then, make again.
i met the same error.
```text
fs/btrfs/extent_io.c: In function ‘btrfs_free_io_failure_record’:
fs/btrfs/extent_io.c:2166:13: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
make[2]: *** No rule to make target `fs/btrfs/volumes.o', needed by `fs/btrfs/btrfs.o'.  Stop.
make[1]: *** [fs/btrfs] Error 2
make: *** [fs] Error 2
```

P33M, 2015-02-05:
Github is notorious for corrupting .zip images of large repositories.

Retry pulling the source code from Github with `git clone https://github.com/raspberrypi/linux.git`.

zaazbb, 2015-02-05:
ok, I am cloning use git, i'll try again later.

zaazbb, 2015-02-06:
now, build finished,  how i can use new kernel?

pelwell, 2015-02-06:
Please close this issue, then go to the RPi forums and ask there.

zaazbb, 2015-02-06:
ok. thank you.

closed 2015-02-06 by zaazbb