raspberrypi/linux #798: setting /sys/class/gpio/gpio<xx>/direction to 'high' no longer works as expected in 3.18.5

s8mlu, 2015-02-05:
Before upgrading to 3.18.5, the following 2 commands would correctly export the pin and set the direction to output with a value of HIGH (or 3.3v on the pin). But now, after these 2 commands, the output is  LOW!
<code>samlu@rpi1 ~ $ sudo sh -c "echo 27 >/sys/class/gpio/export"</code>
<code>samlu@rpi1 ~ $ sudo sh -c "echo high >/sys/class/gpio/gpio27/direction"</code>

The [kernel documentation](https://www.kernel.org/doc/Documentation/gpio/sysfs.txt) is pretty clear on what 'high' in direction should mean.

Thanks for looking into this.

P33M, 2015-02-05:
@pelwell - do you want to triage or shall I?

pelwell, 2015-02-05:
I will do it, but not today.

notro, 2015-02-05:
Judging from gpiolib-sysfs.c logic: http://lxr.free-electrons.com/source/drivers/gpio/gpiolib-sysfs.c#L72
Then this commit should have fixed the issue: https://github.com/raspberrypi/linux/commit/c288c0ec3edd40e090f49c5575c15cd0d9dac413

I don't know if there has been a kernel build since that commit. Try ```rpi-update``` and see if you get a new kernel.

popcornmix, 2015-03-31:
@s8mlu is this still an issue with latest kernel?

s8mlu, 2015-04-01:
I just tried it on kernel version 3.18.7 and it now works properly. Thanks!

closed 2015-04-01 by popcornmix