raspberrypi/linux #800: kernel 3.18: /sys/class/leds/led0 no longer exists; has been renamed to ACT

IvanExpert, 2015-02-06:
Prior to 3.18, the ACT/OK led trigger was at /sys/class/leds/led0. As of 3.18, this is now /sys/class/leds/ACT. This breaks compatibility with existing scripts/code that reference "led0". I would imagine a symlink that also points "led0" to the same place ACT does (or points to ACT itself) would be a way of resolving the issue.

The first rpi-firmware commit to use 3.18 (20-Jan-2015, cad071ac50ddf8a9f3c7a5d808cfa0daf5effb45) is the first one to exhibit this change, and it persists through the latest.

pelwell, 2015-02-06:
Shame - "ACT" was a nicer name.

You can grab updated DTBs now from the firmware repo, in the boot folder. Copy them to your /boot directory and reboot. They will be shipped as standard in the next update.

pelwell, 2015-02-09:
This change has now rolled out to the latest "rpi-update" release, if you'd care to try that.

IvanExpert, 2015-02-27:
Verified fixed on Raspbian 2015-02-16. (And, I do share your sympathies, but I don't see why there couldn't be a symlink to ACT, if led0 is deemed too prosaic.)

closed 2015-02-27 by popcornmix