raspberrypi/linux #801: kernel 3.18: udev rule won't run command during startup

IvanExpert, 2015-02-06:
This may be more broadly applicable than my specific example, but: prior to 3.18, a udev rule could and would run a command, even during startup. In Raspbian with kernel 3.18, it will create symlinks on startup if requested, but not run a command. It will, however, run a command on hotplug, or on manual trigger from the command line.

To reproduce:
* install Raspbian 2014-12-25 (kernel 3.12.35) on a B or B+
* create a udev rule by typing:
    echo 'KERNEL=="ttyUSB*", KERNELS=="1-1.3:1.0", SYMLINK+="ttyUSBlower", RUN+="/bin/touch /usr/local/udev-succeeded"' | sudo tee /etc/udev/rules.d/50-usb.rules

* Attach a USB to serial adapter to the lower port (or lower left port on a B+). I used a PL2303 based adapter.
* type `ls /usr/local` and you should see "udev-successful", meaning the script was executed on device hotplug
* type `rm /usr/local/udev-successful`

* reboot: `sudo shutdown -r now`
* type `ls /usr/local` and you should still see "udev-successful", meaning the command ran during boot as expected
* type `sudo rm /usr/local/udev-successful`

* upgrade the kernel to 3.18: `sudo rpi-update`
* reboot: `sudo shutdown -r now`
* type `ls /usr/local` and you will no longer see "udev-successful", meaning the command didn't run during startup
* type `ls -l /dev/tty*` and you will see "ttyUSBlower" meaning the the rule did in fact match because the symlink exists -- just the command didn't run

* type `sudo udevadm trigger --action=change`
* type `ls /usr/local` and you will see "udev-succesful", meaning the script did execute when udev was manually triggered
* sudo type `sudo rm /usr/local/udev-successful`

The workaround I have come up with is to have the script run by udev drop a semaphore in /tmp; then code added to /etc/rc.local checks to see if the semaphore is there and perfroms the manual trigger if not.

Also, I checked this against Ubuntu Server 14.10 32-bit x86, with kernel 3.18, and it did not exhibit the same issue (that is, the USB-to-Serial adapter was recognized during startup, just as it was with the 3.12 and before Raspbian kernels).

pelwell, 2015-02-06:
I presume your "udev->succeeded"/"udev-successful" difference is just a typo.

Can you show the details of your workaround - it might shed some light on the proble,

IvanExpert, 2015-02-06:
Yes, that was a typo. The workaround looks like this:

udev rule:
`KERNEL=="ttyUSB*", KERNELS=="1-1.3:1.0", SYMLINK+="ttyUSBlower", RUN+="/usr/local/sbin/ttyusbhandler add ttyUSBlower"`

The contents of /usr/local/sbin/ttyUSBhandler look like:
`(do stuff)
rm /tmp/udev-ttyUSBlower-added &> /dev/null
touch /tmp/udev-ttyUSBlower-added`

The contents of /etc/rc.local contain before `exit 0`:
`[ -e /dev/ttyUSBlower ] && ! [ -f /tmp/udev-ttyUSBlower-added ] && udevadm trigger --action=change`

This means that if /etc/rc.local finds the symlink for the device (meaning the udev rule matched), but not the semaphore created by the script that's supposed to run (meaning the udev rule failed to do the other half of its job), /etc/rc.local manually triggers udev. When it does, udev operates correctly, and the script runs, since the symptom only seems to happen during startup.

Ruffio, 2016-08-10:
@IvanExpert has this issue been resolved? If yes, please close this issue. Thanks.

IvanExpert, 2016-08-10:
I don't know that it has been resolved, as there's no further comment here since I posted my workaround, but I can attempt to reproduce on whatever is current and see if it still exists.

closed 2017-05-17 by JamesH65