raspberrypi/linux #803: tinc 1.0.19 as included in raspbian gives a segmentation fault and is unuseable

maxheise, 2015-02-06:
Hello,

I installed raspbian on a raspberry pi b+ and tried to install tinc, which is a vpn package.

The version 1.0.19 which is installable from the raspbian repository gets a segmentation fault after a short period of a few seconds of use. That version of tinc is totally unuseable.

I then added a deb-src for wheezy-backports to sources.list and compiled the tinc version from the wheezy backports and created a .deb, which I installed and which runs fine for a day now.

$ apt-cache showpkg tinc
Package: tinc
Versions: 
1.0.24-2~bpo70+1 (/var/lib/dpkg/status)
 Description Language: 
                 File: /var/lib/apt/lists/mirrordirector.raspbian.org_raspbian_dists_wheezy_main_binary-armhf_Packages
                  MD5: 927734357d793d7ebde564ee023d38d2

1.0.19-3 (/var/lib/apt/lists/mirrordirector.raspbian.org_raspbian_dists_wheezy_main_binary-armhf_Packages)
 Description Language: 
                 File: /var/lib/apt/lists/mirrordirector.raspbian.org_raspbian_dists_wheezy_main_binary-armhf_Packages
                  MD5: 927734357d793d7ebde564ee023d38d2

Reverse Depends: 
Dependencies: 
1.0.24-2~bpo70+1 - libc6 (2 2.13-28) liblzo2-2 (0 (null)) libssl1.0.0 (2 1.0.0) zlib1g (2 1:1.1.4) dpkg (18 1.15.4) install-info (0 (null)) 
1.0.19-3 - libc6 (2 2.13-28) liblzo2-2 (0 (null)) libssl1.0.0 (2 1.0.0) zlib1g (2 1:1.1.4) dpkg (18 1.15.4) install-info (0 (null)) 
Provides: 
1.0.24-2~bpo70+1 - 
1.0.19-3 - 
Reverse Provides: 

BR
Max.

maxheise, 2015-02-06:
The seg fault that I am getting is probably this one, since the conditions explained there match my setup.
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=749677

pelwell, 2015-02-06:
This isn't a kernel problem. Please close this issue and use the Raspbian bug reporting page: http://www.raspbian.org/RaspbianBugs

maxheise, 2015-02-06:
I am sorry if I wrongly filed the bug report.

However, in my defense, I must say that http://www.raspbian.org/RaspbianBugs never says anything about that only kernel problem should be filed here.

Anyway, I'll close this one immediately.

closed 2015-02-06 by maxheise