raspberrypi/linux #804: i2c_bcm2708 crashed

oklitz, 2015-02-06:
Hello !

Raspberry Pi Model B with 3.18.5-2-ARCH armv6l GNU/Linux
i2c driver crashes after random time with this message:

[Feb 6 11:40] BUG: scheduling while atomic: ms5611_d/157/0x00000002
[  +0,000185] Modules linked in: rpcsec_gss_krb5 snd_soc_wm8804 snd_soc_pcm512x_i2c snd_soc_pcm512x regmap_spi snd_soc_tas5713 regmap_i2c r8712u(C) snd_soc_bcm2708_i2s regmap_mmio leds_gpio led_class snd_soc_core spi_bcm2708 snd_compress i2c_bcm2708 snd_pcm_dmaengine sch_fq_codel snd_bcm2835 snd_pcm snd_timer snd rtc_ds1307 i2c_dev bcm2708_rng rng_core ipv6
[  +0,001056] Preemption disabled at:[<  (null)>]   (null)

[  +0,000195] CPU: 0 PID: 157 Comm: ms5611_d Tainted: G         C     3.18.5-2-ARCH #1
[  +0,000216] [<c0013d0c>] (unwind_backtrace) from [<c0011a28>] (show_stack+0x10/0x14)
[  +0,000202] [<c0011a28>] (show_stack) from [<c0044b20>] (__schedule_bug+0x58/0xa8)
[  +0,000188] [<c0044b20>] (__schedule_bug) from [<c075c0f0>] (__schedule+0x588/0x658)
[  +0,000182] [<c075c0f0>] (__schedule) from [<c075c56c>] (schedule_preempt_disabled+0x14/0x20)
[  +0,000196] [<c075c56c>] (schedule_preempt_disabled) from [<c075dbd0>] (__mutex_lock_slowpath+0x6c/0x168)
[  +0,000211] [<c075dbd0>] (__mutex_lock_slowpath) from [<c066febc>] (clk_prepare_lock+0x3c/0xdc)
[  +0,000195] [<c066febc>] (clk_prepare_lock) from [<c0670b08>] (clk_get_rate+0xc/0x5c)
[  +0,000200] [<c0670b08>] (clk_get_rate) from [<bf0d7480>] (bcm2708_i2c_master_xfer+0x44/0x2d0 [i2c_bcm2708])
[  +0,000251] [<bf0d7480>] (bcm2708_i2c_master_xfer [i2c_bcm2708]) from [<c05ca27c>] (__i2c_transfer+0x68/0x2b8)
[  +0,024581] [<c05ca27c>] (__i2c_transfer) from [<c05ca560>] (i2c_transfer+0x94/0xc4)
[  +0,025442] [<c05ca560>] (i2c_transfer) from [<c05cac00>] (i2c_smbus_xfer+0x5d4/0x8a0)
[  +0,025631] [<c05cac00>] (i2c_smbus_xfer) from [<bf07140c>] (i2cdev_ioctl_smbus+0x148/0x218 [i2c_dev])
[  +0,025883] [<bf07140c>] (i2cdev_ioctl_smbus [i2c_dev]) from [<bf071af0>] (i2cdev_ioctl+0xb0/0x268 [i2c_dev])
[  +0,025837] [<bf071af0>] (i2cdev_ioctl [i2c_dev]) from [<c01384d0>] (do_vfs_ioctl+0x3f0/0x5dc)
[  +0,026138] [<c01384d0>] (do_vfs_ioctl) from [<c01386f0>] (SyS_ioctl+0x34/0x5c)
[  +0,013329] [<c01386f0>] (SyS_ioctl) from [<c000e300>] (ret_fast_syscall+0x0/0x30)

I have posted this error already in the raspberry pi forum:
http://www.raspberrypi.org/forums/viewtopic.php?f=44&t=98176

 jeanleflambeur said:
> It's because of a mutex lock inside a spin lock protected section.

he posted me a patch:http://pastebin.com/zhz81fpM
and said i have to compile a new kernel with it. But i don't want this way.

I want to know if this is a kernel(driver) problem or my fault that the driver crashes ??
And if it is a driver bug, can you fix it in the mainline kernel ?
(Sorry I'm no kernel hacker, just programming c in userspace)

regards
Marco Oklitz

P33M, 2015-02-06:
The Raspberry Pi kernel is already updated to prevent this from happening. You appear to be running Archlinux, is that correct?

popcornmix, 2015-02-06:
Actually I think the fix is https://github.com/raspberrypi/linux/pull/780 which I've only just merged.
Will be fixed in next kernel build.

popcornmix, 2015-02-08:
Can you run rpi-update and test? That includes #780.

popcornmix, 2015-03-03:
Closing due to no response. Please reopen if still an issue with updated firmware.

closed 2015-03-03 by popcornmix