raspberrypi/linux #807: Mouse Oddity

AndrewdAzotus, 2015-02-06:
I have an interesting problem. I have two SD cards, the older image with the menu button at the bottom and everything works fine. The second image with the menu button at the top and the mouse does not work within x and dmesg shows the mouse disconnecting every minute or so. Everything is the same except for the SD card image.

has anyone else encountered this problem?

popcornmix, 2015-02-08:
What does "uname -a" report on each system?

ckuehnel, 2015-02-08:
See "man uname".

AndrewdAzotus, 2015-02-08:
Upon trying to reboot the Raspberry Pi into the new image it now dos not boot at all, whereas inserting the new image SD Card into another Pi (these are both Model B) the image boots successfully and the mouse works normally.

So, it would seem to be a hardware fault with either the Pi itself or the specific SD card or that particular combination.

So, it would seem I have my solution but many thanks for you suggestions.

FYI, The uname -a output in question is:

Xanadu - 32GB branded Transcend with NOOBS and about 7 images on it.
Old Image
Linux xanadu 3.12.28+ #709 PREEMPT Mon Sep 8 15:28:00 BST 2014 armv6l GNU/Linux

raspberrypi - 32GB branded '7 day shop' [Mouse does not work or even boot now in 1 Pi B]
New Image
Linux raspberrypi 3.18.5+ #744 PREEMPT Fri Jan 30 18:19:07 GMT 2015 armv6l GNU/Linux

gibberd - 64GB microSDXC in SD Adaptor [boots successfully and mouse works]
New style Image
Linux gibberd 3.18.5+ #744 PREEMPT Fri Jan 30 18:19:07 GMT 2015 armv6l GNU/Linux

closed 2015-02-08 by AndrewdAzotus