raspberrypi/linux #808: Interrupts can't be configured for GPIO outputs with 3.18.5

fivdi, 2015-02-07:
With previous versions of the kernel it was possible to configure interrupts for GPIO outputs. With 3.18.5 this is no longer possible. This can be seen in the commands sequence below. Any attempt to set the edge to rising, falling, or both results in an error. 

```bash
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 3.18.5-v7+ #225 SMP PREEMPT Fri Jan 30 18:53:55 GMT 2015 armv7l GNU/Linux
pi@raspberrypi ~ $ cd /sys/class/gpio/
pi@raspberrypi /sys/class/gpio $ echo 4 > export 
pi@raspberrypi /sys/class/gpio $ ls gpio4/
active_low  device  direction  edge  power  subsystem  uevent  value
pi@raspberrypi /sys/class/gpio $ echo out > gpio4/direction 
pi@raspberrypi /sys/class/gpio $ cat gpio4/direction 
out
pi@raspberrypi /sys/class/gpio $ echo rising > gpio4/edge 
-bash: echo: write error: Input/output error
pi@raspberrypi /sys/class/gpio $ echo falling > gpio4/edge 
-bash: echo: write error: Device or resource busy
pi@raspberrypi /sys/class/gpio $ echo both > gpio4/edge 
-bash: echo: write error: Device or resource busy
pi@raspberrypi /sys/class/gpio $ cat gpio4/edge 
none
pi@raspberrypi /sys/class/gpio $ 
```

popcornmix, 2015-02-07:
Duplicate of https://github.com/raspberrypi/linux/issues/791

fivdi, 2015-02-07:
@popcornmix could you reopen this issue please? It's definitely not a duplicate of #791. #791 is about permissions, this issue has nothing to do with permissions, it's about a feature that was available with previous kernels but no longer available in 3.18.5. In particular, it's no longer possible to configure interrupts for a GPIO output in 3.18.5, even then when the user has the permission to access all the appropriate sysfs files.

In the sequence of command shown in the first post above, the user has the permission to access all the appropriate sysfs files and the permissions are functioning correctly. The use has these permission because udev rules were defined to give the user the permissions.

Below is the same sequence of commands, this time for root. As can be seen, the errors also occur for root, so it's not a permission issue. 

```
pi@raspberrypi ~ $ sudo -s
root@raspberrypi:/home/pi# uname -a
Linux raspberrypi 3.18.5-v7+ #225 SMP PREEMPT Fri Jan 30 18:53:55 GMT 2015 armv7l GNU/Linux
root@raspberrypi:/home/pi# cd /sys/class/gpio/
root@raspberrypi:/sys/class/gpio# echo 4 > export 
root@raspberrypi:/sys/class/gpio# ls gpio4/
active_low  device  direction  edge  power  subsystem  uevent  value
root@raspberrypi:/sys/class/gpio# echo out > gpio4/direction 
root@raspberrypi:/sys/class/gpio# cat gpio4/direction 
out
root@raspberrypi:/sys/class/gpio# echo rising > gpio4/edge 
bash: echo: write error: Input/output error
root@raspberrypi:/sys/class/gpio# echo falling > gpio4/edge 
bash: echo: write error: Device or resource busy
root@raspberrypi:/sys/class/gpio# echo both > gpio4/edge 
bash: echo: write error: Device or resource busy
root@raspberrypi:/sys/class/gpio# cat gpio4/edge 
none
root@raspberrypi:/sys/class/gpio# 
```

pelwell, 2015-02-07:
Could you try the same experiment but with Device Tree disabled? Just add "device_tree=" to config.txt and reboot.

fivdi, 2015-02-07:
Below is the output for root with the device tree disabled. The errors are still there. 

```
Linux raspberrypi 3.18.5-v7+ #225 SMP PREEMPT Fri Jan 30 18:53:55 GMT 2015 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Feb  7 19:00:09 2015 from wherever
pi@raspberrypi ~ $ tail -5 /boot/config.txt 

gpu_mem=16

device_tree=

pi@raspberrypi ~ $ cd /sys/class/gpio/
pi@raspberrypi /sys/class/gpio $ sudo -s
root@raspberrypi:/sys/class/gpio# echo 4 > export 
root@raspberrypi:/sys/class/gpio# ls gpio4/
active_low  direction  edge  power  subsystem  uevent  value
root@raspberrypi:/sys/class/gpio# echo out > gpio4/direction 
root@raspberrypi:/sys/class/gpio# cat gpio4/direction 
out
root@raspberrypi:/sys/class/gpio# echo rising > gpio4/edge 
bash: echo: write error: Input/output error
root@raspberrypi:/sys/class/gpio# echo falling > gpio4/edge 
bash: echo: write error: Device or resource busy
root@raspberrypi:/sys/class/gpio# echo both > gpio4/edge 
bash: echo: write error: Device or resource busy
root@raspberrypi:/sys/class/gpio# cat gpio4/edge 
none
root@raspberrypi:/sys/class/gpio# 
```

fivdi, 2015-02-07:
And thank you for reopening the issue :)

pelwell, 2015-02-07:
Interesting - I really thought it was going to be due to a bug/difference in the gpio driver; we use pinctrl-bcm2835 when DT is enabled.

There are a number of issues at the moment - jumping up to 3.18, enabling DT and launching the Pi 2 all on the same day has made for a busy week - but this will get fixed ASAP.

fivdi, 2015-02-07:
Great, many thanks!

Yes, I'd say you all had a busy week, but there were a hell of a lot more ups than downs.

notro, 2015-02-07:
Why do you set the direction to output before setting irq flags?
```
$ sudo -s
# cd /sys/class/gpio/
# echo 4 > export
# cat gpio4/direction
in
# echo rising > gpio4/edge
# echo both > gpio4/edge
# echo none > gpio4/edge
# echo out > gpio4/direction
# echo both > gpio4/edge
bash: echo: write error: Input/output error
#
```

fivdi, 2015-02-07:
Because I want an interrupt to be generated when the output changes. I realize that most people only use interrupts on inputs, but it's also possible for outputs on many machines. This also worked with previous versions of the kernel on the Pi.

notro, 2015-02-07:
I get this error in the kernel log:
```
[  116.540282] GPIO chip pinctrl-bcm2835: gpio_lock_as_irq: tried to flag a GPIO set as output for IRQ
[  116.540296] gpio-4 (sysfs): failed to flag the GPIO for IRQ
```

Call chain: gpio_edge_store() -> gpio_setup_irq() -> gpio_lock_as_irq()

gpio_lock_as_irq() was added in 3.13:
```c
int gpio_lock_as_irq(struct gpio_chip *chip, unsigned int offset)
{
        if (offset >= chip->ngpio)
                return -EINVAL;

        if (test_bit(FLAG_IS_OUT, &chip->desc[offset].flags)) {
                chip_err(chip,
                          "%s: tried to flag a GPIO set as output for IRQ\n",
                          __func__);
                return -EIO;
        }

        set_bit(FLAG_USED_AS_IRQ, &chip->desc[offset].flags);
        return 0;
}
```
Ref: http://lxr.free-electrons.com/ident?i=gpio_lock_as_irq

pelwell, 2015-02-07:
Odd as the request is, it seems a bit draconian to disallow what must be a valid operation for the hardware.

notro, 2015-02-07:
Here's the commit that introduced this: https://github.com/torvalds/linux/commit/d468bf9ecaabd3bf3a6134e5a369ced82b1d1ca1

pelwell, 2015-02-07:
So fivdi, it looks as though it will also fail to work on other machines after 3.13. Are you wanting the ability to monitor any changes on any outputs, or one in particular? Would using another gpio as input just to monitor that output not work?

We have the ability to patch the generic kernel code, but we would want a very good reason to do so.

fivdi, 2015-02-08:
The software I have can currently monitor any changes on any outputs. It will be possible to modify that software to detect when the outputs are changed per software rather than per hardware as is currently the case. The timing will be a little different, but it should work.

Although I don't actually agree with the claim that detecting interrupts on outputs is a bogus usage scenario ([https://github.com/torvalds/linux/commit/d468bf9ecaabd3bf3a6134e5a369ced82b1d1ca1](https://github.com/torvalds/linux/commit/d468bf9ecaabd3bf3a6134e5a369ced82b1d1ca1)), I don't have a very good reason for patching the generic kernel code either.

Once again, thank you all for your help.

closed 2015-02-08 by fivdi