raspberrypi/linux #812: please enable ipt_CLUSTERIP for netfilter

miekg, 2015-02-08:
Hello,

Could you enable in ipt_CLUSTERIP in the default build? This way the -j CLUSTERIP target in iptables will work by default.

popcornmix, 2015-02-09:
This should be in now - can you rpi-update and test?

miekg, 2015-02-09:
Thanks!

Sorry, for the noob question, but rpi-update does no show any updates (yet)?

popcornmix, 2015-02-09:
What does "uname -a" report?
Should be
```
Linux version 3.18.6+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #754 PREEMPT Sun Feb 8 20:22:45 GMT 2015
```
See: https://github.com/Hexxeh/rpi-firmware/commit/5ab8621f4062ac500f63624e21141cb39d14f6be

miekg, 2015-02-09:
Ah,  no Linux pi0 3.18.6+ #753 PREEMPT Sun Feb 8 14:47:22 GMT 2015 armv6l GNU/Linux,

OK, I should just update to a specific commit. Ok got it.

popcornmix, 2015-02-09:
No - my fault.
raspberrypi/firmware had been updated.
Hexxeh/rpi-firmware had not been pushed - but has now.

A normal rpi-update should work now.

miekg, 2015-02-09:
```
% sudo iptables -A INPUT -d 192.168.1.10 -j CLUSTERIP -i wlan0 --new --hashmode sourceip-sourceport --clustermac 01:00:5E:00:00:01 --total-nodes 3 --local-node 1
% echo $?
0
```

Yes works \o/ Thanks!

popcornmix, 2015-02-09:
Cool - feel free to close if you're happy.

closed 2015-02-09 by miekg