raspberrypi/linux #813: Please enable CONFIG_VIDEO_EM28XX_V4L2=m

fbutler, 2015-02-08:
Hi,
Please enable CONFIG_VIDEO_EM28XX_V4L2=m. 
I believe that this is required to enable V4L2 to work correctly with EM28XX devices since version 0.2.1 of the driver

thanks,

Fergal

popcornmix, 2015-02-09:
This should be in now - can you rpi-update and test?

fbutler, 2015-02-10:
Thanks,

The device is now appearing as video0 in /dev and responds to v4l2-ctl 

pi@pi2 ~ $ v4l2-ctl --all
Driver Info (not using libv4l2):
        Driver name   : em28xx
        Card type     : Pinnacle Dazzle DVC 90/100/101/
        Bus info      : usb-bcm2708_usb-1.2.1.3
        Driver version: 3.18.6
        Capabilities  : 0x85220001
                Video Capture
                Audio
                Read/Write
                Streaming
                Device Capabilities
        Device Caps   : 0x05220001
                Video Capture
                Audio
                Read/Write
                Streaming
Priority: 2
Video input : 0 (Composite1: ok)
Audio input : 1 (Television)
Video Standard = 0x000000ff
        PAL-B/B1/G/H/I/D/D1/K
Format Video Capture:
        Width/Height  : 720/576
        Pixel Format  : 'YUYV'
        Field         : Interlaced
        Bytes per Line: 1440
        Size Image    : 829440
        Colorspace    : Broadcast NTSC/PAL (SMPTE170M/ITU601)
        Custom Info   : feedcafe
Streaming Parameters Video Capture:
        Frames per second: 25.000 (25/1)
        Read buffers     : 4

User Controls

                     brightness (int)    : min=0 max=255 step=1 default=128 value=128 flags=slider
                       contrast (int)    : min=0 max=127 step=1 default=64 value=64 flags=slider
                     saturation (int)    : min=0 max=127 step=1 default=64 value=64 flags=slider
                            hue (int)    : min=-128 max=127 step=1 default=0 value=0 flags=slider
                         volume (int)    : min=0 max=31 step=1 default=31 value=31 flags=slider
                           mute (bool)   : default=1 value=1
                    red_balance (int)    : min=-48 max=48 step=1 default=0 value=0 flags=slider
                   blue_balance (int)    : min=-48 max=48 step=1 default=0 value=0 flags=slider
                      sharpness (int)    : min=0 max=15 step=1 default=0 value=0 flags=slider
                     chroma_agc (bool)   : default=1 value=1 flags=update
                    chroma_gain (int)    : min=0 max=127 step=1 default=40 value=15 flags=inactive, slider, volatile 

I'll test how it behaves later, but at least it is accessible now.

thanks again for enabling it.

Fergal

closed 2015-02-21 by popcornmix