raspberrypi/linux #818: patch missing from rtl8192cu driver.

MrEngman, 2015-02-10:
A number of people have reported the D-Link DWA 131 wifi module has stopped working when upgrading from 3.12 to 3.18

In the out of tree rtl8192cu wifi driver added back into 3.18, ref #755, it appears that a USB VID:PID is missing that was included in the version in 3.12.

This stops the D-Link DWA 131 wifi module being recognised by the driver. ref https://github.com/raspberrypi/firmware/issues/359

The following patch should fix it.

    
    --- linux/drivers/net/wireless/rtl8192cu/os_dep/linux/usb_intf.c.old    2015-02-10 02:32:12.779287926 +0000
    +++ linux/drivers/net/wireless/rtl8192cu/os_dep/linux/usb_intf.c        2015-02-10 02:37:52.512361391 +0000
    @@ -138,6 +138,7 @@
            {USB_DEVICE(0x2001, 0x3307)},/* D-Link - Cameo */ \
            {USB_DEVICE(0x2001, 0x330A)},/* D-Link - Alpha */ \
            {USB_DEVICE(0x2001, 0x3309)},/* D-Link - Alpha */ \
    +       {USB_DEVICE(0x2001, 0x330d)},/* D-Link - DWA 131 */ \
            {USB_DEVICE(0x0586, 0x341F)},/* Zyxel - Abocom */ \
            {USB_DEVICE(0x7392, 0x7822)},/* Edimax - Edimax */ \
            {USB_DEVICE(0x2019, 0xAB2B)},/* Planex - Abocom */ \

popcornmix, 2015-02-10:
Added. Should be in next update.

popcornmix, 2015-02-12:
Please update and test.

closed 2015-02-21 by popcornmix