raspberrypi/linux #819: OpenWrt 3.18 SD card detected but no partitions

Noltari, 2015-02-10:
Hello, I'm having issues adding 3.18 support for OpenWrt.
There's something wrong with the new MMC driver, since it detects SD card insertions/extractions, but for some odd reason no partitions are loaded:
http://pastebin.com/GWFYrXnT
After that I managed to boot from USB:
http://pastebin.com/b4dX0X7B
And I can't see what's wrong with the SD card:
http://pastebin.com/G61ALvLX

Any ideas?

Regards,
Álvaro.

popcornmix, 2015-02-10:
Can you test the sdcard with a raspbian image?
If it fails there we can help.

If it works there and fails with openwrt, you really need to report it to openwrt.

One gotcha - make sure CONFIG_DMA_CMA is enabled. See:
https://github.com/raspberrypi/linux/commit/b83416ee2870dc481abfa2dffedc1fc1afb8e7d2

Noltari, 2015-02-10:
The SD card is working, since I can succesfully boot openwrt 3.14 with it.

I enabled CMA as you said. However, nothing related to the SD changed:
http://pastebin.com/MFnUBEt2

popcornmix, 2015-02-10:
Can you post your .config (after you have built) or the output of:
```
zcat /proc/config.gz
```

Noltari, 2015-02-10:
There you go:
http://pastebin.com/kYt3tyi2

popcornmix, 2015-02-10:
Can you confirm if sdcard works with raspbian kernel?

Now there are hundreds of differences between your kernel config and ours, so it's hard to say what is the important difference. Possibly these are worth expermienting with: 
CONFIG_PREEMPT=y
CONFIG_DMA_SHARED_BUFFER=y
CONFIG_MMC_BLOCK=y
CONFIG_MMC_BLOCK_MINORS=32
CONFIG_MMC_BLOCK_BOUNCE=y

Noltari, 2015-02-10:
Just found the issue, CONFIG_MMC_BLOCK=y got lost at some point :$...
It's working now, thank you very much for your help.

closed 2015-02-10 by Noltari