raspberrypi/linux #821: Pi <-> USB <-> Sata-Controller Issue

MPW1412, 2015-02-11:
Hello,

I'm running a raspberry pi revision 0004 and have a serious stability issue with my usb sata controller:

```Bus 001 Device 008: ID 05e3:0718 Genesys Logic, Inc. IDE/SATA Adapter```

It works between three days and just a couple of minutes. Then the usb device disappears from lsusb and can only be recovered by physically unplugging it from the hub and plugging it back in after a few moments. It does not make any differences weather the controller is plugged directly into the pi or through and external powered usb hub. And the error happens even if no other usb devices are plugged in.

Any help appreciated, I have no idea where to start. Here the dmesg output. As far as I understand, the device is shut down because of some errors.

```
[ 8830.358520] net_ratelimit: 17 callbacks suppressed
[ 8830.358567] ieee80211 phy0: invalid plcp cck rate (0).
[ 8830.663673] ieee80211 phy0: invalid plcp cck rate (0).
[ 8830.665717] ieee80211 phy0: invalid plcp cck rate (0).
[ 8830.711313] ieee80211 phy0: invalid plcp cck rate (0).
[ 8831.028653] ieee80211 phy0: invalid plcp cck rate (0).
[ 8831.033116] ieee80211 phy0: invalid plcp cck rate (0).
[ 8831.072431] ieee80211 phy0: invalid plcp cck rate (0).
[ 8831.147142] ieee80211 phy0: invalid plcp cck rate (0).
[ 8833.937631] ieee80211 phy0: invalid plcp cck rate (0).
[ 8834.224362] ieee80211 phy0: invalid plcp cck rate (0).
[ 8836.456422] net_ratelimit: 1 callbacks suppressed
[ 8836.456515] ieee80211 phy0: invalid plcp cck rate (0).
[ 8836.897683] ieee80211 phy0: invalid plcp cck rate (0).
[ 8837.669283] ieee80211 phy0: invalid plcp cck rate (0).
[ 8907.702274] ieee80211 phy0: invalid plcp cck rate (0).
[ 9137.679036] usb 1-1.3.4: reset high-speed USB device number 7 using dwc_otg
[ 9168.626586] usb 1-1.3.4: reset high-speed USB device number 7 using dwc_otg
[ 9199.674009] usb 1-1.3.4: reset high-speed USB device number 7 using dwc_otg
[ 9230.582277] sd 0:0:0:0: Device offlined - not ready after error recovery
[ 9230.582388] sd 0:0:0:0: [sda]  
[ 9230.582411] Result: hostbyte=0x05 driverbyte=0x00
[ 9230.582430] sd 0:0:0:0: [sda] CDB: 
[ 9230.582446] cdb[0]=0x28: 28 00 14 6a fd f0 00 00 10 00
[ 9230.582547] blk_update_request: I/O error, dev sda, sector 342556144
[ 9230.582684] sd 0:0:0:0: rejecting I/O to offline device
[ 9230.582728] sd 0:0:0:0: [sda] killing request
[ 9230.582773] sd 0:0:0:0: rejecting I/O to offline device
[ 9230.582890] sd 0:0:0:0: [sda]  
[ 9230.582913] Result: hostbyte=0x01 driverbyte=0x00
[ 9230.582930] sd 0:0:0:0: [sda] CDB: 
[ 9230.582946] cdb[0]=0x28: 28 00 14 6a fe 00 00 00 f0 00
[ 9230.583041] blk_update_request: I/O error, dev sda, sector 342556160
[ 9230.583332] sd 0:0:0:0: rejecting I/O to offline device
[ 9230.583542] sd 0:0:0:0: rejecting I/O to offline device
[ 9230.601908] usb 1-1.3.4: USB disconnect, device number 7
[ 9424.771788] ieee80211 phy0: invalid plcp cck rate (0).
[ 9443.484455] ieee80211 phy0: invalid plcp cck rate (0).
[ 9710.019762] ieee80211 phy0: invalid plcp cck rate (0).
[ 9727.921916] EXT4-fs error (device sda1): ext4_find_entry:1289: inode #2: comm java: reading directory lblock 0
[ 9890.108491] ieee80211 phy0: invalid plcp cck rate (0).
```

```$ uname -a
Linux raspberrypi 3.18.5+ #744 PREEMPT Fri Jan 30 18:19:07 GMT 2015 armv6l GNU/Linux```

Thanks for reading,
Greetings from Germany
MPW

foreverguest, 2015-02-12:
Just my personal experience, but I was having similar errors with my SATA to USB adapter, and was able to stop them adding 'dwc_otg.nak_holdoff=2' to cmdline.txt.

P33M, 2015-02-12:
What is the make/model of the USB/SATA adapter? Post a picture if possible.

foreverguest, 2015-02-12:
Seems to be generic maker 'COMTAC':
https://www.digimer.com.br/image/cache/data/produtos/1839-700x700.jpg

Detected as JMicron:
<pre>
[    8.365511] usb 1-1.3: new high-speed USB device number 5 using dwc_otg
[    8.476581] usb 1-1.3: New USB device found, idVendor=152d, idProduct=2338
[    8.484052] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=5
[    8.505398] usb 1-1.3: Product: USB to ATA/ATAPI bridge
[    8.512843] usb 1-1.3: Manufacturer: JMicron
[    8.535347] usb 1-1.3: SerialNumber: 108F81912002
</pre>

MPW1412, 2015-02-14:
Hello,

thanks for your replys. It's close to foreverguest's photo, just the sata port is a a different position, so it might be a different revision. Picture of my adapter:

http://s28.postimg.org/q0j8trxzx/20150214_210409.jpg

I'm on holidays right now, have the adapter with me, but not the pi. I'm gonna try that kernel parameter when I'll be back home.

What does it do, actually?

Bye
MPW

P33M, 2015-02-17:
The NAK holdoff should have no effect on high-speed transfers. If it's some dodgy timing effect at work then this may change timings such that you no longer see the bug.

foreverguest, 2015-02-19:
After a couple weeks stable, it failed again. The NAK holdoff seems to decrease the chances of triggering the problem.

Got different message errors this time:
<pre>
[13049.147997] NYET/NAK/ACK/other in non-error case, 0x00000002
[13049.148067] NYET/NAK/ACK/other in non-error case, 0x00000002
[13049.148115] NYET/NAK/ACK/other in non-error case, 0x00000002
[13049.148202] hub 1-1:1.0: hub_port_status failed (err = -71)
</pre>
repeat lines above some times, then:
<pre>
[13049.235581] smsc95xx 1-1.1:1.0 eth0: Failed to read reg index 0x00000114: -71
[13049.235607] smsc95xx 1-1.1:1.0 eth0: Error reading MII_ACCESS
[13049.235626] smsc95xx 1-1.1:1.0 eth0: MII is busy in smsc95xx_mdio_read
[13049.248133] sd 1:0:0:0: [sdb]  
[13049.248171] Result: hostbyte=0x07 driverbyte=0x00
[13049.248189] sd 1:0:0:0: [sdb] CDB: 
[13049.248201] cdb[0]=0x2a: 2a 00 0e 3d 25 a0 00 00 f0 00
[13049.248261] blk_update_request: I/O error, dev sdb, sector 238888352
[13049.248291] EXT4-fs warning (device sdb1): ext4_end_bio:317: I/O error -5 writing to inode 14160911 (offset 724811776 size 802816 starting block 29861074)
[13049.248310] buffer_io_error: 1412 callbacks suppressed
[13049.248325] Buffer I/O error on device sdb1, logical block 29860788
...
</pre>

After that usb hub seems to reset, but network and storage are down.

MPW1412, 2015-02-19:
The controller sometimes fails at a real pc too. Maybe the issue is in the linux driver for this device.

MPW1412, 2015-03-02:
Here someone wrote, that this might be an issue with power supply, but I'm using an external usb hub: http://www.raspberrypi.org/forums/viewtopic.php?p=143582

Maybe I will try a different power supply adapter, if it fails again.

Right now, I changed the cmdline.txt and will watch what happens.

MPW1412, 2015-03-02:
Failed again, it just worked a couple of hours and after a few hundred GB I get that i/o error again and can't remount the disk.

MPW1412, 2015-03-06:
After a few days, the adapter failed at a x86 system, too. Maybe it's not pi related.

foreverguest, 2015-03-18:
@P33M, is there any parameter to limit usb hub throughput? As it seems to crash only with very high activity.
Using USB1.1 through a external hub or with dwc_otg.speed=1 works but is too slow.
Any parameter described here http://www.cl.cam.ac.uk/~atm26/ephemeral/rpi/dwc_otg/doc/html/module%20parameters.html ?

MPW1412, 2015-03-18:
What kind of „hight throughtput“, number of i/o operations or traffic speed. I guess the first one?

foreverguest, 2015-03-18:
My guess too, as it happens when downloading a few torrents to the external storage or coping large files between folders in the same storage.

MPW1412, 2015-03-19:
Hm, but I copied a 600 GB archive file via ethernet recently and it worked properly up to almost 590 GB. So I guess it must be the number of i/o operations...

MPW1412, 2015-06-18:
I switched to Sharkoon SATA QuickPort U883 XT bl. which works perfecty with the PI. (Use the single not the double docking station, the double has issues with the driver and switches of both drives if only the first isn't needed.)

closed 2015-06-18 by MPW1412