raspberrypi/linux #822: 3.18.x kernel using which gcc to compile?

amazon83, 2015-02-11:
I try a lot driver having problem, but upgrade the compiler gcc to 4.8 fix all the problem, by default 3.18.x gcc was 4.6 version. Any different?

f1vefour, 2015-02-18:
You just need the headers from the kernel you want to compile the module against, either gcc **should** work.

msperl, 2015-02-20:
I tried this - it does not work!

When using gcc-4.8.3 on my RPI (to match the version used on the kernel) and creating the corresponding softlinks  in /lib/modules/... (and using the exact checkout) using https://github.com/notro/rpi-source

I can compile a module, BUT what does not work is loading it afterwards - it aborts with the following message: "disagrees about version of symbol module_layout"

Maybe with RPI2 we could really start compiling the kernels on the pi itself and start sharing the build and source directory (or at least the portions that are necessary).

lurch, 2015-02-23:
Looks like there's more (possibly helpful) discussion about this in #758

msperl, 2015-02-23:
I believe there are 2 distinct points:
a) version between 4.8.0 and 4.8.3 are not allowed  to compile the kernel
b) the other problem is that even when using the identical kernel sources and also the 4.8.3 gcc (but on the RPI) the (external) modules do compile but the do not load.

Maybe provide a kernel that is compiled on the RPI(2) itself with the firmware and deliver the build/sources directory as an optional install?

Ruffio, 2016-08-10:
@R00t8ug has this issue been resolved? If yes, please close this issue. Thanks.

closed 2017-05-17 by JamesH65