raspberrypi/linux #825: Please enable 6LoWPAN support

csett86, 2015-02-12:
Please enable 6LoWPAN support

    CONFIG_6LOWPAN=m
    CONFIG_BT_6LOWPAN=m

This helps for IoT applications like 6LoWPAN over Bluetooth Low Energy  as in https://www.nordicsemi.com/eng/Products/Bluetooth-Smart-Bluetooth-low-energy/nRF51-IoT-SDK

The modification to the kernel linked above are only these two additional modules.

csett86, 2015-02-21:
Is there anything I can help to get these two kernel modules into Raspbian? I can create a pull request for the two defconfigs, if that helps.

PeterPablo, 2015-02-21:
They were just added (https://github.com/raspberrypi/linux/commit/690878e23a8b7a7625514da6d9b005b5c7d178b3) and thus should soon be available via `rpi-update`.

popcornmix, 2015-02-21:
These modules are now added. Please run rpi-update and test.

csett86, 2015-02-21:
Thanks a lot for the quick response, both modules work like expected.

closed 2015-02-21 by csett86

DerDipper, 2015-09-08:
why am i not able to write to the 6lowpan_psm file after the update, is this common or did i do wierd things?