raspberrypi/linux #827: lirc-rpi (out) broken (RP2)

giddyhup, 2015-02-12:
lirc out doesn't work, neither with default nor with explicitly defined pin. Pin can be addressed via regular GPIO commands/libraries. See [thread] (http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=99085).

pelwell, 2015-02-12:
I've posted on the thread, but in brief: although I don't have an IR output LED, attaching a regular red LED and running irsend I can see it flashing. It would be interesting if you could do the same, or monitor the output in some other way, to try and determine if the output is completely dead or just not compliant in some way.

giddyhup, 2015-02-13:
I can also see activity but it is either too weak or the frequency isn't right.

pelwell, 2015-02-13:
I'll get the 'scope on it, but it won't be until Monday.

giddyhup, 2015-02-13:
Great, thanks!

giddyhup, 2015-02-18:
@pelwell: I noticed a commit mentioning this issue, does this mean I could rpi-update and try again?

pelwell, 2015-02-18:
You could try building your own kernel, but at the moment getting a fix into rpi-update is waiting on a new release from @popcornmix.

giddyhup, 2015-02-18:
Thank's @pelwell, I'll wait for the release then.

popcornmix, 2015-02-18:
Please run rpi-update and test.

theceej, 2015-02-18:
I can confirm that the fix has worked, irsend is working well once again. Thanks!

giddyhup, 2015-02-18:
Yes, the update fixed it. Thanks a lot! Issue closed.

(My pw controlled common cathode RGB LED seems to be acting strange. It is less of an issue to me and I have to observe if it is perhaps an electrical/wiring problem.)

closed 2015-02-18 by giddyhup

HotIssue, 2015-06-03:
i don't think so
mode2, irrecord is nice work but
irsend isn't
i was "sudo rpi-update" and apt-get update, upgrade

irsend LIST "" ""
irsend: could not connect to socket
irsend: No such file or directory

can you explain what happen...
i was read all about article