raspberrypi/linux #829: Gerts VGA

Matty11611, 2015-02-13:
Hi, had Gerts VGA adapter for about a week or 2 now and struggling to get it working on the new kernel, any ideas what this might be and how i can fix it so it works on the 3.18 kernel. Thanks, please a noob simple responce, im not to familiar with linux code

popcornmix, 2015-02-13:
You can revert to pre-3.18 kernel (and firmware) with:
```
sudo rpi-update f74b92120e0d469fc5c2dc85b2b5718d877e1cbb
```
However it's unlikely to be a kernel change - this is all handled by the GPU firmware,
but will be interesting if older firmware/kernel works for you.

Matty11611, 2015-02-13:
Hi thanks will try it when I'm back, if not is there an older version of raspbian avaliable for download to try that?

popcornmix, 2015-02-13:
Older raspbian images are here: http://downloads.raspberrypi.org/raspbian/images/

But just reverting the firmware/kernel will be more useful if something has broken - we'd prefer to narrow down exactly where it broke.

Matty11611, 2015-02-13:
yeah, kernel downgrade didnt work, just going to try a lower version of raspbian, unless you know how to downgrade the GPU firmware?

Matty11611, 2015-02-14:
found the september version of raspbian works with this add on but the newer two dont seem to work, i personally dont understand very much linux and coding to work out what causes it to stop working on the latter versions, also struggling on how to get dual output, any tutorials/manuals?

popcornmix, 2015-02-14:
For dual displays, you need to explicitly request a non-default display from the host code.
E.g. omxplayer can do this. See: https://github.com/popcornmix/omxplayer/commit/8bfa622d5efadf9d433efcc7311b0b3c2fedde6e

Starting from a latest raspbian image, you can downgrade both GPU firmware and kernel with rpi-update.

Looks here: https://github.com/Hexxeh/rpi-firmware/commits/master
Click on a commit and the hash (the hex number at end of url) can be used as a parameter for rpi-update. You can use this to identify exactly when it stopped working for you.

DirkUK, 2015-02-15:
FYI: I finally got round to testing Gert's VGA666 adapter with the latest kernel / firmware and it seems to work fine on both Pi B+ and Pi2B.
This was *not* on a fresh install, but updated / upgraded from 3.12.36. and firmware from around the end of december, with which the VGA666 was working fine.
I did not make any changes in config.txt

pelwell, 2015-02-15:
That's good to know. Out of interest, is that with Device Tree enabled (does /proc/device-tree exist)?

DirkUK, 2015-02-15:
@pelwell 
Yes. I have not explicitly disabled DT, and /proc/device-tree does indeed exist

DirkUK, 2015-02-15:
Also tried it using a fresh install with the Raspbian from the RPF download page (kernel version 3.18.5).
config.txt contains
```
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=35
```
everything else removed from config.txt.
VGA output works on first boot on both B+ and Pi2B

Can't see yet why @Matty11611 has problems getting it to work with the new kernel.

Matty11611, 2015-02-15:
thats what i was using with no luck as soon as i installed it on a fresh image from the RPI download page i dont get any image

popcornmix, 2015-03-31:
@Matty11611 is this still an issue for you?
I'm not seeing any other reports of this being broken.

Matty11611, 2015-03-31:
Hi @popcornmix no i completly forgot about this and have managed to get it working, just had a fit so started again, still waiting for the global dual screen though so it could be embedded into my project!

closed 2015-04-01 by popcornmix

procount, 2015-04-02:
I fixed the "sh: can't access tty; job control turned off"  error by adding 'disablesafemode' as the VGA666 is affecting a GPIO input pin for safemode. But the screen settings and missing cursor remain an issue. 
Now the recovery program starts, but it disables the VGA output. I suspect this is because of a call to tvservice that it uses to switch display mode. Is it possible to add an option to tvservice to switch to VGA output aswell?

popcornmix, 2015-04-02:
Kernel is not involved in DPI/VGA driving - it's purely firmware. The NOOBS firmware should be the same as raspbian firmware - recovery.elf is just a renamed copy of start_cd.elf.

What's the contents of config.txt? Do you have the same config.txt on NOOBS boot partition and raspbian boot partition?

procount, 2015-04-02:
Yes, it is identical as I copied it from the Raspbian partition.

=======================config.txt
# NOOBS Auto-generated Settings:
#hdmi_force_hotplug=1
config_hdmi_boost=4
overscan_left=24
overscan_right=24
overscan_top=16
overscan_bottom=16
disable_overscan=0

max_usb_current=1

#For DELL screen at work on VGA
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=86
# Overscan settings. Written by set_overscan.sh
disable_overscan
overscan_top=0
overscan_bottom=0
overscan_left=176
overscan_right=48
================

I commented out all tvservice output commands in recovery except for the tvservice -s commands where I thought it was just reading information, and VGA still got disabled. Then I replaced main() with return 0; and it didn't get disabled, so I'm in a process of elimination at the moment....

popcornmix, 2015-04-02:
Are you saying "tvservice -s" disables the DPI/VGA output?
Even on a raspbian image?
If that occurs, that would be a bug.

procount, 2015-04-02:
No. it was a supposition that I was trying to eliminate. But I have just discovered that it was the access to the GPIO pins in NOOBS that caused the VGA to become disabled. Having removed that functionality I can now see the noobs screen on the VGA output !! Yay!
Going to the ctl-alt-f2 terminal I can get a flashing cursor too.
So I'll re-enable all the tvservice calls and see if the screen is now fixed.

procount, 2015-04-02:
re-enabling tvservice calls made no difference. In the ctrl-alt-f2 terminal. fbset returns
mode "1142x768" 
  geometry 1142 768 1142 768 16
  timings 0 0 0 0 0 0 0
  rgba 5/11,6/5,5/0,0/164
endmode

tvservice -s returns
state 0x120016 [DVI DMT (4) RGB full 4:3], 640x480 @ 60Hz, progressive.
I'm not sure what resolution the noobs screen is using but it has a 4:3 aspect ratio, so I have black bands down the right and left hand sides of my widescreen monitor. I suppose this is because noobs is using the output of tvservice to select its screen size.

So the remaining questions are:
If I boot the noobs kernel, but drop straight into the shell instead of running the recovery program,
1) why is there no flashing cursor? How can I enable it (the recovery program can enable it).
2) Why does the screen not fit the requested resolution(?)/aspect ratio specified in config.txt like it does in Raspbian?

procount, 2015-04-02:
Pressing keys 1,2,3,4 on the noobs display with VGA fitted give some very strange displays, so
3) is it possible to control the VGA screen size using tvservice and get it to return the correct status information?

popcornmix, 2015-04-02:
Cursor behaviour is just standard linux stuff and not related to DPI/VGA output: 
http://unix.stackexchange.com/questions/3759/how-to-stop-cursor-from-blinking

tvservice is not the tool for changing framebuffer size. Look at fbset.

procount, 2015-04-02:
guess it's NOOBS/recovery that needs adapting for the VGA in a way that still works with HDMI and Composite.
Thanks.

procount, 2015-04-02:
Whilst fbset changes the framebuffer size, it does not affect the physical VGA resolution set by dpi_group/mode in config.txt. Is there a way to change dpi_group/mode after boot like tvservice -e "CEA 86" that will change the DPI output resolution? With no HDMI attached, this tvservice function fails.

popcornmix, 2015-04-02:
No, tvservice doesn't affect DPI/VGA output.
Only one set of pixel timings is supported and that's in config.txt.