raspberrypi/linux #830: possible racecondition in dwc_otg driver

flipreverse, 2015-02-14:
Hi all,

i discovered a seriuos bug in the dwc driver. When i connect multiple usb devices and each of them generates a serious amount of traffic, i get a null pointer dereference error after a while.
The error message itself is quoted below. I narrowed the bug down to drivers/usb/host/dwc_otg/dwc_otg_hcd.c line 1205. To be more precise, the assignement of the new predecessor of the next element fails.
(elm)->field.cqe_next->field.cqe_prev =                 \
                    (elm)->field.cqe_prev;
This might happen while a certain context tries to remove an item from the free list another context already removed. Hence, the first context fails during the DWC_CIRECLEQ_REMOVE_INIT macro.
As a first guess, i secured the list using a spinlock. I'm currently testing my fix.
I'll inform you as soon as i know if i fixed that issue.
Meanwhile, can you please tell me if there might be other sources for this bug?

Thanks!

Greetings
flip

[ 4989.166141] Unable to handle kernel NULL pointer dereference at virtual address 00000034                                                                                                
[ 4989.174235] pgd = ede30000                                                                                                                                                              
[ 4989.176936] [00000034] *pgd=2cef2831, *pte=00000000, *ppte=00000000                                                                                                                     
[ 4989.183214] Internal error: Oops: 817 [#1] PREEMPT SMP ARM                                                                                                                              
[ 4989.188689] Modules linked in: ipv6 snd_bcm2835 sg uas evdev rc_su3000 ts2022 ds3103 ts2020 ds3000 dvb_usb_dw2102 dvb_usb dvb_core rc_core snd_soc_wm8804 snd_soc_pcm512x_i2c regmap_sp8
[ 4989.221576] CPU: 0 PID: 3261 Comm: mpegts_input_th Not tainted 3.18.5-v7-al-gddabc66-dirty #1                                                                                           
[ 4989.230085] task: ec17c500 ti: ecf2e000 task.ti: ecf2e000                                                                                                                               
[ 4989.235480] PC is at assign_and_init_hc+0x60/0x610                                                                                                                                      
[ 4989.240264] LR is at dwc_otg_hcd_select_transactions+0x244/0x370                                                                                                                        
[ 4989.246261] pc : [<c03d76fc>]    lr : [<c03d9850>]    psr: 00000193                                                                                                                     
[ 4989.246261] sp : ecf2f928  ip : ecf2f960  fp : ecf2f95c                                                                                                                                 
[ 4989.257716] r10: edced828  r9 : ee6c9000  r8 : e1b334c0                                                                                                                                 
[ 4989.262929] r7 : ee6c9000  r6 : edced800  r5 : e1b33cc0  r4 : ee79c000                                                                                                                  
[ 4989.269442] r3 : ee6c9060  r2 : 00000000  r1 : 00000000  r0 : ee6c9000                                                                                                                  
[ 4989.275957] Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user                                                                                                          
[ 4989.283163] Control: 10c5387d  Table: 2de3006a  DAC: 00000015                                                                                                                           
[ 4989.288895] Process mpegts_input_th (pid: 3261, stack limit = 0xecf2e238)                                                                                                               
[ 4989.295669] Stack: (0xecf2f928 to 0xecf30000)                                                                                                                                           
[ 4989.300019] f920:                   ecf2f94c ecf2f938 c0533c6c c00d14ec ee78a3c0 c081248a                                                                                               
[ 4989.308183] f940: edced800 ee6c9014 ee6c9060 ee6c9000 ecf2f9a4 ecf2f960 c03d9850 c03d76a8                                                                                               
[ 4989.316347] f960: ee78a3c0 00000008 ee6c9038 ee6c901c ecf2f994 20000193 c0533c6c ee6c9000                                                                                               
[ 4989.324511] f980: ee780f00 edced800 ee78a3c0 00000002 e1b33dc0 00000001 ecf2f9cc ecf2f9a8                                                                                               
[ 4989.332675] f9a0: c03dc758 c03d9618 0c004c14 20000193 ee6c9000 ee780f00 e1b33e00 ee6c9000                                                                                               
[ 4989.340839] f9c0: ecf2f9f4 ecf2f9d0 c03dc930 c03dc5d4 00000002 ecf2f9e0 c03af8cc e1b33e00                                                                                               
[ 4989.349003] f9e0: ee780f00 f3980568 ecf2fa24 ecf2f9f8 c03dca6c c03dc88c 00000000 c03dd7f8                                                                                               
[ 4989.357168] fa00: ecf2fa24 e1b33e00 ee780f00 f3980560 ee6c9000 00000002 ecf2fa64 ecf2fa28                                                                                               
[ 4989.365331] fa20: c03ddaec c03dc9f4 00000002 c045ba3c ecf2fa84 ee78ee00 00000000 00000002                                                                                               
[ 4989.373496] fa40: ee6c9000 ee780f00 00000000 f3980560 e1b33e00 00000001 ecf2fabc ecf2fa68                                                                                               
[ 4989.381660] fa60: c03defe4 c03dd844 ee6c1800 c03d9c54 ee6c901c ec033628 ecf2fac4 ecf2fa88                                                                                               
[ 4989.389825] fa80: 00000023 00000000 f398056c f3980568 f3980000 00000003 00000008 ee6c9000                                                                                               
[ 4989.397988] faa0: 00000000 c08122b8 f301080e 00000000 ecf2fadc ecf2fac0 c03df3f4 c03dec8c                                                                                               
[ 4989.406153] fac0: ee6c9000 ee6c8c00 00000001 02000008 ecf2fb04 ecf2fae0 c03df6a4 c03df350                                                                                               
[ 4989.414316] fae0: ee78a600 c07cd95c 00000020 00000000 00000000 00000000 ecf2fb14 ecf2fb08                                                                                               
[ 4989.422481] fb00: c03dbbf4 c03df440 ecf2fb24 ecf2fb18 c03af870 c03dbbe4 ecf2fb64 ecf2fb28                                                                                               
[ 4989.430645] fb20: c0069a1c c03af848 c052dd98 c00d13c0 c052dd90 c07cd900 01400000 c07cd900                                                                                               
[ 4989.438809] fb40: c07cd95c ee78a040 00000000 00000000 00000000 00000002 ecf2fb84 ecf2fb68                                                                                               
[ 4989.446973] fb60: c0069c6c c00699e8 00020000 c07cd900 c07cd95c ecf2ffb0 ecf2fb9c ecf2fb88                                                                                               
[ 4989.455137] fb80: c006ccdc c0069c2c 00000020 c07c6d44 ecf2fbc4 ecf2fba0 c0069384 c006cc3c                                                                                               
[ 4989.463302] fba0: ecf2fbd8 c04ac6c0 60000113 f300b200 ecf2fc0c ec0e45b4 ecf2fbd4 ecf2fbc8                                                                                               
[ 4989.471466] fbc0: c0008564 c006930c ecf2fc94 ecf2fbd8 c05345f4 c0008544 000005a8 e19bc600                                                                                               
[... cut ...]
[ 4990.074191] ffa0:                                     b0100010 00000001 000005e8 00000000                                                                                               
[ 4990.082355] ffc0: 018cd1b0 000000bc 00000010 54df62ab 00000000 018cd15c 000002ff 80000000                                                                                               
[ 4990.090517] ffe0: 000cfc28 b1a82688 b6b473f4 b6b47408 a0000010 ffffffff                                                                                                                 
[ 4990.097123] Code: e1520003 05942034 15941034 05872064 (15821034)

P33M, 2015-02-14:
For reference, which list is being modified?

flipreverse, 2015-02-14:
Oh, sry. I'm talking about *dwc_otg_hc_t.free_hc_list*.

flipreverse, 2015-02-14:
I pushed my first fix to my github account: https://github.com/flipreverse/linux/commits/dwcotgfix

P33M, 2015-02-15:
You're adding a lock to every circleq_* call site. Many of these are in different contexts - particularly with regard to the top-level HCD lock being held or not. This makes the possibility of an A-B-B-A type deadlock far more likely...

There should be no need to have an additional lock. Holding the HCD spinlock while manipulating the list should be sufficient.

flipreverse, 2015-02-15:
I've been running my fix for a few hours. No deadlock happened. Instead, the same null pointer panic occured again. So, why is the cqe_next pointer NULL?
This is a circular list and if doesn't get modified somewhere else, this should not happen.

P33M, 2015-02-15:
Please post a lsusb -v and dmesg since boot.

flipreverse, 2015-02-17:
May i have your email address to send you the requested stuff?

P33M, 2015-02-17:
Sent an email to the address you have listed on Github.

flipreverse, 2015-04-04:
A lot of crashs later, i've got a bit more information about the error itself.
The kernel used to crash in assign_and_init() in drivers/usb/host/dwc_otg/dwc_otg_hcd.c when the drivers tries to remove an item from the list: DWC_CIRCLEQ_REMOVE_INIT(&hcd->free_hc_list, hc, hc_list_entry);.
Nothing new so far.
I figured out that the list has at least two elements. The first one, which gets peeked by DWC_CIRCLEQ_FIRST(&hcd->free_hc_list), has an empty prev and next pointer.
I placed a BUG_ON((elm)->field.cqe_prev == NULL || (elm)->field.cqe_next == NULL) at the end of every DWC_CIRCLEQ_* macro which modifies a list. None of them triggered a panic.
In addition to the free list spinlock, i disabled the fiq before modifying the free list and enabled it afterwards with respect to the fiq_enable variable.

Any further advices what i could do to find the source of this bug?

P33M, 2015-04-05:
Hm.

Thinking about this a bit - what happens if we have no host channels left and try to assign one?

Ordinarily this won't happen because it's very unlikely that you will have multiple transfers active on the bus at the same time - you would have to have 8 simultaneous outstanding transactions. With your particular use case I can see this happening.

flipreverse, 2015-04-05:
Yeah, this might happen. But the hc list isn't empty.
I added the following code to get a slight insight into the situation:
```
if(hc->hc_list_entry.cqe_prev == NULL || hc->hc_list_entry.cqe_next ==  NULL) {
		printk(KERN_ERR "%s:%d hc=0x%p, hc->hc_num=%d, hc->hc_list_entry.cqe_prev=0x%p, hc->hc_list_entry.cqe_next=0x%p,\n isEmpty=%d hcd->free_hc_list.cqh_first=0x%p, hcd->free_hc_list.cqh_last=0x%p\n",
						__func__,
						__LINE__,
						hc,
						hc->hc_num,
						hc->hc_list_entry.cqe_prev,
						hc->hc_list_entry.cqe_next,
						DWC_CIRCLEQ_EMPTY(&hcd->free_hc_list),
						hcd->free_hc_list.cqh_first,
						hcd->free_hc_list.cqh_last);
		if (DWC_CIRCLEQ_EMPTY(&hcd->free_hc_list) || ((hc->hc_list_entry.cqe_prev == NULL) ^ (hc->hc_list_entry.cqe_next == NULL))) {
			BUG();
		}
	}
```
This produces the following output:
```
[ 7409.347717] assign_and_init_hc:1219 hc=0xee7a0980, hc->hc_num=4, hc->hc_list_entry.cqe_prev=0x  (null), hc->hc_list_entry.cqe_next=0x (null),
[ 7409.347717]  isEmpty=0 hcd->free_hc_list.cqh_first=0xee7a0980, hcd->free_hc_list.cqh_last=0xee7a0900
```

jkuek, 2015-09-07:
I'm seeing a similar crash running Raspbian Jessie

P33M, 2015-09-07:
Can you post the output of `sudo lsusb -v`?

jkuek, 2015-09-07:
Bus 001 Device 006: ID 0d8c:0016 C-Media Electronics, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0d8c C-Media Electronics, Inc.
  idProduct          0x0016 
  bcdDevice            1.00
  iManufacturer           1 C-Media Electronics Inc.
  iProduct                2 USB Audio Device
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          255
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength          102
        bInCollection           2
        baInterfaceNr( 0)       1
        baInterfaceNr( 1)       2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             2
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
[... 1019 lines cut ...]
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered

jkuek, 2016-01-21:
I saw this happen again today
[image]

I wonder whether this isn't related to https://github.com/raspberrypi/linux/issues/1256

bogardusj, 2016-06-01:
I have also encountered this problem.  I have encountered it on kernels 4.1.19-v7+ and 4.4.11-v7+ (built from git; with some extra printks).  I have encountered it with various combinations of USB devices, but for me it seems to require 2 USB cameras (tried with multiple UVC and Playstation Eye cameras) and a custom pic24 based full-speed USB device.  Throwing in continuous reads of an old, full-speed USB flash drive seems to help it occur more quickly.

I have determined that the free_hc_list list is being corrupted because an hc is being appended to the list twice.  Occasionally, dwc_otg_hcd_handle_hc_intr() is reading that a given host channel has a pending interrupt, but the corresponding dwc_hc_t structure is already on the free_hc_list - at the end of the list in every case I've seen.  When dwc_otg_hcd_handle_hc_n_intr() calls release_channel(), the dwc_hc_t structure gets appended to the free_hc_list, resulting in it effectively being in the free_hc_list twice (since its own cqe_next points to itself).  Later, it will get removed from the free_hc_list twice.  The first time clears the cqe_next pointer, so it contains NULL when removed a second time - resulting in the NULL pointer dereference.

I don't know enough about the hardware to know if/why there is either an extra interrupt or the original interrupt isn't being cleared.

[lsusb.txt](https://github.com/raspberrypi/linux/files/292516/lsusb.txt)
[Uploading lsusb.txt…]()

P33M, 2016-06-01:
If that's the case, then there is a path in the driver that is re-entrant in some strange way (two hardware interrupts arriving for the same channel, for example).

With there being a double-entry at the end of the list, I'd suspect bogus interrupts occurring in quick succession could cause this.

P33M, 2017-05-15:
fixed in #2010

closed 2017-05-15 by P33M

popcornmix, 2017-05-15:
rpi-update firmware contains latest kernel fixes. Please update and test.