raspberrypi/linux #835: HDMI audio enablement is inconsistent for custom HDMI resolutions

andrewsil1, 2015-02-16:
Please bear with me as this is a rather lengthy description, but I have spent many hours nailing down the exact behavior to the best of my ability and I'm going to try to describe it as precisely as possible.  

Adafruit in NYC sells a variety of LCD monitors for the Pi, most of which come with a driver board based on a Realtek RTD2668 HDMI->LVDS controller.  Some of the available variants of the board include an onboard 2-channel audio amplifier that can drive speakers directly if the HDMI stream contains 2-channel PCM audio.  Their forum turns out to have numerous users of these displays who appear to have encountered the same situation.

I have both a B+ and a Pi 2, and both exhibit the same behavior when used with this monitor/driver board combination, as follows:

Either Pi can be connected to this driver board via HDMI, and the monitor has a native 1280x800 resolution. As long as the Pi is told to use one of the predefined CEA video modes, (e.g. mode 4, 1280x720), then HDMI audio works completely fine, as expected, however the video will be scaled in hardware and therefore loses some quality.  If using a DMT mode (either 27 or 28, to match the native monitor resolution), then there is no audio, also as expected. (tvservice reports DVI mode in those cases.)  Adafruit's instructions, understandably, recommend setting HDMI_DRIVE=2, but this does not seem to have the intended effect at all. Even though tvservice reports "HDMI DMT (28)" in this case, attempting to test audio with speaker-test or aplay does not produce audio when using the group-2, mode-27 or 28 in the config.txt.

However, I have found a means by which I _can_ get the Pi to produce HDMI audio when in the native 1280x800 mode, as follows, but this requires some very odd workarounds that are neither easily explained or desirable, but which seem to work only in very specific circumstances.

First: In the config.txt, define a custom CEA mode 65 mode as per the instructions in the troubleshooting forum, copying the timings from DMT mode 27 or 28, as follows:
hdmi_timings=1280 0 72 128 200 800 1 3 6 22 0 0 0 60 0 83500000 5
hdmi_group=1
hdmi_mode=65

Then, after boot, I go through the following sequence at the console"
tvservice -e "CEA 4 HDMI" && sleep 1 && fbset -depth 8 && fbset -depth 16
aplay <Any wave file, for example the Front_Center.wav. (Plays as expected)>
tvservice -e "CEA 65 HDMI" && sleep 1 && fbset -depth 8 && fbset -depth 16
aplay <the same wave file again>
aplay <the same wave file for a third time>

If I don't follow this exact sequence, then after the mode 65 switch, I will either get no audio, audio that alternates between normal and muted at about 1 second intervals, or audio that works fine for about 3 or 4 seconds and then goes back to muted (permanently.)  There doesn't appear to be any consistency as to which undesirable form of behavior will occur.

If I DO follow this exact sequence, then as long as no additional mode switches are requested, the HDMI audio will continue to work normally with a variety of ALSA or SDL applications, (e.g. the retropie emulators) as long as the monitor remains in the desired native (non-scaled) video mode.  Any additional mode switching to other resolutions may require going through this sequence again to get HDMI audio to play normally again once returning to the custom CEA mode.

I really can't say whether this is a problem with the Pi (e.g. it's not outputting a consistent HDMI audio stream even when it should, given the settings that have been created), or whether it's a problem with the Realtek chip somehow deciding to mute an otherwise valid audio stream due to the incoming resolution.  However, I'm more suspicious of the underlying BCM ALSA driver, given that there are these arcane sequences that seem to get the audio to work properly for some period of time.

I'm happy to provide any additional information you think you might need. I really hope a more permanent solution can be found.

Thanks,
Andy

popcornmix, 2015-02-16:
I've just tried:
```
hdmi_group=2
hdmi_drive=2
hdmi_mode=27
```
with a Yamaha RX-V375 receiver. Now running 
```
/opt/vc/src/hello_pi/hello_audio/hello_audio.bin 1
```
or
```
aplay audio/norah.wav
```
And I hear the audio as expected. That makes me feel we are outputting valid audio, and the Adafruit board is doing something wrong.

Can you repeat your experiments with a normal HDMI connection to a TV or receiver that supports audio and the 1280x800 DMT mode?

andrewsil1, 2015-02-16:
Good point, I'll try that and report back.

andrewsil1, 2015-02-17:
I was able to verify that with a different receiver/monitor that HDMI audio worked as expected after removing my hacks to the startup process, so since there's no obvious action here for the RPi team I'll close this issue.

closed 2015-02-17 by andrewsil1