raspberrypi/linux #842: MMC driver not enabled in bcmrpi_quick_defconfig

rdnetto, 2015-02-19:
In 3.16 and later, the kernel fails to find the root file system as the MMC driver is not included in bcmrpi_quick_defconfig. This is particularly tricky to debug as many images (e.g. Raspbian) specify rootwait as default, which waits indefinitely without an error message.

Proposed fix:
Add the following to bcmrpi_quick_defconfig:
+CONFIG_MMC_BCM2835=y
+CONFIG_MMC_BCM2835_DMA=y

Tested on first generation Model B.

popcornmix, 2015-02-19:
bcmrpi_quick_defconfig has been removed: https://github.com/raspberrypi/linux/commit/852d3bd74b49f8f31a688eaa5b388f09ddd29ec5
Only the standard configs: bcmrpi_defconfig and bcm2709_defconfig are actively maintained, so it's recommended you start with those and cut down if you need a custom config.

closed 2015-02-19 by rdnetto

lurch, 2015-02-23:
I realise this issue has now been closed, but along a similar line, do `bcm2835_sdcard_defconfig` and / or `bcm2835_defconfig` still need to be present, or should/could they be removed too to avoid confusion?
(looks like the former was added during the Pi2 support, which doesn't make sense?)

popcornmix, 2015-02-23:
No, bcm2835*_defconfig are from upstream kernel:
https://github.com/raspberrypi/linux/commits/rpi-3.18.y/arch/arm/configs/bcm2835_defconfig
we have never touched them.

popcornmix, 2015-02-23:
Correction - bcm2835_defconfig is from upstream.
bcm2835_sdcard_defconfig was ours and I've now deleted it.