raspberrypi/linux #845: device tree: bcm2708.dts, node clocks has no lable, can't overlay clocks node

taudac, 2015-02-21:
I am developing a device driver for a HAT that has two crystal oscillators. I have made a device tree overlay file to describe the oscillators which looks like the following:

    ...
	/ {
	compatible = "brcm,bcm2708";
	
	fragment@0 {
		target = <&clocks>;
		__overlay__ {
		
			osc22: osc@22.5792M {
				compatible = "fixed-clock";
				#clock-cells = <0>;
				clock-frequency = <22579200>;
			};
			
			osc24: osc@24.576M {
				compatible = "fixed-clock";
				#clock-cells = <0>;
				clock-frequency = <24576000>;
			};
		};
	};
    ...

Loading of this `dtb` file fails with the error `dterror: can't find symbol 'clocks'`. If I understand it correctly, this error is caused by the fact that the node `clocks` in the file `bcm2708.dtsi` has no label.

In the file `bcm2708.dtsi`, can we replace the line 

    	clocks {

by

    	clocks: clocks {

?

Or am I missing something?

pelwell, 2015-02-21:
Try using:
```
target-path="/clocks";
```

taudac, 2015-02-21:
@pelwell Thanks, using `target-path="/clocks";` works.

closed 2015-02-21 by pelwell

taudac, 2020-12-27:
@pelwell This (`target-path="/clocks";`) seems not to work with kernel **5.4.79**. Seems like my clocks are not being probed. Last known working kernel is **4.19.118**. Any idea? What would be the right way to define external clocks in an DT overlay for a recent kernel?