raspberrypi/linux #846: Hifiberry DAC+  ---  Linux 3.12.36  --- snd_soc_pcm512x module

syntheticpp, 2015-02-22:
I build Linux 3.12.36/Rasp B+ with a cut-down kernel defconfig and it could not load snd_soc_pcm512x:

[ 34.734191] bcm2708-dmaengine bcm2708-dmaengine: Load BCM2835 DMA engine driver
[ 34.803058] bcm2708_spi bcm2708_spi.0: master is unqueued, this is deprecated
[ 34.852349] bcm2708_spi bcm2708_spi.0: SPI Controller at 0x20204000 (irq 80)
[ 34.909897] i2c /dev entries driver
[ 34.942189] bcm2708_i2c_init_pinmode(1,2)
[ 34.946273] bcm2708_i2c_init_pinmode(1,3)
[ 34.972325] bcm2708_i2c bcm2708_i2c.1: BSC1 Controller at 0x20804000 (irq 79) (baudrate 100000)
[ 35.056795] pcm512x 1-004d: Failed to get supply ‘AVDD': -517
[ 35.066081] pcm512x 1-004d: Failed to get supplies: -517
[ 35.071727] i2c 1-004d: Driver pcm512x requests probe deferral
[ 35.104583] snd-rpi-hifiberry-dacplus snd-rpi-hifiberry-dacplus.0: ASoC: CODEC pcm512x.1-004d not registered
[ 35.126124] snd-rpi-hifiberry-dacplus snd-rpi-hifiberry-dacplus.0: snd_soc_register_card() failed: -517
[ 35.136162] platform snd-rpi-hifiberry-dacplus.0: Driver snd-rpi-hifiberry-dacplus requests probe deferral

with this module order (/etc/modules, no depmod installed)
bcm2708-dmaengine
virt-dma
regmap-mmio
spi-bcm2708
i2c-dev
i2c-bcm2708
snd-soc-bcm2708-i2s
snd_soc_pcm512x
snd_soc_hifiberry_dacplus

Has someone an idea what’s going on?
I see no errors using Volumio (3.18), but I assume it could also work with 3.12

syntheticpp, 2015-02-22:
When I build all modules statically but snd-soc-pcm512x/snd-soc-hifiberry-dacplus the error looks like this:

modprobe snd_soc_pcm512x

Jan  1 00:00:40 juplym kern.err kernel: [   40.223460] pcm512x 1-004d: Failed to get supply 'AVDD': -517
Jan  1 00:00:40 juplym kern.err kernel: [   40.229222] pcm512x 1-004d: Failed to get supplies: -517
Jan  1 00:00:40 juplym kern.info kernel: [   40.246831] i2c 1-004d: Driver pcm512x requests probe deferral

AVDD supply: have i missed some power modules?

pelwell, 2015-02-22:
We don't use regulators on Raspberry Pi - try removing OF_REGULATOR from your config.

pelwell, 2015-02-22:
Correction - that should have been CONFIG_REGULATOR.

syntheticpp, 2015-02-23:
Thanks, after removing pcm512x loads:

[   45.054314] pcm512x 1-004d: No SCLK, using BCLK: -2
[   45.068245] pcm512x 1-004d: Completed initialisation - pcm512x_probe

But now the final DAC+ driver fails toload:
[   69.109326] snd-rpi-hifiberry-dacplus snd-rpi-hifiberry-dacplus.0: ASoC: CPU DAI bcm2708-i2s.0 not registered
[   69.131065] snd-rpi-hifiberry-dacplus snd-rpi-hifiberry-dacplus.0: snd_soc_register_card() failed: -517
[   69.141039] platform snd-rpi-hifiberry-dacplus.0: Driver snd-rpi-hifiberry-dacplus requests probe deferral

cat /lib/modules/3.12.36-juplym/modules.builtin | grep i2s
kernel/sound/soc/bcm/snd-soc-bcm2708-i2s.ko

Is maybe a wrong name used in snd-rpi-hifiberry-dacplus?

pelwell, 2015-02-23:
What do `dmesg | grep i2s` and `lsmod` show?

syntheticpp, 2015-02-23:
Only above error message multiple times:

```javascript
#dmesg|grep i2s
[    1.842250] snd-rpi-hifiberry-dacplus snd-rpi-hifiberry-dacplus.0: ASoC: CPU DAI bcm2708-i2s.0
...

#dmesg|grep i2c
[    1.710450] i2c /dev entries driver
[    1.714102] bcm2708_i2c_init_pinmode(1,2)
[    1.718109] bcm2708_i2c_init_pinmode(1,3)
[    1.722708] bcm2708_i2c bcm2708_i2c.1: BSC1 Controller at 0x20804000 (irq 79) (baudrate 100000)

```

What does '.0' mean at the end of bcm2708-i2s.0? I2S bus 0?

pelwell, 2015-02-23:
And `lsmod`?

Unfortunately the i2s module is silent during a successful boot, and there is no equivalent to spidev or i2c-dev that creates /dev/ entries.

I can't spend any more time on this, since you are creating a non-standard configuration, but I suggest you try using bcmrpi_defconfig, verify that it works, then work out what the significant differences are.

syntheticpp, 2015-02-23:
All is build statically, so lsmod shows nothing.
I'll try with bcmrpi_defconfig.

pelwell, 2015-02-23:
Note that it is normal to get some -517/probe deferral "errors" because load order is not guaranteed, but eventually the all components should be ready and the ALSA device will be created.

syntheticpp, 2015-02-23:
OK, I've build with bcmrpi_defconfig and after startup only the module ipv6 is loaded.
Then I call
```C
# modprobe -v i2c-bcm2708
# modprobe -v snd-soc-pcm512x
# modprobe -v snd-soc-bcm2708-i2s
# modprobe -v snd-soc-hifiberry-dacplus
# lsmod
Module                  Size  Used by    Not tainted
snd_soc_hifiberry_dacplus     1843  0 
snd_soc_bcm2708_i2s     5087  2 
regmap_mmio             1967  1 snd_soc_bcm2708_i2s
snd_soc_pcm512x         7329  1 
snd_soc_core           91366  3 snd_soc_hifiberry_dacplus,snd_soc_bcm2708_i2s,snd_soc_pcm512x
snd_compress            6742  1 snd_soc_core
regmap_i2c              1101  2 snd_soc_pcm512x,snd_soc_core
snd_pcm_dmaengine       2968  1 snd_soc_core
regmap_spi              1311  2 snd_soc_pcm512x,snd_soc_core
snd_pcm                63385  3 snd_soc_core,snd_pcm_dmaengine
snd_page_alloc          3167  1 snd_pcm
snd_timer              16456  1 snd_pcm
snd                    44121  7 snd_soc_core,snd_compress,snd_pcm,snd_timer
i2c_bcm2708             4131  0 
ipv6                  275085 14 
#
#dmesg
...
[   29.792681] bcm2708_i2c_init_pinmode(1,2)
[   29.797107] bcm2708_i2c_init_pinmode(1,3)
[   29.850292] bcm2708_i2c bcm2708_i2c.1: BSC1 Controller at 0x20804000 (irq 79) (baudrate 100000)
[   30.151377] pcm512x 1-004d: No SCLK, using BCLK: -2
[   30.188009] pcm512x 1-004d: Completed initialisation - pcm512x_probe
[   30.196992] pcm512x 1-004c: Failed to reset device: -5
[   30.209908] pcm512x: probe of 1-004c failed with error -5
[   30.368288] snd-rpi-hifiberry-dacplus snd-rpi-hifiberry-dacplus.0:  pcm512x-hifi <-> bcm2708-i2s.0 mappin
```
And it WORKS!

But when I configure above probed modules as static modules, DAC does not work any more.
I assume it misses some dependencies. Shouldn't these dependencies be resolved as static modules  by KConfig?

syntheticpp, 2015-02-23:
Seems static modules are broken. I've checked if here is a .o file for each .ko module when module support is disabled. All drivers are there but loading fails.

So I switched to loadable modules and use this modules:

```C
regmap-mmio
regmap-i2c
regmap-spi
i2c-bcm2708
spi-bcm2708
#
soundcore
snd
snd-timer
snd-compress
#
snd-page-alloc
snd-pcm
snd-pcm-dmaengine
#
snd-soc-core
snd-soc-pcm512x
snd-soc-bcm2708-i2s
snd-soc-hifiberry-dacplus
```

Many thanks for your hints Phil!

closed 2015-02-23 by syntheticpp