raspberrypi/linux #847: ALSA: dedicated devices for HDMI and analog audio

julianscheel, 2015-02-23:
I wonder if it would be possible to use HDMI and analog audio in parallel via alsa. I think it should be possible as it can be done through OMX, right?
If this should be supported by the vchiq core I might give a try at reworking the alsa driver to support it.

popcornmix, 2015-02-23:
Yes, the hardware supports it. (e.g. running two instances of hello_audio in parallel should work).
I believe the vchiq audio service (used by alsa) supports it - just open two instances and send control messages with different dest values.

The only thing I'm not sure of is if the alsa driver can switch a single subchannel's dest (e.g. through amixer cset). I think it should be trivial to implement - it's just knowing what the best way of doing it through alsa is.

julianscheel, 2015-02-24:
@popcornmix Thanks, I'll try to implement this sometime soon.

ashleyw-gh, 2015-03-05:
does this mean that separate audio streams could in theory be directed to both the HDMI and analog audio devices simultaneous? If so, this will be a great fix (I can think of squeezelite/Kodi and a host of other potential uses).

julianscheel, 2015-03-10:
@popcornmix I implemented this in a proof-of-concept style and it works flawless so far: https://github.com/julianscheel/linux/commit/8b77ad48c78b0c9b6d44dbfb0a58199c61476f66
I just create two pcm devices now and split the 8 available substreams to them. Additionally the playback device control is duplicated so that each pcm device can be configured independently. So it allows the user to do things like this:

```
amixer cset numid=3 1
amixer cset numid=4 2
speaker-test -Dhw:0,0 &
speaker-test -Dhw:0,1 -t sine
```

Which would play white noise on  analog out and a sine on hdmi out.

While it works well the design feels not really good, actually I'd prefer something like a static setup with a subdevice per output - but doing this castrates the total number of usable substreams per output. Do you have any thoughts on a proper approach for this?

popcornmix, 2015-03-10:
I agree allowing hdmi and analogue to be driven independently could be useful, but I'm not sure how it's best to expose this though alsa. It's also nice to support multiple sub-channels, to get mixing "for free".
Anyone with any better suggestions?

@pelwell any thoughts?

julianscheel, 2015-03-10:
Actually after thinking a bit more about it I think it's possible to create both device with 8 substreams (is 8 a hard limit of the hardware, btw?). In the pcm_open we just need to dynamically maintain a pool of total open substreams and abort with EBUSY if all 8 channels are in use. This should be quite straight forward as there is no hard linkage between alsa substreams and vchiq channels anyway, but all is maintained dynamically.
So we could easily have two pcm devices, one for analog, one for hdmi by default.
Or we could even create 3 - pcm 0 staying as it is now, so that no userspace change is required for existing applications. pcm 1 could be explicit analog/headphone out and pcm 2 explicit hdmi output.

What do you think about this? If you think it's sane I'll take a go at implementing it.

popcornmix, 2015-03-10:
8 sub channels is a software limit - it may handle more, but there is obviously some point where it will fail.
There is also ~32K per sub channel of (static) GPU memory used (we could possibly trim that), so making it arbitrarily large would be undesirable.

I could try increasing it in a test build to check it behaves okay with 16 (or more) sub channels.

I like the idea of keeping backwards compatibility, but do we let all three devices be used together? (i.e. possibly requiring 24 sub-channels)

julianscheel, 2015-03-10:
I think we can stay with 8 channels - we just dynamically share them between the three pcms. So we'd actually have 24 alsa substreams - but as soon as 8 are used in total (and hence 8 vchiq sub channels are in use) pcm_open would fail with EBUSY.

popcornmix, 2015-03-10:
Yes, sounds okay. Once the 8 channels are in use you should find:
https://github.com/raspberrypi/linux/blob/rpi-3.18.y/sound/arm/bcm2835-vchiq.c#L289
returns a fail, and so bcm2835_audio_open_connection returns a failure, so you may be able to avoid hard coding the number of substreams. That way the firmware could support more in future.

BigMacAttack, 2015-12-28:
What is the status on this? Has a decision been made on how to handle this in ALSA? I'd love to be able to output audio via HDMI and analog simultaneously using ALSA.

root42, 2016-07-29:
Same here. I have a RetroPie with a Shairport-Server. I would like the RetroPie to output via HDMI because of image/sound sync and the Shairport to output via analog so I don't have to turn on the TV to hear music.

giddyhup, 2016-09-13:
I second the request. It works with an USB DAC (sound card) and HDMI audio; being able to direct audio to the analog port alternatively makes sense.

JamesH65, 2017-05-17:
@popcornmix @julianscheel is there anything to add to this? Close or continue?

julianscheel, 2017-05-17:
@JamesH65 I still do have some dirty (outdated) kernel tree lying around somewhere, which implements this... I personally don't currently need this, but if there is someone who really needs it I could take some time and search for that tree and possibly do a bit of cleanup...

JamesH65, 2018-06-27:
@popcornmix @julianscheel is there anything to add to this? Close or continue?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

Druw2000, 2018-09-12:
Hello, please keep this issue open. I imagine multi channel PCM is possible with PulseAudio, though?

Holzhaus, 2021-11-29:
Any news on this? Currently I need to use an additional USB sound card due to this issue.

6by9, 2021-11-29:
HDMI and analogue audio are switched to being independent ALSA devices with https://github.com/raspberrypi/linux/commit/8e6c2ab786e7b183ca0b652ce67ff061eae20fce back in July 2020.

closed 2021-11-29 by 6by9