raspberrypi/linux #849: Gstreamer 1.4.5 and bcm2835-v4l2 don't get along

sphaero, 2015-02-24:
Using a fdsrc works but talking directly to it through v4l2src does nothing. Not a bit is send out through the network. It seems the pipeline doesn't flow.... Kernel errors at the bottom.

Using latest Raspbian (Linux 2.18.7+) and a compiled Gstreamer 1.4.5

This one works
```
$ raspivid -t 0 -w 1080 -h 720 -fps 25 -hf -b 2000000 -o - | gst-la
unch-1.0 -v fdsrc ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=192.168.1.1 port=5000
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = "video/x-h264\,\ width\=\(int\)1080\,\
height\=\(int\)720\,\ framerate\=\(fraction\)0/1\,\ parsed\=\(boolean\)true\,\ stream-format\=\(string\)a
vc\,\ alignment\=\(string\)au\,\ level\=\(string\)4\,\ profile\=\(string\)high\,\ codec_data\=\(buffer\)0
1640028ffe1000f27640028ac2b402202df2f00f1226a01000528ee025cb0"
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:src: caps = "application/x-rtp\,\ media\=\(string
\)video\,\ clock-rate\=\(int\)90000\,\ encoding-name\=\(string\)H264\,\ sprop-parameter-sets\=\(string\)\
"J2QAKKwrQCIC3y8A8SJq\\\,KO4CXLA\\\=\"\,\ payload\=\(int\)96\,\ ssrc\=\(uint\)473859634\,\ timestamp-offs
et\=\(uint\)1501147395\,\ seqnum-offset\=\(uint\)23019"
/GstPipeline:pipeline0/GstUDPSink:udpsink0.GstPad:sink: caps = "application/x-rtp\,\ media\=\(string\)vid
eo\,\ clock-rate\=\(int\)90000\,\ encoding-name\=\(string\)H264\,\ sprop-parameter-sets\=\(string\)\"J2QA
KKwrQCIC3y8A8SJq\\\,KO4CXLA\\\=\"\,\ payload\=\(int\)96\,\ ssrc\=\(uint\)473859634\,\ timestamp-offset\=\
(uint\)1501147395\,\ seqnum-offset\=\(uint\)23019"
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:sink: caps = "video/x-h264\,\ width\=\(int\)1080\
,\ height\=\(int\)720\,\ framerate\=\(fraction\)0/1\,\ parsed\=\(boolean\)true\,\ stream-format\=\(string
\)avc\,\ alignment\=\(string\)au\,\ level\=\(string\)4\,\ profile\=\(string\)high\,\ codec_data\=\(buffer
\)01640028ffe1000f27640028ac2b402202df2f00f1226a01000528ee025cb0"
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0: timestamp = 1501147395
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0: seqnum = 23019
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

^Cmmal: Aborting program

handling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:10.363012001
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```
This one doesn't
```
$ gst-launch-1.0 -v v4l2src do-timestamp=true ! video/x-h264,width=640,height=480,framerate=30/1 ! h264parse ! rtph264pay ! udpsink host=192.168.1.1 port=5000 sync=false Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:sink: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:01:16.673993691
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

Just noticed some kernel errors as well:
```
[ 9638.442183] Linux video capture interface: v2.00
[ 9638.485342] bcm2835-v4l2: scene mode selected 0, was 0
[ 9638.485580] bcm2835-v4l2: Work-around for gstreamer issue is active.
[ 9638.492510] bcm2835-v4l2: V4L2 device registered as video0 - stills mode > 1280x720
[ 9638.498031] bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.
[ 9661.282377] bcm2835-v4l2: error 0 waiting for frame completion
[ 9934.824835] bcm2835-v4l2: error 0 waiting for frame completion
[ 9969.360650] ------------[ cut here ]------------
[ 9969.360774] WARNING: CPU: 0 PID: 4036 at drivers/media/v4l2-core/videobuf2-core.c:2135 __vb2_queue_cancel+0xf4/0x160 [videobuf2_core]()
[ 9969.360793] Modules linked in: bcm2835_v4l2 videobuf2_vmalloc videobuf2_memops videobuf2_core v4l2_common videodev media snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd joydev evdev uio_pdrv_genirq uio [last unloaded: videobuf2_memops]
[ 9969.360892] CPU: 0 PID: 4036 Comm: gst-launch-1.0 Not tainted 3.18.7+ #755
[ 9969.360964] [<c00151fc>] (unwind_backtrace) from [<c0012710>] (show_stack+0x20/0x24)
[ 9969.361006] [<c0012710>] (show_stack) from [<c0555b30>] (dump_stack+0x20/0x28)
[ 9969.361052] [<c0555b30>] (dump_stack) from [<c0022ef4>] (warn_slowpath_common+0x7c/0x9c)
[ 9969.361085] [<c0022ef4>] (warn_slowpath_common) from [<c0022fd0>] (warn_slowpath_null+0x2c/0x34)
[ 9969.361144] [<c0022fd0>] (warn_slowpath_null) from [<bf13aa1c>] (__vb2_queue_cancel+0xf4/0x160 [videobuf2_core])
[ 9969.361229] [<bf13aa1c>] (__vb2_queue_cancel [videobuf2_core]) from [<bf13c7f8>] (vb2_queue_release+0x28/0x48 [videobuf2_core])
[ 9969.361300] [<bf13c7f8>] (vb2_queue_release [videobuf2_core]) from [<bf13c874>] (_vb2_fop_release+0x5c/0x84 [videobuf2_core])
[ 9969.361366] [<bf13c874>] (_vb2_fop_release [videobuf2_core]) from [<bf13c8d0>] (vb2_fop_release+0x34/0x38 [videobuf2_core])
[ 9969.361561] [<bf13c8d0>] (vb2_fop_release [videobuf2_core]) from [<bf0fd460>] (v4l2_release+0x48/0x88 [videodev])
[ 9969.361687] [<bf0fd460>] (v4l2_release [videodev]) from [<c013b920>] (__fput+0x90/0x20c)
[ 9969.361723] [<c013b920>] (__fput) from [<c013bb04>] (____fput+0x18/0x1c)
[ 9969.361759] [<c013bb04>] (____fput) from [<c003d498>] (task_work_run+0x9c/0xcc)
[ 9969.361797] [<c003d498>] (task_work_run) from [<c00250a8>] (do_exit+0x2ec/0xa58)
[ 9969.361829] [<c00250a8>] (do_exit) from [<c00258b0>] (do_group_exit+0x50/0xf0)
[ 9969.361858] [<c00258b0>] (do_group_exit) from [<c0025970>] (__wake_up_parent+0x0/0x30)
[ 9969.361892] [<c0025970>] (__wake_up_parent) from [<c000e8c0>] (ret_fast_syscall+0x0/0x48)
[ 9969.361909] ---[ end trace 73848e747642fe10 ]---
[10169.845176] bcm2835-v4l2: error 0 waiting for frame completion
[10409.715748] bcm2835-v4l2: error 0 waiting for frame completion
[10411.325881] bcm2835-v4l2: error 0 waiting for frame completion
[10759.471245] bcm2835-v4l2: error 0 waiting for frame completion
[10761.161298] bcm2835-v4l2: error 0 waiting for frame completion
[11200.460611] bcm2835-v4l2: error 0 waiting for frame completion
[11824.398182] bcm2835-v4l2: error 0 waiting for frame completion
```
Reverting back to the Raspbian provided Gstreamer 1.2 does work!

sphaero, 2015-02-24:
With Gst 1.2 I'm also seeing kernel errors and freezing of the Gstreamer pipeline:
```[ 1399.165362] bcm2835-v4l2: error 0 waiting for frame completion```

Last message from Gstreamer:
```
0:00:24.729681401  2235  0x1cd6c00 DEBUG             GST_MEMORY gstmemory.c:138:gst_memory_init: new memory 0x1cf4940, maxsize:521 offset:0 size:521
0:00:24.730995343  2235  0x1cd6c00 DEBUG              h264parse gsth264parse.c:166:gst_h264_parse_reset_frame:<h264parse0> reset frame
0:00:24.732203290  2235  0x1cd6c00 DEBUG              h264parse gsth264parse.c:814:gst_h264_parse_handle_frame:<h264parse0> last parse position 0
0:00:24.733166247  2235  0x1cd6c00 DEBUG      codecparsers_h264 gsth264parser.c:494:set_nalu_datas: Nal type 1, ref_idc 1
0:00:24.733482233  2235  0x1cd6c00 DEBUG      codecparsers_h264 gsth264parser.c:494:set_nalu_datas: Nal type 1, ref_idc 1
0:00:24.734349195  2235  0x1cd6c00 DEBUG      codecparsers_h264 gsth264parser.c:1296:gst_h264_parser_identify_nalu: Nal start 4, No end found
0:00:24.735609139  2235  0x1cd6c00 DEBUG              h264parse gsth264parse.c:850:gst_h264_parse_handle_frame:<h264parse0> not a complete nal found at offset 4
0:00:24.735937125  2235  0x1cd6c00 DEBUG             GST_MEMORY gstmemory.c:88:_gst_memory_free: free memory 0x1cf4940
0:00:24.736357106  2235  0x1cd6c00 DEBUG                basesrc gstbasesrc.c:2388:gst_base_src_get_range:<v4l2src0> calling create offset -1 length 4096, time 0
0:00:24.737390061  2235  0x1cd6c00 DEBUG                   v4l2 gstv4l2bufferpool.c:859:gst_v4l2_buffer_pool_acquire_buffer:<v4l2bufferpool0> acquire
0:00:24.739087986  2235  0x1cd6c00 DEBUG               GST_POLL gstpoll.c:1200:gst_poll_wait: timeout :99:99:99.999999999
```

6by9, 2015-02-24:
Kernel warning, not error. Already logged as https://github.com/raspberrypi/linux/issues/817

Do you have a latest version of GStreamer? There was a bug with it when handling V4L2 devices https://bugzilla.gnome.org/show_bug.cgi?id=726521
Workaround was added to the V4L2 driver by adding gst_v4l2src_is_broken=1 when loading bcm2835-v4l2. It is fixed in the latest GStreamer.

I know gstreamer works as jamesh has been asking me questions about it today.

sphaero, 2015-02-24:
Yes I'm using the gstreamer workaround. Also tried without the workaround.  If you mean latest == 1.4.5 yes I'm using that version. 

Is Jamesh using 1.4.5 or master git?

JamesH65, 2015-02-24:
I'm using the standard one from the repo, not sure of the version. Running
on Pi2.

sphaero, 2015-02-24:
That's version 1.2. It works for me as well although only for a while. See my comment about 1.2 which shows the logs.

Using rpicamsrc is stable.

6by9, 2015-02-24:
Hmm, 
````gsth264parser.c:1296:gst_h264_parser_identify_nalu: Nal start 4, No end found
Sounds like we're possibly needing a buffer bigger than that allocated. When piping raspivid via stdin then it can just read until it decides to say stop. Via V4L2 it'll probably get given buffer sizes.

You could try dropping the bitrate, otherwise I haven't got many ideas. I'd need to have a look at changes in v4l2src if I can find the appropriate branch.

sphaero, 2015-02-28:
Do you mean this branch? http://cgit.freedesktop.org/gstreamer/gst-plugins-good/tree/sys/v4l2?h=1.4

sphaero, 2015-02-28:
Btw, how does one set the bit rate on a v4l2src. I tried using v4l2-ctl --set-ctrl video_bitrate=100000 but that doesn't seem to work.

fnoop, 2016-05-13:
Using up to date raspbian kernel:
Linux raspberrypi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux
And up to date gstreamer from default repos:
```
Package: libgstreamer1.0-0
Status: install ok installed
Priority: optional
Section: libs
Installed-Size: 3779
Maintainer: Maintainers of GStreamer packages <pkg-gstreamer-maintainers@lists.alioth.debian.org>
Architecture: armhf
Multi-Arch: same
Source: gstreamer1.0
Version: 1.4.4-2
```
I get exactly the same problem as before, trying to use a simple v4l2src hangs the pipline at the same stage, with the same kernel entry:
```
$ gst-launch-1.0 -vvv v4l2src device=/dev/video0 ! 'video/x-h264,width=640,height=480,framerate=3/1' ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=192.168.1.65 port=5000
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:sink: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:01.568543550
Setting pipeline to PAUSED ...
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)avc\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1\,\ parsed\=\(boolean\)true\,\ level\=\(string\)4\,\ profile\=\(string\)high\,\ codec_data\=\(buffer\)01640028ffe1000e27640028ac2b40501ed00f1226a001000428ee1f2c"
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

The raspberry and camera itself work absolutely fine, for example this pipeline works:
```
gst-launch-1.0 -vvv v4l2src device=/dev/video0 ! 'video/x-raw,width=1280,height=720,framerate=30/1' ! h264parse ! omxh264enc ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=192.168.1.65 port=5000 sync=false
```
It's requesting h-264 through v4l2src that hangs, no matter what caps or parameters I use.  Kernel entry each time is:
```
May 13 07:52:33 raspberrypi kernel: [  263.336191] bcm2835-v4l2: error 0 waiting for frame completion
```
This seems to be a pretty major bug, or am I doing something really wrong?  There are quite a few references on the interweb suggesting that lots of other people are doing this successfully.
I've tried with and without the kernel module workaround.

fnoop, 2016-05-13:
Tried updating to the 4.4 kernel with rpi-update, same problem:
```
Linux raspberrypi 4.4.9-v7+ #884 SMP Fri May 6 17:28:59 BST 2016 armv7l GNU/Linux
```

6by9, 2016-05-13:
Just tried
```GST_DEBUG=v4l2src:7 gst-launch-1.0 -vvv v4l2src device=/dev/video0 ! 'video/x-h264,width=640,height=480,framerate=30/1' ! fakesink```
and that reports lots of frames being generated, although it looks like at only 15fps. ```v4l2-ctl -V``` shows that the driver is at 640x480 H264.

Insert H264 parse into the pipeline and things grind to a halt
```GST_DEBUG=h264parse:7,v4l2src:7 gst-launch-1.0 -vvv v4l2src device=/dev/video0 ! 'video/x-h264,width=640,height=480,framerate=30/1' ! queue ! h264parse ! fakesink```

```gst_h264_parse_handle_frame:<h264parse0> Looking for more```
after 3 buffers have been passed. v4l2src stalls because it never gets the buffers back to fill.

I haven't studied the source, but what do you believe h264parse is doing for you? v4l2src should already be providing correctly framed H264 packets, so unlike fdsrc there is no need for it to be creating the correct framing.
Dig into h264parse to determine why it wants more data - what is it waiting for?

6by9, 2016-05-13:
Load the V4L2 driver with extra debug enabled (sudo modprobe bcm2835-v4l2 debug=1) and I can see we get 3 buffers allocated. There are 4 callbacks with respective lengths of 26 (headers), 1820 (I-frame), 421 (P-frame), and 358 (P-frame) bytes. Only 1 buffer ever gets returned and refilled with the 358 byte frame.

On ctrl-C'ing gst-launch I suddenly get 3 buffer-prepare/buffer_queue calls as the buffers finally get returned for filling with something new.

fnoop, 2016-05-13:
If I try without h264parse I get errors:
```
gst-launch-1.0 -vvv v4l2src device=/dev/video0 ! 'video/x-h264,width=640,height=480,framerate=30/1' ! rtph264pay config-interval=1 pt=96 ! udpsink host=192.168.1.6 port=5000
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:sink: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:src: caps = "application/x-rtp\,\ media\=\(string\)video\,\ payload\=\(int\)96\,\ clock-rate\=\(int\)90000\,\ encoding-name\=\(string\)H264\,\ ssrc\=\(uint\)3848481716\,\ timestamp-offset\=\(uint\)3188774680\,\ seqnum-offset\=\(uint\)3403"
/GstPipeline:pipeline0/GstUDPSink:udpsink0.GstPad:sink: caps = "application/x-rtp\,\ media\=\(string\)video\,\ payload\=\(int\)96\,\ clock-rate\=\(int\)90000\,\ encoding-name\=\(string\)H264\,\ ssrc\=\(uint\)3848481716\,\ timestamp-offset\=\(uint\)3188774680\,\ seqnum-offset\=\(uint\)3403"
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:src: caps = "application/x-rtp\,\ media\=\(string\)video\,\ clock-rate\=\(int\)90000\,\ encoding-name\=\(string\)H264\,\ sprop-parameter-sets\=\(string\)\"J2QAKKwrQFAe0A8SJqAA\\\,KO4fLA\\\=\\\=\"\,\ payload\=\(int\)96\,\ ssrc\=\(uint\)3848481716\,\ timestamp-offset\=\(uint\)3188774680\,\ seqnum-offset\=\(uint\)3403"
/GstPipeline:pipeline0/GstUDPSink:udpsink0.GstPad:sink: caps = "application/x-rtp\,\ media\=\(string\)video\,\ clock-rate\=\(int\)90000\,\ encoding-name\=\(string\)H264\,\ sprop-parameter-sets\=\(string\)\"J2QAKKwrQFAe0A8SJqAA\\\,KO4fLA\\\=\\\=\"\,\ payload\=\(int\)96\,\ ssrc\=\(uint\)3848481716\,\ timestamp-offset\=\(uint\)3188774680\,\ seqnum-offset\=\(uint\)3403"
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0: timestamp = 3188815765
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0: seqnum = 3403
WARNING: from element /GstPipeline:pipeline0/GstUDPSink:udpsink0: Error sending UDP packet
Additional debug info:
gstmultiudpsink.c(602): gst_multiudpsink_render (): /GstPipeline:pipeline0/GstUDPSink:udpsink0:
Reason: Error sending message: Bad address
Caught SIGSEGV
#0  0x76beac80 in poll () at ../sysdeps/unix/syscall-template.S:81
#1  0x76cd5528 in ?? () from /lib/arm-linux-gnueabihf/libglib-2.0.so.0
Spinning.  Please run 'gdb gst-launch-1.0 5286' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:06.091175240
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
```

If I try use x-raw or image/jpeg cap and encode but otherwise the same pipeline then it sends the video correctly to the client:
```
gst-launch-1.0 -vvv v4l2src device=/dev/video0 ! 'video/x-raw,width=640,height=480,framerate=30/1' ! omxh264enc ! rtph264pay config-interval=1 pt=96 ! udpsink host192.168.1.65 port=5000
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = "video/x-raw\,\ format\=\(string\)I420\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ interlace-mode\=\(string\)progressive\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = "video/x-raw\,\ format\=\(string\)I420\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ interlace-mode\=\(string\)progressive\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstOMXH264Enc-omxh264enc:omxh264enc-omxh264enc0.GstPad:sink: caps = "video/x-raw\,\ format\=\(string\)I420\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ interlace-mode\=\(string\)progressive\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = "video/x-raw\,\ format\=\(string\)I420\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ interlace-mode\=\(string\)progressive\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstOMXH264Enc-omxh264enc:omxh264enc-omxh264enc0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ profile\=\(string\)high\,\ level\=\(string\)4\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1\,\ framerate\=\(fraction\)30/1"
[... cut ...] )1990417924\,\ timestamp-offset\=\(uint\)1483664605\,\ seqnum-offset\=\(uint\)6682"
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:06.879109301
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```
I thought I usually needed h264parse but seems not even after omxh264enc which is good.  Interestingly if I pipe v4l2src x-h264 through mpegtsmux to a filesink then it seems to create a good file recording.

6by9, 2016-05-13:
h264parse is in gst-plugins-bad - there may be a reason for that.

Sorry, but the issue doesn't appear to be in the Pi code unless you can produce some evidence as to why h264parse thinks the first IDR frame is invalid and doesn't apparently consume any more of the data (it has 2 more buffers sitting there waiting).

fnoop, 2016-05-13:
gstreamer-omx depends on plugins-bad.  If you try and connect v4l2src to just decodebin it tries to run it through h264parse and hangs.
Connecting rtph264pay to tcpserversink gives the same SIGSEGV, any idea where this is coming from?
```
GST_DEBUG=udpsink:7 gst-launch-1.0 -vvv v4l2src device=/dev/video0 ! 'video/x-h264,width=640,height=480,framerate=30/1' ! rtph264pay confi-interval=1 pt=96 ! tcpserversink port=5000
Setting pipeline to PAUSED ...
/GstPipeline:pipeline0/GstTCPServerSink:tcpserversink0: current-port = 5000
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:sink: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = "video/x-h264\,\ stream-format\=\(string\)byte-stream\,\ alignment\=\(string\)au\,\ framerate\=\(fraction\)30/1\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1"
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:src: caps = "application/x-rtp\,\ media\=\(string\)video\,\ payload\=\(int\)96\,\ clock-rate\=\(int\)90000\,\ encoding-name\=\(string\)H264\,\ ssrc\=\(uint\)3107904018\,\ timestamp-offset\=\(uint\)865708876\,\ seqnum-offset\=\(uint\)30852"
/GstPipeline:pipeline0/GstTCPServerSink:tcpserversink0.GstPad:sink: caps = "application/x-rtp\,\ media\=\(string\)video\,\ payload\=\(int\)96\,\ clock-rate\=\(int\)90000\,\ encoding-name\=\(string\)H264\,\ ssrc\=\(uint\)3107904018\,\ timestamp-offset\=\(uint\)865708876\,\ seqnum-offset\=\(uint\)30852"
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:src: caps = "application/x-rtp\,\ media\=\(string\)video\,\ clock-rate\=\(int\)90000\,\ encoding-name\=\(string\)H264\,\ sprop-parameter-sets\=\(string\)\"J2QAKKwrQFAe0A8SJqAA\\\,KO4fLA\\\=\\\=\"\,\ payload\=\(int\)96\,\ ssrc\=\(uint\)3107904018\,\ timestamp-offset\=\(uint\)865708876\,\ seqnum-offset\=\(uint\)30852"
/GstPipeline:pipeline0/GstTCPServerSink:tcpserversink0.GstPad:sink: caps = "application/x-rtp\,\ media\=\(string\)video\,\ clock-rate\=\(int\)90000\,\ encoding-name\=\(string\)H264\,\ sprop-parameter-sets\=\(string\)\"J2QAKKwrQFAe0A8SJqAA\\\,KO4fLA\\\=\\\=\"\,\ payload\=\(int\)96\,\ ssrc\=\(uint\)3107904018\,\ timestamp-offset\=\(uint\)865708876\,\ seqnum-offset\=\(uint\)30852"
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0: timestamp = 865751021
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0: seqnum = 30852
Caught SIGSEGV
#0  0x76bfac80 in poll () at ../sysdeps/unix/syscall-template.S:81
#1  0x76ce5528 in ?? () from /lib/arm-linux-gnueabihf/libglib-2.0.so.0
Spinning.  Please run 'gdb gst-launch-1.0 1620' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:13.682540178
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
```
I'm afraid you've reached the limit of my abilities/knowledge here :)  But as it stands the raspberrypi v4l2src is of limited use with gstreamer, would be great to fix it if possible.

6by9, 2016-05-13:
I didn't say that plugins-bad was inherently bad, but the reasoning behind the name from the GStreamer project (https://gstreamer.freedesktop.org/modules/gst-plugins-bad.html) is 

> GStreamer Bad Plug-ins is a set of plug-ins that aren't up to par compared to the rest. They might be close to being good quality, but they're missing something - be it a good code review, some documentation, a set of tests, a real live maintainer, or some actual wide use. 

At least we're not using plugins-ugly.

No, I haven't looked into their code as to why it SIGSEGVs - as suggested running "gdb gst-launch-1.0 1620" (or whatever the PID changes to on next run). "thread apply all bt" should dump a backtrace of all threads.
That is only of any real use if you have an exact match source tree to compare the code against.

Always start with the most basic pipeline and build up. Start with one component and direct it into fakesink. Then two. Then three. etc. That way you can tell which bit is upset at what it has been presented.

fnoop, 2016-05-17:
Thanks 6by9.  Looks like the problem lies with the interaction between using v4l2src capturing h264 and rtph264pay together (which unfortunately is the only way to stream h264 with v4l2 without decoding).  I suspect v4l2src is outputting a hokey packet somewhere that both h264parse and rtph264pay aren't very good at coping with (they both fail separately with v4l2src h264 output but work with x264 output) .  Using raspivid and piping to the gstreamer fdsrc and through h264parse and rtph264pay is fine, and I've spent a couple of days compiling latest (1.90) gstreamer which works just fine.
The next version of debian comes with gstreamer 1.8.1 so likely the problem will be fixed there, but would be nice to fix raspbian jessie if possible.
Is this the correct place to lodge and investigate raspbian gstreamer / v4l2 problem, or is there a better place? (forums?)

6by9, 2016-05-17:
GST 1.90 works fine using v4l2src to h264parse or rtph264pay?
If so we can look at the history of those and see what change has been made. Whether we can persuade the Raspbian maintainers to accept another patch on GST 1.2 is a different question. Otherwise if we can see what that change did, we may be able to get the V4L2 driver to produce a stream that GST is happier with. 

Most likely it is trying to parse the headers and it doesn't like SPS and PPS being combined into one buffer, or some option in the headers is missing (there are lots of optional fields). Identifying exactly what causes the stall is the bigger task.

Leave this as an issue here for now. It is really a GST issue as the stream is valid, but if we can make a stream that it is happier with then that's a kernel or firmware mod.
Feel free to start a forum thread too as you may pick up others who have a little time to investigate. I'd suggest a link to this issue in the first post on the thread so that they can read what has already been investigated.

6by9, 2016-05-31:
@fnoop Could you confirm if there is still an issue with GST 1.90? 

Is there somewhere that I can download GST 1.8 or 1.9 for Pi to test? I did try building GST myself and gave up after wasting a fair chunk of time on it without success. I haven't got to the bottom of why GST 1.4 doesn't like the stream as yet.

fnoop, 2016-05-31:
@6by9 It seems to work great with 1.9.
There are no precompiled binaries that I know of, if you're determined to build it my recipe is here:
https://github.com/fnoop/maverick/blob/master/manifests/maverick-modules/maverick-vision/manifests/init.pp

It's in 'puppet' code so you'll have to extract the command logic yourself, but it should be reasonably straight forward.

1.9 seems to have deprecated h264parse so you go straight from the source to the pay.

6by9, 2016-05-31:
Thank you, and that is good news.

It sounds like they may have moved the stream parsing logic (which is what h264parse was) into the sink component. That sounds a slightly quirky and possibly backwards step if you're trying to support filesrc from raspivid.
I may take your base manifest and commands and try building 1.4 again so I can throw in huge amounts more logging. Complying with all the build dependencies was half the problem.

Ruffio, 2016-08-14:
@sphaero has this issue been resolved? If yes, please close this issue. Thanks.

sphaero, 2016-08-15:
I have no idea. I have no RPI available currently. Perhaps anybody else can confirm? Otherwise I'll close the issue in a few days.

6by9, 2016-08-15:
I haven't made any changes to address the issue, but fnoop is reporting GST 1.9.0 as working fine (it drops h264parse as a component).

Leave this open for now and if I find a few minutes to dig into h264parse then I will. Memory says it wasn't too pleasant when I looked last time. It may be as simple as throwing an extra couple of buffers into the mix, or GST may be wanting SPS and PPS headers separately.

JamesH65, 2017-05-17:
@6by9 Assign to you, or close?

JamesH65, 2018-06-27:
@6by9 Did you get a change to look at this?  Any ideas what to do with this issue?

6by9, 2018-06-27:
No, and not really. It looked to be H264 header parsing, so potentially that we put SPS and PPS into one V4L2 buffer by themselves.
Stretch is now the standard though, and that bumps GStreamer up to 1.10 IIRC. h264parse certainly still exists in 1.15 (current latest), and I thought it was in the standard Stretch.

I don't suppose you fancy running a quick test, or at least shouting if you have a standard Raspbian build available (mine are all customised in various ways at the moment).

6by9, 2018-12-10:
@JamesH65 Please close. This is a GStreamer bug in an old version of GStreamer. Current versions all seem happy.