raspberrypi/linux #851: DTS naming for RPI board

agherzan, 2015-02-25:
Hi guys,

It seems that the current dts tree is a little confusing. We have some files that are prefixed with the processor family code and some with the actual linux implementation processor code:
bcm2708*.dts
bcm2835*.dts

Are the bcm2835 ones obsolete? Should we remove them to avoid this confusion?

pelwell, 2015-02-25:
No - bcm2835 is an independent, upstreamed implementation of a BCM2835 port using Device Tree best practices. Current RPi kernels are using some of the bcm2835 modules, and we would like to use more and perhaps even merge the two worlds, but for now they are intentionally separate.

closed 2015-02-27 by pelwell

agherzan, 2015-03-01:
Thank you. That clarifies.