raspberrypi/linux #852: Mouse slow in 5.0.3 RPI2B

Lob0426, 2015-02-26:
Mouse slow/dragging issue in 5.0.3
This issue has cropped back up in 5.0.3 for the Raspberry Pi 2 B. Default settings of OpenElec. Mouse drags in the video window. If the video is in the background the mouse is fine. As soon as the video Window is to the front Mouse slows again.
 "usbhid.mousepoll=0" or "usbhid.mousepoll=1" in cmdline.txt has no effect.

My mouse is a wired USB DELL mouse Model Number: M-UAV-DEL8

kiicia, 2015-02-26:
I've got my new Pi2 yesterday and tried clean installs of raspbian and arch. In both of them my microsoft wireless mouse (mouse + keyboard actually) works so bad I was almost sure I have defective Pi2 unit... while I don't remember ever having such a problem with my old Pi B (not B+).

popcornmix, 2015-02-26:
@Lob0426 I suspect your issue is unrelated to kernel (the kernel doesn't care if the mouse is over video or non-video) and should be reported in kodi forum.

@kiicia Have you added usbhid.mousepoll=0 to cmdline.txt?

closed 2015-02-27 by Lob0426

kiicia, 2015-02-27:
sorry for late reply, I have tried it and it seems to solve the wireless mouse problem (wired mouse was working ok even without this param)

now both mice works almost exactly the same with short cursor jumps (if you move them fast) instead of that unnatural, smooth-yet-laggy motion