raspberrypi/linux #855: gpio_set_value kernel module does not work in Pi 2

sriranjanr, 2015-02-27:
Hello,

We recently bought a raspberry Pi 2 B+model. We used to toggle pin state (gpio4) using gpio_set_value() kernel function but now in the pi2 it is not working.I read somewhere that the base address of gpio has been change from 0x20000000 to 0x3f0000000 and now the kernel function does not work.Could you please fix this.

Thanks in advance

pelwell, 2015-02-27:
GPIOs are working on the latest kernel build. Please say what you do, what used to happen, and what happens now.

laxitg, 2015-02-27:
Hi pelwell,

I am in same team as sriranjanr, and following are the details:

We want to change the GPIO pin status when a frame gets queued to the buffer from the camera, we are using the v4l2 driver, hence we modified v4l2-ioctl.c file and added calls to gpio_request_one for the first function call, and gpio_set_value and gpio_free for all subsequent calls.

On the B+ model, for kernel version rpi-3.12.y it used to work fine, the led attached to the gpio pin was blinking, as it was supposed to and value of gpio pin accessed with gpio_get_value and printed with printk were behaving as expected as well.

We recently purchased pi2 model and made the same changes to kernel version rpi-3.18.y. cross compilation was smooth sailing, once we exectued our code led's weren't blinking and values retrieved with gpio_get_value were always 0, even when they had been explicitly set as 1. Although when we tried with shell scrpit ( http://elinux.org/RPi_Low-level_peripherals#Shell ) the led's were turning on and off when the gpio pin status was changed. We tried the C code by Dom and Gert  ( http://elinux.org/RPi_Low-level_peripherals#C ) as it is and didn't work, then we changed the macro BCM2708_PERI_BASE from 0x20000000 to 0x3f000000 and it started working.

As of now, the same changes which worked for us in rpi-3.12.y on pi B+ are not working in rpi-3.18.y on pi2.

Any help would be greatly appreciated. :)

Thanks,
Laxit

pelwell, 2015-02-27:
The shell script is using the driver via the /sys/class/gpio virtual filesystem, and as you say that works. The change in I/O space address should be completely hidden by the GPIO module, but code which directly mmap()s the I/O space (or uses /dev/mem) will need updating for Pi 2. As you are using the kernel interface, those operations should end up being routed to the GPIO driver, so it ought to work.

My first suggestion would be make sure you are on the latest kernel release - pull and rebase your changes against the latest version (git pull --rebase). If the problem persists, please upload your patch somewhere - pastebin, or even here if it isn't that large - and I'll read it through.

sriranjanr, 2015-02-27:
We did a "git pull rebase" on the kernel and it worked like old.Thank you for your time Phil Elwell.

closed 2015-02-27 by pelwell