raspberrypi/linux #859: Repeatable kernel crash in #737

studio-dude, 2015-02-28:
I have a node.js app that repeatable causes the following kernel error:

pi@pi ~ $ uname -a
Linux pi 3.12.36+ #737 PREEMPT Wed Jan 14 19:40:07 GMT 2015 armv6l GNU/Linux

Feb 27 19:33:22 pi kernel: [169472.321768] Modules linked in: snd_bcm2835 snd_pcm snd_page_alloc snd_seq snd_seq_device snd_timer snd pl2303 usbserial leds_gpio led_class spi_bcm2708
Feb 27 19:33:22 pi kernel: [169472.339986] CPU: 0 PID: 2472 Comm: node Not tainted 3.12.36+ #737
Feb 27 19:33:22 pi kernel: [169472.347687] task: d6178c80 ti: d61ee000 task.ti: d61ee000
Feb 27 19:33:22 pi kernel: [169472.354723] PC is at vfp_save_state+0x0/0x28
Feb 27 19:33:22 pi kernel: [169472.360637] LR is at vfp_sync_hwstate+0x70/0x7c
Feb 27 19:33:22 pi kernel: [169472.366797] pc : [<c00099c4>]    lr : [<c0009588>]    psr: 60000013
Feb 27 19:33:22 pi kernel: [169472.366797] sp : d61efeb0  ip : 00000018  fp : beb11de0
Feb 27 19:33:22 pi kernel: [169472.381537] r10: 01771044  r9 : d6179028  r8 : 00000000
Feb 27 19:33:22 pi kernel: [169472.388375] r7 : d61effb0  r6 : beb11cd0  r5 : 80000780  r4 : d61ee030
Feb 27 19:33:22 pi kernel: [169472.396549] r3 : d61ee0f8  r2 : c062ce60  r1 : c0000780  r0 : d61ee0f8
Feb 27 19:33:22 pi kernel: [169472.404701] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Feb 27 19:33:22 pi kernel: [169472.414917] Control: 00c5387d  Table: 16140008  DAC: 00000015
Feb 27 19:33:22 pi kernel: [169472.563714] [<c00099c4>] (vfp_save_state+0x0/0x28) from [<c0009588>] (vfp_sync_hwstate+0x70/0x7c)
Feb 27 19:33:22 pi kernel: [169472.576493] [<c0009588>] (vfp_sync_hwstate+0x70/0x7c) from [<c0009794>] (vfp_preserve_user_clear_hwstate+0x20/0x8c)
Feb 27 19:33:22 pi kernel: [169472.590885] [<c0009794>] (vfp_preserve_user_clear_hwstate+0x20/0x8c) from [<c0010a8c>] (setup_sigframe+0x190/0x1a0)
Feb 27 19:33:22 pi kernel: [169472.605262] [<c0010a8c>] (setup_sigframe+0x190/0x1a0) from [<c0010ee8>] (do_signal+0x2e8/0x440)
Feb 27 19:33:22 pi kernel: [169472.617874] [<c0010ee8>] (do_signal+0x2e8/0x440) from [<c00111d8>] (do_work_pending+0xa4/0xb4)
Feb 27 19:33:22 pi kernel: [169472.630384] [<c00111d8>] (do_work_pending+0xa4/0xb4) from [<c000df00>] (work_pending+0xc/0x20)
Feb 27 19:33:23 pi kernel: [169472.661933] ---[ end trace ca4423d502cf7049 ]---
Feb 27 19:33:23 pi kernel: [169472.675291] note: node[2472] exited with preempt_count 2
Feb 27 19:33:23 pi kernel: [169472.705002] Modules linked in: snd_bcm2835 snd_pcm snd_page_alloc snd_seq snd_seq_device snd_timer snd pl2303 usbserial leds_gpio led_class spi_bcm2708
Feb 27 19:33:23 pi kernel: [169472.748913] CPU: 0 PID: 2472 Comm: node Tainted: G      D      3.12.36+ #737
Feb 27 19:33:23 pi kernel: [169472.776293] [<c00143d4>] (unwind_backtrace+0x0/0xec) from [<c00116b8>] (show_stack+0x10/0x14)
Feb 27 19:33:23 pi kernel: [169472.805157] [<c00116b8>] (show_stack+0x10/0x14) from [<c04459b8>] (__schedule_bug+0x48/0x64)
Feb 27 19:33:23 pi kernel: [169472.833943] [<c04459b8>] (__schedule_bug+0x48/0x64) from [<c044aebc>] (__schedule+0x518/0x5ac)
Feb 27 19:33:23 pi kernel: [169472.862904] [<c044aebc>] (__schedule+0x518/0x5ac) from [<c0045758>] (__cond_resched+0x24/0x34)
Feb 27 19:33:23 pi kernel: [169472.891872] [<c0045758>] (__cond_resched+0x24/0x34) from [<c044b6c4>] (_cond_resched+0x3c/0x44)
Feb 27 19:33:23 pi kernel: [169472.920995] [<c044b6c4>] (_cond_resched+0x3c/0x44) from [<c00c3004>] (__get_user_pages.part.91+0xa0/0x408)
Feb 27 19:33:23 pi kernel: [169472.950871] [<c00c3004>] (__get_user_pages.part.91+0xa0/0x408) from [<c00b8214>] (get_user_pages_fast+0x60/0x78)
Feb 27 19:33:23 pi kernel: [169472.981397] [<c00b8214>] (get_user_pages_fast+0x60/0x78) from [<c0065af8>] (get_futex_key+0xec/0x244)
Feb 27 19:33:23 pi kernel: [169473.010922] [<c0065af8>] (get_futex_key+0xec/0x244) from [<c0065d00>] (futex_wake+0x30/0x160)
Feb 27 19:33:23 pi kernel: [169473.039707] [<c0065d00>] (futex_wake+0x30/0x160) from [<c0067e0c>] (do_futex+0x120/0xb78)
Feb 27 19:33:23 pi kernel: [169473.068342] [<c0067e0c>] (do_futex+0x120/0xb78) from [<c00688e4>] (SyS_futex+0x80/0x160)
Feb 27 19:33:23 pi kernel: [169473.097280] [<c00688e4>] (SyS_futex+0x80/0x160) from [<c001d328>] (mm_release+0xfc/0x138)
Feb 27 19:33:23 pi kernel: [169473.126477] [<c001d328>] (mm_release+0xfc/0x138) from [<c0021774>] (do_exit+0x110/0x970)
Feb 27 19:33:23 pi kernel: [169473.155673] [<c0021774>] (do_exit+0x110/0x970) from [<c00119f0>] (die+0x334/0x390)
Feb 27 19:33:23 pi kernel: [169473.184507] [<c00119f0>] (die+0x334/0x390) from [<c0008334>] (do_undefinstr+0x1b0/0x1dc)
Feb 27 19:33:23 pi kernel: [169473.213890] [<c0008334>] (do_undefinstr+0x1b0/0x1dc) from [<c044d0ec>] (__und_svc_finish+0x0/0x34)
Feb 27 19:33:23 pi kernel: [169473.244100] Exception stack(0xd61efe28 to 0xd61efe70)
Feb 27 19:33:23 pi kernel: [169473.259851] fe20:                   d61ee0f8 c0000780 c062ce60 d61ee0f8 d61ee030 80000780
Feb 27 19:33:23 pi kernel: [169473.288918] fe40: beb11cd0 d61effb0 00000000 d6179028 01771044 beb11de0 00000018 d61efeb0
Feb 27 19:33:23 pi kernel: [169473.317875] fe60: c0009588 c00099c4 60000013 ffffffff
Feb 27 19:33:23 pi kernel: [169473.333576] [<c044d0ec>] (__und_svc_finish+0x0/0x34) from [<c00099c4>] (vfp_save_state+0x0/0x28)
Feb 27 19:33:23 pi kernel: [169473.363003] [<c00099c4>] (vfp_save_state+0x0/0x28) from [<c0009588>] (vfp_sync_hwstate+0x70/0x7c)
Feb 27 19:33:23 pi kernel: [169473.392348] [<c0009588>] (vfp_sync_hwstate+0x70/0x7c) from [<c0009794>] (vfp_preserve_user_clear_hwstate+0x20/0x8c)
Feb 27 19:33:23 pi kernel: [169473.423359] [<c0009794>] (vfp_preserve_user_clear_hwstate+0x20/0x8c) from [<c0010a8c>] (setup_sigframe+0x190/0x1a0)
Feb 27 19:33:23 pi kernel: [169473.454368] [<c0010a8c>] (setup_sigframe+0x190/0x1a0) from [<c0010ee8>] (do_signal+0x2e8/0x440)
Feb 27 19:33:23 pi kernel: [169473.483620] [<c0010ee8>] (do_signal+0x2e8/0x440) from [<c00111d8>] (do_work_pending+0xa4/0xb4)
Feb 27 19:33:23 pi kernel: [169473.512866] [<c00111d8>] (do_work_pending+0xa4/0xb4) from [<c000df00>] (work_pending+0xc/0x20)

Happy to help debug this more, but I'm not sure exactly how. As I said, it's very repeatable.

popcornmix, 2015-02-28:
3.12 kernel is no longer supported. Can you apt-get upgrade to get current 3.18 kernel and confirm if the issue is still present?

studio-dude, 2015-02-28:
OK thanks, upgraded. I'll update if the problem persists...

studio-dude, 2015-03-02:
OK problem still occurs:

    [44897.980959] BUG: unsupported FP instruction in kernel mode
    [44897.988358] Internal error: Oops - undefined instruction: 0 [#1] PREEMPT ARM
    [44897.998862] Modules linked in: snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd pl2303 usbserial spi_bcm2708 uio_pdrv_genirq uio
    [44898.014916] CPU: 0 PID: 2415 Comm: node Not tainted 3.18.7+ #755
    [44898.022729] task: d6321b00 ti: d62ec000 task.ti: d62ec000
    [44898.029912] PC is at vfp_save_state+0x0/0x28
    [44898.035972] LR is at vfp_sync_hwstate+0x7c/0x88
    [44898.042276] pc : [<c0009d20>]    lr : [<c00098b0>]    psr: 60000113
    [44898.042276] sp : d62ede70  ip : d62ede88  fp : d62ede84
    [44898.057358] r10: d6321f58  r9 : d62ec000  r8 : 00000000
    [44898.064377] r7 : d62edec0  r6 : bea1acd8  r5 : 80000780  r4 : d62ec030
    [44898.072714] r3 : d62ec0f8  r2 : c0855904  r1 : c0000780  r0 : d62ec0f8
    [44898.081022] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
    [44898.091720] Control: 00c5387d  Table: 172e8008  DAC: 00000015
    [44898.099297] Process node (pid: 2415, stack limit = 0xd62ec1b0)
    [44898.106934] Stack: (0xd62ede70 to 0xd62ee000)
    [44898.113119] de60:                                     bea1abd0 d62ec000 d62edea4 d62ede88
    [44898.124980] de80: c0009ad4 c0009840 00000120 00000000 bea1aae0 00000000 d62edebc d62edea8
    [44898.136965] dea0: c0011874 c0009ab0 d62edfb0 bea1aae0 d62edf84 d62edec0 c0011cb8 c00116e0
    [44898.148962] dec0: 0028fa94 04000000 b6cbcb20 7ffbfeff fffffffe 00000011 00000000 00040001
    [44898.160958] dee0: 00004d1e 000003e8 00000000 00000001 00000003 c0556c74 a0317588 000028d5
    [44898.173013] df00: d62edf5c d62edf10 c00266a8 c0026234 00010000 c0811200 00000001 00400000
    [44898.185199] df20: c0823388 60000193 00000000 00000000 d62edf5c d62edf40 c0069544 c00c3f98
    [44898.197503] df40: c08104c0 d62ec000 c0837108 d62ec008 d62ec000 00000011 d62edfb0 d62ec008
    [44898.209942] df60: d62ec000 00000000 d62edfb0 00000000 d62ec000 01d40044 d62edfac d62edf88
    [44898.222522] df80: c0011fcc c0011a04 00c5387d 00000001 0030db24 60000010 f200b200 00c5387d
    [44898.235135] dfa0: 00000000 d62edfb0 c000e918 c0011f18 01d40040 00000000 01d40038 330080a8
    [44898.247828] dfc0: 00000002 003eedd4 bea1adec 4f6080c9 4610b3e9 4c6096c4 01d40044 bea1ade0
    [44898.260625] dfe0: bea1add4 bea1add0 39d0a35c 0030db24 60000010 ffffffff 00000000 00000000
    [44898.273491] [<c0009d20>] (vfp_save_state) from [<c0009ad4>] (vfp_preserve_user_clear_hwstate+0x30/0x9c)
    [44898.287607] [<c0009ad4>] (vfp_preserve_user_clear_hwstate) from [<c0011874>] (setup_sigframe+0x1a0/0x1b0)
    [44898.301888] [<c0011874>] (setup_sigframe) from [<c0011cb8>] (do_signal+0x2c0/0x400)
    [44898.314226] [<c0011cb8>] (do_signal) from [<c0011fcc>] (do_work_pending+0xc0/0x110)
    [44898.326552] [<c0011fcc>] (do_work_pending) from [<c000e918>] (work_pending+0xc/0x20)
    [44898.338944] Code: e12fff1e e1a0200d e1a0e009 eafffe76 (eca00b20)
    [44898.363805] ---[ end trace de1f09b100f292e5 ]---
    [44898.377259] note: node[2415] exited with preempt_count 2

popcornmix, 2015-03-02:
Can you provide a tar file containing something that can be run on raspbian to produce this error?

popcornmix, 2015-03-02:
Your problem looks similar to:
https://github.com/raspberrypi/linux/issues/600
Interestingly that appeared to get fixed by an update to BOINC:
http://boinc.berkeley.edu/dev/forum_thread.php?id=9222
not by a kernel change, so I suspect the bug is in your user code.

Ferroin, 2015-03-04:
If you are hitting an illegal instruction in kernel mode, it's not a userspace issue, it's a kernel bug.  The only reason that it doesn't happen with the updated version of BOINC is that it was modified to not hit that codepath in the kernel, thus avoiding triggering the bug.

popcornmix, 2015-03-04:
@Ferroin Do you know what the exact change that was made to BOINC? It might help understand the issue.

Ferroin, 2015-03-04:
No, but It shouldn't take me long to figure out, I use a local build of the BOINC client on all my systems, so I have a local copy of the upstream repository anyway.  I'll see if I can figure it out and hopefully be back with an answer shortly.

Ferroin, 2015-03-04:
Hmm, I'm not seeing any tag for the release hey are saying is working, despite having just updated my local copy today.

In theory, it shouldn't be too hard to find out what is causing this, as there isn't a huge ammount of stuff in the kernel that actually uses VFP.  Based on the backtraces it looks like something in the kernel's signal handling code is what's having the issues, although why that would be I have no idea.  The actual stacktrace appears to be almost identical to the one seen in #600, and I believe the BOINC client changed how it's using signals to manage individual apps between 7.2 and 7.4, so the signal handling code would make some sense as a common issue.

It would be interesting though to see if the same thing happens on a Pi2, as arm7a has a different superset of the ISA than armv6.

popcornmix, 2015-03-04:
The should be *no* VFP code in the kernel.

Ferroin, 2015-03-04:
In the core kernel there isn't any that uses it for floating point calculations, although there are some SIMD optimizations that use it (CryptoAPI has some modules that do this if they detect VFP support, but I don't know for certain of anywhere else that uses it).  Also, some non-FP op-codes might get used during context switch when saving the FPU state, and that looks like what is causing the issue here, although why it is only triggering during the context switch for the signal handler I have no idea.

P33M, 2015-03-06:
It's an undefined instruction exception around the VFP save/restore state.

From http://lxr.free-electrons.com/source/arch/arm/vfp/vfpmodule.c#L516

It appears that the macros are using MCR/MRC accesses to talk to VFP registers. Pi 1 is "unusual" in having a VFPv2 implementation which isn't widely used.

While waiting on availability of the node.js implementation that causes this - perhaps a printk inside vfp_sync_hwstate() will tell us if this function is ever called without crashing?

studio-dude, 2015-03-06:
Re sample code, it's a rather large project that polls to a lot of different equipment. It'd be rather difficult to run externally. I'll see if I weed it down to a smaller utility that is more portable while still exhibiting the crash. 

Since my last post I had node to the latest v0.10.28 and I had thought the problem went away. Now after 3 days, the problem recurred:

    [327065.395401] BUG: unsupported FP instruction in kernel mode
    [327065.402939] Internal error: Oops - undefined instruction: 0 [#1] PREEMPT ARM
    [327065.413619] Modules linked in: snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd pl2303 usbserial spi_bcm2708 uio_pdrv_genirq uio
    [327065.429854] CPU: 0 PID: 4453 Comm: node Not tainted 3.18.8+ #763
    [327065.437801] task: d631a880 ti: d6010000 task.ti: d6010000
    [327065.445117] PC is at vfp_save_state+0x0/0x28
    [327065.451313] LR is at vfp_sync_hwstate+0x7c/0x88
    [327065.457752] pc : [<c0009c4c>]    lr : [<c00097dc>]    psr: 60000113
    [327065.457752] sp : d6011e70  ip : d6011e88  fp : d6011e84
    [327065.473102] r10: d631acd8  r9 : d6010000  r8 : 00000000
    [327065.480260] r7 : d6011ec0  r6 : bea9ec68  r5 : 80000780  r4 : d6010030
    [327065.488732] r3 : d60100f8  r2 : c083cf44  r1 : c0000780  r0 : d60100f8
    [327065.497177] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
    [327065.507988] Control: 00c5387d  Table: 163dc008  DAC: 00000015
    [327065.515650] Process node (pid: 4453, stack limit = 0xd60101b0)
    [327065.523400] Stack: (0xd6011e70 to 0xd6012000)
    [327065.529666] 1e60:                                     bea9eb60 d6010000 d6011ea4 d6011e88
    [327065.541613] 1e80: c0009a00 c000976c 00000120 00000000 bea9ea70 00000000 d6011ebc d6011ea8
    [327065.553692] 1ea0: c00117b4 c00099dc d6011fb0 bea9ea70 d6011f84 d6011ec0 c0011bf8 c0011620
    [327065.565774] 1ec0: 00529b4c 04000000 b6c69b20 7ffbfeff fffffffe 00000011 00000000 00040001
    [327065.577862] 1ee0: 00007eea 000003e8 00000000 00000003 00000003 c05476e4 d8a878b0 00012976
    [327065.590006] 1f00: d6011f5c d6011f10 c0026250 c0025ddc 00010000 c07fac0c 00000001 00400000
    [327065.602274] 1f20: c080b520 60000193 00000000 00000000 d6011f5c d6011f40 c006742c c00c1dac
    [327065.614657] 1f40: c07f9fa0 d6010000 c081f228 d6010008 d6010000 00000011 d6011fb0 d6010008
    [327065.627183] 1f60: d6010000 00000000 d6011fb0 00000000 d6010000 01153044 d6011fac d6011f88
    [327065.639847] 1f80: c0011f0c c0011944 00c5387d 00000001 001fa8e0 60000010 f200b200 00c5387d
    [327065.652544] 1fa0: 00000000 d6011fb0 c000e858 c0011e58 01153040 00000000 01153038 229080a8
    [327065.665319] 1fc0: 00000000 01153040 006e5768 590080c9 234dab1d 455afdc4 01153044 bea9ed7c
    [327065.678202] 1fe0: bea9ed84 bea9ed60 2400a35c 001fa8e0 60000010 ffffffff 00000000 00000000
    [327065.691147] [<c0009c4c>] (vfp_save_state) from [<c0009a00>] (vfp_preserve_user_clear_hwstate+0x30/0x9c)
    [327065.705338] [<c0009a00>] (vfp_preserve_user_clear_hwstate) from [<c00117b4>] (setup_sigframe+0x1a0/0x1b0)
    [327065.719687] [<c00117b4>] (setup_sigframe) from [<c0011bf8>] (do_signal+0x2c0/0x400)
    [327065.732082] [<c0011bf8>] (do_signal) from [<c0011f0c>] (do_work_pending+0xc0/0x110)
    [327065.744478] [<c0011f0c>] (do_work_pending) from [<c000e858>] (work_pending+0xc/0x20)
    [327065.756948] Code: e12fff1e e1a0200d e1a0e009 eafffe76 (eca00b20)
    [327065.777848] ---[ end trace a6e0bc6dd3e22dcd ]---
    [327065.791353] note: node[4453] exited with preempt_count 2

johnspackman, 2015-03-12:
We have a lot of Pis in production (70+) and this seems to crop up from time to time in recent builds; I now have a couple in my office that are doing it more frequently and one which does it every morning.  The odd thing is that with the Pi which does it every day, I have replaced the SD card *and* the Pi itself (both brand new out of the packaging, the SD card is supplied by Raspberry themselves), leaving only the PSU, HDML cable, and the TV.  The PSU is a standard PSU from RS components.

Last night I restarted it and by 2:35am it had gone again.  

Stack trace below

```
pi@player-00002:~$ uname -a
Linux player-00002 3.12.28+ #709 PREEMPT Mon Sep 8 15:28:00 BST 2014 armv6l GNU/Linux

pi@player-00002:~$ /opt/vc/bin/vcgencmd version
Sep  8 2014 19:02:48
Copyright (c) 2012 Broadcom
version 3f2f2607186be72e4945cfa8edc77872dfc73195 (clean) (release)

pi@player-00002:~$ node -v
v0.10.26
```

```
[Thu Mar 12 02:36:35 2015] BUG: unsupported FP instruction in kernel mode
[Thu Mar 12 02:36:35 2015] Internal error: Oops - undefined instruction: 0 [#1] PREEMPT ARM
[Thu Mar 12 02:36:35 2015] Modules linked in: tun snd_bcm2835 leds_gpio snd_soc_bcm2708_i2s led_class regmap_mmio snd_soc_core snd_compress regmap_i2c snd_pcm_dmaengine regmap_spi snd_pcm snd_page_alloc snd_timer snd
[Thu Mar 12 02:36:35 2015] CPU: 0 PID: 2161 Comm: node Not tainted 3.12.28+ #709
[Thu Mar 12 02:36:35 2015] task: d71bcb00 ti: d609c000 task.ti: d609c000
[Thu Mar 12 02:36:35 2015] PC is at vfp_save_state+0x0/0x28
[Thu Mar 12 02:36:35 2015] LR is at vfp_sync_hwstate+0x70/0x7c
[Thu Mar 12 02:36:35 2015] pc : [<c00099c4>]    lr : [<c0009588>]    psr: 60000013
[Thu Mar 12 02:36:35 2015] sp : d609deb0  ip : 00000018  fp : bec84690
[Thu Mar 12 02:36:35 2015] r10: 02550044  r9 : d71bce90  r8 : 00000000
[Thu Mar 12 02:36:35 2015] r7 : d609dfb0  r6 : bec84590  r5 : 80000780  r4 : d609c030
[Thu Mar 12 02:36:35 2015] r3 : d609c0f8  r2 : c061e920  r1 : c0000780  r0 : d609c0f8
[Thu Mar 12 02:36:35 2015] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[Thu Mar 12 02:36:35 2015] Control: 00c5387d  Table: 17248008  DAC: 00000015
[Thu Mar 12 02:36:35 2015] Process node (pid: 2161, stack limit = 0xd609c1b8)
[Thu Mar 12 02:36:35 2015] Stack: (0xd609deb0 to 0xd609e000)
[Thu Mar 12 02:36:35 2015] dea0:                                     d609dec0 bec84488 d609c000 c0009794
[Thu Mar 12 02:36:35 2015] dec0: 00000000 bec84398 00000000 c0010a48 5ac3c35a d609dfb0 bec84398 c0010ea4
[Thu Mar 12 02:36:35 2015] dee0: 00529190 04000000 b6ce9f20 7ffbfeff fffffffe 00000011 00000000 00040001
[Thu Mar 12 02:36:35 2015] df00: 00007d64 000003e8 00000000 00000000 00000000 c061fa90 c061fc90 c0011114
[Thu Mar 12 02:36:35 2015] df20: c67429a0 00001883 00000001 00000001 00000004 c061f128 c061f120 d609c000
[Thu Mar 12 02:36:35 2015] df40: 00000000 00000008 c061f12c c0023a8c d609c000 00000000 0000000a 0028abd1
[Thu Mar 12 02:36:35 2015] df60: d609c010 00400000 bec84690 60000193 00000003 00000011 d609c018 d609c000
[Thu Mar 12 02:36:35 2015] df80: 00000000 d609dfb0 00000000 d609c000 02550044 c0011194 00000000 2ae4dc40
[Thu Mar 12 02:36:35 2015] dfa0: 60000010 f200b200 00000002 c000df00 449080a5 449080a5 449080a4 00002be4
[Thu Mar 12 02:36:35 2015] dfc0: 449080a5 00000002 000010bf 00000002 58e241c5 b6ba8101 02550044 bec84690
[Thu Mar 12 02:36:35 2015] dfe0: 2ae4dc00 bec84688 2ae9e9fc 2ae4dc40 60000010 ffffffff 00000000 00000000
[Thu Mar 12 02:36:35 2015] [<c00099c4>] (vfp_save_state+0x0/0x28) from [<c0009588>] (vfp_sync_hwstate+0x70/0x7c)
[Thu Mar 12 02:36:35 2015] [<c0009588>] (vfp_sync_hwstate+0x70/0x7c) from [<c0009794>] (vfp_preserve_user_clear_hwstate+0x20/0x8c)
[Thu Mar 12 02:36:35 2015] [<c0009794>] (vfp_preserve_user_clear_hwstate+0x20/0x8c) from [<c0010a48>] (setup_sigframe+0x190/0x1a0)
[Thu Mar 12 02:36:35 2015] [<c0010a48>] (setup_sigframe+0x190/0x1a0) from [<c0010ea4>] (do_signal+0x2e8/0x440)
[Thu Mar 12 02:36:35 2015] [<c0010ea4>] (do_signal+0x2e8/0x440) from [<c0011194>] (do_work_pending+0xa4/0xb4)
[Thu Mar 12 02:36:35 2015] [<c0011194>] (do_work_pending+0xa4/0xb4) from [<c000df00>] (work_pending+0xc/0x20)
[Thu Mar 12 02:36:35 2015] Code: e1a0f00e e1a0200d e1a0e009 eafffe89 (eca00b20) 
[Thu Mar 12 02:36:35 2015] ---[ end trace ac93b69b8274a1e0 ]---
[Thu Mar 12 02:36:35 2015] note: node[2161] exited with preempt_count 2
[Thu Mar 12 02:36:35 2015] BUG: scheduling while atomic: node/2161/0x40000003
[Thu Mar 12 02:36:35 2015] Modules linked in: tun snd_bcm2835 leds_gpio snd_soc_bcm2708_i2s led_class regmap_mmio snd_soc_core snd_compress regmap_i2c snd_pcm_dmaengine regmap_spi snd_pcm snd_page_alloc snd_timer snd
[Thu Mar 12 02:36:35 2015] CPU: 0 PID: 2161 Comm: node Tainted: G      D      3.12.28+ #709
[Thu Mar 12 02:36:35 2015] [<c001444c>] (unwind_backtrace+0x0/0xec) from [<c0011730>] (show_stack+0x10/0x14)
[Thu Mar 12 02:36:35 2015] [<c0011730>] (show_stack+0x10/0x14) from [<c043d490>] (__schedule_bug+0x48/0x64)
[Thu Mar 12 02:36:35 2015] [<c043d490>] (__schedule_bug+0x48/0x64) from [<c0442b54>] (__schedule+0x518/0x5ac)
[Thu Mar 12 02:36:35 2015] [<c0442b54>] (__schedule+0x518/0x5ac) from [<c004566c>] (__cond_resched+0x24/0x34)
[Thu Mar 12 02:36:35 2015] [<c004566c>] (__cond_resched+0x24/0x34) from [<c044335c>] (_cond_resched+0x3c/0x44)
[Thu Mar 12 02:36:35 2015] [<c044335c>] (_cond_resched+0x3c/0x44) from [<c00c2030>] (__get_user_pages.part.91+0xa0/0x408)
[... cut ...]
[Thu Mar 12 02:36:36 2015] [<c0009588>] (vfp_sync_hwstate+0x70/0x7c) from [<c0009794>] (vfp_preserve_user_clear_hwstate+0x20/0x8c)
[Thu Mar 12 02:36:36 2015] [<c0009794>] (vfp_preserve_user_clear_hwstate+0x20/0x8c) from [<c0010a48>] (setup_sigframe+0x190/0x1a0)
[Thu Mar 12 02:36:36 2015] [<c0010a48>] (setup_sigframe+0x190/0x1a0) from [<c0010ea4>] (do_signal+0x2e8/0x440)
[Thu Mar 12 02:36:36 2015] [<c0010ea4>] (do_signal+0x2e8/0x440) from [<c0011194>] (do_work_pending+0xa4/0xb4)
[Thu Mar 12 02:36:36 2015] [<c0011194>] (do_work_pending+0xa4/0xb4) from [<c000df00>] (work_pending+0xc/0x20)
```

P33M, 2015-03-12:
Just built from raspberrypi/linux with bcmrpi_defconfig.

Some disassembly:
```
(gdb) disassemble  vfp_sync_hwstate
Dump of assembler code for function vfp_sync_hwstate:
   0xc0009718 <+0>:     mov     r12, sp
   0xc000971c <+4>:     push    {r4, r5, r11, r12, lr, pc}
   0xc0009720 <+8>:     sub     r11, r12, #4
   0xc0009724 <+12>:    push    {lr}            ; (str lr, [sp, #-4]!)
   0xc0009728 <+16>:    bl      0xc000eba0 <__gnu_mcount_nc>
   0xc000972c <+20>:    mov     r3, sp
   0xc0009730 <+24>:    bic     r4, r3, #8128   ; 0x1fc0
   0xc0009734 <+28>:    bic     r4, r4, #63     ; 0x3f
   0xc0009738 <+32>:    ldr     r3, [r4, #4]
   0xc000973c <+36>:    add     r3, r3, #1
   0xc0009740 <+40>:    str     r3, [r4, #4]
   0xc0009744 <+44>:    ldr     r2, [pc, #76]   ; 0xc0009798 <vfp_sync_hwstate+128>
   0xc0009748 <+48>:    add     r3, r0, #248    ; 0xf8
   0xc000974c <+52>:    ldr     r0, [r2, #4]
   0xc0009750 <+56>:    cmp     r0, r3
   0xc0009754 <+60>:    beq     0xc0009780 <vfp_sync_hwstate+104>
   0xc0009758 <+64>:    ldr     r3, [r4, #4]
   0xc000975c <+68>:    sub     r3, r3, #1
   0xc0009760 <+72>:    cmp     r3, #0
   0xc0009764 <+76>:    str     r3, [r4, #4]
   0xc0009768 <+80>:    ldmne   sp, {r4, r5, r11, sp, pc}
   0xc000976c <+84>:    ldr     r3, [r4]
   0xc0009770 <+88>:    tst     r3, #2
   0xc0009774 <+92>:    ldmeq   sp, {r4, r5, r11, sp, pc}
   0xc0009778 <+96>:    bl      0xc0537420 <preempt_schedule>
   0xc000977c <+100>:   ldm     sp, {r4, r5, r11, sp, pc}
   0xc0009780 <+104>:   vmrs    r5, fpexc
   0xc0009784 <+108>:   orr     r1, r5, #1073741824     ; 0x40000000
   0xc0009788 <+112>:   vmsr    fpexc, r1
   0xc000978c <+116>:   bl      0xc0009bf4 <vfp_save_state>
   0xc0009790 <+120>:   vmsr    fpexc, r5
   0xc0009794 <+124>:   b       0xc0009758 <vfp_sync_hwstate+64>
   0xc0009798 <+128>:   addgt   pc, r2, r4, lsl #8
End of assembler dump.
(gdb) disassemble  vfp_save_state
Dump of assembler code for function vfp_save_state:
   0xc0009bf4 <+0>:     vstmia  r0!, {d0-d15}
   0xc0009bf8 <+4>:     vmrs    r2, fpscr
   0xc0009bfc <+8>:     tst     r1, #-2147483648        ; 0x80000000
   0xc0009c00 <+12>:    beq     0xc0009c14 <vfp_save_state+32>
   0xc0009c04 <+16>:    vmrs    r3, fpinst      @ Impl def
   0xc0009c08 <+20>:    tst     r1, #268435456  ; 0x10000000
   0xc0009c0c <+24>:    beq     0xc0009c14 <vfp_save_state+32>
   0xc0009c10 <+28>:    vmrs    r12, fpinst2    @ Impl def
   0xc0009c14 <+32>:    stm     r0, {r1, r2, r3, r12}
   0xc0009c18 <+36>:    bx      lr
End of assembler dump.
```

popcornmix, 2015-03-12:
@P33M 
> While waiting on availability of the node.js implementation that causes this - perhaps a printk inside vfp_sync_hwstate() will tell us if this function is ever called without crashing?

I did add that printk and it occurs many times a second (without crashing).

johnspackman, 2015-03-16:
still happens on the latest kernel:

```
pi@player-00002:~$ uname -a
Linux player-00002 3.18.7+ #755 PREEMPT Thu Feb 12 17:14:31 GMT 2015 armv6l GNU/Linux

pi@player-00002:~$ /opt/vc/bin/vcgencmd version
Feb 14 2015 22:23:03
Copyright (c) 2012 Broadcom
version 7789db485409720b0e523a3d6b86b12ed56fd152 (clean) (release)

pi@player-00002:~$ node -v
v0.10.26
```

stack dump
```
[Sat Mar 14 13:57:27 2015] BUG: unsupported FP instruction in kernel mode
[Sat Mar 14 13:57:27 2015] Internal error: Oops - undefined instruction: 0 [#1] PREEMPT ARM
[Sat Mar 14 13:57:27 2015] Modules linked in: tun snd_bcm2835 snd_pcm snd_timer snd evdev uio_pdrv_genirq uio
[Sat Mar 14 13:57:27 2015] CPU: 0 PID: 2205 Comm: node Not tainted 3.18.7+ #755
[Sat Mar 14 13:57:27 2015] task: d7370d80 ti: d59a2000 task.ti: d59a2000
[Sat Mar 14 13:57:27 2015] PC is at vfp_save_state+0x0/0x28
[Sat Mar 14 13:57:27 2015] LR is at vfp_sync_hwstate+0x7c/0x88
[Sat Mar 14 13:57:27 2015] pc : [<c0009d20>]    lr : [<c00098b0>]    psr: 60000013
[Sat Mar 14 13:57:27 2015] sp : d59a3e70  ip : d59a3e88  fp : d59a3e84
[Sat Mar 14 13:57:27 2015] r10: d73711d8  r9 : d59a2000  r8 : 00000000
[Sat Mar 14 13:57:27 2015] r7 : d59a3ec0  r6 : bef82590  r5 : 80000780  r4 : d59a2030
[Sat Mar 14 13:57:27 2015] r3 : d59a20f8  r2 : c0855904  r1 : c0000780  r0 : d59a20f8
[Sat Mar 14 13:57:27 2015] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[Sat Mar 14 13:57:27 2015] Control: 00c5387d  Table: 17344008  DAC: 00000015
[Sat Mar 14 13:57:27 2015] Process node (pid: 2205, stack limit = 0xd59a21b0)
[Sat Mar 14 13:57:27 2015] Stack: (0xd59a3e70 to 0xd59a4000)
[Sat Mar 14 13:57:27 2015] 3e60:                                     bef82488 d59a2000 d59a3ea4 d59a3e88
[Sat Mar 14 13:57:27 2015] 3e80: c0009ad4 c0009840 00000120 00000000 bef82398 00000000 d59a3ebc d59a3ea8
[Sat Mar 14 13:57:27 2015] 3ea0: c0011874 c0009ab0 d59a3fb0 bef82398 d59a3f84 d59a3ec0 c0011cb8 c00116e0
[Sat Mar 14 13:57:27 2015] 3ec0: 00529190 04000000 b6c73b20 7ffbfeff fffffffe 00000011 00000000 00040001
[Sat Mar 14 13:57:27 2015] 3ee0: 000004c4 000003e8 00000000 00000000 00000001 c0556c74 8bb01020 0000ad6d
[Sat Mar 14 13:57:27 2015] 3f00: d59a3f5c d59a3f10 c00266a8 c0026234 00000000 c0811200 00000001 00400000
[Sat Mar 14 13:57:27 2015] 3f20: c0823388 60000193 00000000 00000000 d59a3f5c d59a3f40 c0069544 c00c3f98
[Sat Mar 14 13:57:27 2015] 3f40: c08104c0 d59a2000 c0837108 d59a2008 d59a2000 00000011 d59a3fb0 d59a2008
[Sat Mar 14 13:57:27 2015] 3f60: d59a2000 00000000 d59a3fb0 00000000 d59a2000 0129b044 d59a3fac d59a3f88
[Sat Mar 14 13:57:27 2015] 3f80: c0011fcc c0011a04 00c5387d 00000001 4ea4dc40 60000010 f200b200 00c5387d
[Sat Mar 14 13:57:27 2015] 3fa0: 00000000 d59a3fb0 c000e918 c0011f18 5b7080a5 5b7080a5 5b7080a4 00002cf2
[Sat Mar 14 13:57:27 2015] 3fc0: 5b7080a5 00000002 0000008a 00000002 2ca241c5 2b8bfe61 0129b044 bef82690
[Sat Mar 14 13:57:27 2015] 3fe0: 4ea4dc00 bef82688 4ea9e9fc 4ea4dc40 60000010 ffffffff 177fa821 177fac21
[Sat Mar 14 13:57:27 2015] [<c0009d20>] (vfp_save_state) from [<c0009ad4>] (vfp_preserve_user_clear_hwstate+0x30/0x9c)
[Sat Mar 14 13:57:27 2015] [<c0009ad4>] (vfp_preserve_user_clear_hwstate) from [<c0011874>] (setup_sigframe+0x1a0/0x1b0)
[Sat Mar 14 13:57:27 2015] [<c0011874>] (setup_sigframe) from [<c0011cb8>] (do_signal+0x2c0/0x400)
[Sat Mar 14 13:57:27 2015] [<c0011cb8>] (do_signal) from [<c0011fcc>] (do_work_pending+0xc0/0x110)
[Sat Mar 14 13:57:27 2015] [<c0011fcc>] (do_work_pending) from [<c000e918>] (work_pending+0xc/0x20)
[Sat Mar 14 13:57:27 2015] Code: e12fff1e e1a0200d e1a0e009 eafffe76 (eca00b20) 
[Sat Mar 14 13:57:27 2015] ---[ end trace 0073c74fe454cde0 ]---
[Sat Mar 14 13:57:27 2015] note: node[2205] exited with preempt_count 2

```

popcornmix, 2015-03-16:
We really need a test case for this. Clear instructions of what to type from a clean raspbian install to provoke the error. Ideally just "run this executable" and see panic.

johnspackman, 2015-03-16:
I will be able to work on a reproducable test case later this week; the problem is that it can take hours to occur and I can't trap the signal (AFAIK) to resolve it to a line of JS code.  The only clue I have is that every time this happens, there is a zombie process for tput (spawned by by node app).  

However, this is starting to look like it's related to the screen - we've got quite a few of these in production, and I'm reasonably sure that there are quite a few examples of *exactly* same software running on them without this error (or if it does occur then it's nowhere near as frequent).  What differs is the screen on each site, and I am having difficulties creating a 100% reliable mode detection & reset particularly on old LG's.

Is there some kind of debugging setup I can hook up to my Pi here so that I can report back when it crops up?  I'm happy to open up ports in the firewall for someone to get direct access to the box if necessary

catschulze, 2015-03-18:
I think I have found a hint what might be happening here. See the ARM1176JZF-S Technical Reference Manual, DDI0301H_arm1176jzfs_r0p7_trm.pdf (freely available on infocenter.arm.com), section 20.4.3, "Floating-point exception register, FPEXC" on page 20-16: The VFP11 coprocessor might detect a situation that requires reporting an exception too late to notify the ARM11 core, and in these cases the VFP11 sets the EX bit (0x8000'0000) in register FPEXC so that the exception is triggered when the next VFP operation is executed. As you can see from P33M's disassembly, the FPEXC register has been read and saved to ARM register r5 at the time of the kernel panic (and ARM register r1 contains the same value ORed with 0x4000'000, the EN(able) bit - r1 is what will be actually saved to the task state in the stm opcode at the end of vfp_save_state). From johnspackman's register dump, I see that r5 = 0x8000'0780, therefore the EX bit is apparently set. And the VFP11 now does exactly what the technical reference manual specifies, it bounces the vstmia opcode at the start of vfp_save_state. The kernel seems to be unprepared to deal with this exception in kernel space and dies. (Note that vfp_save_state actually contains code that checks for the EX bit, but unfortunately only to decide whether the FPINST and potentially also FPINST2 register(s) need to be saved to the task's context besides FPEXC as well.)

From my understanding of the VFP11 technical reference manual, it would be necessary to clear the EX bit in FPEXC before trying to execute VFP opcodes in the kernel (but saving the unmodified value of FPXEC to the task's context, so that the set EX bit will be restored when switching back to the task that had it set). To quote from the TRM: "EX must be cleared by the exception handling routine" - apparently this is also true for the state saving routine during context switch.

I still wonder what causes this to trigger only in rare cases (I have never observed it on my systems). Perhaps you need to have a "pending VFP exception" just at the time of context switch, i.e. the last VFP operation executed in userspace before a task switch has to cause a "delayed exception", and no other user space VFP instructions (that would cause the exception to be triggered in user space) come in between, so that the exception sits and waits to bomb when the first VFP instruction is executed in kernel space (vstmia to save all VFP11 registers during task switch)? (It would be interesting to see the FPINST and FPINST2 register values as well, I wonder what VFP instruction causes this exception.)

P33M, 2015-03-18:
Thanks for that detailed analysis.

This may also help with creating a test case - have two userspace programs that make extensive use of the VFP unit and have them trigger FP exceptions regularly. Have them both running at the same priority so lots of context switches happen and see if we get some crashes...

studio-dude, 2015-03-18:
OP here, I've learned a few things that might help diagnose the problem. The node.js daemon that exhibited the crash was, among other things, forking two processes every few seconds: vcgencmd and raspistill. By removing both of those, I've been running for 12 days now with no problems.

I'm guessing a small node.js script that repeatable forks those tools every 10 sec or so might repeat the problem (after a few days of running).

PeterPablo, 2015-03-18:
**Disclaimer:** I have hardly any clue about the kernel but wanted to share my thought. As such, handle it with care! Hopefully I am not writing pure nonsense. 

By reading the acronym `FPU` I remembered the [infamous *music bug*](http://forum.xda-developers.com/showpost.php?p=57643086&postcount=829) in the linux kernel that affected the Android smartphone Galaxy 2 i9100 that was eventually fixed by @oberon00 (aka `zeitferne` on XDA). I followed that discussion and it seems that one of the tough challenges was to create a reliable test case. The linked thread offers both insight to how that test case was in the end created and patches relating to fixing the power management. As far as I understood some FPU registers were not restored, after switching some specific states. I am unable to judge whether those patches relate to the described problem here and whether they landed upstream.

johnspackman, 2015-03-19:
I've just made it happen with a test script in under 2 hours; here's the script:
````
var child_process = require("child_process");

var log = { trace: console.log, debug: console.log, warn: console.log, error: console.log, info: console.log };

var util = {
process: {
  exec : function(cmd/* , args, opts, callback */) {
    var tmp = [].slice.call(arguments, 1);
    var args = Array.isArray(tmp[0]) ? tmp.shift() : [];
    var callback = typeof tmp[tmp.length - 1] === 'function' ? tmp.pop() : null;
    var opts = (typeof tmp[0] === 'object' ? tmp.shift() : null)||{};

    var proc = child_process.spawn(cmd, args, {
      encoding: "utf8"
    });
    log.debug("spawn: " + cmd + " " + JSON.stringify(args) + ", PID=" + proc.pid);
    var lnrStdout, lnrStderr;
    var outputReceived = false;
    proc.stdout.on("data", lnrStdout = function(data) {
      outputReceived = true;
      log.trace(cmd + " stdout: " + data);
      if (opts.onStdout)
        opts.onStdout(data);
      if (opts.onConsoleOutput)
        opts.onConsoleOutput(data);
      if (opts.copyToConsole)
        process.stdout.write(data);
    });
    proc.stderr.on("data", lnrStderr = function(data) {
      outputReceived = true;
      log.trace(cmd + " stderr: " + data);
      if (opts.onStderr)
        opts.onStderr(data);
      if (opts.onConsoleOutput)
        opts.onConsoleOutput(data);
      if (opts.copyToConsole)
        process.stdout.write(data);
    });
    var closed = false;
    var exited = false;
    var exitCode = null;
    var exitSignal = null;
    proc.on("close", function() {
      log.trace(cmd + " closed");
      proc.stdout.removeListener("data", lnrStdout);
      proc.stderr.removeListener("data", lnrStderr);
      closed = true;
      if (closed && exited && callback) {
        callback(null, exitCode, exitSignal);
        callback = null;
      }
    });
    proc.on("exit", function(code, signal) {
      log.trace(cmd + " exit code=" + code + ", signal=" + signal);
      exited = true;
      exitCode = code;
      exitSignal = signal;
      if (closed && exited && callback) {
        callback(null, exitCode, signal);
        callback = null;
      }
    });
    proc.on("error", function(err) {
      log.trace(cmd + " error=" + err);
      exited = !outputReceived;
      exitCode = -1;
      exitSignal = null;
      if (closed && exited && callback) {
[... 67 lines cut ...]
[ 6230.914775] de80: c0009ad4 c0009840 00000120 00000000 be894430 00000000 d736debc d736dea8
[ 6230.923108] dea0: c0011874 c0009ab0 d736dfb0 be894430 d736df84 d736dec0 c0011cb8 c00116e0
[ 6230.931438] dec0: 00529190 04000000 b6c43b20 7ffbfeff fffffffe 00000011 00000000 00040001
[ 6230.939770] dee0: 00002efa 000003e8 00000000 00000000 00000003 c0556c74 b93c0690 000005aa
[ 6230.948101] df00: d736df5c d736df10 c00266a8 c0026234 00010000 c08121dc 00000001 00400000
[ 6230.956432] df20: c0823388 60000193 00000000 00000000 d736df5c d736df40 c0069544 c00c3f98
[ 6230.964765] df40: c08104c0 d736c000 c0837108 d736c008 d736c000 00000011 d736dfb0 d736c008
[ 6230.973095] df60: d736c000 00000000 d736dfb0 00000000 d736c000 01c57044 d736dfac d736df88
[ 6230.981425] df80: c0011fcc c0011a04 00c5387d 00000001 001fa83c 60000010 f200b200 00c5387d
[ 6230.989755] dfa0: 00000000 d736dfb0 c000e918 c0011f18 01c57040 00000000 01c57038 334080a8
[ 6230.998086] dfc0: 00000002 00301de8 be89473c 229080c9 b6a58dfd 26b4e068 01c57044 be894730
[ 6231.006417] dfe0: be894724 be894720 2ea0a35c 001fa83c 60000010 ffffffff 00000000 00000000
[ 6231.014769] [<c0009d20>] (vfp_save_state) from [<c0009ad4>] (vfp_preserve_user_clear_hwstate+0x30/0x9c)
[ 6231.024363] [<c0009ad4>] (vfp_preserve_user_clear_hwstate) from [<c0011874>] (setup_sigframe+0x1a0/0x1b0)
[ 6231.034130] [<c0011874>] (setup_sigframe) from [<c0011cb8>] (do_signal+0x2c0/0x400)
[ 6231.041951] [<c0011cb8>] (do_signal) from [<c0011fcc>] (do_work_pending+0xc0/0x110)
[ 6231.049772] [<c0011fcc>] (do_work_pending) from [<c000e918>] (work_pending+0xc/0x20)
[ 6231.057672] Code: e12fff1e e1a0200d e1a0e009 eafffe76 (eca00b20)
[ 6231.063891] ---[ end trace 6d5e72d57e400b9c ]---
[ 6231.068611] note: node[2393] exited with preempt_count 2
````  

Although this might seem contrived, it turns out that during out of hours our app was needlessly calling blankScreen() every couple of seconds.

I'm running it again now to see if it happens again :)

johnspackman, 2015-03-19:
that script's just done it a second time, this time 30 minutes after I turned the screen off

catschulze, 2015-03-19:
I have tried to prepare a small user space program in C which could potentially trigger this bug. As my RPi is currently busy (and I cannot risk crashing it now), I have tried to perform the tests in QEmu. However, while I can actually see undefined operation exceptions hitting the CPU, QEmu apparently does not let me single step in exception handlers which have interrupts disabled (?), and it also seems that it does not model the delayed (asynchronous) exceptions the VFP11 uses. Perhaps someone else may be interested in compiling and running this test code on the real hardware? (It might be a good idea to run at least two instances in parallel.)

```C
#include <math.h>
#include <stdio.h>

int main()
{
        float x; /* sqrt(-1.0) test, input */
        float y; /* sqrt(-1.0) test, output */
        float a; /* 1.0/0.0 test, input 1 */
        float b; /* 1.0/0.0 test, input 2 */
        float c; /* 1.0/0.0 test, output */
        int i;   /* busy loop counter */
        asm ( /* Activate all 6 exception traps in FPSCR */
                "vmrs\tr0, fpscr\n\t"
                "orr\tr0, r0, #0x9f00\n\t"
                "vmsr\tfpscr, r0"
                :
                :
                : "r0"
        );
        while(1) {
                x = -1.0;
                y =  0.0;
                asm ( /* Perform sqrt(-1.0) test */
                        "fsqrts\t%0, %1"
                        : "=w" (y)
                        : "w" (x)
                        :
                );
                for(i=0; i<=0x00ffffff; i++);
                printf("Result: x=%f y=%f\n", x, y);
                fflush(0);
                a = 1.0;
                b = 0.0;
                c = 0.0;
                asm ( /* Perform 1.0/0.0 test */
                        "fdivs\t%0, %1, %2"
                        : "=w" (c)
                        : "w" (a), "w" (b)
                        :
                );
                for(i=0; i<=0x00ffffff; i++);
                printf("Result: a=%f b=%f c=%f\n", a, b, c);
                fflush(0);
        }
        return 0;
}
```

The idea behind this is to configure the VFP11 to report all exceptions as interrupts (no quiet NaNs or INFs), and then provide a series of sqrt(-1.0) and 1.0/0.0 operations, each followed by a busy loop, so that the VFP11 delayed exception status is kept alive during this loop and the interrupt is only triggered at the time the VFP11 is used to print the results to stdout.

P33M, 2015-03-19:
Interesting. If I do this on a Pi 2 everything works fine. I can spawn multiple instances.

If I do this on a Pi 1, I get
````
pi@raspberrypi:~$ gdb argh
GNU gdb (GDB) 7.4.1-debian
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /home/pi/argh...done.
(gdb) run
Starting program: /home/pi/argh

Program received signal SIGFPE, Arithmetic exception.
0x00008434 in main () at test.c:23
23                      asm ( /* Perform sqrt(-1.0) test */

```

catschulze, 2015-03-19:
The SIGFPE didn't show up in QEmu, either (emulating a vexpress-a9 board, which has a ARM Cortex A9 CPU, closer to the ARM Cortex A7 CPU of the Pi 2 than the ARM1176 CPU of the Pi 1). At least we know that the illegal operations really trigger an exception to be signalled from VFP11 to the ARM core (causing the kernel to signal SIGFPE to the program). I don't know why this doesn't happen on the Pi 2, but I also only have a TRM for fhe ARM11, not for the Cortex CPUs, so I cannot check whether they behave differently or need other flags to be set up. Apparently, you get the "correct" results NaN and +INF, so it might also caused by the kernel's VFP support handler just setting these values and returning to user space without signalling SIGFPE...

OK, having the program terminated by the SIGFPE is not what was intended. Can you install a signal handler for SIGFPE that just ignores this signal (calling signal() for SIGFPE with handler set to SIG_IGN), so that we can see what happens if the kernel performs a context switch while the delayed exception flag EX in register FPEXC is set?

studio-dude, 2015-03-19:
I can confirm that johnspackman's node sample did indeed reproduce the crash pretty quickly (30min for me).

Not sure what it means, but the c utility just does this:

    pi@pi /tmp $ ./go
    Floating point exception

catschulze, 2015-03-19:
@studio-dude The "Floating point exception" message seems to happen only on the Pi 1, as P33M found out. My tests were done in QEmu and didn't show this message, so my code was not prepared to handle it. If you'd like to try, here's the updated code:

```C
#include <math.h>
#include <stdio.h>
#include <signal.h>

int main()
{
        float x; /* sqrt(-1.0) test, input */
        float y; /* sqrt(-1.0) test, output */
        float a; /* 1.0/0.0 test, input 1 */
        float b; /* 1.0/0.0 test, input 2 */
        float c; /* 1.0/0.0 test, output */
        int i;   /* busy loop counter */

        signal(SIGFPE, SIG_IGN); /* Floating point exceptions are expected and not fatal */

        asm ( /* Activate all 6 exception traps in FPSCR */
                "vmrs\tr0, fpscr\n\t"
                "orr\tr0, r0, #0x9f00\n\t"
                "vmsr\tfpscr, r0"
                :
                :
                : "r0"
        );

        while(1) {
                x = -1.0;
                y =  0.0;
                asm ( /* Perform sqrt(-1.0) test */
                        "fsqrts\t%0, %1"
                        : "=w" (y)
                        : "w" (x)
                        :
                );
                for(i=0; i<=0x00ffffff; i++);
                printf("Result: x=%f y=%f\n", x, y);
                fflush(0);
                a = 1.0;
                b = 0.0;
                c = 0.0;
                asm ( /* Perform 1.0/0.0 test */
                        "fdivs\t%0, %1, %2"
                        : "=w" (c)
                        : "w" (a), "w" (b)
                        :
                );
                for(i=0; i<=0x00ffffff; i++);
                printf("Result: a=%f b=%f c=%f\n", a, b, c);
                fflush(0);
        }
        return 0;
}
```

This should now prevent the FPE from being reported (and the program subsequently aborted), but I could not try it on real ARM11/VFP11 hardware (it works flawlessly on an ARM Cortex A9 simulated in QEmu, however), so please report back if there are still issues with the code.

studio-dude, 2015-03-19:
OK here's what that new version does:

    pi@pi /tmp $ ./new
    Result: x=-1.000000 y=-1.000000
    Segmentation fault
    pi@pi /tmp $ ./new
    Segmentation fault
    pi@pi /tmp $ ./new
    Result: x=-1.000000 y=nan
    Segmentation fault
    pi@pi /tmp $ ./new
    Result: x=-1.000000 y=nan
    Segmentation fault

     pi@pi /tmp $ dmesg
    [1101281.083856] VFP: Error: unhandled bounce
    [1101281.093060] VFP: EXC 0x40000000 SCR 0x00009f01 INST 0xed5b7a07
    [1101281.104121] VFP: s 0: 0x00000000 s 1: 0x7ff80000
    [1101281.113767] VFP: s 2: 0x00000000 s 3: 0x00000000
    [1101281.123105] VFP: s 4: 0x00000000 s 5: 0x00000000
    [1101281.132096] VFP: s 6: 0x00000000 s 7: 0x00000000
    [1101281.140736] VFP: s 8: 0x00000000 s 9: 0x00000000
    [1101281.149072] VFP: s10: 0x00000000 s11: 0x00000000
    [1101281.157155] VFP: s12: 0x00000000 s13: 0xbff00000
    [1101281.165022] VFP: s14: 0x00000000 s15: 0x7ff80000
    [1101281.172551] VFP: s16: 0x7fc00000 s17: 0x00000000
    [1101281.179949] VFP: s18: 0x00000000 s19: 0x00000000
    [1101281.187249] VFP: s20: 0x00000000 s21: 0x00000000
    [1101281.194422] VFP: s22: 0x00000000 s23: 0x00000000
    [1101281.201644] VFP: s24: 0x00000000 s25: 0x00000000
    [1101281.208887] VFP: s26: 0x00000000 s27: 0x00000000
    [1101281.216133] VFP: s28: 0x00000000 s29: 0x00000000
    [1101281.223270] VFP: s30: 0x00000000 s31: 0x00000000
    [1101281.453740] VFP: Error: unhandled bounce
    [1101281.460452] VFP: EXC 0x40000000 SCR 0x00009f01 INST 0xed5b7a07
    [1101281.469259] VFP: s 0: 0x00000000 s 1: 0x7ff80000
    [1101281.476987] VFP: s 2: 0x00000000 s 3: 0x00000000
    [1101281.484619] VFP: s 4: 0x00000000 s 5: 0x00000000
    [1101281.492364] VFP: s 6: 0x00000000 s 7: 0x00000000
    [1101281.500157] VFP: s 8: 0x00000000 s 9: 0x00000000
    [1101281.507891] VFP: s10: 0x00000000 s11: 0x00000000
    [1101281.515668] VFP: s12: 0x00000000 s13: 0xbff00000
    [1101281.523336] VFP: s14: 0x00000000 s15: 0x7ff80000
    [1101281.530893] VFP: s16: 0x7ff80000 s17: 0x00000000
    [1101281.538323] VFP: s18: 0x00000000 s19: 0x00000000
    [1101281.545555] VFP: s20: 0x00000000 s21: 0x00000000

catschulze, 2015-03-19:
These kernel messages originate from vfp_raise_exceptions(), called from VFP_bounce(), in arch/arm/vfp/vfpmodule.c (see https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/tree/arch/arm/vfp/vfpmodule.c?id=refs/tags/v3.18.9). Unfortunately, there are three locations where vfp_raise_exceptions() can be invoked, and we do not have sufficient information where our call comes from. (Perhaps P33M could provide an instrumented kernel that keeps track of this information.)

The first call is associated with lines 371..388 in vfpmodule.c, and would be caused by some illegal combination of status bits in FPEXC, in particular, EX(ception) = 0x8000'0000 not being set. This would be rather strange, because I should think that EX should be set in our case (otherwise we should not enter VFP_bounce() at all).

The remaining two calls are from error handling after instruction emulation, lines 409..411 and 428..430. It would be interesting to see which values are returned into variable "exceptions" here, as only the value VFP_EXCEPTION_ERROR could lead to the kernel message you observed.

Note that the value of EXC (aka FPEXC) in the dump is misleading, as it is taken from a new read of this register, and it had been modified in line 355 (in particular, the EX bit has already been cleared). Therefore, this part of the dump cannot provide sufficient information for our case.

However, the INST(ruction) might contain something useful: 0xed5b7a07 corresponds to vldr s15, [fp, #-28], a load of a VFP11 register from a memory location on the stack, so neither the sqrt(-1.0) nor the 1.0/0.0 operation that might legitimately cause an exception. This might actually hint at one of the last two calls of vfp_raise_exceptions() from VFP_bounce() (associated with instruction emulation). I am beginning to wonder how instructions associated with *delayed* exceptions can be emulated, if they contain references to  the stack or frame pointer, which might have changed since these instructions were issued. Perhaps this is the reason why we see a SIGSEGV (segmentation fault) now?

@P33M: Can you provide a test kernel with instrumentation, or, maybe even better, can you check with a kernel debugger yourself where the call to vfp_raise_exceptions() (or vfp_panic()) originates from? (I am particularly interested in a stack trace leading to the invocation of vfp_panic() that causes studio-dude's kernel messages.)

studio-dude, 2015-03-19:
Also, FWIW on my pi 2, the above node.js code runs just fine.

catschulze, 2015-03-19:
In case this is in fact related to the emulation of the instruction in the FPINST2 slot (i.e., two instructions need to be emulated), the following change to arch/arm/vfp/vfpmodule.c might be relevant:

ARM: VFP: fix emulation of second VFP instruction
author: Russell King <rmk+kernel@arm.linux.org.uk> 2013-02-25 16:09:12 (GMT)
https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/commit/arch/arm/vfp/vfpmodule.c?id=5e4ba617c1b584b2e376f31a63bd4e734109318a

It is particularly interesting that the comment of this commit explicitly refers to the Raspberry Pi, for which this flaw was initially reported.

From my current understanding of the emulation code, the vldr s15, [fp, #-28] instruction mentioned in studio-dude's log messages is *not* emulated by the current kernel, see vfp_emulate_instruction() lines 300..331 in https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/tree/arch/arm/vfp/vfpmodule.c?id=refs/tags/v3.18.9 and note that vfp_single_cpdo() apparently contains no code for such load/store operations. Or am I missing something?

catschulze, 2015-03-19:
Another attempt at fixing my test code - I have included the delay loop in the inline assembler fragment and made sure that the first VFP11 instruction following the fsqrts and fdivs instructions are serializing reads of FPSCR, which should synchronize the VFP unit and force out delayed exceptions immediately, hopefully eliminating any FPINST2 related problems we seem to have experienced previously. Please compile without optimization options...

```C
#include <math.h>
#include <stdio.h>
#include <signal.h>

int main()
{
        /* Memory allocations for inputs and outputs */
        float x; /* sqrt(-1.0) test, input    */
        float y; /* sqrt(-1.0) test, output   */
        float a; /* 1.0/0.0    test, input #1 */
        float b; /* 1.0/0.0    test, input #2 */
        float c; /* 1.0/0.0    test, output   */
        int i;   /* busy loop counter         */

        /* Set signal handler for floating point exceptions
         * Floating point exceptions are expected and will be ignored.
         */
        signal(SIGFPE, SIG_IGN);

        /* Set up VFP11 configuration, trap all exceptions */
        asm (
                "vmrs\tr0, fpscr\n\t"
                "orr\tr0, r0, #0x9f00\n\t"
                "vmsr\tfpscr, r0"
                :
                :
                : "r0"
        );

        /* Set busy loop count */
        i = 0x00ffffff;

        while(1) {

                /* First test, sqrt(-1.0) */
                x = -1.0;
                y =  0.0;
                /* Perform test
                 * (1) Execute the fsqrts operation
                 * (2) Perform the delay loop
                 * (3) Execute a serializing operation
                 *     (read FPSCR) to synchronize VFP11
                 *     to avoid hitting another kernel bug
                 */
                asm (
                        "fsqrts\t%0, %1\n\t"
                        "mov\tr0, %2\n"
                        "1:\tcmp\tr0, #0\n\t"
                        "beq\t2f\n\t"
                        "sub\tr0, r0, #1\n\t"
                        "b\t1b\n"
                        "2:\tvmrs\tr0, fpscr"
                        : "=w" (y)
                        : "w" (x), "r" (i)
                        : "r0"
                );
                /* Show results to check whether it worked */
                printf("Result: x=%f y=%f\n", x, y);
                fflush(0);

                /* Second test, 1.0/0.0, division by zero */
                a = 1.0;
                b = 0.0;
                c = 0.0;
                /* Perform test
                 * (1) Execute the fdivs operation
                 * (2) Perform the delay loop
                 * (3) Execute a serializing operation
                 *     (read FPSCR) to synchronize VFP11
                 *     to avoid hitting another kernel bug
                 */
                asm (
                        "fdivs\t%0, %1, %2\n\t"
                        "mov\tr0, %3\n"
                        "1:\tcmp\tr0, #0\n\t"
                        "beq\t2f\n\t"
                        "sub\tr0, r0, #1\n\t"
                        "b\t1b\n"
                        "2:\tvmrs\tr0, fpscr"
                        : "=w" (c)
                        : "w" (a), "w" (b), "r" (i)
                        : "r0"
                );
                /* Show results to check whether it worked */
                printf("Result: a=%f b=%f c=%f\n", a, b, c);
                fflush(0);

        }

        /* Never reached */
        return 0;

}
```

Again, tested only in QEmu against a simulated ARM Cortex A9 CPU, any test results on real VFP11 hardware welcome...

studio-dude, 2015-03-19:
```
pi@pi /tmp $ ./newer
Segmentation fault
pi@pi /tmp $ dmesg
[1119475.237532] VFP: Error: unhandled bounce
[1119475.244332] VFP: EXC 0x40000000 SCR 0x00009f01 INST 0xed2d8b02
[1119475.253169] VFP: s 0: 0x00000000 s 1: 0xbff00000
[1119475.260833] VFP: s 2: 0x00000000 s 3: 0x00000000
[1119475.268566] VFP: s 4: 0x00000000 s 5: 0x00000000
[1119475.276337] VFP: s 6: 0x00000000 s 7: 0x00000000
[1119475.284096] VFP: s 8: 0x00000000 s 9: 0x00000000
[1119475.291851] VFP: s10: 0x00000000 s11: 0x00000000
[1119475.299512] VFP: s12: 0x00000000 s13: 0xbff00000
[1119475.307325] VFP: s14: 0x00000000 s15: 0x7ff80000
[1119475.314981] VFP: s16: 0x7fc00000 s17: 0x00000000
[1119475.322408] VFP: s18: 0x00000000 s19: 0x00000000
[1119475.329580] VFP: s20: 0x00000000 s21: 0x00000000
[1119475.336839] VFP: s22: 0x00000000 s23: 0x00000000
[1119475.344093] VFP: s24: 0x00000000 s25: 0x00000000
[1119475.351265] VFP: s26: 0x00000000 s27: 0x00000000
[1119475.358575] VFP: s28: 0x00000000 s29: 0x00000000
[1119475.365807] VFP: s30: 0x00000000 s31: 0x00000000
```

catschulze, 2015-03-20:
OK, last attempt for today - activating traps for all exceptions seems to hit not only where intended, but also in other locations, so I restrict activating the traps just around the instructions where I want to see delayed exceptions (the primary idea still is to check what goes wrong during context switch):

```C
#include <math.h>
#include <stdio.h>
#include <signal.h>

int main()
{
        /* Memory allocations for inputs and outputs */
        float x; /* sqrt(-1.0) test, input    */
        float y; /* sqrt(-1.0) test, output   */
        float a; /* 1.0/0.0    test, input #1 */
        float b; /* 1.0/0.0    test, input #2 */
        float c; /* 1.0/0.0    test, output   */
        int i;   /* busy loop counter         */

        /* Set signal handler for floating point exceptions
         * Floating point exceptions are expected and will be ignored.
         */
        signal(SIGFPE, SIG_IGN);

        /* Set busy loop count */
        i = 0x00ffffff;

        while(1) {

                /* First test, sqrt(-1.0) */
                x = -1.0;
                y =  0.0;
                /* Perform test
                 * (1) Set all exceptions to trap
                 * (2) Execute the fsqrts operation
                 * (3) Perform the delay loop
                 * (4) Execute a serializing operation
                 *     (read FPSCR) to synchronize VFP11
                 *     to avoid hitting another kernel bug
                 * (5) Set all exceptions to not trap
                 */
                asm (
                        "vmrs\tr0, fpscr\n\t"
                        "orr\tr0, r0, #0x9f00\n\t"
                        "vmsr\tfpscr, r0\n\t"
                        "fsqrts\t%0, %1\n\t"
                        "mov\tr0, %2\n"
                        "1:\tcmp\tr0, #0\n\t"
                        "beq\t2f\n\t"
                        "sub\tr0, r0, #1\n\t"
                        "b\t1b\n"
                        "2:\tvmrs\tr0, fpscr\n\t"
                        "bic\tr0, r0, #0x9f00\n\t"
                        "vmsr\tfpscr, r0"
                        : "=w" (y)
                        : "w" (x), "r" (i)
                        : "r0"
                );
                /* Show results to check whether it worked */
                printf("Result: x=%f y=%f\n", x, y);
                fflush(0);

                /* Second test, 1.0/0.0, division by zero */
                a = 1.0;
                b = 0.0;
                c = 0.0;
                /* Perform test
                 * (1) Set all exceptions to trap
                 * (2) Execute the fdivs operation
                 * (3) Perform the delay loop
                 * (4) Execute a serializing operation
                 *     (read FPSCR) to synchronize VFP11
                 *     to avoid hitting another kernel bug
                 * (5) Set all exceptions to not trap
                 */
                asm (
                        "vmrs\tr0, fpscr\n\t"
                        "orr\tr0, r0, #0x9f00\n\t"
                        "vmsr\tfpscr, r0\n\t"
                        "fdivs\t%0, %1, %2\n\t"
                        "mov\tr0, %3\n"
                        "1:\tcmp\tr0, #0\n\t"
                        "beq\t2f\n\t"
                        "sub\tr0, r0, #1\n\t"
                        "b\t1b\n"
                        "2:\tvmrs\tr0, fpscr\n\t"
                        "bic\tr0, r0, #0x9f00\n\t"
                        "vmsr\tfpscr, r0"
                        : "=w" (c)
                        : "w" (a), "w" (b), "r" (i)
                        : "r0"
                );
                /* Show results to check whether it worked */
                printf("Result: a=%f b=%f c=%f\n", a, b, c);
                fflush(0);

        }

        /* Never reached */
        return 0;

}
```

BTW, INST(ruction) 0xed2d8b02 from your kernel log represents the operation vpush {d8}, which is again a VFP11 instruction operating on a memory location (and also apparently not emulated in the kernel). Seems we might have found a general problem in the kernel affecting VFP11 behavior, which is triggered when an application activates trapping exceptions - these dmesg messages should not be there in the first place, they are indicating a problem in the kernel's VFP support code...

Please check whether this still produces "unhandled bounce" messages (and the segmentation faults, which are probably a consequence of the bounces, as e.g. failure to emulate a vpush operation will cause stack pointer corruption).

studio-dude, 2015-03-20:
This one took a little bit longer to exhibit the problem:
```
pi@pi /tmp $ dmesg
[1124129.874948] VFP: Error: unhandled bounce
[1124129.881699] VFP: EXC 0x40000700 SCR 0x20009f03 INST 0xeef10a10
[1124129.890446] VFP: s 0: 0x00000000 s 1: 0x7ff00000
[1124129.898193] VFP: s 2: 0x00000000 s 3: 0x00000000
[1124129.905964] VFP: s 4: 0x00000000 s 5: 0x00000000
[1124129.913727] VFP: s 6: 0x00000000 s 7: 0x00000000
[1124129.921492] VFP: s 8: 0x00000000 s 9: 0x00000000
[1124129.929158] VFP: s10: 0x00000000 s11: 0x3ff00000
[1124129.936946] VFP: s12: 0x00000000 s13: 0x00000000
[1124129.944773] VFP: s14: 0x00000000 s15: 0xbf800000
[1124129.952467] VFP: s16: 0x7fc00000 s17: 0x00000000
[1124129.959841] VFP: s18: 0x00000000 s19: 0x00000000
[1124129.967115] VFP: s20: 0x00000000 s21: 0x00000000
[1124129.974397] VFP: s22: 0x00000000 s23: 0x00000000
[1124129.981684] VFP: s24: 0x00000000 s25: 0x00000000
[1124129.988912] VFP: s26: 0x00000000 s27: 0x00000000
[1124129.996173] VFP: s28: 0x00000000 s29: 0x00000000
[1124130.003465] VFP: s30: 0x00000000 s31: 0x00000000
[1124134.056011] VFP: Error: unhandled bounce
[1124134.063115] VFP: EXC 0x40000700 SCR 0x20009f03 INST 0xeef10a10
[1124134.072025] VFP: s 0: 0x00000000 s 1: 0x7ff00000
[1124134.079738] VFP: s 2: 0x00000000 s 3: 0x00000000
[1124134.087531] VFP: s 4: 0x00000000 s 5: 0x00000000
[1124134.095331] VFP: s 6: 0x00000000 s 7: 0x00000000
[1124134.103158] VFP: s 8: 0x00000000 s 9: 0x00000000
[1124134.110875] VFP: s10: 0x00000000 s11: 0x3ff00000
[1124134.118767] VFP: s12: 0x00000000 s13: 0x00000000
[1124134.126616] VFP: s14: 0x00000000 s15: 0xbf800000
[1124134.134255] VFP: s16: 0x7fc00000 s17: 0x00000000
[1124134.141753] VFP: s18: 0x00000000 s19: 0x00000000
[1124134.148884] VFP: s20: 0x00000000 s21: 0x00000000
[1124134.156089] VFP: s22: 0x00000000 s23: 0x00000000
[1124134.163375] VFP: s24: 0x00000000 s25: 0x00000000
[1124134.170568] VFP: s26: 0x00000000 s27: 0x00000000
[1124134.177818] VFP: s28: 0x00000000 s29: 0x00000000
[1124134.185188] VFP: s30: 0x00000000 s31: 0x00000000
```

catschulze, 2015-03-20:
Fine to hear that we got rid of the segmentation violations - that should allow you to run two of these instances in parallel and check whether you get any kernel panic (related to context switching).

It's still odd that we get these "unhandled bounce" reports. FYI, INST(ruction) 0xeef10a10 corresponds to the operation "vmrs r0, fpscr", the serializing instruction I used to synchronize the VFP11's state. From my understanding of the VFP11 Technical Reference Manual, this instruction may bounce, and even should do so directly after my fsqrts and fdivs operations in the tests, but it should never appear in registers FPINST or FPINST2, so it should never need to be emulated, only retried. Retrying is quite simple, so I see no reason why the kernel should complain about an "unhandled bounce" here. Maybe the diagnostics in VFP_bounce() get confused by some peculiarities of the VFP11? (I have found a reference to this on the ARM kernel mailing list, where they talk about having to adjust this code for some ARM CPUs, and called this "opening a can of worms", because many ARMs and their FPUs seem to work in slightly different ways, but the kernel code must deal with all of them.)

@P33M: Can we arrange to have a kernel with custom diagnostics in VFP_bounce() built and provided to studio-dude or other testers? I think this "unhandled bounce" is a different problem than the "invalid opcode" oops that started this thread, but both symptoms seem to hint at problems of the upstream kernel to handle the VFP11 properly, which should all be fixed IMO. If you agree, I might be able to provide some hints what instrumentation to add where in the code later today.

catschulze, 2015-03-20:
OK - there seems to be indeed a bug in the VFP_bounce() code in the current upstream kernel:
```C
   /*
    * Check for the special VFP subarch 1 and FPSCR.IXE bit case
    */
   if ((fpsid & FPSID_ARCH_MASK) == (1 << FPSID_ARCH_BIT)
         && (fpscr & FPSCR_IXE)) {
      /*
       * Synchronous exception, emulate the trigger instruction
       */
      goto emulate;
   }
```
From the kernel messages of my RPi, the VFP identifies as architecture 1, so the code above matches when bit IXE is set (and I set it in my code, check for the value 0x9f00, IXE is bit 0x1000):
```
[    0.132043] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
```
However, the presence of the "interrupt enable" bit IXE does not say anything about *synchronous* or *asynchonous* exception, it's just some sort of *IRQ enable* (we still need to check the EX bit!). My test code produces *asynchronous* exceptions, and the instruction in *FPINST* should be emulated. The kernel chooses to emulate the *trigger instruction* instead - bad, this needs to be fixed! (IMO opinion this is the root cause for all these "unhandled bounce" messages, and while it didn't help to fix studio-dude's oopses, it at least showed yet another bug in VFP_bounce()...)

@studio-dude, @P33M: Please update your copy of my test code and change the *two instances* of "orr\tr0, r0, #0x9f00\n\t" to "orr\tr0, r0, #0x0300\n\t" (but leave the two bic operations at 0x9f00, these don't do any harm), then compile and run again. This should avoid setting IXE, and we should hopefully not see any "unhandled bounce" complaints any more.

catschulze, 2015-03-20:
Update: Further study of the ARM1176JZF-S TRM shows that the kernel code is actually perfectly OK, and *FPSCR_IXE ist not only a "IRQ enable" style flag* (unlike all other exception trap enable flags), but *it forces all exceptions to be synchronous and precise*, so the current kernel code for VFP architecture 1 is actually correct (except for a optional workaraound for an official ARM1176JFZ-S CPU bug that I'll discuss below). See DDI0301H_arm1176jzfs_r0p7_trm.pdf, section 22.3, "Support code", on page 22-5, where the requirements for the implementation of VFP_bounce() are discussed. (I thought at first the mention of IXE might have been a typo and they might have meant the *status bit* IXC instead, but it is apparently really IXE what needs to be checked. This is definitely made clear in section 22.10.1, "Inexact exception enabled", on page 22-18, but this location is very well hidden if you are not really looking for it.)

**BUT:** There exists an **erratum for the ARM1176JZF-S CPU**, discussed in the document UAN0014B_ARM1176_errata_r0p7.pdf on infocenter.arm.com, that affects exactly this functionality. See section 2.3.1, erratum #328593, "VFP can take an inexact exception on non-CDP VFP instructions", on page 2-19 in that document. This erratum is triggered by an user space application setting the IXE flag, and ARM mentions that most application won't do that for performance reasons anyway, so a workaround in VFP_bounce() is considered optional.

Therefore, just apply the update "orr\tr0, r0, #0x9f00\n\t" to "orr\tr0, r0, #0x0300\n\t" as mentioned above and re-run the tests, it should now be fine and not cause any "unhandled bounce" messages any more. However, it might trigger the oops on context switch, what is what I am actually waiting for...

popcornmix, 2015-03-20:
No unhandled bounces with latest code with the two  orr r0, r0, #0x9f00 changed to orr r0, r0, #0x0300. Running on Pi1.
I've launched 4 instances, and no quick oopses (after a few minutes). I'll leave it running a bit longer.

@catschulze If you provide me with a patch to grab some useful logging (or otherwise try potential fixes) I can produce a test kernel build.

catschulze, 2015-03-20:
I think the following patch for arch/arm/vfp/vfpmodule.c is worth a try:
```
--- vfpmodule-3.18.9.c  2015-03-20 13:51:32.000000000 +0100
+++ vfpmodule-patched.c 2015-03-20 14:58:31.256135431 +0100
@@ -183,8 +183,11 @@ static int vfp_notifier(struct notifier_
                 * case the thread migrates to a different CPU. The
                 * restoring is done lazily.
                 */
-               if ((fpexc & FPEXC_EN) && vfp_current_hw_state[cpu])
+               if ((fpexc & FPEXC_EN) && vfp_current_hw_state[cpu]) {
+                       /* vfp_save_state oopses on VFP11 if EX bit set */
+                       fmxr(FPEXC, fpexc & ~FPEXC_EX);
                        vfp_save_state(vfp_current_hw_state[cpu], fpexc);
+               }
 #endif

                /*
@@ -454,13 +457,16 @@ static int vfp_pm_suspend(void)
        /* if vfp is on, then save state for resumption */
        if (fpexc & FPEXC_EN) {
                pr_debug("%s: saving vfp state\n", __func__);
+               /* vfp_save_state oopses on VFP11 if EX bit set */
+               fmxr(FPEXC, fpexc & ~FPEXC_EX);
                vfp_save_state(&ti->vfpstate, fpexc);

                /* disable, just in case */
                fmxr(FPEXC, fmrx(FPEXC) & ~FPEXC_EN);
        } else if (vfp_current_hw_state[ti->cpu]) {
 #ifndef CONFIG_SMP
-               fmxr(FPEXC, fpexc | FPEXC_EN);
+               /* vfp_save_state oopses on VFP11 if EX bit set */
+               fmxr(FPEXC, (fpexc & ~FPEXC_EX) | FPEXC_EN);
                vfp_save_state(vfp_current_hw_state[ti->cpu], fpexc);
                fmxr(FPEXC, fpexc);
 #endif
@@ -523,7 +529,8 @@ void vfp_sync_hwstate(struct thread_info
                /*
                 * Save the last VFP state on this CPU.
                 */
-               fmxr(FPEXC, fpexc | FPEXC_EN);
+               /* vfp_save_state oopses on VFP11 if EX bit set */
+               fmxr(FPEXC, (fpexc & ~FPEXC_EX) | FPEXC_EN);
                vfp_save_state(&thread->vfpstate, fpexc | FPEXC_EN);
                fmxr(FPEXC, fpexc);
        }
@@ -689,7 +696,8 @@ void kernel_neon_begin(void)
        cpu = get_cpu();

        fpexc = fmrx(FPEXC) | FPEXC_EN;
-       fmxr(FPEXC, fpexc);
+       /* vfp_save_state oopses on VFP11 if EX bit set */
+       fmxr(FPEXC, fpexc & ~FPEXC_EX);

        /*
         * Save the userland NEON/VFP state. Under UP,
```
I decided to patch all callers of vfp_save_state() to disable the EX bit that causes our oops - I decided against patching vfp_save_state() itself, because that would lead to the code doing something differently than its name suggests, but this might be an alternative that would require fewer changes.

Note that I changed the EX bit on entry to vfp_save_state(), but I did not do anything special to FPEXC after the call to vfp_save_state() returns. In the majority of cases, the caller sets FPEXC explicitly after the call to vfp_save_state() anyway, and with EN(able) cleared there seems to be not much sense in clearing also EX(ception). Let's just see what the patch as provided above does.

If this should really work, I would strongly suggest to consult some other ARM FPU kernel experts before committing this to the RPi kernel tree, because my experience in this context is quite limited, and I cannot guarantee that it might introduce undesired and potentially harmful side effects.

BTW, I think I know now what caused this vfp_safe_state() calls in the oops that started this thread: do_signal() hints at signal delivery, and in fact in order to pass execution to a signal handler, you will need to provide a structure of all CPU and FPU registers at the time the signal was delivered. In order to get these values out of the FPU and into memory, you will need to call some sync function - which is precisely vfp_save_state(), called from vfp_sync_hwstate(). So perhaps we were looking for the wrong trigger, and it's not context switches, but custom signal handlers...

BTW2, to get more debugging output related to VFP, you could just #define DEBUG at the top of arch/arm/vfp/vfphw.S - but I am not sure how much output that will generate.

catschulze, 2015-03-20:
Updated code that uses custom signal handlers (SIGALRM, triggering each second), might be better suited to provoke the oops (on an unpatched kernel):
```C
#include <math.h>
#include <stdio.h>
#include <signal.h>
#include <unistd.h>

void alarm_handler(int dummy)
{
        printf("SIGALRM...\n");

        /* Retrigger in one second */
        alarm(1);
}

int main()
{
        /* Memory allocations for inputs and outputs */
        float x; /* sqrt(-1.0) test, input    */
        float y; /* sqrt(-1.0) test, output   */
        float a; /* 1.0/0.0    test, input #1 */
        float b; /* 1.0/0.0    test, input #2 */
        float c; /* 1.0/0.0    test, output   */
        int i;   /* busy loop counter         */

        /* Set signal handler for floating point exceptions
         * Floating point exceptions are expected and will be ignored.
         */
        signal(SIGFPE, SIG_IGN);

        /* Install handler for alarm() calls */
        signal(SIGALRM, alarm_handler);

        /* Trigger alarm in one second */
        alarm(1);

        /* Set busy loop count */
        i = 0x00ffffff;

        while(1) {

                /* First test, sqrt(-1.0) */
                x = -1.0;
                y =  0.0;
                /* Perform test
                 * (1) Set all exceptions to trap
                 * (2) Execute the fsqrts operation
                 * (3) Perform the delay loop
                 * (4) Execute a serializing operation
                 *     (read FPSCR) to synchronize VFP11
                 *     to avoid hitting another kernel bug
                 * (5) Set all exceptions to not trap
                 */
                asm (
                        "vmrs\tr0, fpscr\n\t"
                        "orr\tr0, r0, #0x0300\n\t"
                        "vmsr\tfpscr, r0\n\t"
                        "fsqrts\t%0, %1\n\t"
                        "mov\tr0, %2\n"
                        "1:\tcmp\tr0, #0\n\t"
                        "beq\t2f\n\t"
                        "sub\tr0, r0, #1\n\t"
                        "b\t1b\n"
                        "2:\tvmrs\tr0, fpscr\n\t"
                        "bic\tr0, r0, #0x9f00\n\t"
                        "vmsr\tfpscr, r0"
                        : "=w" (y)
                        : "w" (x), "r" (i)
                        : "r0"
                );
                /* Show results to check whether it worked */
                printf("Result: x=%f y=%f\n", x, y);
                fflush(0);

                /* Second test, 1.0/0.0, division by zero */
                a = 1.0;
                b = 0.0;
                c = 0.0;
                /* Perform test
                 * (1) Set all exceptions to trap
                 * (2) Execute the fdivs operation
                 * (3) Perform the delay loop
                 * (4) Execute a serializing operation
                 *     (read FPSCR) to synchronize VFP11
                 *     to avoid hitting another kernel bug
                 * (5) Set all exceptions to not trap
                 */
                asm (
                        "vmrs\tr0, fpscr\n\t"
                        "orr\tr0, r0, #0x0300\n\t"
                        "vmsr\tfpscr, r0\n\t"
                        "fdivs\t%0, %1, %2\n\t"
                        "mov\tr0, %3\n"
                        "1:\tcmp\tr0, #0\n\t"
                        "beq\t2f\n\t"
                        "sub\tr0, r0, #1\n\t"
                        "b\t1b\n"
                        "2:\tvmrs\tr0, fpscr\n\t"
                        "bic\tr0, r0, #0x9f00\n\t"
                        "vmsr\tfpscr, r0"
                        : "=w" (c)
                        : "w" (a), "w" (b), "r" (i)
                        : "r0"
                );
                /* Show results to check whether it worked */
                printf("Result: a=%f b=%f c=%f\n", a, b, c);
                fflush(0);

        }

        /* Never reached */
        return 0;

}
```

studio-dude, 2015-03-20:
catschulze, you're right. That one crashed rather quickly:

```
[1175175.559613] BUG: unsupported FP instruction in kernel mode
[1175175.575578] Internal error: Oops - undefined instruction: 0 [#3] PREEMPT ARM
[1175175.603220] Modules linked in: snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd pl2303 usbserial spi_bcm2708 uio_pdrv_genirq uio
[1175175.636407] CPU: 0 PID: 17473 Comm: newererer Tainted: G      D        3.18.8+ #763
[1175175.664530] task: d625b600 ti: d73a0000 task.ti: d73a0000
[1175175.680207] PC is at vfp_save_state+0x0/0x28
[1175175.694631] LR is at vfp_sync_hwstate+0x7c/0x88
[1175175.709072] pc : [<c0009c4c>]    lr : [<c00097dc>]    psr: 60000113
[1175175.709072] sp : d73a1e70  ip : d73a1e88  fp : d73a1e84
[1175175.740106] r10: d625ba58  r9 : d73a0000  r8 : 00000000
[1175175.754881] r7 : d73a1ec0  r6 : befc95c8  r5 : 80000701  r4 : d73a0030
[1175175.779908] r3 : d73a00f8  r2 : c083cf44  r1 : c0000701  r0 : d73a00f8
[1175175.804767] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[1175175.830079] Control: 00c5387d  Table: 09800008  DAC: 00000015
[1175175.844972] Process newererer (pid: 17473, stack limit = 0xd73a01b0)
[1175175.860432] Stack: (0xd73a1e70 to 0xd73a2000)
[1175175.873783] 1e60:                                     befc94c0 d73a0000 d73a1ea4 d73a1e88
[1175175.899365] 1e80: c0009a00 c000976c 00000120 00000000 befc93d0 00000000 d73a1ebc d73a1ea8
[1175175.924875] 1ea0: c00117b4 c00099dc d73a1fb0 befc93d0 d73a1f84 d73a1ec0 c0011bf8 c0011620
[1175175.950732] 1ec0: 000084a0 14000000 b6dcbb20 00002000 00000000 0000000e 00000000 00000080
[1175175.976774] 1ee0: 00000000 00000000 00000000 00000002 00000003 d73a1f00 c005dad8 c005feb4
[1175176.002915] 1f00: d73a1f5c d73a1f10 c0026250 c07fac0c 00000000 c07fac0c d73a0000 c07f4a00
[1175176.029158] 1f20: c080b520 60000193 00000000 00000000 d73a1f5c d73a1f40 c006742c c00c1dac
[1175176.055643] 1f40: c07f9fa0 d73a0000 c081f228 00000000 d73a1f74 0000000e c0026720 d73a0008
[1175176.082760] 1f60: d73a0000 00000000 d73a1fb0 00000000 d73a0000 b6f0c000 d73a1fac d73a1f88
[1175176.110377] 1f80: c0011f0c c0011944 00c5387d 00000001 0000853c 20000010 f200b200 00c5387d
[1175176.138461] 1fa0: 00000000 d73a1fb0 c000e858 c0011e58 008badfe 00000000 00000001 00ffffff
[1175176.166679] 1fc0: 00000000 00000000 000083f4 00000000 00000000 00000000 b6f0c000 befc96f4
[1175176.195079] 1fe0: b6ecc210 befc96c0 b6e0aca8 0000853c 20000010 ffffffff 00000000 00000000
[1175176.223661] [<c0009c4c>] (vfp_save_state) from [<c0009a00>] (vfp_preserve_user_clear_hwstate+0x30/0x9c)
[1175176.253886] [<c0009a00>] (vfp_preserve_user_clear_hwstate) from [<c00117b4>] (setup_sigframe+0x1a0/0x1b0)
[1175176.284737] [<c00117b4>] (setup_sigframe) from [<c0011bf8>] (do_signal+0x2c0/0x400)
[1175176.313723] [<c0011bf8>] (do_signal) from [<c0011f0c>] (do_work_pending+0xc0/0x110)
[1175176.342645] [<c0011f0c>] (do_work_pending) from [<c000e858>] (work_pending+0xc/0x20)
[1175176.371670] Code: e12fff1e e1a0200d e1a0e009 eafffe76 (eca00b20)
[1175176.388538] ---[ end trace 0f7eb96c4b666b02 ]---
[1175176.403807] note: newererer[17473] exited with preempt_count 2
```

popcornmix, 2015-03-20:
I think these kernels should be drop replacements for current "rpi-update" kernels:
https://dl.dropboxusercontent.com/u/3669512/temp/vfp/kernel.img
https://dl.dropboxusercontent.com/u/3669512/temp/vfp/kernel7.img

They include @catschulze's suggested patch.

studio-dude, 2015-03-20:
OK running @popcornmix's kernel.img seems to have fixed the problem. I'm running the latest code from @catschulze and it's running without problems (assuming below is the desired output):

```
pi@pi /tmp $ uname -a
Linux pi 3.18.9+ #769 PREEMPT Fri Mar 20 15:04:30 GMT 2015 armv6l GNU/Linux
pi@pi /tmp $
pi@pi /tmp $ ./go
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
SIGALRM...
Result: x=-1.000000 y=inf
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
SIGALRM...
Result: a=1.000000 b=0.000000 c=nan
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
SIGALRM...
Result: x=-1.000000 y=inf
Result: a=1.000000 b=0.000000 c=inf
Result: x=-1.000000 y=nan
```

catschulze, 2015-03-20:
Well, it doesn't crash any more, which is good. **However, the results are sometimes wrong**. Expected results are:
```
Result: x=-1.000000 y=nan
Result: a=1.000000 b=0.000000 c=inf
```
However, sometimes I see
```
Result: x=-1.000000 y=inf
```
or
```
Result: a=1.000000 b=0.000000 c=nan
```
Both are clearly wrong, as sqrt(-1.0) should give y = NaN, while 1.0 / 0.0 should give c = +INF.

It is very interesting to see that these wrong results seem to happen always after a SIGALRM event. There is still something wrong in the fix, I guess. Will have a look at it - maybe we need some additional tracing.

@popcornmix: Was #define DEBUG included in arch/arm/vfp/vfphw.S as I suggested? If yes, then:
@studio-dude: Could you check dmesg for debugging output from vfphw.S?

If #define DEBUG was not included, I would suggest building another test kernel, so that we get some more hints what is going on in ther kernel.

popcornmix, 2015-03-20:
Will build another kernel with DEBUG enabled...

catschulze, 2015-03-20:
I am currently trying to analyze the situation by reading the source code. I would need some information about the configuration of the kernel
```
Linux pi 3.18.9+ #769 PREEMPT Fri Mar 20 15:04:30 GMT 2015 armv6l GNU/Linux
```
tested by @studio-dude. Find below some CONFIG variables an the settings I think they have for this kernel - please comment (corrections or affirmation):

CONFIG variable | setting
----------------------|---------
CONFIG_SMP | not set
CONFIG_CPU_PM | not set
CONFIG_KERNEL_MODE_NEON | not set

If all three are not set, then the only effective part of my patch affects vfp_sync_hwstate(), and I can restrict myself to checking what might be going wrong there.

@popcornmix: I just noticed that the instrumentation in vfphw.S (by #define DEBUG) is not really helpful for tracking bugs related to signal delivery. I might need to provide another patch that adds some diagnostic output later (and a new kernel build, of course).

catschulze, 2015-03-20:
New patch for vfpmodule.c, applies to upstream source (not the already patched version), includes debug output of all saved VFP state (to check if some bits are missing). Not tested to compile cleanly, you might need to tweak it a bit if the compiler complains...
```
--- vfpmodule-3.18.9.c  2015-03-20 13:51:32.000000000 +0100
+++ vfpmodule-patched.c 2015-03-20 20:44:27.070873564 +0100
@@ -31,6 +31,19 @@
 #include "vfpinstr.h"
 #include "vfp.h"

+static void dump_vfp_state(const int *state, const char *location)
+{
+       int i;
+       printk(KERN_NOTICE "VFP state dump for [%s]:\n", location);
+       for(i=0; i<16; i++) {
+               printk(KERN_NOTICE "  d%02i:   0x%08x 0x%08x\n", i, state[2*i], state[2*i+1]);
+       }
+       printk(KERN_NOTICE "  fpexc:   0x%08x\n", state[32]);
+       printk(KERN_NOTICE "  fpscr:   0x%08x\n", state[33]);
+       printk(KERN_NOTICE "  fpinst:  0x%08x\n", state[34]);
+       printk(KERN_NOTICE "  fpinst2: 0x%08x\n", state[35]);
+}
+
 /*
  * Our undef handlers (in entry.S)
  */
@@ -183,8 +196,12 @@ static int vfp_notifier(struct notifier_
                 * case the thread migrates to a different CPU. The
                 * restoring is done lazily.
                 */
-               if ((fpexc & FPEXC_EN) && vfp_current_hw_state[cpu])
+               if ((fpexc & FPEXC_EN) && vfp_current_hw_state[cpu]) {
+                       /* vfp_save_state oopses on VFP11 if EX bit set */
+                       fmxr(FPEXC, fpexc & ~FPEXC_EX);
                        vfp_save_state(vfp_current_hw_state[cpu], fpexc);
+                       dump_vfp_state((const int*)(vfp_current_hw_state[cpu]), "vfp_notifier");
+               }
 #endif

                /*
@@ -454,14 +471,19 @@ static int vfp_pm_suspend(void)
        /* if vfp is on, then save state for resumption */
        if (fpexc & FPEXC_EN) {
                pr_debug("%s: saving vfp state\n", __func__);
+               /* vfp_save_state oopses on VFP11 if EX bit set */
+               fmxr(FPEXC, fpexc & ~FPEXC_EX);
                vfp_save_state(&ti->vfpstate, fpexc);
+               dump_vfp_state((const int*)(&ti->vfpstate), "vfp_pm_suspend EN");

                /* disable, just in case */
                fmxr(FPEXC, fmrx(FPEXC) & ~FPEXC_EN);
        } else if (vfp_current_hw_state[ti->cpu]) {
 #ifndef CONFIG_SMP
-               fmxr(FPEXC, fpexc | FPEXC_EN);
+               /* vfp_save_state oopses on VFP11 if EX bit set */
+               fmxr(FPEXC, (fpexc & ~FPEXC_EX) | FPEXC_EN);
                vfp_save_state(vfp_current_hw_state[ti->cpu], fpexc);
+               dump_vfp_state((const int*)(vfp_current_hw_state[ti->cpu]), "vfp_pm_suspend ~EN");
                fmxr(FPEXC, fpexc);
 #endif
        }
@@ -523,8 +545,10 @@ void vfp_sync_hwstate(struct thread_info
                /*
                 * Save the last VFP state on this CPU.
                 */
-               fmxr(FPEXC, fpexc | FPEXC_EN);
+               /* vfp_save_state oopses on VFP11 if EX bit set */
+               fmxr(FPEXC, (fpexc & ~FPEXC_EX) | FPEXC_EN);
                vfp_save_state(&thread->vfpstate, fpexc | FPEXC_EN);
+               dump_vfp_state((const int*)(&thread->vfpstate), "vfp_sync_hwstate");
                fmxr(FPEXC, fpexc);
        }

@@ -689,17 +713,21 @@ void kernel_neon_begin(void)
        cpu = get_cpu();

        fpexc = fmrx(FPEXC) | FPEXC_EN;
-       fmxr(FPEXC, fpexc);
+       /* vfp_save_state oopses on VFP11 if EX bit set */
+       fmxr(FPEXC, fpexc & ~FPEXC_EX);

        /*
         * Save the userland NEON/VFP state. Under UP,
         * the owner could be a task other than 'current'
         */
-       if (vfp_state_in_hw(cpu, thread))
+       if (vfp_state_in_hw(cpu, thread)) {
                vfp_save_state(&thread->vfpstate, fpexc);
+               dump_vfp_state((const int*)(&thread->vfpstate), "kernel_neon_begin A");
 #ifndef CONFIG_SMP
-       else if (vfp_current_hw_state[cpu] != NULL)
+       } else if (vfp_current_hw_state[cpu] != NULL) {
                vfp_save_state(vfp_current_hw_state[cpu], fpexc);
+               dump_vfp_state((const int*)(vfp_current_hw_state[cpu]), "kernel_neon_begin B");
+       }
 #endif
        vfp_current_hw_state[cpu] = NULL;
 }
```
BTW, I saw that CONFIG_CPU_PM is set in my running config here. Doesn't really matter, this code should not be called in our test setup (and if it is, my new instrumentation in the patch above will notice it anyway).

popcornmix, 2015-03-20:
Current rpi-update firmware (3.18.9) has SMP, NEON and CPU_PM all unset for Pi1.
I'll build a kernel with extra debug.

catschulze, 2015-03-20:
OK, I begin to see what is causing this. It's again vfpmodule.c, this time vfp_restore_user_hwstate(), which is used in combination with vfp_preserve_user_clear_hwstate() to save the current CPU and FPU state to memory for signal processing and read it back later (if you look at the oops traces from @studio-dude, you will find the latter function). Strangely enough, vfp_restore_user_hwstate() contains the following fragment:
```C
   /* Ensure the VFP is enabled. */
   fpexc |= FPEXC_EN;  /*  <== THIS IS OK */

   /* Ensure FPINST2 is invalid and the exception flag is cleared. */
   fpexc &= ~(FPEXC_EX | FPEXC_FP2V);    /*  <== THIS IS STRANGE */
   hwstate->fpexc = fpexc;
```
in lines 618..623. Here, both FPEXC_EX and FPEXC_FP2V are explicitly masked out from the bitmask read back from memory (and I'm pretty sure that we write FPEXC_EX = 1 in my test code), and therefore VFP state before and after signal delivery does not match any more.

This is really strange, and I can see no reason for this. When building the new kernel, could you perhaps just comment out the line
```
   fpexc &= ~(FPEXC_EX | FPEXC_FP2V);
```
so that we can see whether this is actually the cause for the wrong results in my test program? I do not say that we want this line to be removed in the final patch, as there might be legitimate reasons for it, but I just want to see whether we have found the root cause.

P33M, 2015-03-20:
I think it may be illuminating if a little history digging is done around this code segment. I suspect that the mask line may have been introduced as a fix for a bug triggered from elsewhere - hardware errata notwithstanding.

https://github.com/torvalds/linux/commits/master/arch/arm/vfp/vfpmodule.c

popcornmix, 2015-03-20:
https://dl.dropboxusercontent.com/u/3669512/temp/vfp/kernel.img
https://dl.dropboxusercontent.com/u/3669512/temp/vfp/kernel7.img
updated with debug patch

catschulze, 2015-03-20:
The original commit was in arch/arm/kernel/signal.c, and this code was later moved to arch/arm/vfp/vfpmodule.c:

(1) Commit to signal.c, introducing VFP state on the signal stack:
https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/commit/arch/arm/kernel/signal.c?id=82c6f5a5b3e91ef4d2fb8725de4b8cf7affd4d61

(2) Commit that moved code from signal.c to vfpmodule.c
https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/commit/arch/arm/?id=2498814fcb3068f19b82b1519b4038721f61af43

To quote from (1):
```
Introduce also a new user struct for VFP exception registers. In
this too fields not relevant to the current VFP architecture are
ignored.
```
"too fields" should obviously read "two fields", and these two fields are FPEXC_EX and FPEXC_FP2V. "Relevance to the current VFP architecture" does not include VFP11 (which was already somewhat old at that point in time), I guess ;-)

From our experience with the RPi 2 vs. RPi 1 I would guess that only VFP11 can set FPEXC_EX "in the background" without synchronously triggering an exception. On newer VFPs (e.g., ARM Cortex A), FPEXC_EX is most likely only set at the same time an exception triggers, and the exception handler in the kernel will process the exception and clear the bit, so no user space code should ever see this bit set. If that's correct, then masking these two bits out makes sense - for ARM Cortex A. But obviously not for VFP11 on the RPi 1.

Allowing user space code to set FPEXC_EX and FPECX_FP2V on VFP11 should be relatively harmless, as user space code can do this anyway by executing corresponding opcodes (that's just what my code does). The only potential problem is that user space code has control over FPINST and FPINST2 as well and could provide any opcode it desires, even opcodes that would never be stored in these registers by the FPU (e.g., ARM CPU opcodes). However, the only thing the VFP support code does with these registers is emulate the instructions found in them, and it will notice when it cannot emulate an opcode (this is what caused the "unhandled bounce" instructions in our first attempts to reproduce the problem).

My current suggestion: Surround the line
```C
fpexc &= ~(FPEXC_EX | FPEXC_FP2V);
```
with a check for VFP11, similar to line 363 in vfpmodule.c:
```C
   if ((fpsid & FPSID_ARCH_MASK) != (1 << FPSID_ARCH_BIT)) { ... }
```
and skip masking out the two bits for architecture #1. But we should probably better ask on the ARM kernel mailing list for the opinion of other developers who know the plethora of ARM cores better...

popcornmix, 2015-03-20:
https://dl.dropboxusercontent.com/u/3669512/temp/vfp2/kernel.img
also contains:
```
	/* Ensure FPINST2 is invalid and the exception flag is cleared. */
	if ((fpsid & FPSID_ARCH_MASK) != (1 << FPSID_ARCH_BIT))
		fpexc &= ~(FPEXC_EX | FPEXC_FP2V);
```

studio-dude, 2015-03-21:
OK rebooted into @popcornmix's latest kernel. (Got a bit of a scare doing this remotely, since it took about 5 min for the system to come back up - I guess the debugging slowed things down?). 

Anyway, I ran the most recent c-code above, here's what I get:

```
[  233.012289] VFP state dump for [vfp_sync_hwstate]:
[  233.012335]   d00:   0x00000000 0x00000000
[  233.012350]   d01:   0x00000000 0x00000000
[  233.012362]   d02:   0x00000000 0x00000000
[  233.012373]   d03:   0x00000000 0x00000000
[  233.012384]   d04:   0x00000000 0x00000000
[  233.012395]   d05:   0x00000000 0x00000000
[  233.012406]   d06:   0x00000000 0x00000000
[  233.012417]   d07:   0x00000000 0x00000000
[  233.012428]   d08:   0x00000000 0x00000000
[  233.012439]   d09:   0x00000000 0x00000000
[  233.012450]   d10:   0x00000000 0x00000000
[  233.012462]   d11:   0x00000000 0x00000000
[  233.012474]   d12:   0x00000000 0x00000000
[  233.012485]   d13:   0x00000000 0x00000000
[  233.012496]   d14:   0x00000000 0x00000000
[  233.012508]   d15:   0x00000000 0x00000000
[  233.012518]   fpexc:   0x40000000
[  233.012527]   fpscr:   0x00000000
[  233.012536]   fpinst:  0x40000000
[  233.012546]   fpinst2: 0xd739be28
[  234.144126] VFP state dump for [vfp_sync_hwstate]:
[  234.144206]   d00:   0x00000000 0x00000000
[  234.144220]   d01:   0x00000000 0x00000000
[  234.144231]   d02:   0x00000000 0x00000000
[  234.144246]   d03:   0x00000000 0x00000000
[  234.144258]   d04:   0x00000000 0x00000000
[  234.144269]   d05:   0x00000000 0x00000000
[  234.144280]   d06:   0x00000000 0x00000000
[  234.144292]   d07:   0x00000000 0x00000000
[  234.144303]   d08:   0x00000000 0x00000000
[  234.144314]   d09:   0x00000000 0x00000000
[  234.144325]   d10:   0x00000000 0x00000000
[  234.144336]   d11:   0x00000000 0x00000000
[  234.144347]   d12:   0x00000000 0x00000000
[  234.144358]   d13:   0x00000000 0x00000000
[  234.144371]   d14:   0x00000000 0x00000000
[  234.144382]   d15:   0x00000000 0x00000000
[  234.144392]   fpexc:   0x40000000
[  234.144401]   fpscr:   0x00000000
[  234.144411]   fpinst:  0x40000000
[  234.144421]   fpinst2: 0xd739be28
[  235.453072] VFP state dump for [vfp_sync_hwstate]:
[  235.453116]   d00:   0x00000000 0x00000000
[  235.453130]   d01:   0x00000000 0x00000000
[  235.453142]   d02:   0x00000000 0x00000000
[  235.453153]   d03:   0x00000000 0x00000000
[  235.453164]   d04:   0x00000000 0x00000000
[  235.453177]   d05:   0x00000000 0x00000000
[  235.453188]   d06:   0x00000000 0x00000000
[  235.453199]   d07:   0x00000000 0x00000000
[  235.453210]   d08:   0x00000000 0x00000000
[  235.453221]   d09:   0x00000000 0x00000000
[  235.453232]   d10:   0x00000000 0x00000000
[  235.453244]   d11:   0x00000000 0x00000000
[  235.453255]   d12:   0x00000000 0x00000000
[  235.453266]   d13:   0x00000000 0x00000000
[  235.453277]   d14:   0x00000000 0x00000000
[  235.453289]   d15:   0x00000000 0x00000000
[  235.453301]   fpexc:   0x40000000
[  235.453310]   fpscr:   0x00000000
[  235.453319]   fpinst:  0x40000000
[  235.453329]   fpinst2: 0xd739be28
[  236.742808] VFP state dump for [vfp_sync_hwstate]:
[  236.742854]   d00:   0x41e1e82f 0x3ec05e0c
[... 2200 lines cut ...]
[  339.900190]   fpscr:   0x00000000
[  339.900200]   fpinst:  0x40000000
[  339.900209]   fpinst2: 0xd6311e28
[  341.028696] VFP state dump for [vfp_sync_hwstate]:
[  341.028740]   d00:   0x00000000 0x00000000
[  341.028754]   d01:   0x00000000 0x00000000
[  341.028768]   d02:   0x00000000 0x00000000
[  341.028780]   d03:   0x00000000 0x00000000
[  341.028791]   d04:   0x00000000 0x00000000
[  341.028802]   d05:   0x00000000 0x00000000
[  341.028813]   d06:   0x00000000 0x00000000
[  341.028824]   d07:   0x00000000 0x00000000
[  341.028835]   d08:   0x00000000 0x00000000
[  341.028846]   d09:   0x00000000 0x00000000
[  341.028857]   d10:   0x00000000 0x00000000
[  341.028868]   d11:   0x00000000 0x00000000
[  341.028880]   d12:   0x00000000 0x00000000
[  341.028893]   d13:   0x00000000 0x00000000
[  341.028904]   d14:   0x00000000 0x00000000
[  341.028915]   d15:   0x00000000 0x00000000
[  341.028926]   fpexc:   0x40000000
[  341.028935]   fpscr:   0x00000000
[  341.028944]   fpinst:  0x40000000
[  341.028954]   fpinst2: 0xd6311e28
```

catschulze, 2015-03-21:
OK, to sum things up, as there are quite a lot of redundancies in the dmesg output: I'm really happy to see
```
[  291.086021] VFP state dump for [vfp_sync_hwstate]:
[  291.086064]   d00:   0x00000000 0x7ff00000
[  291.086077]   d01:   0x00000000 0x00000000
[  291.086090]   d02:   0x00000000 0x00000000
[  291.086104]   d03:   0x00000000 0x00000000
[  291.086116]   d04:   0x00000000 0x00000000
[  291.086127]   d05:   0x00000000 0x3ff00000
[  291.086138]   d06:   0x00000000 0x00000000
[  291.086149]   d07:   0x00000000 0xbf800000
[  291.086161]   d08:   0x7f800000 0x00000000
[  291.086172]   d09:   0x00000000 0x00000000
[  291.086183]   d10:   0x00000000 0x00000000
[  291.086194]   d11:   0x00000000 0x00000000
[  291.086205]   d12:   0x00000000 0x00000000
[  291.086216]   d13:   0x00000000 0x00000000
[  291.086229]   d14:   0x00000000 0x00000000
[  291.086241]   d15:   0x00000000 0x00000000
[  291.086251]   fpexc:   0xc0000701
[  291.086260]   fpscr:   0x20000303
[  291.086270]   fpinst:  0xeeb18ae7
[  291.086279]   fpinst2: 0xd637de88
```
as this is indeed showing that the VFP11 state seems to be saved correctly. FPEXC is 0xc0000701; this means EX | EN | VECITR(0) | IOC, what I would expect for a trapping scalar sqrt(-1.0). FPSCR is 0x20000303; this means C | DZE | IOE | DZC | IOC. DZE and IOE are set by my code (exception trap enable for Divide by Zero and Invalid Operation), C is Carry, and DZC and IOC indicate that the code has already experienced Divide by Zero and Invalid Operation exceptions. FPINST is 0xeeb18ae7; this is the operation "vsqrt.f32 s16, s15", the first of the two set up to trap. FPINST is set, but not meaningful, as FP2V is not set in FPEXC.

There are two different values for FPINST reported in traces with FPEXC_EX set:
```
[  291.086270]   fpinst:  0xeeb18ae7
[  292.087340]   fpinst:  0xee878a87
```
The first is "vsqrt.f32 s16, s15", as already said, the second is "vdiv.f32 s16, s15, s14", matching the second operation set up in my code to cause a trap. Fine!

In other words, it seems that the vfp_sync_hwstate() is doing exactly what it is supposed to do, and I also see only vfp_sync_hwstate() traces in the log. This seems to confirm that the problematic part is indeed the masking of FPEXC_EX and FPEXC_FP2V in upstream's vfp_restore_user_hwstate().

I think we now have sufficient dmesg trace data, and I would suggest we try with a kernel that is reduced again to clearing FPEXC_EX before entering vfp_save_state(), and avoiding clearing FPEXC_EX and FPEXC_F2VP for architecture 1 VFP modules when restoring VFP state after signal handling. The following patch should be applied to an upstream vfpmodule.c and contains all required fixes.
```
--- ../vfpmodule-3.18.9.c       2015-03-20 13:51:32.000000000 +0100
+++ vfpmodule-patched.c 2015-03-21 09:58:03.027806844 +0100
@@ -183,8 +183,11 @@ static int vfp_notifier(struct notifier_
                 * case the thread migrates to a different CPU. The
                 * restoring is done lazily.
                 */
-               if ((fpexc & FPEXC_EN) && vfp_current_hw_state[cpu])
+               if ((fpexc & FPEXC_EN) && vfp_current_hw_state[cpu]) {
+                       /* vfp_save_state oopses on VFP11 if EX bit set */
+                       fmxr(FPEXC, fpexc & ~FPEXC_EX);
                        vfp_save_state(vfp_current_hw_state[cpu], fpexc);
+               }
 #endif

                /*
@@ -454,13 +457,16 @@ static int vfp_pm_suspend(void)
        /* if vfp is on, then save state for resumption */
        if (fpexc & FPEXC_EN) {
                pr_debug("%s: saving vfp state\n", __func__);
+               /* vfp_save_state oopses on VFP11 if EX bit set */
+               fmxr(FPEXC, fpexc & ~FPEXC_EX);
                vfp_save_state(&ti->vfpstate, fpexc);

                /* disable, just in case */
                fmxr(FPEXC, fmrx(FPEXC) & ~FPEXC_EN);
        } else if (vfp_current_hw_state[ti->cpu]) {
 #ifndef CONFIG_SMP
-               fmxr(FPEXC, fpexc | FPEXC_EN);
+               /* vfp_save_state oopses on VFP11 if EX bit set */
+               fmxr(FPEXC, (fpexc & ~FPEXC_EX) | FPEXC_EN);
                vfp_save_state(vfp_current_hw_state[ti->cpu], fpexc);
                fmxr(FPEXC, fpexc);
 #endif
[... 23 lines cut ...]
-       fpexc &= ~(FPEXC_EX | FPEXC_FP2V);
+       /* Mask FPXEC_EX and FPEXC_FP2V if not required by VFP arch */
+       if ((fpsid & FPSID_ARCH_MASK) != (1 << FPSID_ARCH_BIT)) {
+               /* Ensure FPINST2 is invalid and the exception flag is cleared. */
+               fpexc &= ~(FPEXC_EX | FPEXC_FP2V);
+       }
+
        hwstate->fpexc = fpexc;

        __get_user_error(hwstate->fpinst, &ufp_exc->fpinst, err);
@@ -689,7 +701,8 @@ void kernel_neon_begin(void)
        cpu = get_cpu();

        fpexc = fmrx(FPEXC) | FPEXC_EN;
-       fmxr(FPEXC, fpexc);
+       /* vfp_save_state oopses on VFP11 if EX bit set */
+       fmxr(FPEXC, fpexc & ~FPEXC_EX);

        /*
         * Save the userland NEON/VFP state. Under UP,
```
@popcornmix or @P33M: Could you build a new kernel to test?
@studio-dude: Could you run another test with this new kernel, and include the *console output* of a test run (*not dmesg traces*, I need to see the actual values produced by the test code to see whether it works)?

If this should finally solve our problem, I am happy to send it to the ARM linux kernel mailing list for further review and potential inclusion in the upstream kernel.

popcornmix, 2015-03-21:
Kernels with latest @catschulze patch:
https://dl.dropboxusercontent.com/u/3669512/temp/vfp3/kernel.img
https://dl.dropboxusercontent.com/u/3669512/temp/vfp3/kernel7.img

[22 more replies]

closed 2016-08-15 by studio-dude