raspberrypi/linux #860: crypto/crc32.c  typo

ShorTie8, 2015-02-28:
While trying to load the btrfs.ko module I get

insmod /lib/modules/3.18.7-v7+/kernel/fs/btrfs/btrfs.ko
Error: could not insert module /lib/modules/3.18.7-v7+/kernel/fs/btrfs/btrfs.ko: Unknown symbol in module

doing some research I found  https://github.com/manjaro/packages-core/issues/8
saying "crypto/crc32c.c is missing the MODULE_ALIAS_CRYPTO("crc32c")."
and adds in
+MODULE_ALIAS_CRYPTO("crc32c"); to it

when looking at  https://github.com/raspberrypi/linux/commit/cc1b377756e1d52297022fb231458b9ca78e3c0f#diff-be6e4ae5b11572a04e1e8ef1d5fee241

I see "+MODULE_ALIAS_CRYPTO("crc32");" has been added to crypto/crc32c.c

I am wondering if there was a typo and this should have been crc32c instead of just crc32 ??

Thankz for looking/checking
Have A Great Day
Jeff

pelwell, 2015-02-28:
Interesting idea, but I think it is intentional - the module names appear to match the filenames.

popcornmix, 2015-02-28:
I think we have the corresponding patch:
https://github.com/raspberrypi/linux/commit/080c3875761ff370d000b94b9513bebe22bf1dae#diff-7f383596cc16b0c6ab588def4757fd61R174
and for me:
```sudo modprobe btrfs```
runs without complaint.

ShorTie8, 2015-03-01:
Yes i see it does.
I updated kernel with rpi-update and all seems well.
Sorry for being such a Dum Dum.
Thankz for all your hard work and efforts.

Have A Great Day
Jeff

closed 2015-03-01 by ShorTie8