raspberrypi/linux #861: puppetmaster-passenger package installation broken

DamienGarrido, 2015-02-28:
During the `puppetmaster-passenger` package installation, the apache2 module passenger `libapache2-mod-passenger` is not enabled (whereas the `ssl` and `headers` ones are !), thus the following error occurs:

> Syntax error on line 3 of /etc/apache2/sites-enabled/puppetmaster:
> Invalid command 'PassengerHighPerformance', perhaps misspelled or defined by a module not > included in the server configuration
> Action 'configtest' failed.
> The Apache error log may have more information.
>  failed!
> invoke-rc.d: initscript apache2, action "restart" failed.

An alternative is to manually enable the module during the installation process by starting the command  `a2enmod passenger` before the apache2 server is restarted.

popcornmix, 2015-03-01:
This github repo is for the linux kernel. I'm guessing your issue is related to the packages installed from the raspbian repo. Best to ask here:
http://www.raspberrypi.org/forums/viewforum.php?f=66

closed 2015-03-01 by popcornmix

DamienGarrido, 2015-03-01:
Sorry !
I misunderstood the link from the http://www.raspbian.org/RaspbianBugs Raspbian page...
Thank you for your help !