raspberrypi/linux #863: 3.18.8 initrd booting issue again

ha5di, 2015-03-01:
I had an initrd booting issue back to 3.18.6 but it was solved, see

It was working fine with 3.8.6 and dated kernel. Now with 3.18.8 and the current firmware it doesn't work anymore. With root=/dev/ram0 in cmdline.txt and ramfsfile=... in config.txt kernel do not find root file system. Adding ramfsaddr=-1 kernel do not boot, no messages only the rainbow coloured screen.

How to boot initrd now?

mr-berndt, 2015-03-01:
Same problem here, too. I am fighting for a week with this problem now. Until yesterday, I thought, I might have overlooked something in the kernel-configuration but actuelly I am sure meanwhile that my kernel sould boot with an initrd.

My setup is quite the same as picore's, cmdline.txt, root=/dev/ram0 etc. so I will not go into details right now but I will help testing.

amtssp, 2015-03-01:
Hi Belá I'm away from my development computer, but did you try something like this in the config.txt file:
initramfs piCore3187mm.gz followkernel

And the delete the old ramfsfile=...  and ramfsaddr=-1 from the config.txt file?

Steen

mr-berndt, 2015-03-01:
Tried that over here and it doesn't work.

popcornmix, 2015-03-01:
Can you post contents of config.txt and cmdline.txt, and output of:
```
vcgencmd version
uname -a
vcgencmd get_config str
vcgencmd get_config int
```
Can you also identify the first firmware update that doesn't work. You can get older versions of firmware with
```sudo rpi-update <hash>```
where hash comes from end of URLs here:
https://github.com/Hexxeh/rpi-firmware/commits/master

mr-berndt, 2015-03-01:
Hmm, a bit difficult, I only have a very minimal busybox-system with very few commands (it is an audio-streamer) and vcgencmd and rpi-update are not amongs them. ;(

I don't know if the PiCore guys will be able to come up with more, since they are also a self-built busybox-based distro, but maybe they compiled some comfort-features into it..

uname -a would output "Linux qstreamer 3.18.7-v7+ #9 SMP PREEMPT Sat Feb 28 20:27:44 CET 2015 armv7l GNU/Linux".

So you think it is a firmware-issue and not kernel-related?

In order to get older firmware I guess I would have to get the entire git-repo and checkout the according hases, right?

popcornmix, 2015-03-01:
We need to establish whether it is kernel or firmware.
The history of firmware files is available on github.
Try downloading start.elf and fixup.dat from a specific point in time and replace the ones on sdcard, and check if that fixes things.

If it makes no difference then we know it's kernel. If a specific version fixes things then we know it's firmware.

Note a bug has been fixed with initrd, but versions https://github.com/Hexxeh/rpi-firmware/commit/9e69b79f484fe005d7eca698962f3ee27fa00aaf to https://github.com/Hexxeh/rpi-firmware/commit/0c9001b568d1ac9593925fb76795bcc25cc59558 will not work correctly.

Make sure your current version is not in that range.

mr-berndt, 2015-03-01:
OK, I will try to get the older versions and i forgot to post my cmdline.txt and config.txt.
Here we go:

cmdline.txt:
loglevel=1 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 console=tty1 initrd=0xa00000,11764110 root=/dev/ram0 rw rootfstype=ext4 elevator=deadline rootwait sdhci_bcm2708.enable_llm=0 dwc_otg.speed=0 init=/bin/init

config.txt:
disable_splash=1
kernel=zImage
device_tree=

# used to work on rasberry < 2 with linux-3.16.y
# doesn't on raspberry 2 with 3.18.8
ramfsfile=qstreamerimage.cpio.gz
ramfsaddr=0x00a00000

# Doesn't work on raspberry 2
#ramfsaddr=-1

# Doesn't work on R2 as well
#initramfs qstreamerimage.cpio.gz
#followkernel

gpu_mem=256
disable_overscan=1
hdmi_force_hotplug=1

pelwell, 2015-03-01:
Try:
```
initramfs qstreamerimage.cpio.gz followkernel
```
on one line, which is at least correct.

popcornmix, 2015-03-01:
Yes, ramfsaddr=0x00a00000 is just wrong for Pi 2.
ramfsaddr is a bus address, which needs to start with 0xC (e.g. 0xc0a00000) on Pi2.
Technically it should be 0x40a00000 on Pi1 to avoid allocation in GPU's L1 cache which arm can't see.

But really you should try to use  "followkernel" and let that get set automatically.

mr-berndt, 2015-03-01:
Thx popcornmix, so I tried putting "initramfs qstreamerimage.cpio.gz followkernel" into config.txt instead of the two lines "ramfsaddr..." and "ramfsaddr..."  but I still get "unable to mount root fs on unknown-block(1,0).

OK, on to firmware. I cloned the repository but I don't really know how to get a specific snapshot now. In the kernel-repository I know how to navigate to a certain point in history and get the tag I need to checkout a version. How do do that in the fw-repo? Also I saw a "bump to 3.18" two days ago. Does that mean, trying a fw older that that is senseless with kernel 3.18?

amtssp, 2015-03-01:
Final idea from my side.  Did you use the mkknlimg utility on your new kernel before using it?

If you are using a self compiled kernel with DT enabled you need to follow the advice from Ross on this page: http://www.raspberrypi.org/forums/viewtopic.php?f=29&t=98009 

As far as I understand you still need to do "./mkknlimg zImage kernel.img" on your newly build kernel in order to make it DT compliant

popcornmix, 2015-03-01:
Go here:
https://github.com/Hexxeh/rpi-firmware/commits/master/start.elf
Click on the "<>" icon, then click the "raw" button to download that specific version.
Do the same with fixup.dat.

mr-berndt, 2015-03-01:
I went through all files backwards from the latest down to Feb 7th "Bump to 3.18.6" with no luck. I don't know how old Bela's firmware was, but he mentioned geting problems when switching from 3.18.6 to a newer kernel, so I stopped there. Also my daughter claims me now, so I will be away for a while. What makes sense to try next? I would go and compile a kernel 3.18.6 and see, if I get that working.

Bela, got a .config for me to make sure I use a configuration that is proven to boot using initramfs?

popcornmix, 2015-03-01:
Well we need some evidence that something has been broken.
The obvious way is to point at a version (of gpu firmware or kernel) that works, and a subsequent version that doesn't work.

If you are building custom kernels that don't work (and may never have worked) then we can't really help.

 Can I just state the obvious. A Pi1 kernel won't boot on Pi2. You need to build the kernel with Pi2 .config. You are probably well aware of that, but you haven't explicitly said that.

mr-berndt, 2015-03-01:
I built dozens of kernel that worked on Pi1 and trying to switch to Pi2 leaves me with a kernel that works, exept for the initramfs-feature. That is why I asked for Belas .config, I never had it running with initramfs on the Pi2. Anyway, I will compile a 3.18.6 tonight and see where that gets me with firmware from around that time. If I get that to boot then we can see what needs to change in order to break it.

mr-berndt, 2015-03-01:
Well, kernels were never my strong suit but this is puzzling me:

Having checked out the sources from git for 3.18.6 I don't find any options for bcm2708 or bcm2709. In 3.18.7 they are there. Checking one of my older kernels (3.16.5) I find bcm2708 just as I remembered. So the option seems to have disappeared somewhen and came back? A bit strange for kernel-sources pulled from the raspberry github. I wonder if something has gone wrong with my checkout but I can configure and even compile the sources for 3.16.6 just fine.

Now checking for the defconfigs I find bcm2709_defconfig in 3.18.7 but nothing even similar in 3.18.6, in 3.16.5 there is a bcmrpi_defconfig.

So, sorry, I cannot try and compile 3.16.8 to check, if it is working.

popcornmix, 2015-03-01:
bcmrpi_defconfig should be available with any kernel version.
bcm2709_defconfig will only be available with versions of the kernel newer than release of Pi2.

mr-berndt, 2015-03-01:
On my 3.18.6 sources wich I pulled from https://github.com/raspberrypi/linux:
ls arch/arm/configs/bcm*
arch/arm/configs/bcm2835_defconfig  arch/arm/configs/bcm_defconfig

I used:
git-init
git clone https://github.com/raspberrypi/linux.git
opened the history of the Makefile under main and got the hashtag
git checkout 9960914954298eafb09f21ac1b042c79032f271f
and ended up with these sources. Did I do wrong?

mr-berndt, 2015-03-01:
Ok, I can confirm now that "initramfs qstreamerimage.cpio.gz follow kernel" does the trick for kernel 3.18.7. Can't say about 3.18.8 though, didn't bother to compile one, yet. I still had "initrd=0xa00000,11764110  rootfstype=ext4" next to "root=/dev/ram0" left over in my cmdline.txt and that didn't seem to go well together. I am not exactly sure which firmware I had on the sdcard when all the lights suddenly went up, since I copied around a lot of files over the day, trying to make it work, but if there is interest, I can report back on the latest firmware tomorrow.

How I ended up with the weird kernel-sources still puzzles me, though! Never really used git to checkout something not-recent..

ha5di, 2015-03-02:
It is really strange. Now kernels do not boot at all, including my first 3.18.6 DT kernel, latest Raspbian image and my dev system on a different card, get only rainbow colors. An hour ago 3.18.6 was booting fine. Maybe it is just a random coincidence and I got just a hw defect independently of tests, but the situation is suscipious. It is a Model B, Rev 2 board. At the moment I do not have another armv6 board in hand, will grab one. WIll recheck these stuff on RPi2.

popcornmix, 2015-03-02:
Try a stock raspbian image to rule out a hardware fault (but I'm imagine that is very unlikely).

ha5di, 2015-03-02:
No one is booting, including stock Raspbian and our distribution images with 3.18.6 and previous 3.12.36, with different SD cards. Board looks dead.

ha5di, 2015-03-02:
Found. It's the SD card socket, unrealible contacts. Board is from the early series and used a lot. This socket was a week part of the construction. Will see how can I fix it.
.

popcornmix, 2015-03-02:
Ah - a bit of card behind the sdcard, or gentle prising up of the pins on the socket are the usual ways of getting a better grip.

ha5di, 2015-03-02:
Yes, I will glue a small plastic to cover the open top of the 'socket' and clean the contacts.

ha5di, 2015-03-07:
Now I have a working board again. Connected a serial cable and found it booting my 3.18.8 kernel:

...
...
[    2.363920] [vc_sm_connected_init]: end - returning 0
[    2.373633] F2FS-fs (ram0): Magic Mismatch, valid(0xf2f52010) - read(0x0)
[    2.382552] Indeed it is in host mode hprt0 = 00021501
[    2.394126] mmc0: host does not support reading read-only switch, assuming write-enable
[    2.405382] F2FS-fs (ram0): Can't find valid F2FS filesystem in 1th superblock
[    2.426177] F2FS-fs (ram0): Magic Mismatch, valid(0xf2f52010) - read(0x0)
[    2.434680] F2FS-fs (ram0): Can't find valid F2FS filesystem in 2th superblock
[    2.435002] mmc0: new high speed SDHC card at address b368
[    2.466039] F2FS-fs (ram0): Magic Mismatch, valid(0xf2f52010) - read(0x0)
[    2.474528] F2FS-fs (ram0): Can't find valid F2FS filesystem in 1th superblock
[    2.496072] mmcblk0: mmc0:b368 SMI   3.74 GiB
[    2.503300] F2FS-fs (ram0): Magic Mismatch, valid(0xf2f52010) - read(0x0)
[    2.512208]  mmcblk0: p1 p2
[    2.527444] F2FS-fs (ram0): Can't find valid F2FS filesystem in 2th superblock
[    2.542884] List of all partitions:
[    2.548933] b300         3929088 mmcblk0  driver: mmcblk
[    2.556404]   b301           57344 mmcblk0p1 0009bf4f-01
[    2.563690]   b302         3138560 mmcblk0p2 0009bf4f-02
[    2.571051] No filesystem could mount root, tried:  ext3 ext2 ext4 squashfs vfat msdos f2fs
[    2.583596] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(1,0)
[    2.595728] CPU: 0 PID: 1 Comm: swapper Not tainted 3.18.8-piCore+ #1
[    2.604224] [<c0014b64>] (unwind_backtrace) from [<c00120ec>] (show_stack+0x20/0x24)
[    2.615863] [<c00120ec>] (show_stack) from [<c055e80c>] (dump_stack+0x20/0x28)
[    2.625150] [<c055e80c>] (dump_stack) from [<c055a78c>] (panic+0xa4/0x214)
[    2.634082] [<c055a78c>] (panic) from [<c07b42c0>] (mount_block_root+0x210/0x254)
[    2.645460] [<c07b42c0>] (mount_block_root) from [<c07b44f0>] (mount_root+0xf0/0x114)
[    2.657280] [<c07b44f0>] (mount_root) from [<c07b467c>] (prepare_namespace+0x168/0x1cc)
[    2.669342] [<c07b467c>] (prepare_namespace) from [<c07b3e68>] (kernel_init_freeable+0x178/0x1bc)
[    2.682306] [<c07b3e68>] (kernel_init_freeable) from [<c05596e0>] (kernel_init+0x18/0xf4)
[    2.694718] [<c05596e0>] (kernel_init) from [<c000ea48>] (ret_from_fork+0x14/0x20)
[    2.706702] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(1,0)
[    4.713062] random: nonblocking pool is initialized

Hm..... f2fs file system? cmdline:

dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0xe bcm2708.serial=0xbea4fd44 smsc95xx.macaddr=B8:27:EB:A4:FD:44 bcm2708_fb.fbswap=1 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/ram0 elevator=deadline rootwait nortc

initrd is the usual cpio with -H newc gzipped.

pelwell, 2015-03-07:
What is in your config.txt now?

ha5di, 2015-03-07:
initramfs 6.1alpha3.gz followkernel

and

ramfsfile=6.1alpha3.gz
ramfsaddr=-1

give the same result. I will check the initrd file itself now.

ha5di, 2015-03-07:
Rebuilt initrd, result is the same. So I have no idea how to fix it.

mr-berndt, 2015-03-07:
I don't know if this will help, but I had a similar issue along my way. In my case the ignited was faulty. Strangely the script I used to build the initrd from a regular root-filesystem contained a portion which used dd to make an image of the first partition of the sd-card in question. Always worked with Pi1 and failed with Pi2. The problem was that "dd if=/dev/sdb bs=512 count=$number_of_512k-blocks_of_filesystem1_of _disk" had to be changed to "count=$number_of_512k-blocks_of_filesystem1_of _disk +1. It makes no sense to me at all but it fixed my problem.

Btw, out of curiosity I downloaded your picore alpha version a couple of days ago, put a 3.17 kernel (bcm2709_defconfig) onto the partition, changed the options to initramfs 6.1alpha3.gz follow kernel and bootet just fine.

ha5di, 2015-03-08:
Rebuilt system with 3.18.9 kernel and current firmware. Everything is working as expected. Still no idea what was wrong with the 3.18.8 version, it was built on the same way as 3.18.6 and 3.18.9 

Anyhow, problem is gone, ticket closed.

closed 2015-03-08 by ha5di