raspberrypi/linux #865: i2c-bcm2708 now gives timeout on 513 byte transfer

notro, 2015-03-01:
Branch: rpi-3.19.y

I'm driving an i2c oled display.
2 and 3 byte transfer is fine (initialization), but sending 513 bytes (prefix + pixel data) leads to timeout:
```
[    7.907176] i2c i2c-1: transfer timed out
[    7.913685] ada-ssd1306fb: probe of 1-003c failed with error -110
```

Reverting to the previous commit fixes the issue: i2c_bcm2708: Fix clock reference counting

Pinging patch author @jeanleflambeur:
[Fix grabbing lock from atomic context in i2c driver](https://github.com/raspberrypi/linux/commit/0ca4c1f57c84bb114641176d63d16ce345f5a95d)

jeanleflambeur, 2015-03-01:
It could be because of this:
#define I2C_WAIT_LOOP_COUNT 40

Can you increase it to some huge number, say  2000000000?

notro, 2015-03-01:
That didn't help.

jeanleflambeur, 2015-03-01:
What i2c frequency are you using?

notro, 2015-03-01:
I'm using the default which seems to be: ```clock-frequency = <100000>;```

This is the overlay I use:
```
/*
 * Raspberry Pi Device Tree overlay
 *
 * Adafruit
 * Monochrome 128x32 I2C OLED graphic display
 * Product id: 931
 *
 */

/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2708";
	
	fragment@0 {
		target = <&gpio>;
		__overlay__ {
			ssd1306_pins: ssd1306_pins {
				brcm,pins = <4>;
				brcm,function = <1>; /* out */
			};
		};
	};

	fragment@1 {
		target = <&i2c1>;
		__overlay__ {
			status = "okay";

			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;

			ssd1306: ssd1306@1{
				compatible = "ada,ssd1306-128x32";
				reg = <0x3c>;
				pinctrl-names = "default";
				pinctrl-0 = <&ssd1306_pins>;

				reset-gpios = <&gpio 4 0>;
			};
		};
	};
};
```

notro, 2015-03-01:
And this is my transfer function:
```
static int lcdreg_i2c_send(struct i2c_client *client, unsigned index, void *buf, size_t len)
{
	u8 *txbuf;
	int ret;

	txbuf = kmalloc(1 + len, GFP_KERNEL);
	if (!txbuf)
		return -ENOMEM;

	txbuf[0] = index ? 0x40 : 0x80;
	memcpy(&txbuf[1], buf, len);

	ret = i2c_master_send(client, txbuf, 1 + len);
	kfree(txbuf);

	return ret < 0 ? ret : 0;
}
```

jeanleflambeur, 2015-03-01:
Can you pls try with the bold part commented out:
/* did we time out or some error occured? */
if (wait_loops < 0 **|| (s & (BSC_S_ERR | BSC_S_CLKT))**) {
    return -1;
}

jeanleflambeur, 2015-03-01:
There are 3 parts in this commit:
1. Moved clk_get_rate from bcm2708_bsc_setup to bcm2708_i2c_functionality
2. Added a timeout check in bcm2708_bsc_setup
3. Added some missing error checking in bcm2708_bsc_setup
You already tested the timeout from 2. and the problem is not there. So it can be either 1. or 3.

jeanleflambeur, 2015-03-21:
notro - did you get any chance to test this further? I don't have any i2c device that can receive so much data unfortunately.

notro, 2015-03-24:
The project I use this in is currently on hold and I don't know when I can return to it.
BTW this issue illustrates one reason why we should never do many changes in the same commit.
It's hard to bisect inside a commit :-(

jeanleflambeur, 2015-03-24:
Lesson learned about multiple changes in the same commit.
I ordered an i2c oled display today so hopefully I'll be able to test during the weekend.

notro, 2015-07-25:
I have returned to this project now and it's working fine on 4.0.
I thought maybe [i2c-bcm2708: Increase timeouts to allow larger transfers](https://github.com/raspberrypi/linux/commit/285fcbbdecb0ab26d0a76f200a12ff6a4ce1419f) had solved it for me, but reverting it still makes it work.
So I don't know what fixed this problem for me, my driver code is exactly the same as on 3.19.
Anyway, this issue is closed.

closed 2015-07-25 by notro